tsduck/tsduck #550: Is DTA-2131 supported? (DTAPI status 4126)

rezna11, 2020-03-09:
Hello, I'm wondering if DTA-2131 is unsupported by TSDuck or if I'm doing something wrong...

```
tsp -v -I dektec -d 2 -f 490000000 --dvbt-bandwidth 8 -m DVB-T2 -O play
* dektec: using Dektec device 2, input channel 0 (DTA-2131 port 1)
* Error: dektec: capture error on Dektec device 2: DTAPI_E_INTERNAL (DTAPI status 4126)
```
Device info:

```
* Device 2: DTA-2131 (Multi-standard VHF/UHF receiver)
  Physical ports: 1
  Channels: input: 1, output: 0
  Input 0: DTA-2131 port 1, Demodulator (Advanced demodulation, VHF-band
      47-470MHz, UHF-band 400-862MHz, Uni-directional input, Transparent mode,
      MPEG-2 transport stream, Demodulation, ATSC 8-VSB reception, ATSC3.0
      reception, DAB reception, DVB-C2 reception, DVB-T reception, DVB-T2
      reception, I/Q sample reception, ISDB-T reception, QAM-A reception, QAM-B
      reception, QAM-C reception, T2MI reception)
  Subsystem id: 0x0000 (DTA-2131)
  Subsystem vendor id: 0x0000
  Device id: 0x0853
  Vendor id: 0x1A0E
  Serial number: 000000007F047FF5
  Firmware version: 2 (0x00000002)
  Firmware variant: 0 (0x00000000)
  PCI bus: 31, slot: 81
  Engineering change level: Rev 1
  Manufacture id: 04
  Production date: 2017.09
  Part number: DTA-2131
```

I tried with DTAPI version 3.7.0.83 and TSDuck version 3.19-1520.

lelegard, 2020-03-09:
Hello,

I have never used DTA-2131 myself but I think that it has been used by other TSDuck users. Maybe there is an inconsistency between the DTAPI and the drivers.

rezna11, 2020-03-09:
Well if anyone managed to tune in DVB-T2 with DTA-2131, I'd like to know. I'm not sure if DTAPI and drivers are the problem because they're both downloaded from Dektec and installed on the same day, so I'd assume compatibility. And also there's no problem with DTA-2131 in the software tools from DekTec, they can all tune in DVB-T2 without issues.

lelegard, 2020-03-09:
What is the OS, Windows or Linux?

> I'm not sure if DTAPI and drivers are the problem because they're both downloaded from Dektec and installed on the same day,

So, you mean that you recompiled TSDuck with the new DTAPI?

DTAPI is a static library which is embedded with the application. So, if you download and install the latest drivers and use the public binaries for TSDuck 3.19 (built 3 months ago), there can be an inconsistency between the driver and the DTAPI in the TSDuck binaries.

rezna11, 2020-03-10:
It's Windows.
So I downloaded DekTec WinSDK and installed it. Then I downloaded TSDuck and built using build-installer.ps1. Then I installed TSDuck. I also downloaded and installed Dta driver.
Still the same result (error 4126).

Sefy76, 2023-06-12:
I see this is from 3 years ago, is there any update on this? i've tried using this card and i get the following message:

C:\Program Files\TSDuck\bin>tsp.exe -v -I dektec -c 0 --atsc3-bandwidth 6 --frequency 213000000 -P regulate --pcr-synchronous -O ip 234.250.1.6:6000
* regulate: minimum wait: 2,000,000 nano-seconds, using 50,000,000 ns
* dektec: using Dektec device 0, input channel 0 (DTA-2131 port 1)
* Error: dektec: error enabling Dektec LNB controller: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)

lelegard, 2023-06-12:
> Error: dektec: error enabling Dektec LNB controller: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)

Trying to setup an LNB for ATSC reception, and more generally a cable/terrestrial receiver, is obviously an error. LNB are used for satellite reception only.

I will have a look at the code. However, I no longer have any Dektec device to test.

Sefy76, 2023-06-12:
I can do tests for you, I have a system that is currently available for this, as we are trying to use this model for ATSC 3.0 reception

lelegard, 2023-06-12:
~~~
$ tsp -I dektec --help
...
  -m value
  --modulation value
      For demodulators, indicate the modulation type. The supported modulation
      types depend on the device model. The default modulation type is DVB-S.
      Must be one of "128-QAM", "16-QAM", "256-QAM", "32-QAM", "64-QAM",
      "ATSC-3.0", "ATSC-VSB", "DAB", "DVB-C2", "DVB-S", "DVB-S-QPSK", "DVB-S2",
      "DVB-S2-16APSK", "DVB-S2-32APSK", "DVB-S2-8PSK", "DVB-S2-QPSK", "DVB-T",
      "DVB-T2", "ISDB-T", "QAM".
~~~

The default modulation is `DVB-S`. This is why it tries to setup the LNB.

Please try with `--modulation ATSC-3.0`.

There are many options in the `dektec` input and output plugins. In case of trouble, make sure to review them all in the user's guide.

Sefy76, 2023-06-12:
Is there a specific location I need to put the modulation? i've tried it, i'm getting the following:

tsp.exe -v -I dektec -c 0 --atsc3-bandwidth 6 --modulation ATSC-3.0 --frequency 213000000 -P regulate --pcr-synchronous -O ip 234.250.1.7:6007
* regulate: minimum wait: 2,000,000 nano-seconds, using 50,000,000 ns
* dektec: using Dektec device 0, input channel 0 (DTA-2131 port 1)
* Error: dektec: device SetRxControl error: DTAPI_E_INVALID_MODE (DTAPI status 4108)

lelegard, 2023-06-12:
> Is there a specific location I need to put the modulation?

This is an option of the `dektec` input plugin, so your command is correct.

Maybe you need some other modulation parameters to make ATSC 3 work. Being located on a DVB continent, I have never used ATSC reception. Please review all options of this plugin in the user's guide. 

Also try
~~~
tsp --debug -I dektec -c 0 --atsc3-bandwidth 6 --modulation ATSC-3.0 --frequency 213000000 -O drop
~~~

Make sure to surround your log with markdown code delimiters `~~~` to make it more readable.

Sefy76, 2023-06-12:
How's this? 
~~~
tsp --debug -I dektec -c 0 --atsc3-bandwidth 6 --modulation ATSC-3.0 --frequency 213000000 -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec -c 0 --atsc3-bandwidth 6 --modulation ATSC-3.0 --frequency 213000000
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 0, input channel 0 (DTA-2131 port 1)
* Debug: dektec: attaching to device DTA-2131 serial 0x000000007F0481F6
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting RxControl, mode: 0
* Debug: dektec: clearing FIFO and flags
* Debug: dektec: getting FIFO max size
* Debug: dektec: max FIFO size: 67,108,864 bytes
* Debug: dektec: using FIFO size: 67,108,864 bytes
* Debug: dektec: tuning to frequency 213,000,000
* Debug: dektec: setting RxMode, mode: 17
* Debug: dektec: setting RxControl, mode: 1
* Error: dektec: device SetRxControl error: DTAPI_E_INVALID_MODE (DTAPI status 4108)
* Debug: start() error in plugin dektec
* Debug: Report logging thread terminated
* ~~~

lelegard, 2023-06-12:
We need an advice from someone at Dektec.

@EricDeJong, is there any specific setup for ATSC 3 demodulation on a DTA-2131 ?

I will add more debug trace, using `DtDemodPars::ToXml()` before `DtInpChannel::Tune()`.

lelegard, 2023-06-13:
> I will add more debug trace, using DtDemodPars::ToXml() before DtInpChannel::Tune().

Please retry the debug command with the latest nightly build 3.35-3227 [here](https://tsduck.io/download/prerelease/). Then, let's wait for a feedback from Dektec.

EricDeJong, 2023-06-13:
RxMode 17 is DTAPI_RXMODE_ST188

From what I know about ATSC3.0 this is an IP based stream, not ASI.

I will ask around here which mode you will need to receive the ATSC3.0 data.

> * Debug: dektec: setting RxMode, mode: 17

EricDeJong, 2023-06-14:
To receive ATSC-3.0, the DtAdvDemod class from the DekTec SDK is needed in stead of the DtInpChannel class.
The reason is that with ATSC-3.0 ALP data, multiple PLP’s are received in a callback, the DtinpChannel class is not suitable for this task.
TsDuck is a transport stream analyzer, I’m afraid it cannot handle ATSC-3.0 ALP packets (yet?).

Of course, TsDuck can be used with an DTA-2131 for DVB-C/DVB-T and other transport stream related modulation standards.

Sefy76, 2023-06-14:
Hi @EricDeJong , thank you, unfortunately we need the card specifically for ATSC 3.0 purpose
@lelegard, i'm assuming due to what is said above ( (i assume ALP packets) that it will not be supported ?

EricDeJong, 2023-06-14:
Hi @Sefy76, if you contact me at Dektec.com, I can send you a sample code (c++) which uses the DekTec SDK to write PCAP packets to disk.

Sefy76, 2023-06-14:
@EricDeJong , is there any specific email to contact you there? or just use the info@dektec.com?

EricDeJong, 2023-06-14:
Info@ is the address. The idea is to not put email addresses in a mailing list to prevent spambots to send even more spam.

Sefy76, 2023-06-14:
My apologies @EricDeJong , i've edited my post and removed the full email
I'll send you an email within a few min, thank you very much!

lelegard, 2023-06-14:
@Sefy76, @EricDeJong, I am pretty ignorant in ATSC-3.0. From your responses, I understand that the low-level packets are ALP and not TS and that there are several PLP. But what is the format of each individual PLP? Is this a stream of TS packets which can be decapsulated?

TSDuck is, as its names implies, a toolkit which manipulates MPEG transport streams, and exclusively MPEG-TS. It does not make sense to try to work on something else. However, the input plugins are designed to extract a TS from some input binary magma. For instance, DVB-T2 and DVB-S2 can encapsulate several PLP's. When you demodulate a DVB-T2 or DVB-S2 carrier, you get something which is not a TS but which encapsulates several TS. The `dvb` input plugin can extract _one_ PLP out of it, which becomes _the_ TS which is manipulated by TSDuck. The same principle applies to the `ip` input plugin: it extracts one TS out of UDP datagrams.

So, if each PLP inside an ATSC-3 transport can be extracted as a stream of TS packets, it makes sense to enhance the `dektec` input plugin to get it. Just like DVB-T2 or DVB-S2, TSDuck will never analyze the lower-level packets nor all PLP at the same time. But it can work on one PLP.

In the other hand, if ATSC-3 does not use the MPEG-TS format anywhere, not even inside a PLP, there is no point using TSDuck in such a context.

EricDeJong, 2023-06-14:
The ATSC 3.0 is defined in the ATSC A/300 specification. In the ATSC 3.0 Link-Layer Protocol (ALP) the IP packets are encapsulated, which are used for transporting the media.
ALP does define the possibility to transport an MPEG-2 transport stream, but we have not heard of anyone using this. More important, this is currently not implemented/supported by the DekTec drivers.
As ATSC 3.0 is mainly used for IP based streams (and no, this is not an MPEG-2 transport stream over IP), and TsDuck is exclusively MPEG-TS, I agree that TsDuck cannot be used here.

lelegard, 2023-06-14:
Thanks @EricDeJong for making this clear.

Currently, the `dektec` input plugin maps all configuration options from DTAPI as command line parameters in the plugin. I must admit that this was done blindly, without knowing exactly what was behind all of them, with the idea that it was better to have too many parameters and options, some of them being useless in an MPEG-TS context, rather than missing an option which could be useful to a user.

Considering this topic, as well as #1222 (using a DTA-2144 in SDI mode), it seems that this was not such a good idea. If an option is present, users try to use it and, if it does not make sense in the TSDuck context, it creates some trouble and misunderstanding.

At the beginning, when Dektec devices were merely ASI input/output and modulators, that was not a bad idea since everything was MPEG-TS. Today, this is no longer the case.

So, I think we should cleanup and remove any option or configuration from the `dektec` input and output plugins when they are not compatible with MPEG-TS mode.

@EricDeJong, would is be possible for you to browse through the documentation of the Dektec plugins ([user's guide](https://tsduck.io/download/docs/tsduck.pdf), page 203 and following) and spot which options are not compatible with MPEG-TS input and output? In short, any input option which is not compatible with `SetRxMode(DTAPI_RXMODE_ST188)` and any output option which is not compatible with `SetTxMode(DTAPI_TXMODE_188)`. Then, I will do the cleanup in the code and documentation.

Thanks for your help.

EricDeJong, 2023-06-21:
The following items in the documentation are not compatible with mode DTAP_RXMODE_ST188 and make no sense for TS-duck. These items can be removed:

Page 203+ Table 6:
All items with ‘SDI’ (SDI-receiver, SDFI-525i/59.94, HD-SDI-xxx etc)
GPS-clock
HDMI
RS422
encoder
phase-noise

IP, demodulator and modulator are used with RxMode DTAP_RXMODE_ST188, provided that the standard supports a transport stream (e.g., modulation/demodulation and IP for ATSC-3 do not)

Page 205, Demodulators options:
Remove –atsc3-bandwidth

Page 208/209 Table 8:
Remove CMMB
(Note ISDB-T would require DTAP_RXMODE_ST204, and additional data in the extra 16 bits)

Page 212:
Remove –cmmb-area-id, –cmmb-bandwidth, --cmmb-pid, --cmmb-transmitter-id

Page 215: in –modulation value
Remove CMMB

lelegard, 2023-06-21:
Thanks @EricDeJong. I will do the cleanup according to your list.

With ISDB-T, do we need to receive 204 bytes? What are the extra 16 bytes? Not a Reed-Solomon code as in DVB I assume. And for ISDB-T output, do we need to provide those 16 bytes? With what inside?

lelegard, 2023-06-24:
The cleanup has been done.

closed 2023-11-26 by github-actions[bot]