tsduck/tsduck #551: Guidance for coding a new standalone tstool

lars18th, 2020-03-11:
Hi @lelegard,

I need your help to develop a new simple tstool. Furthermore, and if you agree with that, my idea is to provide it as an extension to the TSDuck using a PR. However, I need some guidance to make it easy and simple.

Let me to explain first my objective and after the concept of the tool:

- My objective is to develop one standalone EIT _merger_. As a concept the tool will read some “*.bin” file with the “schedule” tables in binary format for a group of transponders. And the tool will connect with two FIFOs. The first for input and the second for output. The idea is to insert EIT pid 18 packets with p/f tables from one specific TS, and read at the output the new pid 18 packets with the merge of current and schedule tables. This input/output will be done in real-time and with 1-to-1 schedule; so for each input packet you will get one output packet. Then it will be possible to use this standalone EIT merger with a tool that is processing a TS. The processing tool then only needs to filter all pid 18 packets, copy them to the input fifo and read from the output fifo the packets to replace the input.

- For implementing this standalone EIT _merger_ I think that the best is to extend the current `TSDuck/sample-app/myexec.cpp.c`. More or less the change will be oriented to read/write the TS from the FIFOs. But I need also a solution for the 1-to-1 schedule mode. I feel that other users will appreciate this change, as with this alternative running mode will be easy to implement some TSDuck processing inside other tools with a weak coupling between them.

What you think?
Any idea to develop such tool?
You agree to create a PR?

Please let me know what you think before I start coding.
Thank you!

lelegard, 2020-03-12:
Why not making it a `tsp` plugin? In the stream, keep EIT p/f, remove EIT schedule (if any), insert new EIT schedule, stealing stuffing if necessary.

The real-time EIT schedule can be inserted using .bin files using a polling mechanism (same as in `-P inject --poll-files` or `-P spliceinject --files`).

If the source is a real-time stream which is processed by `tsp`, you could use UDP:
- The source EIT schedule are extracted using `-P tables --all-sections --ip-udp ... --no-encapsulation --pid ... --tid ...`).
- In your EIT merge plugin, you can receive them the same way as `-P spliceinject --udp`).

BTW, the [transparency of contributions](https://tsduck.io/doxy/contributing.html#transparency) is still required. No anonymous PR will be merged.

lars18th, 2020-03-27:
Hi @lelegard ,

With the current implementation of the FIFOs input/output and the 1-to-1 processing I feel I don't need at all to develop a new tool. My current approach is then to use some scripting to filter the EIT packets, and pass them to the `tsp` processor over the FIFOs. Futhermore, with the `tspcontrol` it will be easy to control the process.

But, now I have an additional problem. For sure more than one solution can be used. And I prefer to ask you before. The problem is how to "insert" the schedule tables if the `tsp` processor will only receive the packets from the current PID 18 ? As the overall architecture is an external processing of the EIT stream (pid 18), we can "replace" packets, but not add them (remember the 1-to-1 processing mode). So we don't have free space to inject the new data and after that merge it with the `psimerge` plugin.

One idea that I have is this: Why not add a new option to the `inject` plugin called `--preload` that reads the `input-file` when the plugin is started? In this case, all table stuff will be preloaded and it can be merged with the current EIT data using the `psimerge` plugin.

What you think? Is this a crazy idea or it has sense?
Do you have a different idea?
Tell me your opinion, please.

lelegard, 2020-03-29:
Use tsp options to insert stuffing for instance.

lars18th, 2020-03-29:
> Use tsp options to insert stuffing for instance.

OK. But I feel this way will create troubles, as 1-to-1 implies one input packet one output packet. Then the inserted stuffing should be removed. In any case, I'll try it.

Thank you!

lelegard, 2020-03-29:
Strict 1-to-1 and insert additional data at the same time?
The two are incompatible. You have to choose.

lars18th, 2020-03-30:
> Strict 1-to-1 and insert additional data at the same time?
> The two are incompatible. You have to choose.

That's the problem. However, I'm exploring for alternatives.