tsduck/tsduck #552: Request: Persistent input pipes

lars18th, 2020-03-17:
Hi @lelegard ,

When using named pipes the problem is when the input file is closed because the writing process ends. Because this, when you're using multiple writes you'll be in troubles. Please, see this example:
```
tsp -d -s -I file fifo.in -P count -a -O drop

echo test.ts > fifo.in
```
Here when the echo ends, the `tsp` process ends.
But with this example this changes:
```
exec 3>fifo.in

tsp -d -s -I file fifo.in -P count -a -O drop

echo test.ts > fifo.in
echo test.ts > fifo.in
echo test.ts > fifo.in
echo test.ts > fifo.in
exec 3>&-
```
In this case, you can use as many writing echoes as you want until you close the fifo file.

Then I suggest if you can add some support for this: `tsp -I file --persistent-pipe fifo-file ...`.
In this case the input pipe is opened and running until the `tsp` process exits.

What do you think? This has sense for you?
I take the example from here: https://unix.stackexchange.com/questions/496810/using-exec-3-to-keep-a-named-pipe-open
The problem with this external approach is that the control is not under the running `tsp` process.

Regards.

lelegard, 2020-03-23:
As usual, I fixed it in a different but more generic way (I didn't like the Unix-specific option about FIFO). No need to introduce a new option. Simply use the existing option `--infinite` (or  `--repeat`).

These options rewind the file when the end of file is reached. Rewinding / seeking is possible only on regular files and using these options resulted in an error with non-regular files such as FIFO's.

I just changed this. When the input file is not a regular one but the user explicitly specifies `--infinite` or  `--repeat`, then the file is closed and reopened when end of file is reached. The result is the one you expected on Unix FIFO's. I suspect many other usages can be found either on Unix or Windows.

lars18th, 2020-03-24:
Hi @lelegard ,

And as usual I'm impressed with you! Thank you. It seems to be a more powerful solution.
I'll try it and I'll share the results.

Regards.

lars18th, 2020-03-26:
Hi @lelegard ,

I'm testing this (#552) and the #553, and I have an small request. Please, try this:
```
$ /tsp -d -s --realtime --initial-input-packets 1 --max-flushed-packets 1 --max-input-packets 1 -I file --infinite ./infifo -O file ./outfifo

$ cat ./outfifo | hexdump

$ cat null.ts > ./infifo
```
In this test all is working as expected. So thank you for last enhancements!
However, one small trouble appears. And I feel you can overcome it.

The problem is this: For each `cat null.ts` I can see one output in the `cat outfifo`. And even if I close the second process (the _reader_) and I restart it, nothing disturbs the `tsp` process. I can continue writing and all continues right. But, if I close the _reader_ and I write, then in this case the `tsp` process detects that it can't write to the output. Then this error appears:
```
* Debug: file: write error on ./outfifo, fd=5, error_code=32
* Debug: file: output thread aborted after 15 packets (15 output)
```

I'm not sure if this problem can be solved. I feel the most simple solution is to discard the writing but continue the processing. Or even repeat it the next time.

What you think?

lelegard, 2020-03-26:
OK, this is a completely different issue. The original one was about the `file` _input_ plugin reading non-regular files. This one is about the `file` _output_ plugin writing non-regular files.

It is not possible to avoid the write error when the reader terminates. It is possible to define a specific option for a retry policy on error, such as closing the output file and trying to reopen / recreate it again and again with a retry delay and some optional maximum number of attempts.

For Unix FIFO, the first immediate retry would solve your issues. With disk on removable storage, this could be a way to swap devices and continue when the device is back. There may be other interesting use cases. But there also may be some traps, so I need to think about it first.

lars18th, 2020-03-26:
Hi @lelegard ,

> OK, this is a completely different issue. The original one was about the `file` _input_ plugin reading non-regular files. This one is about the `file` _output_ plugin writing non-regular files.

Sorry for mixing different issues. However, the test described above describes my use case: an external processing using the TSP tool of the EIT tables, all using only FIFOs to interchange data.

> It is not possible to avoid the write error when the reader terminates. It is possible to define a specific option for a retry policy on error, such as closing the output file and trying to reopen / recreate it again and again with a retry delay and some optional maximum number of attempts.

My use case is simple: one streamer _pass_ all PID 18 packets to the TSP processor using the FIFO input. Then read from the FIFO output and replace the original packets. And I don't want to stop the TSP process if the streamer hangs/close/restart/etc. So I prefer to discard the writing to the output FIFO if the other end of the FIFO is not available.

> For Unix FIFO, the first immediate retry would solve your issues. With disk on removable storage, this could be a way to swap devices and continue when the device is back. There may be other interesting use cases. But there also may be some traps, so I need to think about it first.

Perhaps a simple solution can be to start with the dumb action: if the writing fails, then discard the packet and continue, clearing the status of the output (FIFO) file or reopening it.

Anyway if you found a more powerful solution, then please implement what you want.
Thank you for listening our suggestions.
Regards.

lelegard, 2020-03-27:
Added `--reopen-on-error` in `file` output plugin. Tested OK on Linux with FIFO's.

lars18th, 2020-03-27:
Hi @lelegard ,

I'm testing the FIFO external processing. Now it works very robust, but I found another glitch. And as the problem seems to be related, I prefer to not open a new issue and continue here. Check this example similar to previous:
```
$ /tsp -d -s --control-port 12345 --realtime --initial-input-packets 1 --max-flushed-packets 1 --max-input-packets 1 -I file --infinite ./infifo -O file ./outfifo

$ cat ./outfifo | hexdump

$ cat null.ts > ./infifo
```
The only difference here is that the remote control is enabled. Then it will be possible from another process to use the TSP processing on-demand. In my opinion it's very powerful and easy to use. Thank you implement it. However, I see this:
```
* Debug: file: initial buffer load: 1 packets, 188 bytes
* file: unknown initial input bitrate
* Debug: file: opening output file ./outfifo
* Debug: file: output thread started
* Debug: control commands: found 0 packet processor plugins
* Debug: control commands: setting socket reuse address to 0
* Debug: control commands: binding socket to 0.0.0.0:12345
* Debug: control commands: server listen, backlog is 5
* Debug: file: input thread started
* Debug: control commands: control command thread started
* Debug: file: closing and reopening ./infifo
```
The problem is the `tspcontrol` that it's only enabled *after* the first packet is _writed_ and _readed_. Please, check it. It's very easy to see that the debug lines of `control commands` doesn't appear until the first TS packets traveses the entire chain. I suspect this isn't the expected behaviour, as the control plane requires to be active from the execution start. I'm right?

Regards.

lelegard, 2020-03-27:
In fact, _nothing_ start before the initial load of the buffer (one packet in your case). No plugin thread is started, no other thread (monitoring, control) is started. The successful initial load is the original event which triggers everything.

This was a design choice from the beginning a long time ago. As far as I remember, one of the reasons was that input could be invalid or empty (such as tuning on an empty frequency) and I did not want empty output or log files be created for each invalid input source. There were other reasons too but I don't recall immediately.

In short, we start the whole stuff only when the input has proven to be valid, ie. when some packets can be received.

In most cases, real-time streams or offline files, starting the read is immediate. Most input plugins have a `--receive-timeout` option to make sure that the input does not block forever in case of absence of input.

You have a pathological use case where packets are read one by one, potentially at a very low bitrate. I can provide non-default fine tuning to move the cursors to adapt to this very specific case (all options I recently added). But I cannot change the architecture for it.

lars18th, 2020-03-27:
Hi @lelegard ,

First of all, it's not my objective to change the original architecture. This project is very good. Please, don't try to do huge modifications. I only want to comment my troubles. If you prefer to overcome some of my ideas, I'll accept it. You have by far more experience than me!

The second comment is more specific: I agree with the idea to not start until valid input is not available. I feel this is a good design. However, I don't understand why the first packet requires to be writed for triggering the execution of the plugins. Why not starting when the first packet goes over the INPUT? It's quite annoying to wait until the packet arrives to the OUTPUT. Don't you think that's more logical?

And I have a third comment: From my point of view the specific case of the `tspcontrol` requires a different management. As this option can be used to control/kill/restart/etc the complete `tsp` process, then the listener TCP socket needs to be operative when the process is launched. If this _module_ is not started before the rest, the control plane is not available at some point. And this can generate some troubles. So please, think on this specific behaviour and consider whether this module requires different treatment.

I look forward to your comments.
Regads.

lelegard, 2020-03-29:
> However, I don't understand why the first packet requires to be writed for triggering the execution of the plugins. 

This is not the case. You misinterpret what is happening. I said that all plugins are _initialised_ before starting the processing. Initialised means that command line arguments are interpreted, files are opened, etc. The reason is to avoid starting the processing when something goes wrong (impossible to create or open a file for instance). Once everything is successfully initialised and open, once the initial load is there, everything start.

In your case, you block on _opening_ the FIFO. As long as the open of the FIFO does not complete, the processing is waiting.

The answer is there: `man 4 fifo`: _"The FIFO must be opened on both ends (reading and writing) before data can be passed. Normally, opening the FIFO blocks until the other end is opened also."_

Unix FIFO's are very specific beasts with very unusual behaviour. You must carefully understand how they work before using them. Otherwise you will get bad surprises (well, you already did). 

The only interesting unique property of FIFO's is that they have a message-oriented transmission policy instead of a stream policy for pipes and files. As far as I am concerned, this is the only reason I may use FIFO's. Otherwise, I avoid them because they are too weird.

About you last point, I see no reason to start the control server before starting the processing. You won't see any running plugin anyway and it would make internal synchronisation of tsp more complex. Again, in most cases (in fact in all cases except yours), transmission starts immediately. So, there is no gain for the increased complexity.

lars18th, 2020-03-30:
Hi @lelegard ,

OK. I accept your point of view. And I agree with the idea to not increase the complexity if it's not required. Development in this way increases robustness, which I think is the highest priority for this project (secondly there is efficiency, and thirdly functionality). So, don't worry, I'll try to found for an alternative solution to my use case. My current idea is to launch the `tsp` process from a single script that insert a null packet at start and read it from the output. This will trigger a full starting and from this point the process can work asynchronously in a 1-to-1 working mode. That's all that I need.

Thank you!

lars18th, 2021-02-25:
This is outdated, so I'm closing it.

closed 2021-02-25 by lars18th