tsduck/tsduck #553: Suggestion: reduce MIN_BUFFER_SIZE

lars18th, 2020-03-17:
Hi @lelegard ,

I found a solution for a 1-to-1 processing, however one line of code is required to be changed.
See this example:
```
$ tsp -d -s --realtime --buffer-size-mb 0.0004 --max-flushed-packets 1 --max-input-packets 1 -i -I file fifo.in  -P count -a -O file fifo.out

$ tsp -d -s --realtime --buffer-size-mb 0.0004 --max-flushed-packets 1 --max-input-packets 1 -i -I file fifo.out -P count -a -O dop

$ cat empty.ts > fifo.in
```
Here the `empty.ts` file is one TS packet with NULL stuff. And the buffer size is 2 packets long.

So, for every `cat` you receive one output from the first `tsp` process for _each_ input packet.

However, the problem is this line of code:
https://github.com/tsduck/tsduck/blob/3e6625043995ca36621ab241f6518593c781ab24/src/libtsduck/plugin/tsTSProcessorArgs.h#L73
As it enforces a minimum of 100 packets for the buffer.
But, when I change this value to 188 (or 376, as in fact 2 packets buffer is required in this example) and I recompile then all works.

So, please can you accept to reduce this minimum value?
I know that you want to maintain the code clean and robust. However, the TSDuck is made by engineers for engineers, and in some case this 1-to-1 working mode can be very powerful for testing purposes.

I hope you accept it.
Regards.

lelegard, 2020-03-18:
Normally, ` --max-flushed-packets 1 --max-input-packets 1` should be sufficient to make sure that packets are read one by one and propagated one by one to plugins up to output.

There is no need to have a small buffer. Having a small buffer is a bad idea here. The granularity of the propagation (here 1) impacts the latency while the buffer size impacts the flow control and reliability (packet loss when the inputs are datagrams).

lars18th, 2020-03-18:
Hi @lelegard ,

> Normally, ` --max-flushed-packets 1 --max-input-packets 1` should be sufficient to make sure that packets are read one by one and propagated one by one to plugins up to output.
>
> There is no need to [...]

I do the test _before_ writing this request. If you want, please try the test described. Without changing the value of the `MIN_BUFFER_SIZE` the first `tsp` process never writes until 100 packets are processed. However, when changing the value _every time_ for 1 input packet 1 ouput packet is writed. This can be confirmed using the second `tsp` process.

I feel you don't belive me in this case. So for this reason I suggest to check it.

And regarding the idea of "Having a small buffer is a bad idea here." That's a good argument when doing generic processing. However, TSDuck can be used _also_ for testing/development (_by engineers for engineers_). And in different use cases one engineer can want to use a small buffer, despite the performance impact. So, if you want (and I suggest to do it), please write a WARNING message (even with debug deactivated) when using a value below the _recommended_ buffer size. But, please don't enforce this.

I hope you'll want to reconsider this position for this specific issue. I think you're a very reasonable person.
Regards.

lelegard, 2020-03-24:
OK, so you want 1-to-1 immediate throughput and I do not want to reduce the size of the global buffer too much because of the potential performance problems this will create. So, I implemented another solution which solves your problem without reducing the buffer size.

The proposed options `--max-flushed-packets 1 --max-input-packets 1` solve the 1-to-1 immediate throughput, but only after the initial buffer loading phase. So, I simply added an option `--initial-input-packets` to reduce the initial size of the buffer load instead of reducing the buffer size.

The solution to your problem is now 
~~~
tsp --initial-input-packets 1 --max-flushed-packets 1 --max-input-packets 1 ...
~~~

But note that this should be reserved to very special cases like yours. With such a small initial load, the input bitrate is not yet known when the subsequent plugins are started. This my or may not be a problem, depending on the plugins you use.

lars18th, 2020-03-25:
Hi @lelegard ,

This exceeds my expectations! Thank you. For me a simple WARNING when reducing the size over the minimum value will be sufficient, but relaxing the enforcement. However, your solution is clearly superior.

And the _special_ use case of this option is when the bitrate is not relevant at all. For example, when processing _data_ streams, like EIT. In this case, no synchronized streams are involved, like audio or video. So we don't care about the bitrate.

Thank you for implementing this solution. I'll check it soon.
Regards.

lars18th, 2020-03-26:
Hi @lelegard ,

I close this issue as the discussed solution is implemented. :+1: :tada: 

Thank you!

closed 2020-03-26 by lars18th