tsduck/tsduck #554: DTA-2162 problem with VLANs

rezna11, 2020-03-19:
Hello, I configured VLANs for DTA-2162 and tried to use TSDuck to read a multicast but it's not working. When using StreamXpress, it works, so the stream is reaching the Windows machine. The problem with TSDuck is that not only it won't read the TS, it also gets stuck when trying to stop (Ctrl+C). This is my command and output.

Command:
`tsp -v --debug -I ip -l 10.23.11.91 10.23.11.243@232.160.3.6:5000 -O play`

Output:
```
tsp: Debug: debug level set to 1
* Debug: debug level set to 1
* Debug: trying to load "C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll"
* Debug: symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll
* Debug: trying to load "C:\Program Files (x86)\TSDuck\bin\tsplugin_play.dll"
* Debug: symbol tspNewInput not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_play.dll
* Debug: symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_play.dll
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 10.23.11.91:5000
* ip: joining multicast group 10.23.11.243@232.160.3.6:5000 from local address 10.23.11.91
```
and it gets stuck at this
`* tsp: user interrupt, terminating...`

I tried without SSM, with the same result.

10.23.11.91 is the VLAN IP address on the Windows PC
10.23.11.243 is the VLAN IP address on the encoder

When VLANs are not used (so no virtual network adapters are created), TSDuck is able to read multicasts from DTA-2162.

lelegard, 2020-03-21:
Hello,

This hard to tell. I do not have any Dektec device with Ethernet port to test. I do not even have any VLAN-aware switch at home.

The `ip` plugin uses the standard socket interface to talk to the network driver. It is not aware of the type of hardware NIC or port driver. Until now, it has worked on Windows, Linux and macOS using the standard socket interface on all standard NIC's. Dektec devices are very specialised NIC's. I agree that they should behave exactly the same way as standard NIC's, thanks to the Dektec drivers. But maybe not.

Maybe there is a problem in TSDuck but I do not have the hardware to reproduce and analyze the problem. 

More probably, this is a Dektec driver or configuration issue. The fact that `tsp` hangs after Ctrl-C seems to confirm this. When Ctrl-C is hit, a software interrupt is triggered in a separate thread which, first, prints "user interrupt, terminating..." and then terminates all plugin threads. At a higher level, `tsp` is waiting for all threads to cleanly terminate before exiting. In the case of the `ip` input plugin, interrupting the input means closing the socket from the interruption thread. As a consequence, the current input is interrupted by the OS which tells the device driver to abort the input operation with an error status. Here, it seems that the device driver (here the Dektec driver for the DTA device) never terminates the input operation. The input thread never terminates and `tsp` indefinitely waits for it. If you try a second Ctrl-C while `tsp` hangs, this one will not be handled by `tsp` and the process terminates in a dirty way _after all kernel operations complete_. So, if `tsp` does not terminate after the second Ctrl-C, then this is definitely a hanging driver problem.

Concerning StreamXpert, this is a Dektec tool which is designed for Dektec devices. So, it probably directly uses the DTAPI to access the network functions instead of the standard OS network interface over Dektec drivers. Using DTAPI, I assume that StreamXpert has a closer control over the Dektec NIC using proprietary functions. So, this is not surprising that StreamXpert and TSDuck behave differently.

rezna11, 2020-03-21:
Thanks for your input. It doesn't matter how many times I try Ctrl+C, it hangs forever. So basically this means it would be better to access DTA-2162 using DTAPI but from the TSDuck manual I understand that it's not supported. DekTec VLANs are probably too specific, which is why their virtual VLAN interfaces only work with DTAPI.

lelegard, 2020-03-23:
Hanging after several Ctrl-C means that this is a hanging driver problem. TSDuck is just a userland application and it catches only the first Ctrl-C. The second Ctrl-C is directly handled by Windows which abruptly kills all userland processing. If the process hangs, this is the symptom of a hanging kernel activity.

Most probably, the problem does not come from the network upper layers (from the Windows kernel). Otherwise, we would see that with other devices. So, this is probably a Dektec driver issue. You should report that to Dektec. They cannot just say that you should use Dektec tools. The Dektec device is a network interface card and should correctly behave as such. You may mention that the problem is seen with TSDuck, at least some software people at Dektec are aware of it.

lelegard, 2020-03-24:
Concerning Ctrl-C, it seems that this is a general Windows issue with network drivers. The behaviour I described is correct on Linux. But on Windows, when an application is stuck waiting for UDP datagrams, the application never stops with several Ctrl-C. Weird... Just another Windows weirdness....

So, the Ctrl-C issue is probably not a Dektec issue. The VLAN issue, I cannot explain, having no such card.