tsduck/tsduck #556: Accessing API in C

tarelda, 2020-03-26:
I am considering writing tsduck apps using python, but as far as I know simple binding with ctypes works only with C interface. Is there any way to use tsduck in C or I just should build my own?

lelegard, 2020-03-26:
Hi @tarelda,

TSDuck is developped in C++ and heavily relies on the C++ features to enforce clean object isolation, clean resources management and robustness. It is not possible to use the TSDuck library from C.

Just a historical note: I wrote the very first ancestor of TSDuck in 2005 using C. I soon realized it was a huge mistake and I scrapped it in 2007 and rewrote it from scratch in C++. To be clear, it is possible to write good or bad code in either C or C++ but only C++ provides mechanisms which can be used to _enforce_ a good and robust usage of an API.

I have never used Python interface to native code. But I would be surprised if an object-oriented language such as Python would need a prehistoric language such as C as an intermediate layer to access another object oriented language such as C++. In the past, I used JNI, the Java Native Interface. Java is able to call either C or C++ but the C++ interface is way much better than the C one. I let experts comment on the equivalent for Python.

As a matter of fact, having a Python interface to the TSDuck library is something I would have liked to do for a while. I never did it because of lack of time and skills. If you start something, it would be good to share it so that others could benefit from it.

lelegard, 2020-03-26:
[SWIG](http://www.swig.org/) is a binding generator for C++ to Python (and other scripting languages). It looks a bit old but its development is still active (new versions, new commits).

tarelda, 2020-03-26:
> Hi @tarelda,
>
> TSDuck is developped in C++ and heavily relies on the C++ features to enforce clean object isolation, clean resources management and robustness. It is not possible to use the TSDuck [...]

In Python there are generally two ways of binding remote libraries. 
First approach is using ctypes. It is basically mapping of shared library functions into Python code.
Second bases on writing custom extension. This is more cumbersome, but only way that support C++ out of the box. 

Honestly I'd rather advise against using generator, because at least in case of ns-3 it made them non-pythonic thus unnatural to use. So I decided to write plain C++ instead of using them.

lelegard, 2020-10-13:
Hi @tarelda,

Better late than never, I have added a few [Python bindings](https://github.com/tsduck/tsduck/tree/master/src/libtsduck/python) to some high-level features of TSDuck with some [sample application code](https://github.com/tsduck/tsduck/tree/master/sample/sample-python).

This is still experimental and I am not sure how to document and install it. The sample applications have been tested on Linux / Python 3.8 and macOS / Python 3.7.

tarelda, 2020-10-13:
Hello @lelegard,
I highly appreciate your efforts. I will try to play with them this week.

What do you mean by not sure how to install it?  IMO regarding documentation this only needs similiarly to cpp description of available classes, methods and arguments.

After quick skim through the code I have few basic questions:
1. As far as I understand these are neither thread-safe nor "asyncio"-safe ? Preffered way of launching tsprocessor is by using separate process? 
2. I assume reports here are text messages produced by plugins. So when I want to interactively monitor processing and analyze output I have to overload AsyncReport log function?
3. Why not directly plugin hybrid (python/cpp) implementation of plugin event monitoring object?  Or using reports was just means of "testing the waters"?

lelegard, 2020-10-13:
> As far as I understand these are neither thread-safe nor "asyncio"-safe ?

The bindings directly interface the TSDuck C++ library. The TSDuck library is thread safe and all plugins run in their respective threads. I am not a Pythonista and not familiar with asyncio.

> Preffered way of launching tsprocessor is by using separate process?

No, the TS processing session runs in the same process as the Python application, in separate threads (one per plugin).

The Python class `ts.TSProcessor` is exactly equivalent to the C++ class `ts::TSProcessor`. In fact, the former is just a wrapper which calls the latter.

> I assume reports here are text messages produced by plugins.

Using `Report` is just for message logging. The TSDuck C++ library heavily relies on the class [`ts::Report`](https://tsduck.io/doxy/classts_1_1_report.html) and all its subclasses (on files, strings, stderr, asynchronous, etc). This was the only requirement for `ts::TSProcessor` which needs one thread-safe (and non-intrusive) instance of `Report` to log messages. This is why I added Python wrappers to some C++ `Report` subclasses. The `AsyncReport` is the one that logs messages asynchronously in a multi-thread environment. It is the appropriate one for a TSP session to avoid the usual "error avalanche" effect.

So, just consider it as a support for running a TSP session, not as a first-class Python object.

> So when I want to interactively monitor processing and analyze output I have to overload AsyncReport log function?

The methods of the `Report` Python classes are just here to allow the Python applications to inject messages in the same `Report` objects as the TSDuck library. Overloading the Python methods won't change the behaviour of the C++ classes since the C++ virtual methods will be unaffected by the Python overloading.

> Why not directly plugin hybrid (python/cpp) implementation of plugin event monitoring object? Or using reports was just means of "testing the waters"?

The `Report` classes are essentially designed to process log messages in a unified way. There is a [C++ interface for plugin-generated events](https://tsduck.io/doxy/classts_1_1_plugin_event_handler_interface.html) but it is not widely used and not accessible to Python anyway.

I can think of another subclass of `ts::Report` and a corresponding Python wrapper class to handle the log message as callbacks to the Python application. But I have to improve my Python first. 

Remember that TSDuck is essentially a large C++ library with almost 600 classes to handle all aspects of Digital TV. It will never be translated in Python (not by me at least) and only a few high-level TSDuck features can be wrapped in Python for a reasonable cost.

I have learned Python a long time ago and, while I find it a perfect scripting language for quick prototyping, I consider it as an awful language in terms of software engineering for large applications. None of the C++ features I rely on to build the robustness of TSDuck can be found in Python. For the anecdote, I come back a little bit to Python only to support my son's efforts since he learns it at school 😄

lelegard, 2020-10-13:
Forgot these:

> What do you mean by not sure how to install it?

I mean where to install the Python module in the file system and how to make it available for all applications during the TSDuck installation process (rpm, dpkg, homebrew and Windows installers).

> IMO regarding documentation this only needs similiarly to cpp description of available classes, methods and arguments.

I was refering to doxygen vs. Python doc strings. The TSDuck library is documented using doxygen and I wonder how I can integrate the Python bindings there.

lars18th, 2021-03-05:
Hi,

This issue can be closed as the Python binding is already included?

closed 2021-03-05 by lelegard