tsduck/tsduck #558: Suggestion: PSIMERGE individual mode

lars18th, 2020-03-30:
Hi @lelegard ,

Regarding my specific use case described in #551 , I have one suggestion of an _small_ enhancement that can solve the problem.

The objective is this: using the 1-to-1 working mode I want to insert all PID 18 packets, and merge them with binary tables with schedule information. At time is quite simple to complete almost all tasks: I can filter incoming packets that are not p/f, I can convert the binary tables to a TS, etc. However, I miss the option to "merge" the tables using the 1-to-1 working mode.

Proposal: My suggestion is to add a new parameter to the `psimerge` plugin called `--convolution` or `--unique` or `--smooth` (please, select the best name that you prefer). The idea is not to merge _two_ streams, but to merge from _one_ stream. When this have sense? In my case I can _insert_ in the stream the schedule information one time. This is easy translating the binary tables to a TS, and inserting all of these packets in the `tsp` process at start. Futhermore, as you know how many packets are inserted, you can read the same number of packets at output. Then before starting the realtime processing you only need to complete a simple initialization task. That actually fills the internal structures of the library with the schedule data. After I can insert in realtime all PID 18 data with only the p/f data. And I expect to read from the output the "merged" version of all tables. But at time this is not the case. However, if the `psimerge` plugin can work in this way (aka using only one stream) then I can achieve my objective.

What do you think? Does this improvement make enough sense to be implemented, or do you have another suggestion for achieving that goal?

lelegard, 2020-03-30:
> The idea is not to merge two streams, but to merge from one stream. 

Merging things requires more than one thing.

lars18th, 2020-03-31:
> > The idea is not to merge two streams, but to merge from one stream.
> 
> Merging things requires more than one thing.

Ummmm.... "schedule EIT" + "p/f EIT" are two things. What I want to do is to _insert_ the schedule data only _one_ time, and the present/following as regular. Then, in the output I want to obtain the repetition of all EIT tables. You know how to achieve this objective with the current implementation?

lars18th, 2020-04-01:
Hi @lelegard ,

In my current tests I'm trying to mark with a different label the schedule tables. And then merge them with the `psimerge` plugin. However, it doesn't work. And in any case, I think that it has sense to include the option to "smooth" or "re-merge" or "convolve" multiple EIT tables in the same PID 18 stream. Why you mention that is not possible to merge them?

Regards.

lars18th, 2020-04-07:
Hi @lelegard ,

I don't want to be a bother! But I need your help. I need a method to "injest" Schedule EIT tables _only_ 1 time, and p/f EIT tables in realtime, and obtain at the output the "merge" of them.

Any idea to achieve this objective?
Thank you for your effort with this impresive project!

lars18th, 2020-04-15:
Hi @lelegard ,

I'm still waiting for an idea from you.
Please comment about it.
Thank you.

lars18th, 2020-04-18:
Hi @lelegard ,

I don't know how to target this. Please, any idea?

lelegard, 2020-04-18:
I have no idea currently, except developing a dedicated plugin.

Mixing real-time tables and stable scheduled tables does not fit in the current set of TSDuck classes. This is why, for instance, the plugin `zap` can cleanup the list of services in DVB systems but not in ATSC systems. In DVB, each reserved PID is dedicated to either stable tables (SDT/BAT for instance) or real-time tables (TDT/TOT). In ATSC, the PSIP PID 0x1FFB contains a mixture of both and it is very difficult to update stable scheduled tables (such as the xVCT for the service names) in the middle of real-time tables such as the STT (ATSC equivalent of TDT).

The EIT's have always been special beasts : mixture of scheduled and p/f, long sections but incomplete tables and avalanche effect at midnight because of the usage rules on section numbers.

Managing EIT's in all possible ways probably deserves a set of dedicated plugins. But I have no plan to work on this for now.