tsduck/tsduck #559: TR 101 290 analysis

chache35, 2020-03-30:
Hello, 

Do you plan to add a TR 101 290 analysis to TsDuck in the future ?
(full analysis that includes priorities 1/2/3 monitoring)

Best regards.

lars18th, 2020-03-30:
Hi @chache35 ,

It will be desirable. However, quite complex to implement it _inside_ the library. Perhaps it can be an independent tool based on TSDuck.

Meanwhile, I'm using this: https://www.promaxelectronics.com/ing/products/educational-trainers/TSAnalyser/free-software-for-transport-stream-analysis/

And this text only tool for console: https://github.com/Cinegy/TsAnalyser

I hope it helps.
Regards.

chache35, 2020-03-30:
Thank you.
I will have a look to this tools.
Sure it could be great to be able to run such analysis directly over TsDuck.

Regards