tsduck/tsduck #561: alpine compilatio error

enricovittorini, 2020-04-06:
```
uname -a
Linux 3e1aeb44e239 3.10.0-1062.18.1.el7.x86_64 #1 SMP Tue Mar 17 23:49:17 UTC 2020 x86_64 GNU/Linux
```

 ```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-alpine-linux-musl/9.2.0/lto-wrapper
Target: x86_64-alpine-linux-musl
Configured with: /home/buildozer/aports/main/gcc/src/gcc-9.2.0/configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --build=x86_64-alpine-linux-musl --host=x86_64-alpine-linux-musl --target=x86_64-alpine-linux-musl --with-pkgversion='Alpine 9.2.0' --enable-checking=release --disable-fixed-point --disable-libstdcxx-pch --disable-multilib --disable-nls --disable-werror --disable-symvers --enable-__cxa_atexit --enable-default-pie --enable-default-ssp --enable-cloog-backend --enable-languages=c,c++,objc,fortran,ada --disable-libssp --disable-libmpx --disable-libmudflap --disable-libsanitizer --enable-shared --enable-threads --enable-tls --with-system-zlib --with-linker-hash-style=gnu
Thread model: posix
gcc version 9.2.0 (Alpine 9.2.0)
```

```
# g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-alpine-linux-musl/9.2.0/lto-wrapper
Target: x86_64-alpine-linux-musl
Configured with: /home/buildozer/aports/main/gcc/src/gcc-9.2.0/configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --build=x86_64-alpine-linux-musl --host=x86_64-alpine-linux-musl --target=x86_64-alpine-linux-musl --with-pkgversion='Alpine 9.2.0' --enable-checking=release --disable-fixed-point --disable-libstdcxx-pch --disable-multilib --disable-nls --disable-werror --disable-symvers --enable-__cxa_atexit --enable-default-pie --enable-default-ssp --enable-cloog-backend --enable-languages=c,c++,objc,fortran,ada --disable-libssp --disable-libmpx --disable-libmudflap --disable-libsanitizer --enable-shared --enable-threads --enable-tls --with-system-zlib --with-linker-hash-style=gnu
Thread model: posix
gcc version 9.2.0 (Alpine 9.2.0)
```

error:
```

[CXX] base/network/tsUDPSocket.cpp
base/network/tsUDPSocket.cpp: In member function 'ts::SocketErrorCode ts::UDPSocket::receiveOne(void*, size_t, size_t&, ts::SocketAddress&, ts::SocketAddress&, ts::Report&)':
base/network/tsUDPSocket.cpp:566:73: error: zero as null pointer constant [-Werror=zero-as-null-pointer-constant]
  566 |     for (::cmsghdr* cmsg = CMSG_FIRSTHDR(&hdr); cmsg != nullptr; cmsg = CMSG_NXTHDR(&hdr, cmsg)) {
      |                                                                         ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [../../Makefile.common:905: release-x86_64/tsUDPSocket.o] Error 1
make[1]: *** [Makefile:43: default] Error 2
make: *** [Makefile:59: default] Error 2
```

lelegard, 2020-04-06:
Hi,

I cannot reproduce this on my Alpine VM.
The current TSDuck repo compiles correctly and passes all test suites.

My configuration:
~~~
$ cat /etc/alpine-release
3.10.4
$ uname -a
Linux vmalpine 4.19.98-0-virt #1-Alpine SMP Thu Jan 23 11:06:31 UTC 2020 x86_64 GNU/Linux
$ gcc --version
gcc (Alpine 8.3.0) 8.3.0
~~~

As you know, I am not an expert in Alpine. My VM is just a regular installation, probably old. But your kernel is older and your gcc is newer. So, it does not like like neither an older nor a newer version. How did you get there? Is your configuration a supported combination?

lelegard, 2020-04-06:
I can reproduce the error after upgrading to the latest version: Alpine 3.11.5, kernel 5.4.27 and gcc 9.2.0. I will have a look. Probably another oddity of musl libc headers.

lelegard, 2020-04-06:
I confirm that this is an invalid definition in musl libc headers (not found in glibc). I have added a workaround for it in commit 630344ff467f45e431414582f20d7f6260964f29.

I hate musl libc. Those guys don't know how to write code which compiles with all compiler warnings on.

closed 2020-04-16 by lelegard