tsduck/tsduck #562: Labeling by PES Stream ID

synologic, 2020-04-12:
Good evening,

I went thru your extensive documentation (good job there) and i'm trying to figure out if it's possible to filter or set a label based on PES Stream ID.

The scenario would be that, when doing input switching from various sources, i would like to be able to pass *any* video PIDs and any audio pids (would be awesome if it would be possible to select based on language descriptor too) for later renumbering.

It happens that there are several sources of the same content, coming over different mediums, which have different PIDs.

This of course can be done by reading the output of analyzer plugin and having an external process trigger labeling, but would be nice to have this built in.

Any suggestions ?
Thanks

lelegard, 2020-04-13:
Hi,

This is currently not possible. It would be possible to add an option to the `filter` plugin to detect a given stream id on clear PES streams and then let the corresponding PID pass (or set on a label on it).

Selecting by language is different since it needs the filtering and analysis of PMT's while the `filter` plugin is based on raw TS packet content only. Filtering by language is already implemented in `zap` plugin (but this plugin serves a different purpose).

lelegard, 2020-04-13:
Hi,

I just added option `--stream-id` to plugin `filter` because it was trivial to do in a reliable way.

For PID selection based on PMT properties (audio, video, language), this will need more time to do it in a reliable way, especially in use cases like yours. When the TS structure changes (maybe because of a input switch), then you need to reconsider everything. But, in a filter, you are not aware that the TS has changed. With some luck, you will see new PMT's. But what happens with previously selected PID's?

Consider the following example:

- You select all audio PID's
- Receiving the PMT of service 1, PID 100 is audio, you select it.
- The input TS switches
- Receiving the PMT of service 2, PID 200 is audio, you select it. And by the way PID 100 is video in the new TS.

After input switch, you accidentally keep PID 100 which has become video.

Worse, if your two input TS have different but very stable structures, the versions of the PSI sections will remain stable. You will probably have version 0 on both PAT's, or the same version number, whatever it is.

What happens is:

- On PID 0, receive a PAT with version 0 (or N).
- Analyze it, get all PMT PID's, start analysing all PMT's.
- The input TS switches. The new TS has a different structure (services, PMT PID's) but the PAT remains on PID 0 of course.
- Since the PAT version has not changed, then you won't receive the updated PAT (an updated PAT with the same version number, which is illegal)
- So, you won't get the new service structure _at all_.

Generally speaking, switching input is dangerous when you want to analyse the structure of the TS downwards because the standard MPEG rules on sections are broken.

As a conclusion, filtering PID's based on PMT properties is feasible on a legal MPEG stream, following all standard rules. But it may stop working properly after an input switch.

synologic, 2020-04-14:
Hi Thierry

The reason for relabeling would be to renumber the PIDs after the selection, so basically the scenario would allow generation of PSI while grabbing the useful ES at the input and renumbering them. Mostly applied for SPTSs, this way you don't worry about any tables having to change as you don't really change the structure of the stream.

Hope that makes sense :)

I'm going to start working on an UI for your tools ... :)

lelegard, 2020-04-14:
> generation of PSI while grabbing the useful ES at the input 

If each input has its own tsp processing of the TS producing, letting pass only the ES of one service without PMT, then you can identify the streams from the stream id. However, in the case of multiple audio, you cannot identify the language since it comes with the PMT.

So either, you let the PSI in but you may not detect the input switch and incorrectly identify PID"s.
Or you only let the ES in, without PSI, but you cannot identify the language of the audio PID's.

> I'm going to start working on an UI for your tools ... :)

Do you plan to make it open source ?

synologic, 2020-04-14:
> If each input has its own tsp processing of the TS producing, letting pass only the ES of one service without PMT, then you can identify the streams from the stream id. However, in the case of multiple audio, you cannot identify the language since it comes with the PMT.
> 
But if labels pass thru fork, once an ES labelled can be easily identified downstream as being "audio language XXX" ... that was my initial reasoning

> > I'm going to start working on an UI for your tools ... :)
> 
> Do you plan to make it open source ?

Yeah, if i ever finish it or have to keep working on it :)

lelegard, 2020-04-14:
> But if labels pass thru fork

They don't. Forking a process from `tsp` passes a valid TS. The command may be anything reading a TS, a media player for instance. The plugin `play` simply locates a media player and then behaves as the `fork` plugin. Similarly, piping the input or output of `tsp` or `tsswitch` simply uses the `file` plugin on the standard input or output.

Labels are internal metadata to `tsp`. They are not part of the TS data.

lars18th, 2020-04-15:
Hi,

My 5 cents:

- This use case (enhanced detection) can be solved if it can be detected _any_ change in the running PMT/PAT. So checking if new PMT and/or PAT comes, then the filtering can be reseted. This will be very easy if the Trasnport ID is consistent and not repeated, but even in this case, any other change detected can be used as a trigger.

- Labels can be passed using a _simple_ hack: The 0x47 initial TS packet header can be used to add some inter-process data. In fact 254 values can be used as a offset. And when using this hack both parts, the sender and the receiver, requires to agree to use this offset. I feel it will be easy to add options to `fork`, `tsp` and `tsswitch` to support this **hacked mode**.

I hope it helps.