tsduck/tsduck #565: Tsanalyze JSON output?

elettrone1, 2020-04-23:
Hi,
is it possible to output tsanalyze results in JSON format?

Please help me I don't have C++ skills.

Also, I would like to use tsanalyze or whatever tool in TSDuck to analyze a http or rtsp stream, since I need to read from a SAT>IP server. 

Thanks!

lelegard, 2020-04-23:
Hi @elettrone1 

> is it possible to output tsanalyze's results in JSON format?

There is no JSON output. However, there is a _normalized_ output format (option `--normalized`) in tsanalyze and other TSDuck tools which is specifically made for automation.

JSON is too complicated to be analyzed by simple shell scripts. It cannot be analyzed text line by text line in a shell script and needs a parser in an application. Well, you can now use the `jq` tool but it did not exist 15 years ago when TSDuck was started and tsanalyze and its normalized output format formed the initial core of TSDuck.

The normalized format is very simple to use and was designed to be used in shell scripts with `grep` and `sed` as sole tools. See the reference in the TSDuck user's guide in the tsanalyze chapter. Also read a few sample scripts in the example chapter.

> Also, I would like to use tsanalyze or whatever tool in TSDuck to analyze a http or rtsp stream, since I need to read from a SAT>IP server.

HTTP and RTP are possible inputs. See the `http` and `ip` input plugins. RTSP is not supported as such because it is a playout control protocol and the TS itself is sent over RTP. TSDuck just needs a TS as input.

lelegard, 2020-04-23:
> Please help me, I don't have C++ skills.

Also note that I made the effort to write a 370-page user's guide to make sure that no programming skill was required to use TSDuck. So please RTFM 😃

lars18th, 2020-04-23:
Hi @elettrone1 ,

> Also, I would like to use tsanalyze or whatever tool in TSDuck to analyze a http or rtsp stream, since I need to read from a SAT>IP server.

I have some skills with SAT>IP. To use it with TSDuck to _analyze_ only, I recommend to use the HTTP input plugin of TSDuck with a SAT>IP server that supports HTTP streaming (good one is minisatip). And if you need a more robust solution, then install the vtuner module for satip that emulates a DVB tuner in any Linux kernel. That's preferable to use the SAT>IP dialect of RTSP with TSDuck.

I hope it helps.
Regards.

elettrone1, 2020-04-23:
> > Please help me, I don't have C++ skills.
> 
> Also note that I made the effort to write a 370-page user's guide to make sure that no programming skill was required to use TSDuck. So please RTFM

You are right, but please understand the user who is looking for a fast solution and maybe doesn't have the time to read 370 pages.

But of course I'll try to find what I need there, if you don't have the time to make a working example!

Thanks!

elettrone1, 2020-04-23:
> Hi @elettrone1 ,
>
>
> I have some skills with SAT>IP. To use it with TSDuck to _analyze_ only, I recommend to use the HTTP input plugin of TSDuck with a SAT>IP server that supports HTTP streaming (good [...]

Yes, I am using minisatip. Do you have a working example for the input plugin?

I am using this now

curl -s -m 3 http://localhost:8080/... | tsanalyze --normalized

but I am sure it's not the best way.

Thanks for help.

lelegard, 2020-04-23:
> You are right, but please understand the user who is looking for a fast solution and maybe doesn't have the time to read 370 pages.

I wouldn't read them either. This is a matter of method. A reference manual is not a novel. Never read it sequentially. When you know what you want, just read the table of contents, this is by far the most important part of a technical manual. Then, you spot what you want and directly jump into the reference section you want.

I have always been frustrated by the total absence of documentation for complex software such as ffmpeg or openssl that I used a lot and lost a huge amount of time looking for just the right set of options to do something. With a reference manual, I would have found what I want in a matter of minutes. This is the reason why I would never release software without a comprehensive reference manual. I want users to be independent.

lelegard, 2020-04-23:
> curl -s -m 3 http://localhost:8080/... | tsanalyze --normalized

Alternatively:
~~~
tsp -I http http://localhost:8080/... -P analyze --normalized -O drop
~~~

lelegard, 2020-04-23:
For examples on how to use the normalized output of tsanalyze (or the analyze plugin for tsp), see sections 5.2.12, 5.2.12 and 5.2.14.

12 years ago, I wrote a shell script which performed the complete monitoring of all possible parameters in the French DTTV network. Whenever something was out of spec, the script sent me an email. This script was run from time to time using the crontab of a Linux machine.  For the budget of a basic desktop PC and a ~70€ DVB-T tuner, I could get near to similar features as over-expensive monitoring systems.

lars18th, 2020-04-24:
Hi @elettrone1 ,

> Yes, I am using minisatip. Do you have a working example for the input plugin?

Yes, the same as pointed by @lelegard :
`tsp -I http "http://minisatip-server:8080/?src=1&freq=11347&pol=V&msys=DVBS2&sr=22000&pids=0,17,16,6600,6610,66,20,6630" -P analyze --normalized -O drop`
Try it with and witout `--normalized` (without is for human, the with is for machines)

> I am using this now
> curl -s -m 3 http://localhost:8080/... | tsanalyze --normalized
> but I am sure it's not the best way.

Yes, is not the best as TSDuck can handle HTTP natively (HLS too).

> For the vtuner...I think it's not possible to use it on Linux (which I am actually using).
> I would like a lot to use a vtuner on Linux...do yo know how to do it?

vtuner works _only_ in Linux. I'm using it for years without a trouble (with a custom patch to increase the number of useable pids until 64). See this: https://github.com/catalinii/minisatip/wiki/vtuner-pids

My suggestion is simple:
- For _simple_ analyzing (in the sense of short captures without pid changes) then use the native HTTP input of TSDuck.
- For _complex_ analysis (freq changes and/or pid changes) then use the "vtuner" solution.

In any case I don't want to request to @lelegard to create a SAT>IP input plugin. I never suggest reinventing the wheel. However, if someone else wants to implement such a plugin, then they are welcome to do so.

> Thanks for help.

:wink:

lars18th, 2020-04-24:
> 12 years ago, I wrote a shell script which performed the complete monitoring of all possible parameters in the French DTTV network. Whenever something was out of spec, the script sent me an email. [...]

Today a recommend these two options for low cost monitoring:
- A raspberry Pi 4 with a _good_ USB tuner.
- A barebone PC with a network DVB-T/T2 tuner (HDHR or SAT>IP).

The second is the most robust and professional.

elettrone1, 2020-04-24:
> Hi @elettrone1 ,
> 
> > Yes, I am using minisatip. Do you have a working example for the input plugin?
> 
> Yes, the same as pointed by @lelegard :
> 
> ```
> tsp -I http \
>  "http://minisatip:8080/?src=1&freq=11347&pol=V&msys=DVBS2&sr=22000&pids=0,17,16,6600,6610,6620,6630" \
>  -P analyze --normalized -O drop
> ```
> 
> Try it with and without `--normalized` (without is for humans, the with is for machines)
> 
> > I am using this now
> > curl -s -m 3 http://localhost:8080/... | tsanalyze --normalized
> > but I am sure it's not the best way.
> 
> Yes, is not the best as TSDuck can handle HTTP natively (HLS too).
> 
> > For the vtuner...I think it's not possible to use it on Linux (which I am actually using).
> > I would like a lot to use a vtuner on Linux...do yo know how to do it?
> 
> vtuner works _only_ in Linux. I'm using it for years without a trouble (with a custom patch to increase the number of useable pids until 64). See this: https://github.com/catalinii/minisatip/wiki/vtuner-pids
> 
> My suggestion is simple:
> 
> * For _simple_ analyzing (in the sense of short captures without pid changes) then use the native HTTP input of TSDuck.
> * For _complex_ analysis (freq changes and/or pid changes) then use the "vtuner" solution.
> 
> In any case I don't want to request to @lelegard to create a SAT>IP input plugin. I never suggest reinventing the wheel. However, if someone else wants to implement such a plugin, then they are welcome to do so.
> 
> > Thanks for help.

Could you PLEASE give me the source code that you used for building vtuner?
The source code linked in that page doesn't exist anymore.

https://code.google.com/archive/p/satip/

Also, if I try building the module with some forked sources I get millions of errors.
I am using Debian v4.9.0-12-amd64 BUT I AM READY TO SETUP THE SAME SYSTEM THAT YOU ARE USING, to be able to use vtuner.

Thanks!

lars18th, 2020-04-24:
Hi @elettrone1 ,

I updated the wiki to point to the current github mirror: https://github.com/Leatherface75/satip

Regarding my environment, I'm not sure about recent kernels but for this I still use the Debian 7 x64 (debian 3.2.0-4-amd64). Even @lelegard doesn't like to maintain ancient kernels with old gcc versions, the current TSDuck still compiles in this machine when using the GCC 4.9.

elettrone1, 2020-04-24:
> Hi @elettrone1 ,
>
> I updated the wiki to point to the current github mirror: https://github.com/Leatherface75/satip
>
> Regarding my environment, I'm not sure about recent kernels but for this I still [...]

I get this error while compiling vtuner_tweaks

/home/debian/satip/vtuner_tweak/vtunerc_proxyfe.c:445:18: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .get_frontend = dvb_proxyfe_get_frontend,

It is impossibile to build this stuff, I don't know how you did it, maybe kernel 3.2.0.4 is a lot simpler!

elettrone1, 2020-04-24:
> > curl -s -m 3 http://localhost:8080/... | tsanalyze --normalized
> 
> Alternatively:
> 
> ```
> tsp -I http http://localhost:8080/... -P analyze --normalized -O drop
> ```

It doesn't have a timeout...doesn't produce an output until I stop with CTRL-C.
Is it possibile to use a timeout flag?

lars18th, 2020-04-24:
Hi @elettrone1 ,

> > > curl -s -m 3 http://localhost:8080/... | tsanalyze --normalized
> > 
> > 
> > Alternatively:
> > ```
> > tsp -I http http://localhost:8080/... -P analyze --normalized -O drop
> > ```
> 
> It doesn't have a timeout...doesn't produce an output until I stop with CTRL-C.
> Is it possibile to use a timeout flag?

Please, read the documentation. What you need is to add the `until` plugin. In the form of:
`tsp -I http ... -P until -s 10 -P analyze --normalized -O drop`.

The TSDuck has a _lot_ of useful stuff. You only need to search for it in the documentation. :wink:

elettrone1, 2020-04-24:
> Hi @elettrone1 ,
>
>
> Please, read the documentation. What you need is to add the `until` plugin. In the form of:
> `tsp -I http ... -P until -s 10 -P analyze --normalized -O drop`.
>
> The TSDuck has a [...]

Thanks for your patience.

Do you know the bitstream lib?

https://www.videolan.org/developers/bitstream.html

If you build it, in the examples folder you will find dvb_print_si that will analyze ts with XML output.
Ok you don't like JSON, but it could be useful to have an XML output.
The so called "normalized" format is an expression you use to denote your own format, isn't it?
I don't find it elsewhere as a standard.

Thanks!

lelegard, 2020-04-24:
> The so called "normalized" format is an expression you use to denote your own format, isn't it?
I don't find it elsewhere as a standard.

Normalized does not mean any kind of standard. It means that the output structure is stable, will not change and is easy to process in scripts. This is unlike "human-friendly" output that is more complex to parse and changes from time to time to improve the reported data.

> Ok you don't like JSON, but it could be useful to have an XML output.

Don't misinterpret me, I like JSON and XML (and even YAML if you need more). TSDuck includes JSON and XML processors as C++ classes.

At the time, I just needed some stable output format which was extremely easy to process in scripts with classical basic line-oriented text processing tools (namely `sed`  and `grep`). If I had to rewrite it today, I would probably produce some JSON and use `jq` in scripts to extract the data from the output.

elettrone1, 2020-04-24:
> Hi @elettrone1 ,
> vtuner works _only_ in Linux. I'm using it for years without a trouble (with a custom patch to 

I know!

I try to explain what I mean.

The author of vtunerc proposed to include it in the kernel sources but it was rejected because of a licensing issue!

Read this thread for the story please

https://patchwork.kernel.org/patch/894212

elettrone1, 2020-04-24:
@lars18th
> Please, read the documentation. What you need is to add the `until` plugin. In the form of:

I am using an example from pag. 279 to record a ts stream from my dvb usb card.

tsp -I dvb --tune 11766:v:29900 -P until --seconds 3 -O file video.ts

But I get the errors

* Error: dvb: unknown option --tune
* Error: dvb: no parameter allowed, use options only

lars18th, 2020-04-24:
> At the time, I just needed some stable output format which was extremely easy to process in scripts with classical basic line-oriented text processing tools (namely `sed` and `grep`). If I had to [...]

Perhaps one standalone tool to translate from this "normalized" format to JSON will be useful.
What you think @lelegard?

lelegard, 2020-04-24:
> I am using an example from pag. 279 to record a ts stream from my dvb usb card.
> tsp -I dvb --tune  11856: v :0 :27500-P until --seconds 3 -O file video.ts

Sorry, the option `--tune` was removed a few years ago. It used the old "LinuxTV zap format" which is now deprecated (it supported only  a few DVB options while we now support a lot more types of modulation). I will update the example in the manual.

Please have a look at the modulation options in the reference section for the `dvb` input plugin.

I must admit I don't understand your approach. You test a command with specific modulation parameters which has anyway zero chance to work if you don't use the very same transponder on the same satellite. When the command fails with "unknown option", you don't even have a look at the few pages which document modulation options.

elettrone1, 2020-04-24:
@lelegard 
> I must admit I don't understand your approach. You test a command with specific modulation parameters which has anyway zero chance to work if you don't use the very same transponder on the same [...]

Give me the time to read!

tsp -I dvb --lnb 9750,10600,11700 --delivery-system DVB-S2 --fec-inner 3/4 --frequency 11766000000 --modulation 8-PSK --symbol-rate 29900000 --polarity vertical -P until --seconds 3 -O file video.ts

This works.

elettrone1, 2020-04-24:
@lelegard 
I have another question, this is not about TSDuck, but I think you are the right person to ask.

I have a STB (Vu+ Zero) running a SAT>IP server (minisatip). 
I would like to record and analyze the full ts stream from there. 
Since there is a known limitation on STBs that prevents to record the full ts stream (using pid 0x2000), my idea was 

- analyze the ts stream from SAT>IP with only the PAT, CAT, SDT pids and read the PMT pids;
- analyze the ts stream from SAT>IP with PAT, CAT, SDT, PMT pids and read all the other pids;

This works with other STBs but not on this Vu+ Zero because, if I analyze the stream from minisatip that includes PMT pids, I can't find in the stream the video, audio pids ecc. It seems that all Vu+ receivers have this problem. Note that this is not a minisatip issue, because if I record the stream on the STB there is the same problem.

Do you think it is a hardware limitation of Vu+ tuners or I can solve this?

Thank you for your help!

lars18th, 2020-04-25:
Hi @elettrone1 ,

> @lelegard
> I have another question, this is not about TSDuck, but I think you are the right person to ask to.

I'm not @lelegard but I feel I can respond to your request.

- First, note that for a _full transport stream_ analysis I only recommend to use robust tuners. Low cost tuners can't provide sufficient bandwidth, and this is actually true with STBs. Forget them to do this analysis. For DVB-T/C/T2 signals I recommend HDHR boxes, and for DVB-S/S2/S2X I recommend Digital Devices SAT>IP boxes. They provide the full TS without problems (even with multiple tuners) and can work from the network. So you can use any platform to do the analysis.
- Second, if you _only_ need to analyze PSI/SI tables, then you don't need the full TS. It's sufficient to filter pids 0-20. Then use the PAT table to know the pids of all PMTs and then add them to the pid list. So with pids 0-20 + PMT-pids you have sufficient data to analyze the TS. And the total bandwith of this data has low bit rate, so you can use any tuner to provide the TS.
- Third, form my experience with STBs and the SAT>IP protocol, it's preferable to forget _all_ closed source platforms. The only exeception are the STBs based on Linux. On these boxes you can run software SAT>IP servers like minisatip. But in any case it's mandatory to use a platform with a robust SoC and good tuners. If you want to do a _serious_ analysis then I repeat: use robust tuners. If you only want to do homemade tests then any SAT>IP server is valid if you only use pids 0-20 + PMT-pids.
- Finally, I don't recommend to change the pid list from a STB SAT>IP server when doing a TS analysis. In these platforms, even the standard requires to be transparent, it's very common to lost some TS packets when adding/deleting pids (so you will see several CC errors). So my recomendation is to first analyze pids 0-20. Then when you know the PMT pids, do a TEARDOWN, and restart with a SETUP with the final pid list: 0-20 + PMT-pids. And only then do the real analysis.

I hope this helps you.
Regards.

elettrone1, 2020-04-25:
Hi @lars18th 

> * First, note that for a _full transport stream_ analysis I only recommend to use robust tuners. 

I do have a TBS 5927 USB.

> * Second, if you _only_ need to analyze PSI/SI tables, then you don't need the full TS. 

Yes, but as I said, I want to analyse AND record the full ts stream.

> It's sufficient to filter pids 0-20. Then use the PAT table to know the pids of all PMTs and then add them to the pid list. So with pids 0-20 + PMT-pids you have sufficient data to analyze the TS. And the total bandwith of this data has low bit rate, so you can use any tuner to provide the TS.

Yes, this works with my USB card on PC, but if I try to ask PMT pids to my STB I can't find the video and audio pids in the ts.

I make an example to be clear

On my STB Vu+ Zero if I record PAT (0), CAT (1), SDT (17), one PMT (1101)

tsp -I http "http://vuzero:8080/?src=1&freq=11766&sr=29900&pol=v&msys=dvbs2&mtype=8psk&fec=34&pids=0,1,17,1101" -P until -s 3 -P analyze --normalized -O drop

I have this result

service:id=3401:tsid=5200:orignetwid=318:access=clear:pids=1:clearpids=1:scrambledpids=0:packets=0:bitrate=0:bitrate204=0:servtype=1:pmtpid=1101:pidlist=1101:provider=Rai:name=Rai 1 HD

No video, audio ecc. in the pidlist!

If I record PAT (0), CAT (1), SDT (17), one PMT (1101) and a video PID (501) related to the PMT

tsp -I http "http://vuzero:8080/?src=1&freq=11766&sr=29900&pol=v&msys=dvbs2&mtype=8psk&fec=34&pids=0,1,17,1101,501" -P until -s 3 -P analyze --normalized -O drop

I have this result

service:id=3401:tsid=5200:orignetwid=318:access=clear:pids=14:clearpids=14:scrambledpids=0:packets=23433:bitrate=13386617:bitrate204=14525903:servtype=1:pmtpid=1101:pcrpid=501:pidlist=501,567,601,611,621,1101,2011,2012,3011,3012,3101,4225,4425,5425:provider=Rai:name=Rai 1 HD

All the pids (video, audio, ecm ecc.) related to the PMT (1101) I asked for are in the pidlist!

My question is: why does this happen?

Note, I have this problem only on my Vu+ STBs. No problem on PC with a USB card, or with other STBs.

> * Third, form my experience with STBs and the SAT>IP protocol, it's preferable to forget _all_ closed source platforms. The only exeception are the STBs based on Linux. 

Indeed I use Linux or Enigma2 STBs with minisatip!

Thanks!

elettrone1, 2020-04-25:
> > Hi @elettrone1 ,
> > I updated the wiki to point to the current github mirror: https://github.com/Leatherface75/satip
> > Regarding my environment, I'm not sure about recent kernels but for this I still use the Debian 7 x64 (debian 3.2.0-4-amd64). Even @lelegard doesn't like to maintain ancient kernels with old gcc versions, the current TSDuck still compiles in this machine when using the GCC 4.9.
> 
> I get this error while compiling vtuner_tweaks
> 
> /home/debian/satip/vtuner_tweak/vtunerc_proxyfe.c:445:18: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
> .get_frontend = dvb_proxyfe_get_frontend,
> 
> It is impossible to build this stuff, I don't know how you did it, maybe kernel 3.2.0-4 is a lot simpler!

elettrone1, 2020-04-25:
> Hi @elettrone1 ,
> I updated the wiki to point to the current github mirror: https://github.com/Leatherface75/satip
> Regarding my environment, I'm not sure about recent kernels but for this I still use [...]

Successfully compiled on kernel 4.9!

lars18th, 2020-04-25:
Hi @elettrone1 ,

Before anything, try removing the `--normalized` parameter to make the data more easy to read for us (humans).

> Hi @lars18th

> tsp -I http "http://vuzero:8080/?src=1&freq=11766&sr=29900&pol=v&msys=dvbs2&mtype=8psk&fec=34&pids=0,1,17,1101" -P until -s 3 -P analyze --normalized -O drop
> 
> No video, audio ecc. in the pidlist, only the PMT!

Yes, that's the expected behaviour.

> 
> tsp -I http "http://vuzero:8080/?src=1&freq=11766&sr=29900&pol=v&msys=dvbs2&mtype=8psk&fec=34&pids=0,1,17,1101,501" -P until -s 3 -P analyze --normalized -O drop
> 
> All the pids (video, audio, ecm ecc.) related to the PMT (1101) I asked for are in the pidlist!
> 

Yes, I see the problem: you receive all pids of this program but you only request for 501 (perhaps the video)...

> My question is: why does this happen?

I feel that's the reason: Several STB doesn't provide a real SAT>IP server, but a _translator_ from a native streaming service. A lot of STBs what do when streaming it's in fact this: they start to play the requested program in the TV and at the same time they copy the TS packets to the network. Then when you request _some_ pid in the PMT then _all_ pids related to this program are selected to play on TV. And the result is that the streaming contains all pids of this program. That's absurd using the SAT>IP protocol, but it's the stupid solution used by some manufacturers. However, if you want _less_ pids you can filter out using the "filter" plugin of TSDuck. So the only problem is the performance of the STB to support the streaming.

I hope this solve your question.

elettrone1, 2020-04-25:
Hi @lars18th

> > tsp -I http "http://vuzero:8080/?src=1&freq=11766&sr=29900&pol=v&msys=dvbs2&mtype=8psk&fec=34&pids=0,1,17,1101" -P until -s 3 -P analyze --normalized -O drop
> > No video, audio ecc. in the pidlist, only the PMT!
> 
> Yes, that's the expected behaviour.

Are you sure? On my PC I get the video, audio PIDs, i.e., all the pids related to that PMT. I think that this is the expected behavior. If you ask for PMT, you should find all PMT's related PIDs in the ts.

> Yes, I see the problem: you receive all pids of this program but you only request for 501 (perhaps the video)...

No, I am saying that I asked for PMT (1101), I expect to find all PMT's related PIDs in the ts and I get them. This is the expected behavior.

So in summary:

Vu+ Zero: ask for PMT -> get the PMT, but doesn't get the PMT's related PIDs (wrong)
Vu+ Zero: ask for PMT and a video pid of that PMT -> get the PMT's and all PMT related PIDs (right)
PC: ask for PMT -> get the PMT's and all PMT related PIDs (right)

> I feel that's the reason: Several STB doesn't provide a real SAT>IP server, but a _translator_ from a native streaming service.

I don't think the problem is minisatip, which is running on my Vu+. The problem is related to the Vu+'s tuner. I don't understand why, though.

elettrone1, 2020-04-26:
Hi @lars18th

> I don't think the problem is minisatip, which is running on my Vu+. The problem is related to the Vu+'s tuner. I don't know why, though.

Solved! The problem was with the -P until parameter that was too low. You have to wait more time in order to parse the transponder.

lars18th, 2020-04-26:
> Hi @lars18th
> 
> > I don't think the problem is minisatip, which is running on my Vu+. The problem is related to the Vu+'s tuner. I don't know why, though.
> 
> Solved! The problem was with the -P until parameter that was too low. You have to wait more time in order to parse the transponder.

Great if the problem is solved for you. However, I think you have a wrong assumption about pids with the SAT>IP protocol:
- Based on specifications of SAT>IP, when you request a list of pids, then you will receive _only_ these pids. Never more, and never less. However, one exception exist about this: if the stream is shared, then _more_ pids can be inside it; but never _less_ pids. So based on the standard any client requires to expect to receive _almost_ the pids requested.
- Your assumption about receive all pids of a program when you request the PMT or PMT+video is false. This can be true with your device **by the reason I pointed before**: because the tuner in fact is playing the program, and what the SoC is doing is to _copy_ the TS packets to the network. But this is not the expected behaviour. So don't think that standard tools like TSDuck or any SAT>IP server will implement this behaviour.

Regards.

elettrone1, 2020-04-26:
Hi @lars18th 

> * Your assumption about receive all pids of a program when you request the PMT or PMT+video is false.

You misunderstood me. I never said this! I repeat again: I want to READ THE VALUES OF PMT RELATED PIDS (i.e., video, audio etc.) when I request a stream specifying the PMT. 

I don't want to RECEIVE video and audio PIDs.

I need to record the full ts stream from a STB, so I must read the pids in the PMT. 

This is what I did and it worked.

Thanks!

elettrone1, 2020-04-26:
Hi @lelegard 

Is it possibile to retrieve the video resolution for a video PID with TSDuck? I didn't find any reference in the manual. I think I should parse the ES header.

Thanks

lars18th, 2020-04-27:
Hi @elettrone1 ,

> > * Your assumption about receive all pids of a program when you request the PMT or PMT+video is false.
> 
> You misunderstood me. I never said this! I repeat again: I want to READ THE VALUES OF PMT RELATED PIDS (i.e., video, audio etc.) when I request a stream specifying the PMT.

Sorry if I misunderstood you. However...

> I don't want to RECEIVE video and audio PIDs.

Then, only request pids 0-20 **and** the pid of the PMT (you get it from the PAT pid 0).

> I need to record the full ts stream from a STB, so I must read the pids in the PMT.

Sorry, this can't be called the "full ts stream". The full TS is all pids, including the NULL stuff. I feel you want to say "the full pids _of a program_" from the STB. I'm right now?

Regards.

elettrone1, 2020-04-27:
Hi @lars18th  
> Then, only request pids 0-20 **and** the pid of the PMT (you get it from the PAT pid 0).

This is what I did.

> Sorry, this can't be called the "full ts stream". The full TS is all pids, including the NULL stuff. I feel you want to say "the full pids _of a program_" from the STB. I'm right now?

No, I mean the full TS. I read all the PIDs of TS with that procedure. The NULL stuff is PID 8191, right? What other PIDs remain?

lars18th, 2020-04-27:
Hi @elettrone1 ,

> > Then, only request pids 0-20 **and** the pid of the PMT (you get it from the PAT pid 0).
> 
> This is what I did.

OK. :wink: 

> No, I mean the full TS. I read all the PIDs of TS with that procedure. The NULL stuff is PID 8191, right? What other PIDs remain?

Here is your error:
- "Full Transport Stream": _ALL_ packets of a TS, including NULL stuff (pid 8191), and all other pids.
- "Filtered Transport Stream": A full TS _without_ some of the original pids.

Then you can request with the SAT>IP protocol or Linux DVB API:
- The full TS: using `pids=all` with SAT>IP or `pid=8192` with the Linux DVB API (when the tuner driver supports it).
- A filtered TS: when you select only some pids, ex.: 0-20, 1101, 501.

The problems are:
- We don't a have a common API to request "all valid pids", that's all pids without NULL stuff (in fact equivalent to request for pids 0-8190).
- It's common that STBs can't handle a full TS. The reason is because the SoC and the network stack can go over 10-20Mbps in most cases. Then you can expect to receive incomplete packets (CC errors) when you request `pids=all` or the server responds with `unavailable source`.
- Similar problem appears with ancient linux tuners. A lot of them doesn't support the magic pid 8192 for sending the full TS.
- Then is common to only support a limited list of pids. Common list sizes are 16 or 32 pids.

So, your problem is that the tuner you're using (a VU+ acting as a SAT>IP server) can't handle the "full TS" (that's ALL pids in a transponder). Then you need to work _everytime_ with a **list of pids**. You **can't** overcome this problem. It's a hardware limitation. If you want to process the full TS, then use serious tuners.

But, this limitation doesn't limit you to catch all PSI/SI tables from a TS. Or receive all the pids of a program. Even for a multiple programs. The actually limit is the number of pids: you can receive 16 or 32 (or whatever number) based of the hardware that you're using.

I hope this helps you.
Regards.

elettrone1, 2020-04-27:
Hi @lars18th 

> So, your problem is that the tuner you're using (a VU+ acting as a SAT>IP server) can't handle the "full TS" (that's ALL pids in a transponder). Then you need to work _everytime_ with a **list of [...]

This is what I am doing! I am not asking pids=8192. I get the list of all pids and include them in the url!

> But, this limitation doesn't limit you to catch all PSI/SI tables from a TS. Or receive all the pids of a program. Even for a multiple programs. The actually limit is the number of pids: you can receive 16 or 32 (or whatever number) based of the hardware that you're using.

This is indeed a problem. I will make some tests.

elettrone1, 2020-04-27:
Hi @lars18th 

It seems there is a limit of 38 PIDs on the Vu+ Zero!

lars18th, 2020-04-27:
> It seems that there is a limit of 38 PIDs on the Vu+ Zero!

And you need to count too with the problem of no pid 8192 or "pids=all" support!

But, knowing this I'm sure you can do what you need using TSDuck.
Regards.

lars18th, 2020-04-29:
Hi @elettrone1 ,

> Could you PLEASE give me the source code that you used for building vtuner?
> The source code linked in that page doesn't exist anymore.
>
> https://code.google.com/archive/p/satip/
>
> Also, if I try [...]

Sorry! Try this: https://github.com/BackupGGCode/satip

elettrone1, 2020-04-29:
Hi @lars18th

> Solved! The problem was with the -P until parameter that was too low. You have to wait more time in order to parse the transponder.

No it's not solved. On Vu+ Zero.

I make an example to explain this strange behavior

Channel: RAI 1 HD (Hotbird 13E)

I record the stream asking for these PIDs: PAT (0), CAT (1), NIT (16), SDT/BAT (17), EIT (18), TDT/TOT (20), PMT (1101)

curl -m 20 -o video.ts "http://vuzero:8080/?src=1&freq=11766&sr ... 18,20,1001"

I check the recorded TS file with tsanalyze (from TSDuck)

tsanalyze --normalized video.ts

And I find in the stream the PID I requested but I can't read in the PMT the other PIDs values (video, audio, subtitles, ecm etc.)!

service:id=3401:tsid=5200:orignetwid=318:access=clear:pids=1:clearpids=1:scrambledpids=0:packets=0:bitrate=0:bitrate204=0:servtype=1:pmtpid=1101:pidlist=1101:provider=Rai:name=Rai 1 HD

pidlist lists only the PMT pid! Not the video, audio etc.

If request the PMT I don't want to record the video, audio etc., but the PMT should list all the other PIDs values for the service!

elettrone1, 2020-04-29:
> Hi @elettrone1 ,
> 
> > tsanalyze --normalized video.ts
> 
> Please, can you post here the result **without** the `--normalized` parameter? (I'm human, not a machine!)

Log in attach.
[log.txt](https://github.com/tsduck/tsduck/files/4554176/log.txt)

lars18th, 2020-04-30:
Hi @elettrone1 ,

> > > tsanalyze --normalized video.ts
> >
> > Please, can you post here the result **without** the `--normalized` parameter? (I'm human, not a machine!)
> 
> Log in attach.
> [log.txt](https://github.com/tsduck/tsduck/files/4554176/log.txt)

From this analysis it's clear that the INPUT stream doesn't contains _any_ packet of the pid `0x044D (1101)` (the PMT of Rai 1). So for this reason you don't see the content.

So, TSDuck can't do an impossible thing.
Please, verify your STB request.

elettrone1, 2020-04-30:
> Hi @elettrone1 ,
>
>
> From this analysis it's clear that the INPUT stream doesn't contains _any_ packet of the pid `0x044D (1101)` (the PMT of Rai 1). So for this reason you don't see the content.
>
> So, [...]

Sure. I am asking minisatip on Vu+ Zero with the url above. How can I verify the request? Note that if I record the global pids I get also listed the PMT related pids.

lars18th, 2020-05-01:
Hi @elettrone1 ,

> Sure. I am asking minisatip on Vu+ Zero with the url above. How can I verify the request? Note that if I record the global pids I get also listed the PMT related pids.

My suggestion is this:
- Do multiple "recordings" (you can use curl, don't worry about this if the server is minisatip or any other that supports HTTP with SAT>IP) with different pid lists (save the pids in the filename of the output TS).
- Check the content of the recordigns using TSDuck.
- After that you can have an overall overview of the behaviour of your STB.

I'm not sure, but I feel it doesn't respond in accordance of the SAT>IP protocol. Instead it plays the program more similar to the requested pids, and it streams all the content outputed by the tuner. In this case, you will recieve all these unwanted video/audio/other pids. But for TSDuck this isn't a problem. You can filter out them with the `filter` plugin. And if you care about the performance of the streaming, if the STB can handle the program, then it's obvious that it can stream it without problems. The problem with this behaviour is then only if you want to receive _all_ PMTs of the TS. With this behaviour it will be impossible in one shot. If this is your objective, then please belive me: use one _serious_ network tuner and forget a STB!

Regards.

elettrone1, 2020-05-01:
HI @lars18th 

> * Do multiple "recordings" (you can use curl, don't worry about this if the server is minisatip or any other that supports HTTP with SAT>IP) with different pid lists (save the pids in the filename [...]

If I record PAT, PMT and the global PIDs (as listed by tsanalyze --normalized) and then I use tsanalyze I can read the pidlist of each PMT! With this workaround it works.

> But for TSDuck this isn't a problem. You can filter out them with the `filter` plugin.

I think the problem with these STBs is that they provide less PIDs than requested...

Anyway, thank you very much for your patience, detailed explanation and suggestions!

lars18th, 2020-05-01:
Hi @elettrone1 ,

> If I record PAT, PMT and the other global PIDs (as listed by tsanalyze --normalized) and then I use tsanalyze I can read the pidlist of each PMT! With this workaround it works.

Great to hear that you found a solution to your problem.

> > But for TSDuck this isn't a problem. You can filter out them with the `filter` plugin.
> 
> I think the problem with these STBs is that they provide less PIDs than requested...

From my experience it's just the opposite: the STBs provide _more_ PIDs than requested. It's curious!

> Anyway, thank you very much for your patience, detailed explanation and suggestions!

:wink: 

So, please close this issue if you feel it's resolved.
Regards.

closed 2020-05-01 by elettrone1