tsduck/tsduck #567: Satellite MPTS to SPTS, TS discontinuity and random jitter | tsp zap

AlexandreLicinio, 2020-04-25:
Hi, 

I simply try to extract and stream one service from a **mpts** (Astra) but no success. 
```
tsp -I ip 239.127.0.1:5000 -P zap "ONE HD" -P regulate -O ip 239.127.0.1:6000
```
An **Ateme DR5000** is used for decoding the **239.127.0.1:6000** containing only the **ONE HD** service.

[image]

The jitter is jagged. Lots of TS discontinuity in both audio and video.
The **239.127.0.1:5000**, the whole mpts is well decoded without any TS errors and the jitter is very low and straight (2.3ms).

I tried to add **tsresync -c** but no luck.
Thanks for helping.

lelegard, 2020-04-25:
Try to remove `-P regulate`. Your input is already regulated, so the result is probably over regulated. You need to "regulate" (ie. slow down the stream) only when read at faster speed, from a file for instance. When the input is already at playout speed, over regulating can only degrade the stream.

AlexandreLicinio, 2020-04-25:
Thanks for answering so fast. unfortunately, removing the `-P regulate` not fix the issue.
I get a wrong bitrate over the decoder, 3mb/s instead 8mb/s

[image]

thanks

lelegard, 2020-04-25:
OK, then add `-P analyze -i 30 -o report.txt` between `zap` and output. It will produce an analysis report file every 30 seconds. So, you will have a idea of what is received and produced by `tsp`. Especially, you will see if the bitrate of the service is 3 or 8 Mb/s.

You can also try `--enforce-burst` in `-O ip`. Since the 7 TS packets per input UDP packet were fragmented by removing all services, then you enforce 7 TS packets per output UDP packet. Functionally, it does not change anything. But maybe the receiver will need it.

AlexandreLicinio, 2020-04-26:
Thanks. With `--enforce-burst` it may takes up to 1 minute before the **decoder** switch from **probing** state to **decoding** state. The bitrate seems ok but still weird jitter and lots of TS discontinuity.

[image]

The tsp cmd is:
```
tsp -I ip :5000 -P zap 10378 -O ip 10.100.34.241:5000 --enforce-burst
```
And the report:
```
===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ....... 1039 (0x040F)  |  Services: .............. 1  |
|  Bytes: ........................ 21,840,148  |  PID's: Total: ......... 14  |
|  TS packets: ...................... 116,171  |         Clear: ......... 14  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 1  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ......................... 42,584,503 b/s   46,208,716 b/s  |
|  Estimated based on PCR's: ................ 8,523,283 b/s    9,248,669 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ...................................... 4 sec (0 mn 4 sec)  |
|  First TDT UTC time stamp: ........................... 2020/04/26 08:27:50  |
|  Last TDT UTC time stamp: ............................ 2020/04/26 08:28:09  |
|  First TOT local time stamp: ......................... 2020/04/26 10:27:50  |
|  Last TOT local time stamp: .......................... 2020/04/26 10:28:10  |
|  TOT country code: ................................................... DEU  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x288A  SR Fernsehen HD .............................. C   42,497,993 b/s  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 236, PID's: 3 (clear: 3, scrambled: 0)                         |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       86,509 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       86,509 b/s  |
|  0x0000  PAT .......................................... C       30,058 b/s  |
|  0x0011  SDT/BAT ...................................... C       41,422 b/s  |
|  0x0014  TDT/TOT ...................................... C       15,029 b/s  |
|=============================================================================|
|  Service: 0x288A (10378), TS: 0x040F (1039), Original Netw: 0x0001 (1)      |
|  Service name: SR Fernsehen HD, provider: ARD                               |
|  Service type: 0x19 (Advanced codec HD digital television service)          |
|  TS packets: 115,935, PID's: 11 (clear: 11, scrambled: 0)                   |
|  PMT PID: 0x1536 (5430), PCR PID: 0x1537 (5431)                             |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Advanced codec HD digital television service . C   42,497,993 b/s  |
|  0x087B  DSM-CC U-N (HbbTV) ........................... C      748,897 b/s  |
|  0x0BA0  DSM-CC Stream Descriptors .................... C       49,853 b/s  |
|  0x1112  MPEG-2 Private sections (AIT) ................ C       49,853 b/s  |
|  0x1114  DSM-CC U-N (HbbTV) ........................... C    1,997,425 b/s  |
|  0x1536  PMT .......................................... C       30,058 b/s  |
|  0x1537  AVC video (1280x720, high profile, level 4.0,  C   33,858,365 b/s  |
|  0x1538  MPEG-1 Audio (deu, Audio layer II, 192 kb/s,   C      991,198 b/s  |
|  0x1539  MPEG-1 Audio (mis, Audio layer II, 192 kb/s,   C      991,198 b/s  |
|  0x153A  Teletext (deu, Initial Teletext page) ........ C    1,313,777 b/s  |
|  0x153B  Subtitles (deu, DVB subtitles for hard of hea  C      159,823 b/s  |
|  0x153C  AC-3 Audio (deu, AC-3, stereo (L,R), @48,000   C    2,307,541 b/s  |
[... 265 lines cut ...]
|-----------------------------------------------------------------------------|
|  0x3C (60, DSM-CC DDM), TID ext: 0x0001 (1)                                 |
|      Repetition: ..... 280  ms  Section cnt: ...... 147                     |
|      Min repet.: ..... 272  ms  Table cnt: ......... 15                     |
|      Max repet.: ..... 289  ms  Version: ........... 12                     |
[... cut ...]
|  Service: 0x288A (10378) SR Fernsehen HD                                    |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x288A (10378)                                     |
|      Repetition: ...... 50  ms  Section cnt: ....... 82                     |
|      Min repet.: ...... 39  ms  Table cnt: ......... 82                     |
|      Max repet.: ...... 53  ms  Version: ............ 1                     |
===============================================================================
```

lelegard, 2020-04-26:
A few points:

- To get more accurate reports, I should have mentioned to add `-P pcrbitrate` before `-P analyze`.  The TS bitrate was 42.5 Mb/s on input and is used in the report, even though all services but one were removed. So, all bitrates are incorrect. The `pcrbitrate` plugin forces a recomputation of the bitrate based on PCR before analysing the stream. Anyway, the analysis report also includes the bitrate based on an internal analysis of the PCR and it says 8.5 Mb/s. So, it's correct (at least it's not 3 Mb/s).

- The analysis report mentions zero discontinuities in all PID's. So, discontinuities appear downstream.

- Using option `--enforce-burst` in output plugin is a matter of tuning but it does not change anything on a functional standpoint. All TS packets are sent as soon as they appear but they are better grouped in UDP packets, sending less but larger UDP packets.

As a conclusion, your problem is definitely in the receiver/decoder (or maybe - but less likely -  in the network between tsp and the receiver). Your receiver does not seem very robust and requires a very accurate realtime transmission. Of course, `tsp` is a generic lab tool with lots of plugins and options. Output is less smooth than a specialised streamer which focuses on real time accuracy. But most receivers handle that smoothly. Apparently, yours does not. 

To smoothen the processing (at the expense of a higher CPU usage), you may try to reduce `tsp` parameters such as `--max-input-packets` and `--max-flushed-packets` to 100. Or more or less, just tune.

AlexandreLicinio, 2020-04-26:
Thanks, you take time to resolve my issue.
Unfortunately it does not help the **Ateme DR5000** to decode smoothly without any errors.
What i don't get is why a professionnal decoder cannot decode the service without any error while vlc does.

Thanks @lelegard

lelegard, 2020-04-26:
> What i don't get is why a professionnal decoder cannot decode the service without any error while vlc does.

VLC is the Hummer of the decoders, it can decode everything, this is the most robust player software I have ever seen. Your receiver is probably too strict by default. It seems to be a sophisticated product, so maybe it has many fine tuning options.

AlexandreLicinio, 2020-04-27:
Yes, 100% true.
I just tested with another broadcast decoder (OpenBroadcastSystems C100 - Decoder) and it decodes without any errors ! This decoder is based on [UPIPE](https://github.com/Upipe/upipe). 

So i close the issue, thanks for the help and the time you provded to me.

closed 2020-04-27 by AlexandreLicinio