tsduck/tsduck #568: Support for ISDB-T

nuclidjp, 2020-04-28:
Thank you for supporting ISDB-T in Japan.

ISDB has Japan specification ISDB-T and international standard ISDB-Tb.
ISDB-T has MPEG-2 as a video encoding method, ISDB-Tb has AVC and some differences.
Unlike DVB-T, which has been adopted and standardized in many countries, there are several types of ISDB.
Anyone interested in ISDB-T?
There is a Japanese ISDB-T file in the [public stream repository.](https://tsduck.io/streams/)

nuclidjp, 2020-05-07:
Hi,
I used NHK.ts from Streams repository.

tsp -I file NHK.ts -P tables --isdb --pid 17 --tid 66 --max 1 --xml nit.isdb.xml -O drop
tsp -I file NHK.ts -P inject nit.isdb.xml --pid 17 --replace --stuffing -O file NHK.isdb.ts
tsp -I file NHK.isdb.ts -P psi --japan -a -o NHK.isdb.txt -O drop

tsp -I file NHK.ts -P tables --japan --pid 17 --tid 66 --max 1 --xml nit.japan.xml -O drop
tsp -I file NHK.ts -P inject nit.japan.xml --pid 17 --replace --stuffing -O file NHK.japan.ts
tsp -I file NHK.japan.ts -P psi --japan -a -o NHK.japan.txt -O drop

In both texts, the Japanese Name of broadcasting station is garbled.
Is the command wrong?

lelegard, 2020-05-07:
Let me guess, the names are incorrect in the first two commands on each group but correct in the third one, right?

The option `--japan` is a convenient synonym for `--isdb --default-charset ARIB-STD-B24`.

- The option `--isdb` indicates how to interpret the signalization, typically table ids and descriptor tags. There are at least two conflicting table ids between ISDB and ATSC and many conflicting descriptor tags between DVB (private), ATSC and ISDB. So, TSDuck requires some hints from the user on which standard to use.
- The option `--default-charset ARIB-STD-B24` indicates how to interpret characters inside tables and descriptors. This option is not included into `--isdb` because ISDB is used outside Japan but only Japan uses this specific character encoding.

As a conclusion, use option `--japan` in all your plugins above.

nuclidjp, 2020-05-07:
The explanation was insufficient.
sorry

tsp -I file NHK.ts -P tables --japan --pid 17 --tid 66 --max 1 --xml nit.japan.xml -O drop
Japanese in nit.japan.xml is correct.

tsp -I file NHK.ts -P inject nit.japan.xml --pid 17 --replace --stuffing -O file NHK.japan.ts
The Japanese of the TS file is incorrect.
I confirmed it with a Japanese TS player.

tsp -I file NHK.japan.ts -P psi --japan -a -o NHK.japan.txt -O drop
It is the same result as the Japanese TS player.
Japanese is not correct.

lelegard, 2020-05-07:
> tsp -I file NHK.ts -P inject nit.japan.xml --pid 17 --replace --stuffing -O file NHK.japan.ts

You have to use option `--japan` also in `inject` plugin. Otherwise, the NIT will be encoded according to DVB character sets. But... but... but... I unfortunately realize that the plugin `inject` does not have the encoding options of `tstabcomp` . Sorry for the mistake, I am going to fix this right away. It should be trivial.

In the meantime, you can split the operation in two phases: 1) compile the XML file into a binary section, 2) inject the binary section. Since the character encoding occurs during compilation, this is where `--japan` has to be specified. Then, the plugin `inject` simply reads the binary sections without trying to encode anything.

~~~
tstabcomp --japan nit.japan.xml -o nit.japan.bin
tsp ... -P inject nit.japan.bin ...
~~~

lelegard, 2020-05-07:
> Sorry for the mistake, I am going to fix this right away. It should be trivial.

Done, the charset options, including the `--japan` shortcut, have been added to plugin `inject`. You can now inject XML files directly.

nuclidjp, 2020-05-08:
Thank you very much for your quick response.
I confirmed that the Japanese version is correct in version 3.21-1763.