tsduck/tsduck #570: Fix missing PAT, PMT in a scrambled TS?

elettrone1, 2020-05-04:
Hi,
is it possible to fix a scrambled TS file with missing PAT and PMT PIDs?

I notice that tsfixcc fixes CC errors. 

Tools like ffmpeg can fix this problem but only if the TS is not scrambled.

Thanks!

lelegard, 2020-05-04:
Hi,

You must know what to set in the missing PMT. No tool can retrieve everything. Especially, when descriptors are needed to describe the audio and video properties. Since the corresponding streams are scrambled, no tool can guess what they are and which is which.

But if you know what to set in the PMT, then you can create XML files for the PAT and PMT and inject them using the plugin `inject`.

elettrone1, 2020-05-04:
Hi @lelegard 

> But if you know what to set in the PMT, then you can create XML files for the PAT and PMT and inject them using the plugin `inject`.

My scenario is: I have a TS scrambled channel with video, audio, ECM PIDs, but PAT and PMT are missing. I want to descramble (for example with tsdecrypt ) and play this file so I need to fix PAT and PMT. If the channel would be FTA I would use ffmpeg which automatically rebuilds PAT and PMT so that the TS file is playable. In this case I can't so you tell me I have to tell tsp how PAT and PMT are made. But how is it done a xml describing the video and audio PIDs? Is not possibile to just tell tsp what are the video and audio PIDs and then it creates the PAT and PMT?

Thanks for your patience.

lelegard, 2020-05-05:
Hi,

Without XML files, the plugins `pat` and `pmt` can create the tables if they do not exist after a certain timeout. See options `--create` or `--create-after`. This could be fine for a live stream, without begin or end. But in a file, this is a problem because the start of the file won't have tables.

In the PMT, do not forget to add a CA_descriptor with the right CA_system_id and private content. Otherwise, the set-top box won't find the ECM stream.

elettrone1, 2020-05-05:
Hi @lelegard 

> In the PMT, do not forget to add a CA_descriptor with the right CA_system_id and private content. Otherwise, the set-top box won't find the ECM stream.

I am trying with

tsp -I file video.ts -P inject -p 0 input-file pat.xml -i 2 -P inject -p 1101 input-file pmt.xml -i 2 -O file out.ts

but I get the error

* Error: inject: unknown file type for input-file

Xml files in attach.

I don't know if the -i 2 parameter is right, though.

Thanks.

[pat_pmt_xml.zip](https://github.com/tsduck/tsduck/files/4582971/pat_pmt_xml.zip)

elettrone1, 2020-05-05:
Hi @lelegard 

I am trying with

tsp -I file video.ts -P inject -i 2 -p 0 pat.xml -P inject -i 2 -p 1101 pmt.xml -O file out.ts

but the ts file does not contain any PAT or PMT.

I don't know if the parameter -i 2 is right. Which one should I use?

XML files in attach.

Thanks.
[pat_pmt_xml.zip](https://github.com/tsduck/tsduck/files/4583075/pat_pmt_xml.zip)

lelegard, 2020-05-06:
Hi,

Don't worry about the buffer locking issue. It is just informational. TSDuck tries to lock the global packet buffer in physical memory to avoid pagination avoid it. It becomes less useful with recent machines (was done 15 years ago).

The option `-i` or `--inter-packet` is one of the possible ways to indicate how frequently the section are inserted. Inserting PAT and PMT after every one packet in the TS is of course excessive. Specifying the insertion rate this way is reserved to specific low-level operations. For a generic TS, a better way is to specify the target bitrate of the PID. For a PAT or PMT, 15 kb/s is typical. So, use `--bitrate 15000` instead of `-i`.

If you don't get any PAT or PMT on output, you probably have no stuffing to replace. Please be sure to read the documentation of `tsp` in the TSDuck user's guide. It explains the main concepts to understand and pitfalls to avoid about `tsp`. Without at least reading this, you may end up in an endless cycle of trying/error/question. It is a much better investment to spend 10 minutes reading it.

In short, `tsp` nevers _creates_ packets. It processes existing packets, possibly replacing them. So, _insertion_ of new data can be done only by replacing existing null packets. This is typically broadcast-oriented where the modulation imposes a fixed bitrate with internal stuffing. If you don't have enough or no stuffing, then it depends on the context. If you are in a broadcast context, you can't increase the global bitrate and you need to remove - and replace by stuffing - existing data streams or services. If you are in an IP or file context, then you have to artificially create stuffing at input, use it as available replacement for inserted data and remove the remaining stuffing before output.

In your case, this would result in a command similar to this one. I let you read the documentation, understand the command and adapt it if necessary.
~~~
tsp --add-start-stuffing 10 --add-input-stuffing 1/10 \
    -I file video.ts \
    -P inject -p 0x0000 pat.xml --bitrate 15000 \
    -P inject -p 0x044d pmt.xml --bitrate 15000 \
    -P filter -n -p 0x1FFF \
    -O file out.ts
~~~

elettrone1, 2020-05-06:
Hi @lelegard 

> In your case, this would result in a command similar to this one. I let you read the documentation, understand the command and adapt it if necessary.
>
> ```
> tsp --add-start-stuffing 10 [...]

Thank you for the detailed explanation. It works!

closed 2020-05-06 by elettrone1