tsduck/tsduck #571: Add Arch Linux support

piotrserafin, 2020-05-04:
Add Arch Linux Support:

- [ ] PR with changes in install-prerequisites.sh accepted.
- [ ] New TSDuck version with changed install-prerequisites.sh released.
- [ ] Update [AUR:tsduck](https://aur.archlinux.org/packages/tsduck/) package by me.

Above will allow Arch users to install TSDuck automatically from AUR repo.

piotrserafin, 2020-05-05:
[PR#571](https://github.com/tsduck/tsduck/pull/572) merged to master

lelegard, 2020-05-06:
Hi @piotrserafin,

Thanks a lot for your effort to provide TSDuck in Arch Linux. I do not use this distro and do not test on it. So, this is good to have a contributor to provide support on Arch.

For your information, I regularly test on Fedora, Ubuntu, CentOS and Raspbian. I provide binaries on all of them for official releases. I also have an Alpine Linux VM because this distro has a non-standard libc which creates compilation issues from time to time.

Out of curiosity, what is the packaging process for Arch Linux? Does it require specific packaging such as with `rpm` and `dpkg`? In that case, would you like to include the corresponding scripts or configuration files in the main source tree, as currently done to produce `.rpm` and `.deb` packages?

I see that AUR is the "Arch User Repository". Is this different from the main source of packages for Arch? I see that the available version here is 3.15 from November 2018. It is indeed time to upgrade.

piotrserafin, 2020-05-06:
Hello @lelegard ,

Packaging system in Arch is called ABS (Arch Build System) and it is used for compiling code to installable packages typically with .pkg.tar.xz extension. Those packages are maintained by core arch developers and kept in main package repositories. There is also AUR which contains packages maintained by general community, this is the place for tsduck. Arch users use pacman tool (you can see it in PR I've created) to install and manage system packages. 

ABS uses mapkepkg tool which reads PKGBUILD file and generates package based on it's content.
For example PKGBUILD for dvbblast (which can be found AUR) looks following: 

```bash
pkgname=dvblast
pkgver=3.4
pkgrel=1
pkgdesc="A simple and powerful MPEG-2/TS demux and streaming application"
arch=('i686' 'x86_64')
url="https://www.videolan.org/projects/dvblast.html"
license=('GPL2')
depends=('glibc' 'libev')
makedepends=('bitstream')
source=("https://get.videolan.org/dvblast/$pkgver/dvblast-$pkgver.tar.bz2")
sha256sums=('7abd45aff738544017350af87ec7ebad00fcfb4e4effd04f35023d504d9b6f44')

build() {
  cd "$pkgname-$pkgver"

  make
}

package() {
  cd "$pkgname-$pkgver"

  make DESTDIR="$pkgdir" PREFIX="/usr" install
}
```
When user is installing this package from AUR, package manager downloads the sources and compiles it on the fly then copies binaries, licence, man pages into proper paths in rootfs. 

The difference between AUR and main arch repo is that compilation stage is done outside of local machine and in case of AUR compilation is typically done locally (but this is not the rule).

From TSDuck perspective I will update PKGBUILD file with new release version and modify prepare step into something like this:

```bash
pkgname=tsduck
_pkgver=3.20-xyvz
pkgver=${_pkgver/-/_}
pkgrel=1
pkgdesc="An extensible toolkit for MPEG/DVB transport streams"
arch=('x86_64')
url="https://tsduck.io/"
license=('BSD')
depends=(curl pcsclite)
source=("$pkgname-${_pkgver}.tar.gz::https://github.com/tsduck/tsduck/archive/v${_pkgver}.tar.gz")
md5sums=("65b05c93432b2274d38ea2ed65d0d388")

prepare() {
    cd "$pkgname-${_pkgver}/build"
    ./install-prerequisites.sh
}

build() {
    cd "$pkgname-${_pkgver}"
    make
}

check() {
    cd "$pkgname-${_pkgver}"
    make test
}

package() {
    cd "$pkgname-${_pkgver}"

    make SYSROOT="$pkgdir" install install-devel
    install -D -m644 LICENSE.txt "${pkgdir}/usr/share/licenses/${pkgname}/LICENSE"
    mv "$pkgdir/usr/lib64" "$pkgdir/usr/lib"

    local TSPLUGINS_PATH="/usr/lib/tsduck/plugins"
    mkdir -p "$pkgdir/$TSPLUGINS_PATH"
    find "$pkgdir/usr/bin" -type f -name 'tsplugin_*.so' -exec mv -t "$pkgdir/$TSPLUGINS_PATH" {} 

    mkdir -p "$pkgdir/etc/profile.d/"
    echo "export TSPLUGINS_PATH=/${TSPLUGINS_PATH}" > "$pkgdir/etc/profile.d/tsduck.sh"
}
```

In AUR repository only PKGBUILD file is keep and no sources, binaries, etc. Based on this Arch package manager tool is downloading, building and coping build results into proper locations by following PKGBUILD recipe.

Every time new release will show up I will update PKGBUILD file accordingly.

lelegard, 2020-05-06:
Thanks for the information.

It seems a similar system, in principle, as Homebrew for macOS. The equivalent of ABS / AUR is GitHub (all Homebrew repos, aka "taps", are GitHub repos). The repo contains a Ruby module which is the equivalent of your PKGBUILD file. Installing a package means pulling the git repo containing the Ruby file, downloading the source tarball from GitHub and recompiling it locally. Typically, the packages from the main repo are binaries while user-provided packages are locally recompiled. This looks very similar in principle.

Since my iMac is my main development platform, I manage the Homebrew repo for TSDuck myself here: https://github.com/tsduck/homebrew-tsduck. In Homebrew parlance, this URL is simply `tsduck/tsduck` (tap `X/Y` translates to https://github.com/X/homebrew-Y).

I see that you relocate the plugins into `/usr/lib/tsduck/plugins` and define a bash init file with `TSPLUGINS_PATH`. As a coincidence, another user just submitted a similar request in #575.

piotrserafin, 2020-05-07:
Yes it is very similar to Homebrew (one of my machines is also mac).

> I see that you relocate the plugins into /usr/lib/tsduck/plugins and define a bash init file with TSPLUGINS_PATH. As a coincidence, another user just submitted a similar request in #575.

That's actually leftover after previous maintainer but probably I will keep it that way.

I've decided that I will not use install_prerequisites.sh for AUR packages. Why? If I will provide dependencies in PKGBUILD they will be automatically removed (if not used by other packages) when user will remove tsduck. 

install_prerequisites.sh arch part is still useful for ppl like me who like to build from sources directly.

lelegard, 2020-05-07:
> install_prerequisites.sh arch part is still useful for ppl like me who like to build from sources directly.

Yes, definitely. The script was made for developers because it installs the C++ compiler, doxygen etc. Initially, I installed everything manually. But when I started to use GitHub Actions (the CI/CD platform from GitHub), the workflow files became too complicated and I created that script for the sake of simplicity. But it is good to have improved it for Arch.

piotrserafin, 2020-05-07:
HI @lelegard,

FYI tsduck is now up to date AUR repository -  [AUR:tsduck](https://aur.archlinux.org/packages/tsduck/).

In future I will continue to maintain it.

closed 2020-05-07 by piotrserafin

lelegard, 2020-05-07:
Thanks @piotrserafin, I appreciate the effort.

stoyanovgeorge, 2021-04-14:
@piotrserafin I have tried to install today `tsduck v. 3.25-2237-2` from AUR on my Manjaro `pamac build tsduck` and the build failed, so I needed to manually install `jdk-openjdk` maybe you can update the dependencies for this package.

lelegard, 2021-04-14:
I confirm that Java and Python bindings were added in version 3.25 and a JDK is required to build that part. The script `build/install-prerequisistes.sh` was updated accordingly. The Python bindings don't require compilation since this is just a `.py` file.

piotrserafin, 2021-04-14:
Sure, will update PKGBUILD tomorrow morning.

piotr-serafin-epam, 2021-04-17:
Added dependencies provided in install-prerequisites script. @stoyanovgeorge please try now.

stoyanovgeorge, 2021-04-19:
@piotr-serafin-epam, I have already manually installed the Java package and installed tsduck on my machine.

lelegard, 2021-04-21:
@piotrserafin, for your information, version 3.26-2349 was released yesterday.

stoyanovgeorge, 2021-04-21:
@piotr-serafin-epam it worked by the way, but I have jdk15 installed and it pushed jdk11 which is the LTS version, so maybe you can check if any version of Java is installed before pushing the LTS.

piotr-serafin-epam, 2021-04-22:
> @piotrserafin, for your information, version 3.26-2349 was released yesterday.

Dear @lelegard I understand that `NOTEST=true` is now set as default for make & make install ?

piotr-serafin-epam, 2021-04-23:
AUR package updated to 3.26

piotr-serafin-epam, 2021-08-22:
AUR package updated to 3.27-2438

lelegard, 2021-08-22:
Thanks @piotr-serafin-epam !

lelegard, 2021-08-24:
Hi @piotr-serafin-epam ,

For next version (3.28), you may want to add `NOGITHUB=true` to the make command in the build phase.

This new flag disables many options in `tsversion` to check, download and upgrade new versions of TSDuck from GitHub. These features are nice for users installing the generic binary packages from GitHub or who install from sources. However, distro packagers may prefer to limit the distribution and upgrade to the package management of the distro, as in the case of AUR on Arch Linux.

piotr-serafin-epam, 2021-08-24:
Hi @lelegard ,

Thx for info, will do, also will use this ticket to share AUR package update info.

ToDo:

- [ ] Update to 3.28
- [ ] Add  NOGITHUB=true  to make command

piotrserafin, 2021-09-26:
Updated to 3.28_2551

piotr-serafin-epam, 2021-12-12:
Updated to 3.29_2651

lelegard, 2022-07-04:
Hi @piotr-serafin-epam 

Just to let you know that TSDuck version 3.31-2761 is out.

piotrserafin, 2022-07-24:
Updated to 3.31-2716

youns88, 2022-08-14:
Hi @piotr-serafin-epam
can you add support for arm based devices ?
i have S905W chip and would like to have tsduck on my manjaro linux
thanks for previous efforts

piotrserafin, 2022-08-16:
Hi @youns88, let me look at this.