tsduck/tsduck #573: De-jitter udp IP multicast input

mks75, 2020-05-04:
Hi, how can I de-jitter an udp IP multicast stream as input and take corrected IP multicast output using tsp or any other option in tsduck? Please advise.
Thanks - MKS

mks75, 2020-06-17:
Hi, Any advise on my query regarding de-jitter the udp IP MPEG-TS stream?
Thanks - MKS

lelegard, 2020-06-17:
Hi,

As discussed in #587, TSDuck is probably not the best tool to do this.

If the jitter is completely insane, you may reduce it using the plugin `regulate` with extremely fine-tuned buffering parameters.

However, de-jittering should be an extremely precise operation which needs a specialized tool if you want to reach a millisecond precision. TSDuck is a generic and flexible framework which does too many things to be that precise on the realtime side.

mks75, 2020-06-17:
Thanks @lelegard , will try to use "regulate" plugin. 
In this plugin, the --pid-pcr value is in decimal or hex?
Thanks-MKS

lelegard, 2020-06-17:
With TSDuck, all integer values can be decimal or hex, just use "0x" to prefix hex values. See section 3.1.2 in the user's guide.

mks75, 2020-06-17:
ok thanks. Just to mention more into the issue, I want to smoothen or de-jitter the following udp MPEG-TS stream as seen on the analyzer.
[image]

mks75, 2020-06-17:
although this is a CBR stream but doesn't look smooth. How can this be made smooth? Will "regulate" plugin help here? Please advise.