tsduck/tsduck #574: TSP LIVE INPUT, TSANALYZE

jacksonvm, 2020-05-05:
Hi,

We are looking to use TSP to analyse some live udp/rtp inputs for TR101 analysis and currently We are using this command to analyze 

tsp -v -b 1200000 -I ip 225.1.1.99:9900 -l 192.168.25.235 -P analyze -i .1  -O drop

Is there any option to display incremental data of discontinuities, I would like to see how many CC errors have occurred in past till I restart my TSP session. 

And Is there any option to enable this?

Regards,

Jackson

lelegard, 2020-05-05:
Did you have a look at plugin `continuity`? It is quite basic but reports discontinuities.

ldymek, 2020-05-29:
about this, any chance for api, I would like to build web panel for tsduck..

lelegard, 2020-05-30:
> any chance for api, I would like to build web panel for tsduck..

TSDuck is essentially a C++ API, a large library, [fully documented here](https://tsduck.io/doxy/index.html). TSDuck command line tools and plugins are usually only small wrappers on top of this library. So, virtually any DTV application is possible on top of TSDuck. A couple of sample apps are provided in the `sample` directory. This can be used as a starting point to get something built and functional.

closed 2022-09-23 by lelegard