tsduck/tsduck #575: Linux installation directories

APTX, 2020-05-06:
On linux executables, libraries, and configuration files are usually not installed in the same location [1]. The documentation says that everything is installed under `PREFIX/bin`. The linux binary packages (deb, rpm) seem to install everything into `/usr/bin` as well. Is there a way to specify separate directories for exacutables, libraries, and  when installing TSDuck?

One of the more obvious issues with the way things are currently installed is that all _tsplugin_ libraries appear in shell autocomplete when looking for _ts tools_.

[1] https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard

lelegard, 2020-05-06:
Hi,

I know the usages on Linux (on all Unix systems in fact). The way TSDuck is installed is probably not optimal. This is the result of many conflicting requirements:

- Must be portable across Windows and various Unixen (Linux, macOS and potentially others).
- Simple to troubleshoot (no need to mess with different conventions of different OS, e.g. `bin` vs. `lib`, `lib` vs. `lib64`, etc).
- No hardcoded path, neither absolute nor relative. I fundamentally disagree with the pitfalls of `configure` scripts with `PREFIX` options resulting in hardcoded paths in the binaries. I wanted everything to be "moveable" anywhere.
- Each executable must be useable by default anywhere using zero external configuration (no environment variable, no complicated or non-portable trick).

The obvious simplest choice was to store everything at the same place. The code retrieves the file location of the current executable (available in any OS) and uses the same directory to look for plugins and configuration files.

> Is there a way to specify separate directories for executables, libraries, and when installing TSDuck?

You can install TSDuck in a local environment (`/usr/local`, `$HOME/local` or whatever). See [here](https://tsduck.io/doxy/building.html#nonstdinst). But this does not separate executables and plugins.

Additionnally, you may add new plugins or relocate existing ones somewhere else using the `TSPLUGINS_PATH` environment variable (inspired by `LD_LIBRARY_PATH` on Linux), although it may open security issues. It also contradicts the (self-imposed) requirement for "useable by default anywhere using zero external configuration".

> One of the more obvious issues with the way things are currently installed is that all tsplugin libraries appear in shell autocomplete when looking for ts tools.

Here, you score a point. This is the result of a mistake. The `.so` files are installed with execute permission. This is not necessary. I will revert the access modes to 644. They won't appear in shell autocompletion anymore.

lelegard, 2020-05-06:
> The .so files are installed with execute permission. This is not necessary. I will revert the access modes to 644.

Done. This was a mistake with all `tsplugin_*.so` which were installed with 755. The big `tsduck.so` which contains most TSDuck code was already installed with 644.

Note that `gcc` creates shared objects with execute rights by default. On Linux, this is useless. But note that [this article](https://unix.stackexchange.com/questions/40587/why-are-shared-libraries-executable) mentions that this is required on HP-UX for security reasons. This is a very good reason in my opinion: on HP-UX you can't map executable memory pages from a file without execute permission. Linux used to be lousy on security in the beginning compared to more professional OS of that time and never completely caught up, unfortunately.

APTX, 2020-05-06:
Thank you for the quick fix!

Would you be open to adding `make` (or `Makefile`) options to set these directories in the future?

This feature is more for making a packages for distributions, not something you would need by default. Such packages are meant to do all the configuration so everything is ready to go anyway. I realized this issue when writing an [ebuild](https://wiki.gentoo.org/wiki/Ebuild) for my own use. Though I have to say this is a minimal inconvenience at this point.

Installing files in the directories the system expects _should_ remove the need to do any hacks like hardcoding paths. I don't know why `configure` would do that.

> useable by default anywhere using zero external configuration

As TSDuck is perfectly happy with running from a single directory I was surprised that everything isn't built in one build directory. Currently all the build artifacts are scattered around multiple sub-directories in `src`. `setenv.sh` is inconvenient for me as it requires to use `bash`.

lelegard, 2020-05-06:
> Installing files in the directories the system expects should remove the need to do any hacks like hardcoding paths. I don't know why configure would do that.

This is true for `tsduck.so` (common code library) but not for plugins `plugin_*.so` and extensions `tslibext_*.so`. The former is explicitly linked with all application. So the standard system-specific shared object activation applies. But the latter are dynamically looked up and activated using `dlopen()`. So the application needs to know where to find them. `tsp` does not known any list of predefined plugins. It uses plugin names from the command line and lookup the corresponding files. Similarly, the command `tsp -l` searches all possible plugin files. So, it needs to know where to search.

If plugins are located in the same directory on Windows, in relative path `../lib` on macOS and 32-bit Linux, on `../lib64` on 64-bit Linux, etc., then this is no longer transparent to the source because the file lookup must be different.

APTX, 2020-05-10:
I just realized that TSDuck installs the udev rules in the wrong place for my distribution:
```
 * QA Notice: udev rules should be installed in /lib/udev/rules.d:
 *
 *   /etc/udev/rules.d/80-tsduck.rules
 *
```
Is there a way to set the udev path? Alternatively would it be possible to have `make install` skip installing the udev rules?

lelegard, 2020-05-10:
Hi,

Which distribution are you using?

I had used `/etc/udev/rules.d` for TSDuck from the beginning (15 years) without thinking of checking if this was the best place. I just tried to understand the difference between `/etc/udev/rules.d` and `/lib/udev/rules.d`. All posts I read, including [this one](https://linuxconfig.org/tutorial-on-how-to-write-basic-udev-rules-in-linux) and [this one](https://unix.stackexchange.com/questions/333697/etc-udev-rules-d-vs-lib-udev-rules-d-which-to-use-and-why), seem to agree that `/lib/udev/rules.d` is for system-installed non modifiable rules while `/etc/udev/rules.d` is made for custom rules which are manually created by the system administrator. If two files with the same name exist in both directories, the custom one in `/etc/udev/rules.d` is used.

As a consequence, `/lib/udev/rules.d` seems a better place for TSDuck udev rules. I will update this, independently of any distribution.

lelegard, 2020-05-10:
I have pushed the fix moving udev rules to `/lib/udev/rules.d`. I checked the `.rpm` on Fedora and the `.deb` on Ubuntu. The old file in `/etc/udev/rules.d` has disappeared and the new one appeared in `/lib/udev/rules.d`.

lelegard, 2020-06-01:
@APTX,

> As TSDuck is perfectly happy with running from a single directory I was surprised that everything isn't built in one build directory. Currently all the build artifacts are scattered around multiple sub-directories in src. setenv.sh is inconvenient for me as it requires to use bash.

See #588

lelegard, 2020-06-04:
@APTX, @piotrserafin, all,

I plan to solve the file location issue and I would like to collect your feedback.

Requirement:
- On a given platform, one binary should fit all configurations, no need to compile differently depending on the target configuration (i.e. no `configure`-like script).
- It should continue to work as it does today because 1) this is the way it works on Windows, 2) this is the way it works in the build area.

Propose solution:
- No code change:
  - The main big `tsduck.so` is in `/usr/lib`. It is also used by third-party applications.
- Code change:
  - The plugins are searched in the following order:
    - all directories from `$TSPLUGINS_PATH`
    - same directory as main executable
    - Linux and macOS only:
      - directory `../lib/tsduck/` from main executable
    - default system activation for shared libraries (direct use of `dlopen()` on Linux)
  - The configuration files (`tsduck.*.names` and `tsduck.*.xml`) are searched in the following order:
    - all directories from `$TSPLUGINS_PATH`
    - same directory as main executable
    - Linux and macOS only:
      - directory `../../etc/tsduck/` from main executable
      - directory `../lib/tsduck/` from main executable
      - all directories from `$LD_LIBRARY_PATH`
    - all directories from `$PATH`
- On Linux x86_64 platforms, use `lib64` instead of `lib`.

So, instead of copying everything into `/usr/bin`, the package would typically install:
 - commands in `/usr/bin`
 - `tsduck.so` in `/usr/lib[64]`
 - plugins and configuration files in `/usr/lib[64]/tsduck`

@piotrserafin, comparing this with your [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=tsduck) [in AUR for Arch Linux](https://aur.archlinux.org/packages/tsduck/), I see the following differences:
 - use `/usr/lib64` instead of `/usr/lib` on x86_64, does Arch Linux use `/usr/lib64`? why not?
 - move `tsduck.so` in `/usr/lib[64]` (you leave it in `/usr/bin`)
 - move configuration files in `/usr/lib[64]/tsduck` (you leave them in `/usr/bin`)
 - move plugins to `/usr/lib[64]/tsduck` (you move them to `/usr/lib/tsduck/plugins` but I do not see any reason to create an additional level of directory)
 - no need to create profile script to define `TSPLUGINS_PATH` (but that is the original purpose)

Thanks for your feedback.

APTX, 2020-06-04:
If possible, please name it _**lib**tsduck.so_. `-ltsduck` will not find `tsduck.so` (you have to write it like `-l:tsduck.so` or use a full path, which is uncommon knowledge). The static library is already named `libtsduck.a` meaning that usually the static library will be found instead of the dynamic one. This is likely not what is expected. This also makes writing build scripts for 3rd party tools/plugins more complicated.

Consider putting files that are not config files (like the xml models and `*.names` files) or any other files that contain static data that isn't usually modified by the user under `/usr/share`.

I don't think it's that useful to search `$PATH` and `$LD_LIBRARY_PATH` for the config files.

`lib`/`lib64` is not universal among linux distributions so some distributions may need to override it.

Currently I install TSDuck into `/opt/tsduck` so that it can all be in one directory that is also not used by anything else. I add that path to the system environment.

lelegard, 2020-06-04:
Agreed on most points.

> Consider putting files that are not config files (like the xml models and *.names files) or any other files that contain static data that isn't usually modified by the user under `/usr/share`.

There is no such file in TSDuck. 

> `lib/lib64` is not universal among linux distributions so some distributions may need to override it.

Currently, `lib64` is assumed when the OS is Linux and the architecture is x86_64. I should add the condition that `/usr/lib64` exists on the build system (assumed to be a similar distribution).

lelegard, 2020-06-05:
> Consider putting files that are not config files (like the xml models and *.names files) or any other files that contain static data that isn't usually modified by the user under /usr/share.

I am unsure to understand correctly your sentence.
1. Do you mean that `tsduck.*.xml` and `tsduck.*.names` should be in `/usr/share/tsduck`.
2. Or do you mean that _other_ files (there is none) should be there, meaning leaving `tsduck.*.xml` and `tsduck.*.names` in `/usr/lib/tsduck`?

The `tsduck.*.xml` and `tsduck.*.names` files are constant configuration files for TSDuck. Using `/usr/lib/product` or  `/usr/share/product` seems to depend on the product. Some of them, such as grub or systemd, use both without clear rationale explaining why which file is in which zone. Since `/usr/lib/tsduck` needs to be created for plugins anyway, I see no reason to scatter TSDuck files in too many different places.

APTX, 2020-06-05:
I did mean the first one. `/usr/share` is intended for "architecture independent data", which are things like static data files, documentation, example files.

To use grub as an example (at least on my system): Grub puts its modules (or "plugins") into `/usr/lib/grub/<platform>` where `<platform>` are things like `x86_64-efi` (64-bit UEFI BIOS) or `i386-pc` (old non-UEFI BIOS). The plugins are built for their respective platforms. Grub puts things like themes and fonts into `/usr/share/grub`. Those files can be _shared_ between all platforms.

lelegard, 2020-06-05:
Hi,

I made the file reorganisation. The .rpm and .deb package are conformant. You may test if you like.

By renaming `tsduck.so` as `libtsduck.so` and declaring it as a "standard" library, I discovered something I never noticed in 20 years of Linux usage. In the Red Hat family (Fedora, CentOS), all standard shared libraries have executable rights. They _must_ have them. If they haven't, the files work but the shared library is not considered as a provision of the RPM package and this creates dependency issues. On the other hand, in the Debian family (Ubuntu, Raspbian), no shared library has executable rights. Weird....

APTX, 2020-06-06:
I checked the [nightly builds](https://tsduck.io/download/prerelease/), but the latest build does not have the changes yet. Build 1834 should have them, but when I checked the latest one was 1832.

lelegard, 2020-06-06:
1834 is there now. Nightly builds are published every night at 4:10 UTC.

Generally speaking, using nightly builds is essentially useful to Windows users because it takes a long time to rebuild on Windows (typically one hour or more). For Linux or macOS users, TSDuck takes only a few minutes to rebuild, provided that you specify `make -j10` for instance.

lelegard, 2020-06-06:
I still have an issue with `/usr/lib` vs `/usr/lib64`. We must find the right place, distro by distro. If we put `libtsduck.so` in the wrong place for a given distro, then the executables don't start, failing to find `libtsduck.so`.

Note that when everything was in the same place, we never had similar issues thanks to the `-rpath '$ORIGIN'` linker option. I know you did not like it, but it worked and was reliable.

In version 1834, I use a simple principle: when the OS is Linux (ie. not macOS) AND the architecture is x86_64 AND `/usr/lib64` already exists, then we use `/usr/lib64`. Otherwise we use `/usr/lib`. With Fedora, CentOS and Ubuntu, this works. The first two use `/usr/lib64` and the last one uses `/usr/lib`. 

But I just installed a Debian VM and it failed. Debian uses `/usr/lib`, just like Ubuntu, but `/usr/lib64` exists and is not empty:
~~~
$ ls -l /usr/lib64
total 0
lrwxrwxrwx 1 root root 32 May  1  2019 ld-linux-x86-64.so.2 -> /lib/x86_64-linux-gnu/ld-2.28.so
~~~
When `libtsduck.so` is in `/usr/lib64`, as a result of the previous algorithm, it is not found by `ldd`.

Of course, we can test distro by distro; if Debian ... else if Fedora ... etc. But users of other distros will complain. So, I would like to find a more reliable method.

I can try to use `/usr/lib64` only when `/usr/lib64/libc.so` exists, indicating that `/usr/lib64` seems really used for system-wide shareable libraries. But there is not guarantee that it will work on all distros if the file exists but the actual `libc.so` is somewhere else.

I looked at the content of `/etc/ld.so.conf.d/` but it is too different among distros to be usable.

For a given distro, there is always a way. But finding a simple method for all distros is a challenge. It seems such an obvious issue that I thought it would have been answered but Google does not exhibit any solution, just people talking about their favorite distro, claiming this is the _only right way_.

Any advice ?

APTX, 2020-06-07:
I think the best way to handle these cases is to have a "library path" setting and leave value up to whoever packages the TSDuck for a specific distribution. For example, see [cmake `GnuInstallDirs`](https://cmake.org/cmake/help/latest/module/GNUInstallDirs.html) which lets you change the installation paths for various types of files.

lelegard, 2020-06-07:
> the best way to handle these cases is to have a "library path" setting and leave value up to whoever packages the TSDuck for a specific distribution.

It already exists using `make install USRLIBDIR=/foo/bar/usr/lib47`.

I was looking for a generic reliable default. OK, let's make it simple. I will slightly improve the default behaviour and use `lib64` only when `/usr/lib64/libc.so*` exists. That will be adequate for the Red Hat and Debian families. Then, `USRLIBDIR` can always be used in specific cases for specific packaging.

lelegard, 2020-06-07:
One more question, for Debian users.

In the Red Hat family, each package is always identified with the distro (RHEL/CentOS vs. Fedora) and version of the distro in the package file name. Example:

- For RHEL/CentOS 8.x : tsduck-3.21-1819.el8.x86_64.rpm
- For Fedora 32 : tsduck-3.21-1819.fc32.x86_64.rpm

In the Debian family, I don't see any such convention. Currently, in that family, I produce TSDuck binaries for Ubuntu (64-bit Intel) and Raspian (ARM 32-bit) only. So, the difference is implicit on the architecture:

- Assumed to be latest Ubuntu : tsduck_3.21-1819_amd64.deb
- Assumed to be latest Raspbian : tsduck_3.21-1819_armhf.deb

If I decide to produce packages for Debian 10.x and Ubuntu 20.x on the same architecture, how would you identify the various packages? Note that GitHub releases provides assets which are different by file name only. The can't be in different directories (possibly with same name).

I can still decide my own convention as follow, but is there any widely accepted convention?
- tsduck_3.21-1819.ub20_amd64.deb
- tsduck_3.21-1819.deb10_amd64.deb
- tsduck_3.21-1819.rasp10_armhf.deb

piotrserafin, 2020-06-08:
Hello @lelegard,

> @piotrserafin, comparing this with your [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=tsduck) [in AUR for Arch Linux](https://aur.archlinux.org/packages/tsduck/), I see the [...]
In Arch /usr/lib64 is only symbolic link to /usr/lib but it is worth to follow same convention will change it.
>     * move `tsduck.so` in `/usr/lib[64]` (you leave it in `/usr/bin`)
Ok, make sense.
>     * move configuration files in `/usr/lib[64]/tsduck` (you leave them in `/usr/bin`)
Ok.
>     * move plugins to `/usr/lib[64]/tsduck` (you move them to `/usr/lib/tsduck/plugins` but I do not see any reason to create an additional level of directory)
Same with me.
>     * no need to create profile script to define `TSPLUGINS_PATH` (but that is the original purpose)
Agreed.

Will adapt those changes this week.

lelegard, 2020-06-08:
Hi @piotrserafin 

Following the exchanges above with @APTX, I modified the placement of configuration files with architecture-dependent and independent files in distinct places. The file layout is now:
~~~
/etc/security/console.perms.d/80-tsduck.perms
/lib/udev/rules.d/80-tsduck.rules
/usr/bin/tsanalyze
...
/usr/bin/tsversion
/usr/lib/libtsduck.so
/usr/lib/tsduck/tsplugin_aes.so
...
/usr/lib/tsduck/tsplugin_zap.so
/usr/share/tsduck/tsduck.channels.model.xml
...
/usr/share/tsduck/tsduck.tables.model.xml
~~~

piotrserafin, 2020-06-08:
Hello @lelegard ,

Great, PKGBUILD will simplify after those changes. This is not yet part of 3.21-1819? When you are planning new release? 

Best Regards,
Piotr Serafin

APTX, 2020-06-16:
I tested the new way TSDuck installs and it seems to work perfectly out of the box on gentoo.

I noticed two things:
* I can not build with dektec in the ebuild, as the build script is not allowed to download anything. I don't really mind this as I do not own any such devices.
* The prebuilt packages come with documentation files (like `tsduck.pdf`). I checked the documentation and I couldn't find anything about installing those. I could just install the files from the `doc` directory, if there is no install command for documentation.

lelegard, 2020-06-16:
> I can not build with dektec in the ebuild, as the build script is not allowed to download anything. I don't really mind this as I do not own any such devices.

I cannot embed Dektec library in TSDuck source, the static library is in binary form only. Moreover, Dektec regularly releases new versions, typically each time they have new devices. So, you need to download the latest version yourself. This is automated in the `make` process. If you cannot download Dektec libraries, then you can't build Dektec support. This may be fine for you but is not acceptable as a generic delivery method for Gentoo. 

In the DTV industry, a lot of people use Dektec devices. When working in the broadcast area, virtually everyone use Dektec devices. They are relatively inexpensive (for professional devices, compared with production modulators for instance), flexible, versatile, reliable. The perfect lab equipment. (No, I am not paid by Dektec😃). Many people use TSDuck _because_ it directly drives Dektec devices. So, it is not possible to disable Dektec support for all users in a given distro. It can be solely for you own usage.

> The prebuilt packages come with documentation files (like tsduck.pdf). I checked the documentation and I couldn't find anything about installing those. I could just install the files from the doc directory, if there is no install command for documentation.

Unlike binaries or configuration files, each distro has its own way of storing docs and license files. So, the docs are not installed in the generic `make install`. You have to handle that case by case. For RPM-based distros for instance, the docs are provided in a `%doc` directive in `tsduck.spec`. If there is some equivalence with Gentoo, you have to specify it in ebuild. I have never used Gentoo and I have no idea.

APTX, 2020-06-16:
Is there a way to build TSDuck with dektec without TSDuck trying to download anything? The build is done in a sandbox and downloading files is not allowed. I could make a separate ebuild for the dektec library, but I'm not sure how to make TSDuck use that or how that works regarding license. I also can't really test any of it beyond making sure it builds.

I'll see what I can do with the documentation. I assumed I would have to install it "by hand", but I was just wondering if there was an `install-docs` or similar.

lelegard, 2020-06-16:
> Is there a way to build TSDuck with dektec without TSDuck trying to download anything? The build is done in a sandbox and downloading files is not allowed. I could make a separate ebuild for the dektec library, but I'm not sure how to make TSDuck use that or how that works regarding license. 

You may download the Dektec Linux SDK somewhere first. This is a file named `LinuxSDK_v2020.05.1.tar.gz` for instance. The version number is in the file name. Then, build TSDuck with `make DTAPI_ORIGIN=/your/path/LinuxSDK_v2020.05.1.tar.gz`.

I use that variable to rebuild TSDuck from scratch in empty places and avoid downloading the Dektec SDK again and again. You can it for the same reason.

You can use the script `dektec/dtapi-config.sh` to get the URL:
~~~
$ dektec/dtapi-config.sh --url
https://www.dektec.com/products/SDK/DTAPI/Downloads/LinuxSDK_v2020.05.1.tar.gz
~~~

> I also can't really test any of it beyond making sure it builds.

Honestly, if it builds, it will work. And if it does not work, it won't be because of your environment. I started using Dektec device around 2003 with their very first DTA-100 ASI output device and I trust their software.

> I'll see what I can do with the documentation. I assumed I would have to install it "by hand", but I was just wondering if there was an `install-docs` or similar.

There is no `make install-docs`  because, honestly, I have no idea where to install the docs in a place which would satisfy all distros...

APTX, 2020-06-24:
I came across a DekTec modulator and tried using it with TSDuck.

Just building TSDuck with the DekTec modulator is not enough to use it. The SDK also contains sources for linux drivers in the form of kernel modules that need to be installed for the device can be detected by TSDuck. The license in the DekTec linux SDK appears to be 2-clause BSD, so it should be possible to make a package for it as well. Is there a way to make TSDuck use an extracted version of the DekTec SDK? If not I probably could have both packages download it.

I've tried to output to the modulator, but as TSDuck says ISDB-T is not supported I tried DVB-T instead. I couldn't use the output but I assume that it works.

I've solved the docs issue.

lelegard, 2020-06-24:
Although packaged by Dektec in the same zip file, the SDK and the drivers have distinct usages and should have distinct distribution channels or packages.

The SDK is statically linked with the application (Dektec provides a `.o` file only). So, it ends up inside `libtsduck.so` and is distributed everywhere, whether you actually use Dektec devices or not.

The drivers should be packaged as a DKMS module which is installed only on systems where you actually use Dektec devices. The [job is already done](https://github.com/tsduck/dektec-dkms) for `.rpm` and `.deb` packaging. If you want to add another type of packaging for other distros, you may adapt this or, better, contribute to add the new packaging in the [dektec-dkms project](https://github.com/tsduck/dektec-dkms).

APTX, 2020-06-25:
I've decided to follow your suggestion and have a kmod package and have the TSDuck package use the DekTec SDK by itself.

Just to clear up one thing I initially wanted to make a "DekTec SDK" package that would install the kernel modules and DTAPI files, not have TSDuck install the kernel modules.

When building TSDuck with DekTec this warning appears:
```
 * QA Notice: The following files contain writable and executable sections
 *  Files with such sections will not work properly (or at all!) on some
 *  architectures/operating systems.  A bug should be filed at
 *  https://bugs.gentoo.org/ to make sure the issue is fixed.
 *  For more information, see:
 * 
 *    https://wiki.gentoo.org/wiki/Hardened/GNU_stack_quickstart
 * 
 *  Please include the following list of files in your report:
 *  Note: Bugs should be filed for the respective maintainers
 *  of the package in question and not hardened@gentoo.org.
 * --X --- --- usr/lib64/libtsduck.a:DTAPI64.o
 * RWX --- --- usr/lib64/libtsduck.so
```
The warning suggests this already, but I did make sure that the issue is with the DTAPI object file itself. I'm not sure what can be done about this, if anything.

TSDuck and its DekTec support is working despite the warning.

lelegard, 2020-06-25:
The DTAPI library is shipped by Dektec in binary format, "as is". If there is something that does not make the QA check happy, then we have to live with it. No other option.

closed 2022-09-23 by lelegard