tsduck/tsduck #576: TSDuck and OSCam connection?

elettrone1, 2020-05-07:
Hi @lelegard 

Is it possible to connect TSDuck to OSCam in order to watch a scrambled channel?

What I would do is asking for video, audio, ECM  to the DVB device and then have a way to connect TSDuck to OSCam's DVBAPI listening port or socket /tmp/camd.socket. 

Is there a plugin to do this? 

tsp -v -I dvb --frequency 11766000000 --fec-inner 3/4 --modulation 8-PSK --polarity vertical --symbol-rate 29900000 --delivery-system DVB-S2 --device-name /dev/dvb/adapter0:1 -P filter -p 0x0000 -p 0x01f5 -p 0x0259 -p 0x044d -p 0x1081 -O play -m

Thanks!

lelegard, 2020-05-07:
OSCam is a piracy device. Hackers are not welcome here. All discussions about hacking TV programs are banned.

closed 2020-05-07 by lelegard

marian3, 2020-08-06:
OSCam is about as pirate as tsduck. Just do camd.socket and you don't have to copy the oscam code to tssmartcard.

lelegard, 2020-08-06:
TSDuck is developed by TV engineers for TV engineers. All CA-related features are used to test, debug or demonstrate end-to-end TV systems which include CA systems. This is effectively used every day. TSDuck is publicly used in demo systems from Cisco or Widevine CAS for instance. 

As explained several times, hacking topics are banned here and users insisting on promoting hacking systems are banned. As a consequence of offensive comments, user @marian3 has been blocked from tsduck project.