tsduck/tsduck #577: tsp rtp quality versus

fguinet, 2020-05-07:
Hi @lelegard ,

I really don't have the same quality between :
`tsp --verbose --debug=3 -I file toto.ts -P regulate -O ip --rtp 233.233.233.223:6665`
and : 
`./ffmpeg -re -stream_loop -1 -i toto.ts -c copy -map 0 -f rtp_mpegts rtp://233.233.233.223:6665`

Do you know what could be the reason why tsduck rtp stream is glitching so much (on vlc at least).

Thanks for your answer and 'félicitations pour tout le travail accompli sur ce bel outil'

Fabien

lelegard, 2020-05-08:
TSDuck is a very generic tools for experimentation. It will never have the same stability as a dedicated IP streamer or even ffmpeg as you experience. The generic tsp framework works best either on offline file processing or, at the opposite, on stable self-regulated constant bitrate broadcast streams (tuner to modulator). It is known to be poor at VBR streams since the regulation is always late on bitrate variations. So, if you file is CBR with accurate PCR's, you will have a stable output. If your file is VBR, then the regulation won't be so good.

To improve the reactiveness of the chain, you may try this
~~~
tsp --bitrate-adjust-interval 1 --max-flushed-packets 14 \
    -I file toto.ts \
    -P regulate --pcr-synchronous --packet-burst 14 \
    -O ip --enforce-burst --rtp 233.233.233.223:6665
~~~

Also:
- RTFM for fine tuning options
- Remove the verbose and debug options, they introduce additional processing.
- The latest version is 3.20 and you may try a snapshot of the repo but I would not expect much difference.

fguinet, 2020-05-11:
I tried the command line you suggested and it change the behavior of the stream (the quality is now good but a lot of sequence are skipped)

* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
...

Thanks for your answers

Note : Here is the tsanalyse of the ts I used for testing : 

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: .......... 1 (0x0001)  |  Services: .............. 1  |
|  Bytes: ....................... 147,810,488  |  PID's: Total: .......... 5  |
|  TS packets: ...................... 786,226  |         Clear: .......... 5  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 1  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ................ 1,619,172 b/s    1,756,974 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: .................................. 730 sec (12 mn 10 sec)  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x0001  Service01 .................................... C    1,572,767 b/s  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 22,533, PID's: 2 (clear: 2, scrambled: 0)                      |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       46,404 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       46,404 b/s  |
|  0x0000  PAT .......................................... C       38,733 b/s  |
|  0x0011  SDT/BAT ...................................... C        7,671 b/s  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0x0001 (1), Original Netw: 0xFF01 (65281)         |
|  Service name: Service01, provider: FFmpeg                                  |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 763,693, PID's: 3 (clear: 3, scrambled: 0)                     |
|  PMT PID: 0x1000 (4096), PCR PID: 0x0100 (256)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C    1,572,767 b/s  |
|  0x0100  AVC video (854x358, high profile, level 3.0,   C    1,394,373 b/s  |
|  0x0101  MPEG-2 AAC Audio (und, Audio layer 0, dual ch  C      139,659 b/s  |
|  0x1000  PMT .......................................... C       38,733 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================

===============================================================================
|  PIDS ANALYSIS REPORT                                                       |
|=============================================================================|
|  PID: 0x0000 (0)                                                       PAT  |
|-----------------------------------------------------------------------------|
|  Global PID                Transport:                Discontinuities:       |
|  Bitrate: .... 38,733 b/s  Packets: ........ 18,808  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ................. 0  Unit start: .. 18,808  |
|=============================================================================|
[... 42 lines cut ...]
===============================================================================
|  TABLES & SECTIONS ANALYSIS REPORT                                          |
|=============================================================================|
|  PID: 0x0000 (0)                                                       PAT  |
|-----------------------------------------------------------------------------|
[... cut ...]
|  Service: 0x0001 (1) Service01                                              |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x0001 (1)                                         |
|      Repetition: ...... 39  ms  Section cnt: ... 18,808                     |
|      Min repet.: ....... 2  ms  Table cnt: ..... 18,808                     |
|      Max repet.: ...... 39  ms  Version: ............ 0                     |
===============================================================================

closed 2020-05-11 by fguinet