tsduck/tsduck #578: ISDB-T scanning with tsscan gives no results

APTX, 2020-05-08:
Using `tsscan` fails to find any service with my ISDB-T tuner in Japan. Is scanning supported on ISDB-T? I tried both examples, but both give an empty scan result.

In _verbose_ mode `tsscan` says it's scanning:
```scanning channel 21, 474,000,000 Hz
scanning channel 22, 482,000,000 Hz
[...]
scanning channel 68, 850,000,000 Hz
scanning channel 69, 858,000,000 Hz
```

In ISDB-T the physical channels start from 13 and have different frequencies than the ones printed (for example physical channel 21 would be ~521MHz).

In `dvb` plugin `--bandwidth` it says that the default for ISDB-T is "7-MHz", but as far as I know all channels are 6MHz.

`tslsdvb -v` lists the tuner as ISDB-T. It also correctly lists the frequency range of the tuner, but `tsscan` still tries channels outside of that frequency range.

`dvbv5-scan` requires an _initial file_ that lists the frequencies and bandwidths of physical channels to scan. Does `tsscan` require such a file? I haven't found an option to provide one.

I have a file with found services generated by `dvbv5-scan` (in its `dvbv5` format) from a successful scan. Can TSDuck read that file format? Is there an easy way of converting it into the TSDuck format? Is it possible to provide the frequency and bandwidth on the command line?

Nit: [`tsscan` documentation](https://tsduck.io/download/docs/tsduck.pdf) lists a `---first-channel` option (with 3 dashes). I assume that is not intended.

lelegard, 2020-05-08:
Hello,

Thanks for trying `tsscan` in Japan. It seems you are the first one to try. So, I will need your help to make it work correctly. I am based in Europe (a DVB area) and I have no access to any ISDB network. It is only recently that I had confirmation that ISDB tuners worked with TSDuck and received TS files from Japan. I am currently implementing most ISDB signalisation and try to make options more consistent for Japanese users.

The definition of "HF band" depends on the country. By default, the Western Europe definitions for UHF band are used. These are the frequencies you see.

There is a configuration file named `tsduck.hfbands.xml` which contains some definitions of local HF bands. For Japan, it contains the following definitions:
~~~
  <hfband type="VHF">
    <region name="japan"/>
    <channels first_channel="1" last_channel="3" base_frequency="93,000,000" channel_width="6,000,000"/>
    <channels first_channel="4" last_channel="12" base_frequency="173,000,000" channel_width="6,000,000"/>
  </hfband>
~~~

I got those definitions at the time I implemented the HF band configuration. Could you confirm that they are correct? No UHF, VHF only. Only 12 channels in two disjoint ranges of frequencies (1-3 than 4-12). If this is not correct, could you please provide a complete description of the VHF/UHF bands in Japan?

To use the current definitions with `tsscan`, you should use the command:
~~~
tsscan --hf-band-region japan --vhf-band
~~~

To my knowledge, it was never tried yet. Could you please give us the result?

With your cooperation in Japan, I am sure we will make it work.

And thanks for pointing the typo in the doc. I just fixed it.

lelegard, 2020-05-08:
Hi @APTX,

I just read a Wikipedia article mentioning that Japanese terrestrial TV moved to UHF. On [this page](https://www.teleconsult.it/rf-tools/uhf-channel-definitions-usa-canda-japan), I found that UHF channels 13 to 62 were in use in Japan.

So, the definitions in the XML file are now:
~~~
  <hfband type="VHF">
    <region name="japan"/>
    <channels first_channel="1" last_channel="3" base_frequency="93,000,000" channel_width="6,000,000"/>
    <channels first_channel="4" last_channel="12" base_frequency="173,000,000" channel_width="6,000,000"/>
  </hfband>

  <hfband type="UHF">
    <region name="japan"/>
    <channels first_channel="13" last_channel="62" base_frequency="473,000,000" channel_width="6,000,000"/>
  </hfband>
 ~~~

Since these definitions were found on an Italian (!) site, could you confirm that they are correct?

Additionally, I extended to effect of option `--japan` to include `--hf-band-region japan` in its list of equivalences.

APTX, 2020-05-09:
Hi @lelegard,

I don't know if VHF is still used in Japan (probably not). My tuner can not tune to the VHF frequencies, so I could not try them.

ISDB-T / UHF channel scanning is working. I used ` tsscan --verbose --japan --save-channels -`. `tsscan` found all the services `dvbv5-scan` found.

This is the NHK (National Broadcaster) entry:
```
  <network id="0x7FE0" type="ISDB-T">
    <ts id="0x7FE0" onid="0x7FE0">
      <isdbt frequency="557,000,000" bandwidth="6-MHz" transmission="8K" guard="1/8"/>
      <service id="0x0400" name="NHK.Am9g.1..El5~" PMTPID="0x01F0" type="0x01" cas="false"/>
      <service id="0x0401" name="NHK.Am9g.2..El5~" PMTPID="0x03F0" type="0x01" cas="false"/>
      <service id="0x0580" name="NHK.7HBS.G..El5~" PMTPID="0x1FC8" type="0xC0" cas="false"/>
      <service id="0xFFF0" PMTPID="0x1CF0"/>
    </ts>
  </network>
```

The NHK scan result more or less matches with the [`NHK.ts` analyze output](https://tsduck.io/streams/japan-dttv/NHK.analyze.txt) in the [stream repository](https://tsduck.io/streams/?name=japan-dttv).

I assume the names are garbled because they require the special character encoding. In other tools `--japan` sets the encoding. I tried adding `--default-charset ARIB-STD-B24` but that just gave `tsscan: unknown option --default-charset`.

`type="0xC0"` should be [1seg](https://en.wikipedia.org/wiki/1seg). I tried capturing a bit of it:
```
tsp -I dvb -c "NHK.7HBS.G..El5~" -P zap "NHK.7HBS.G..El5~" -O file 1seg.ts
```
`tsanalyze` says it's not scrambled and it has a low video resolution (320x180) which is what 1seg should be (`tsanalyze` calls it `Service type: 0xC0 (Data service)`).

non-1seg services should all be scrambled, `cas` probably should not be `false` on those.

___

I would also want to try scanning ISDB-S. Unfortunately I know a lot less about how it works and I am not sure if `tsscan` supports it at all. Running `tsscan` with an ISDB-S tuner makes it try the ISDB-T frequencies (`tslsdvb` identifies the tuner as ISDB-S). This does not work (as ISDB-S is a completely different frequency range).

I'm not sure how I could put some initial ISDB-S frequencies into `tsduck.hfbands.xml` (or even if it is the right file, but there is no other file). The file doesn't really appear in the documentation.

I got `dvbv5-scan` to find some services on ISDB-S, so I should be able to test `tsscan` if I figure out how to use it with ISDB-S.

lelegard, 2020-05-09:
> I assume the names are garbled because they require the special character encoding. In other tools --japan sets the encoding. I tried adding --default-charset ARIB-STD-B24 but that just gave tsscan: unknown option --default-charset.

Your interpretation is correct. When the SDT is analyzed, characters are interpreted using DVB encoding rules by default. There is a number of predefined options for character sets, standards, HF bands, etc. which can be "plugged" in the various TSDuck applications. To avoid having the syntax of each application to be cluttered by lots of useless options, these predefined options are selectively plugged depending on the requirements of each application. Sometimes, it is easy to miss a non-immediate requirement for an application (such as decoding characters in a frequency scanning application). I will plug the character sets options into `tsscan`. This is easy. Note that `--default-charset ARIB-STD-B24` is automatically selected by `--japan` in applications which use character sets options.

ISDB-S and satellite in general is a completely different story. This is based on my experience of DVB-T/T2 vs. DVB-S/S2 but ISDB is probably not very different in principle.

Here are some differences which explain why scanning is completely different:
- Terrestrial TV:
  - The number of allowed frequencies is limited in each country. They are defined by the _HF bands_ (which also means that the file `tsduck.hfbands.xml` is only valid for terrestrial TV).
  - With DVB-T (and probably ISDB-T since your scanning succeeded), the software part of the receiver does not really care about fine tuning parameters such as guard interval and transmission mode. Just try a frequency and the hardware tuner will find the other parameters by itself.
- Satellite TV:
  - The number of frequencies is much, much larger and you don't really know what they are. There is no predefined channel per frequency. You have to know what each satellite provides.
  - All tuning parameters count. If you have the right frequency but your symbol rate or FEC is incorrect, you won't receive anything.

So, to summarize:
- Terrestrial TV: "Blind" scanning is quite feasible. You just need to try each predefined channel once to find all transports. The number of tuning attempts is in an order of magnitude of 40 to 60.
- Satellite TV: "Blind" scanning is impossible in the general case. The number of combinations of frequencies and tuning parameters is in an order of magnitude of thousands. It would take weeks to blindly scan a satellite network.

This is why, for satellite, you need a first "anchor", the exact tuning parameters of at least one "reference transponder" which never change. The NIT of this transponder is read, all _satellite_delivery_descriptors_ (DVB-S/S2) are interpreted to locate all associated transponders. Each of these transponder is then tested to check what is inside it.

This means that you can check only what is referenced in the NIT of the reference transponder, typically one commercial operator (CanalSat, BskyB, etc. in Europe).

With `tsscan`:
- If frequency and tuning parameters are specified, we assume "NIT scanning". The tuning parameters describe the reference transponder.
- With `--vhf-band`, we use blind scanning of the VHF band.
- With `--uhf-band` (the default), we use blind scanning of the UHF band.

Note that NIT scanning can also be used in terrestrial TV if the _terrestrial_delivery_descriptors_ are all present and accurate. This is the case in the UK but neither in France nor Japan. The NIT of the French DTTV contains all descriptors but the frequencies are dummy since there is no cell-specific reinjection of PSI/SI. The NIT of the Japanese DTTV contains only one single descriptor, the one for the current TS, which is useless because we already know its tuning parameters by definition.

For ISDB, the delivery descriptors are different from the DVB ones. I know the ISDB _terrestrial_delivery_descriptor_ because it is in ARIB STD-10 for which an English version is freely available. But I do not know how is designed the ISDB _satellite_delivery_descriptor_. It is not in ARIB STD-10. It may be in ARIB STD-B16 (ISDB Standard Digital Receiver for Satellite) but this document is not available in English. So, before being able to scan an ISDB-S network, I need access to the specification of the ISDB _satellite_delivery_descriptor_. 

This is an endemic problem of Japanese TV, compared to DVB:
- The latest version of each standard is available only to ARIB members and only in Japanese, _never_ in English, never free access.
- Sometimes, older versions are available for free and in English. This is the case for STD-10 but not STD-16.
- No single central SI reference document (ARIB STD-10 only covers part of what ETSI EN 300 468 covers in DVB).
- Additionally, the wording is sometimes more ambiguous than in DVB standards. See examples in the TSDuck source code for the PCAT table. Also in the broken serialisation of series of _extended_event_descriptors_ in NHK streams. 

Any English version of ARIB STD-B16, anywhere ? On even a Japanese version (Google translate can help if the table of contents clearly points to the various descriptors).

APTX, 2020-05-09:
You can find the _satellite_delivery_descriptor_ (and probably all other ISDB descriptors) in [ARIB STD-B32](https://www.arib.or.jp/english/std_tr/broadcasting/std-b32.html), Fascicle 3, Chapter 3, Section 3.11. Not sure if this is enough information, but the structure is described there. _satellite_delivery_descriptor_  should be on page 77 in the pdf, the page is numbered 39 in the pdf.

lelegard, 2020-05-09:
Thanks for pointing to STD-B32. I checked the document named "service information" (STD-10) but I didn't had the idea to search the service information definitions in a document named "video coding, audio coding, and multiplexing specifications". There are succinct diagrams instead of full definitions but this should be enough.

If you have the opportunity to capture 30 or 60 seconds of a few ISDB-S streams, I would be very interested in checking them.

APTX, 2020-05-09:
Actually it seems that this is already working in TSDuck. I captured a bit of a ISDB-S service that I scanned (with `dvbv5-scan`) and there 26 entries like this one (`tstables` xml output):
```
<satellite_delivery_system_descriptor frequency="11,727,480,000" orbital_position="110.0" west_east_flag="east" polarization="right" roll_off="0.35" modulation_system="DVB-S" modulation_type="auto" symbol_rate="28,860,000" FEC_inner="4/5"/>
```

APTX, 2020-05-09:
I can share the captured ts (it's 44M), but I'm not sure how I can send the file.

lelegard, 2020-05-09:
> I can share the captured ts (it's 44M), but I'm not sure how I can send the file.

Any file transfer site should be OK: wetransfer.com or dropbox or any other

lelegard, 2020-05-09:
> You can find the satellite_delivery_descriptor (and probably all other ISDB descriptors) in ARIB STD-B32, Fascicle 3, Chapter 3, Section 3.11. 

> Actually it seems that this is already working in TSDuck. 

Weird, it should not work in fact. There is something wrong either in the standard or in the actual signalisation.

From ARIB STD-B32, I can see that ISDB uses the same descriptor tag and size as DVB for satellite delivery descriptor but with a slightly different layout and a few differences in the semantics. This is insane, really.
- Frequency:
  - ARIB: _The frequency shall be a field that writes a frequency (GHz). An 8-digit number, each
digit of which consists of a 4-bit BCD code, shall be used to write a frequency. The lower four digits represent the fractional part._
  - DVB: _The frequency is a 32-bit field giving the 4-bit BCD values specifying 8 characters of the frequency value. For the satellite_delivery_system_descriptor the frequency is coded in GHz, where the decimal point occurs after the third character (e.g. 011,75725 GHz)._
  - So, the same binary value should give _f_ x 10 in ISDB for _f_ in DVB. If your scanning works, either the standard is wrong or the values are incorrect for ISDB (but correct for DVB).
- Orbital position:
  - ARIB: _The orbit shall be a field that writes an orbital position (degrees). A 4-digit number, each digit of which consists of a 4-bit BCD code, shall be used to write a position. The lower four digits represent the fractional part._
  - DVB: _The orbital_position is a 16-bit field giving the 4-bit BCD values specifying 4 characters of the orbital position in degrees where the decimal point occurs after the third character (e.g. 019,2°)._
  - The ARIB standard is obviously incorrect. The fractional part of a 4-digit number cannot be the lower 4 digits to represent an angle up to 180°! Probably, we should read lower 4 _bits_, meaning lower 1 digit, making it compatible with DVB.
- Modulation:
  - ARIB: There is a 5-bit field.
  - DVB: 3 fields of 2, 1, 2 bits over those 5 bits. Except for QPSK which ends up with the same binary pattern, everything else is different. 

At least, the ISDB terrestrial delivery descriptor is different from the DVB one. So, you know which is what. 

If I can inspect a few ISDB-S captures, I could maybe have more information.

Additionally, a copy of ARIB STD-B16 may help if it contains a full specification of the descriptors, as found in STD-B10 for terrestrial descriptors. STB-B32 just contains sketches of the descriptors, not a full specification. If the definitions were consistent, that would be enough. But since the accompanying text is inconsistent, a complete specification would probably help.

APTX, 2020-05-09:
I've sent the TS through wetransfer.com to the email in your commit messages.

I did not actually check if the numbers are correct, but you are right, at least the frequency is wrong.

The one _satellite_delivery_system_descriptor_ I pasted is for a service I found in the scan and the frequency is 100 times higher in the descriptor I pasted.

For comparison this is the information `dvbv5-scan` stored for that service:
```
[BS朝日1]
        SERVICE_ID = 151
        NETWORK_ID = 4
        TRANSPORT_ID = 16400
        VIDEO_PID = 320
        AUDIO_PID = 321
        PID_0d = 1244 1243 1242 1241 1240
        PID_06 = 326 325
        LNB = 110BS
        FREQUENCY = 11727480
        STREAM_ID = 16400
        DELIVERY_SYSTEM = ISDBS
```
Most of the values TSDuck is looking for are not stored (or are stored under non-obvious names).

lelegard, 2020-05-09:
Got the file, thanks. The analysis looks good but the satellite delivery descriptors look weird. They are interpreted as the DVB equivalent one. I need to find more about the _exact_ semantics of the ISDB interpretation of this DVB descriptor.

APTX, 2020-05-09:
I found another _satellite_delivery_system_descriptor_ definition in [ARIB STD-B10](https://www.arib.or.jp/english/std_tr/broadcasting/std-b10.html), Part 2, Section 6.2.6. Is this any better? You mentioned B10 before so I'm not sure if you haven't seen it or if it was useless.

lelegard, 2020-05-09:
Yes, it's much better indeed. I originally missed it because it was the DVB one and I thought there was an ISDB one, just like the terrestrial one for which the DVB one and ISDB one are different. 

The modulation option is explained for ISDB. The frequency and orbital position are identical to DVB. As a consequence, the wording in ARIB STD-B32 is completely wrong ! Not quite satisfying for an international standard...

I will implement the ISDB variant based on the context (DVB vs. ISDB).

nuclidjp, 2020-05-10:
I am using version 3.21-1772 and I get no results when I scan.
`tsscan --verbose --japan --save-channels channels.xml`
If you specify the frequency, you can capture normally.

Japanese TV frequency channels
It's Japanese.
https://ja.wikipedia.org/wiki/%E3%83%86%E3%83%AC%E3%83%93%E5%91%A8%E6%B3%A2%E6%95%B0%E3%83%81%E3%83%A3%E3%83%B3%E3%83%8D%E3%83%AB

APTX, 2020-05-10:
Hi @nuclidjp !

I ran the scan again and channels are found for me. Do you get any output at all? If you are on linux, does scanning with `dvbv5-scan` work for you? I have not tried it on windows and I don't really have a way to do so.

nuclidjp, 2020-05-10:
Hi @APTX 

dvb-tools is not installed.
I have TSDuck installed on my Raspberry Pi.
I have tried it several times, but tsscan does not return any results.
Is my command wrong?

APTX, 2020-05-10:
The command is correct.

Do you get any output at all? When I run the command it prints
```
$ tsscan --verbose --japan --save-channels channels.xml
scanning channel 13, 473,000,000 Hz
scanning channel 14, 479,000,000 Hz
[...]
scanning channel 26, 551,000,000 Hz
UHF channel 26 (551 MHz)
* UHF channel 26 (551 MHz)
  Transport stream id: 32737, 0x7FE1
[...]
saving channels.xml
```

I would make sure the tuner is actually working on the Raspberry Pi. `tslsdvb` should print you all the tuners on your system. Check `dmesg` if there are any errors related to the tuner. The tuner might also not get any signal.

lelegard, 2020-05-10:
Hi @APTX,

I have added the interpretation of the satellite delivery descriptor in ISDB context. I have also added ISDB-S entries in XML channel files. So, now, NIT-based scan should work on ISDB-S. Could you please try?

I try to maintain in the TSDuck user's manual a list of tested tuners. No ISDB-S tuner is present for now. Could you please indicate which models you tested with TSDuck? Interesting details are: Linux and/or Windows tests, number of tuners, supported modulations, USB/PCIe, additional useful info such as required firmware files, etc.

Thanks for your cooperation.

APTX, 2020-05-11:
How do I actually make `tsscan` do the scan over ISDB-S? It always tries the UHF frequencies.

When I added `--delivery-system ISDB-S` (which the documentation says is set automatically) I only get:
```
tsscan: DVB frontend FE_SET_TONE error: Unknown error 524
```

I'm using a PCIE card called PT3. It has 2 ISDB-T and ISDB-S tuners (so 4 in total). The linux driver is DVB_PT3. I haven't tried it on windows.

nuclidjp, 2020-05-11:
Mygca S270 did not give tsscan results, but Mygica S2870 did.

lelegard, 2020-05-11:
Hi @APTX 

> How do I actually make tsscan do the scan over ISDB-S? It always tries the UHF frequencies.

As I explained in a long post above, scanning a satellite network cannot be a "blind" scan. It can only be a NIT-based scan (at least with DVB-S/S2, so probably with ISDB-S as well).

You need prior knowledge of a reference transponder containing a complete NIT for the network. Any TS can be used if a complete NIT is present everywhere. The concept of "reference" transponder is only an operational artifact. Large operators change their transponders from time to time (new frequencies, modified more efficient modulation parameters). But they try to keep one well-known transponder on a stable frequency.

So, specify modulation parameters to `tsscan`: frequency and whatever you used with the `dvb` plugin to read that TS. `tsscan` will tune to that TS, read the NIT and all its satellite delivery system descriptors and will then scan all described TS.

lelegard, 2020-05-11:
> Mygca S270 did not give tsscan results, but Mygica S2870 did.

What does `tsscan --verbose` say with the S270? It should at least display all attempted frequencies. Any error message? If the same card can receive a TS using the `dvb` plugin, it should be able to scan. Otherwise, they is a problem with TSDuck and I would like to find it and fix it.

nuclidjp, 2020-05-11:
S270 works with `dvb` plugin.
`tsscan --verbose` will show all frequencies.
All strengths are 0%.
`tsscan` takes about 5 seconds and there is no error message.

APTX, 2020-05-11:
Hi @lelegard,

When you wrote:
> I have also added ISDB-S entries in XML channel files.

I thought that you added that "reference transponder" and the scan would be automatic.

I did not get the ISDB-S .ts I shared using TSDuck, I used `dvbv5-zap`.

___
```tsscan: DVB frontend FE_SET_TONE error: Unknown error 524```
This error is likely because ISDB-S (or at least my tuner) does not use the DiSEqC commands (as that seems to be a European standard). Removing those calls (and `FE_SET_TONE`) got rid of `error 524`.

`dvbv5-scan` also tries to set `FE_SET_TONE`, but ignores the error.

Then I encountered this:
```
// For satellite, Linux DVB API uses an intermediate frequency in kHz
freq = uint32_t(params.lnb.value().intermediateFrequency(params.frequency.value()) / 1000);
```
So it turns out that the "frequency" in the `dvbv5-scan` file is not a real frequency and I had to figure out how to convert it in the other direction. I still don't quite know how to do this conversion, but I got a frequency that works and I got `tsscan` to scan.

I got this channel xml in the end:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <network id="0x0004" type="ISDB-S">
    <ts id="0x4450" onid="0x0004">
      <isdbs frequency="11,727,480,000" symbolrate="28,860,000" polarity="right"/>
      <service id="0x00C0" name="WOWOWライブ" provider="WOWOW" PMTPID="0x0201" type="0x01" cas="true"/>
    </ts>
    <ts id="0x40D0" onid="0x0004">
      <isdbs frequency="11,880,920,000" symbolrate="28,860,000" polarity="right"/>
      <service id="0x008D" name="BS日テレ" provider="BS日本" PMTPID="0x0101" type="0x01" cas="false"/>
      <service id="0x008E" name="BS日テレ" provider="BS日本" PMTPID="0x0201" type="0x01" cas="false"/>
      <service id="0x008F" name="BS日テレ" provider="BS日本" PMTPID="0x0203" type="0x01" cas="false"/>
      <service id="0x02E8" name="　" provider="BS日本" PMTPID="0x0401" type="0xC0" cas="false"/>
      <service id="0x02E9" name="　" provider="BS日本" PMTPID="0x0402" type="0xC0" cas="false"/>
      <service id="0x02EA" name="　" provider="BS日本" PMTPID="0x0403" type="0xC0" cas="false"/>
    </ts>
    <ts id="0x4730" onid="0x0004">
      <isdbs frequency="11,996,000,000" symbolrate="28,860,000" polarity="right"/>
      <service id="0x00EA" name="グリーンチャンネル" provider="グリーンチャンネル" PMTPID="0x0401" type="0x01" cas="true"/>
    </ts>
    <ts id="0x4090" onid="0x0004">
      <isdbs frequency="11,804,200,000" symbolrate="28,860,000" polarity="right"/>
      <service id="0x00D3" name="BS11イレブン" provider="日本BS放送" PMTPID="0x01F0" type="0x01" cas="false"/>
    </ts>
    <ts id="0x40F1" onid="0x0004">
      <isdbs frequency="11,919,280,000" symbolrate="28,860,000" polarity="right"/>
      <service id="0x0065" name="NHKBS1" provider="NHK" PMTPID="0x01F0" type="0x01" cas="false"/>
      <service id="0x0066" name="NHKBS1" provider="NHK" PMTPID="0x02F0" type="0x01" cas="false"/>
      <service id="0x02BC" name="NHKデータ1" provider="NHK" PMTPID="0x10F0" type="0xC0" cas="false"/>
      <service id="0x02BD" name="NHKデータ2" provider="NHK" PMTPID="0x11F0" type="0xC0" cas="false"/>
      <service id="0x02C3" name="707チャンネル" provider="NHK" PMTPID="0x17F0" type="0xC0" cas="false"/>
      <service id="0x03A1" name="Dpaダウンロード" provider="Dpa" PMTPID="0x0701" type="0xA4" cas="false"/>
    </ts>
  </network>
  <network id="0x0000" type="ISDB-S">
    <ts id="0x0000" onid="0x0000">
      <isdbs frequency="12,149,440,000" symbolrate="28,860,000" polarity="right"/>
    </ts>
  </network>
</tsduck>
```

There are still some strange things happening, like the network with id 0. For a bunch of the frequencies no PAT was found (not sure if normal or not). Overall TSDuck found less services than `dvbv5-scan`.

lelegard, 2020-05-11:
Hi @APTX,

Thanks for the feedback. The DiSEqC, set tone (not the same thing) and the intermediate frequency conversions are inherent to satellite reception. Even it the equipment are different, you must use the equivalent for your equipment. `dvbv5-scan` has to use them as well, we should try to figure out how.

First, the frequency of a transponder: There are in fact three different frequencies for each transponder: 1) the "up" one from the satport to the satellite (you don't care), 2) the "down" one from the satellite to your dish (the ~11 GHz frequency here), 3), the "intermediate" one from your dish to your tuner.

The 11 GHz "down" frequency is made to cross a long distance (~36000 km) of space but it wouldn't survive a few meters of coax cable from the dish to your tuner. So, the LNB in your dish transposes the signal to a much lower frequency, more suitable to coax cable. The only well-know frequency is the ~11 GHz "down" one because it is the same for everyone. One the other hand, the only frequency that matters to your tuner is the one that it receives, the intermediate frequency. All receivers must know the type of LNB you use to compute the conversion from down to intermediate frequency, TSDuck and `dvbv5-scan` must do it equally. Each type of LNB may have a different transposition. So, using different dishes, you have to compute a different intermediate frequency. The default transposition which is used in TSDuck is the so-called "universal LNB". You can specify other transpositions using the parameter `--lnb` (RTFM). So, you must know what kind of LNB is used in Japan (I have no idea) and if everyone uses the same. Then, you will be able to build the right `--lnb` parameter.

Then, you have to remotely program the LNB for each transponder. There are 2 parameters to send from the tuner to the LNB: the polarity (mandatory) and high/low band which affects the frequency transposition (optional, depends on the LNB). There are several ways to send those parameters: 1) the old set tone/voltage on the cable and 2) the DiSEqC protocol. Since we don't know in advance which method is accepted by the LNB, most receivers usually send both. This is what TSDuck does. If a tuner and its driver do not support one of the two methods, it would probably be better to ignore the returned error. The downside is that actual errors on equipment which need them will not be reported.

Anyway, transposing the frequency the right way and sending polarity and optional band is required for any kind of satellite reception. So, both TSDuck and `dvbv5-scan` must do it. If your version of `dvbv5-scan` works, then we should have a look in the code to see what is does differently from European or American satellite reception.

lelegard, 2020-05-11:
Hi @nuclidjp,

> S270 works with dvb plugin.
tsscan --verbose will show all frequencies.
All strengths are 0%.

The problem probably comes from the driver which does not report the signal strength. In the past, most drivers used to correctly report all tuning and signal parameters. I have noticed that recent drivers become more and more lousy, not reporting most tuning parameters for instance.

It is probable that the driver for the S270 does not report the signal strength while the one for the S2870 does. 

With the `dvb` plugin, you specify a frequency and parameters. So, the plugin directly tunes there and then, either you receive or you don't, but you don't care about the signal strength since your are interested in one specific transponder only.

With `tsscan`, you don't know in advance where the signals are. So, you try all frequencies and check the strength and quality of the signal. This is particularly important in the presence of "offsets" as found in many European countries. When offset scanning is enabled, `tsscan` scans back and forth around the central frequency of a UHF channel to find the actual offset where the reception is the best. If you get 0 strength, we assume there is no usable signal. This could be changed using some options to simply try to receive without checking the strength. That would solve the issue of a driver not reporting the signal strength. But that would make offset scanning meaningless. So, this should be an option.

nuclidjp, 2020-05-11:
Hi @lelegard 

As far as I know, the new, affordable ISDB-t USB tuner is only the S270 and PX-S1UD V2.0.
I would appreciate it if you could respond with an option.

The S2870 is an old product and is no longer in the Japanese market.
There is probably a problem with the Linux firmware and a large number of drop will occur.
When used in Windows, there is almost no drop.

APTX, 2020-05-12:
I came across a PX-S1UD V2.0 and tried scanning using TSDuck with it. I got basically the same result as @nuclidjp .

```
tsscan --verbose --japan -g --save-channels tsduck-isdb-t.xml
scanning channel 13, 473,000,000 Hz
[...]
scanning channel 17, 497,000,000 Hz
UHF channel 17 (497 MHz), strength: 0%
scanning channel 18, 503,000,000 Hz
UHF channel 18 (503 MHz), strength: 0%
[...]
saving tsduck-isdb-t.xml
```

No channels were found.

Does TSDuck only print the strength when it is low? Scanning works on PT3 and I never saw the `, strength:` part printed.

Just to be sure I also tried scanning with `dvbv5-scan` and it find services.

lelegard, 2020-05-12:
OK, let's fix ISDB-T scanning first, we'll fix ISDB-S afterward.

Since I have no access to your tuners and your TV network, with 7 hours of time zone difference, it is a bit difficult for me to debug. I checked the git log of `dvbv5-scan`. The ISDB support required a substantial amount of changes and was done by a Japanese developer (presumably directly debugging the code while tuning).

First, could you please post the complete log of `tsscan --debug --japan -g` with the S2870 (working) and also the S270 (non working). I would like to check the exact difference in debug traces.

APTX, 2020-05-12:
@lelegard could it be that the tuners are always reporting the strength as `0`?

```
        if (strength >= 0 && strength <= _opt.min_strength) {
            // Strength is supported but too low
            ok = false;
        }
```
`_opt.min_strength` is `0` by default. Due to `<=`, if strength is `0` and the tuner always returns `0`, then `ok` should always be set to `false`.

If the tuner has a lock the strength shouldn't be `0`, right?

PT3 might return that signal strength is unsupported (which is identified as `-1`) and pass this check that way.

nuclidjp, 2020-05-12:
@lelegard 

[S270.txt](https://github.com/tsduck/tsduck/files/4614725/S270.txt)
[S2870.txt](https://github.com/tsduck/tsduck/files/4614726/S2870.txt)

lelegard, 2020-05-12:
Hi @nuclidjp 

Thanks for the log. I notice that all tune requests on the S270 fail with the following error:
~~~
tsscan: error setting filter on /dev/dvb/adapter0/demux0: Connection timed out
~~~

Can you confirm that the S270 can be used with the `dvb` plugin and receives a full TS?
Could you post a full log of the following command (with ... replaced by the tuning options of an actual TS):
~~~
tsp --debug -I dvb ... -P tables -p0 -x1 -O drop
~~~
Thus, we will see what traces are produced when the S270 works.

nuclidjp, 2020-05-12:
Hi @lelegard 

`tsp -I dvb --frequency 521,142,857 -P until --seconds 60 -O file NHK.ts`
Confirmed the reception of the complete TS.

`tsp --debug -I dvb --a 0 --frequency 521142857 -P tables -p0 -x1 -O drop`
[S270.debug.txt](https://github.com/tsduck/tsduck/files/4615225/S270.debug.txt)

lelegard, 2020-05-12:
Thanks for the log. All low-level system parameters are identical, except the frequency.

The working frequency is 521,142,857 Hz. Where did you get that exact value from?

The scanning uses rounded values. The channel 21 uses 521,000,000 Hz and fails.

- Could you please try the same command with the `dvb` plugin with 521,000,000 as frequency? Does it fail with the same "error setting filter"?
- If it fails with the same error, then the definition of the UHF frequencies are probably incorrect. Do you know _exact_ values for the UHF frequencies in Japan?

lelegard, 2020-05-12:
As a test, if the plugin `dvb` fails on 521,000,000 Hz, please modify the file `tsduck.hfbands.xml`, in the last section (UHF for Japan). Replace the base frequency "473,000,000"  with "473,142,857". 
~~~
    <channels first_channel="13" last_channel="62" base_frequency="473,142,857" channel_width="6,000,000"/>
~~~
And then retry the scan.

nuclidjp, 2020-05-12:
The frequency was obtained from this software.
https://github.com/dogeel/recdvb

It also matches this site.
This is a Japanese site.
https://ja.wikipedia.org/wiki/%E3%83%86%E3%83%AC%E3%83%93%E5%91%A8%E6%B3%A2%E6%95%B0%E3%83%81%E3%83%A3%E3%83%B3%E3%83%8D%E3%83%AB

tsp --debug -I dvb --a 0 --frequency 521,000,000 -P tables -p0 -x1 -O drop
[S270.debug1.txt](https://github.com/tsduck/tsduck/files/4615656/S270.debug1.txt)

Wait for the next test,

nuclidjp, 2020-05-12:
Changed to 473,142,857.

[S270.debug2.txt](https://github.com/tsduck/tsduck/files/4615830/S270.debug2.txt)

lelegard, 2020-05-12:
OK, so the 142 kHz offset is not significant.

You did not post the log of the scan with the updated base frequency. The file S270.debug2.txt contains a tsp log, not a tsscan one.

nuclidjp, 2020-05-12:
I made a mistake.
sorry
Tsscan results are not obtained after changing to 473,142,857

lelegard, 2020-05-12:
> Tsscan results are not obtained after changing to 473,142,857

Do you get the same "error setting filter" on all frequencies ?

This is really weird. The same tuner is used with exactly the same low-level parameters (seen in the traces) and it consistently works in one case and consistently fails in the other.

The only difference is the dynamic. After opening the device, we directly tune to the target frequency in the case of the `dvb` plugin but we tune several times without closing / re-opening the device in the case of `tsscan`. 

As a last test, try this:
~~~
tsscan --debug --japan -g --first-channel 21 --last-channel 21
~~~
This will be exactly identical to tuning once to 521,142,857 Hz. Please post the complete log.

If this fails, then a developer would need to debug the problem on site, with the same tuner, receiving the same signal. I cannot do this remotely.

The most surprising part is that the error occurs in the `ioctl()`  to the demux using `DMX_SET_PES_FILTER`. This is a standard call which is made exactly the same way, with the same constant parameters, on all Linux tuners by TSDuck from the beginning. I can understand that a specific error occurs on specific tuning or configuration parameters on a specific tuners. But, here, the `ioctl()` simply says "grab all PID's from that TS".

There is one case where this call is reputed to fail: on some cheap tuners which are made to watch TV only (typically in set-top boxes), it is impossible to grab a complete TS. You have to configure a limited number of PID's. In this case, the call with the "big filter 0x2000" fails. 

But, this cannot be the case here:
- In that case, the `dvb` plugin would also fail on this tuner since it calls the exact same `ioctl()`.
- I have never seen that in 15 years in any USB or PCI tuner. This is known to happen only in open set-top boxes devices with embedded tuners.

nuclidjp, 2020-05-12:
Modified `tsduck.hfbands.txt`
I made it a txt file because I cannot send an xml file.
Please check as I may be wrong.
[tsduck.hfbands.txt](https://github.com/tsduck/tsduck/files/4618440/tsduck.hfbands.txt)

`tsscan --debug --japan -g --first-channel 21 --last-channel 21`
[tsscan --debug.txt](https://github.com/tsduck/tsduck/files/4618442/tsscan.--debug.txt)

`tsscan --verbose --hf-band-region japan --save-channels channels.xml`
[tsscan.txt](https://github.com/tsduck/tsduck/files/4618445/tsscan.txt)

I confirmed that TS can be obtained by `tsp -I dvb ...`.

lelegard, 2020-05-13:
Hi,

There is something that is not consistent between the various tests.

From `S270.txt`:
~~~
tsscan --debug --japan -g

tsscan: Debug: debug level set to 1
scanning channel 13, 473,000,000 Hz
... skipped debug traces
tsscan: Debug: [24] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: error setting filter on /dev/dvb/adapter0/demux0: Connection timed out
scanning channel 14, 479,000,000 Hz
---> continue with next channel
~~~

From `tsscan.--debug.txt`:
~~~
tsscan --debug --japan -g --first-channel 21 --last-channel 21

tsscan: Debug: debug level set to 1
scanning channel 21, 521,000,000 Hz
... skipped debug traces
tsscan: Debug: [24] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
UHF channel 21 (521 MHz), strength: 0%
---> end of scan (first and only channel to scan)
~~~

In the first case, the first tune request fails with this driver error: 'error setting filter'.

In the second case, the first tune request successfully completes. The signal strength is zero, but this is a different issue.

Are you sure that the two tests were run with same tuner on the same system?

APTX, 2020-05-13:
I managed to get PX-S1UD V2.0 to scan with TSDuck. `tsscan` really just skips over frequencies when strength is too low ([code in previous comment](https://github.com/tsduck/tsduck/issues/578#issuecomment-627225700)).

Due to the logic `tsscan` will never look for services if the signal strength is `0`, because signal strength has to be **greater than** `--min-strength` (which is `10` by default, not `0`, I was wrong about that part). `--min-strength` can not be set to a negative value so the check will **always** fail for a signal strength of `0`.

I checked that the [driver tries to return a valid signal strength](https://github.com/torvalds/linux/blob/9c7db5004280767566e91a33445bf93aa479ef02/drivers/media/common/siano/smsdvb-main.c#L1059), but it always comes out as `0` for me (could be a firmware bug). [PT3 doesn't support reading the signal strength](https://github.com/torvalds/linux/blob/89d57dddd7d319ded00415790a0bb3c954b7e386/drivers/media/dvb-frontends/tc90522.c#L756).

I changed the condition in `src/tstools/tsscan.cpp`:
```
        // Old condition
        if (strength >= 0 && strength <= _opt.min_strength) {

        // New condition
        // Consider strength == 0 as strength being unsupported
        // Allow strength to be equal to --min-strength
        if (strength > 0 && strength < _opt.min_strength) {
```
____
@lelegard I think that the error:
```
tsscan: error setting filter on /dev/dvb/adapter0/demux0: Connection timed out
```
is the unrelated error. It is an actual _error_ so should be printed without `--debug` or `--verbose`. When that error appears `strength: 0%` is not printed, but that is what @nuclidjp mentioned first. It's also the only place the error appears.

nuclidjp, 2020-05-13:
Hi, @lelegard 

The configuration has not changed.
Raspberry Pi and S270.
Do you want to control my Raspberry Pi with SSH?
If you control, I will email you the address and required files.

lelegard, 2020-05-13:
Hi @APTX 

> `// Consider strength == 0 as strength being unsupported`

No, we don't.

Strength being unsupported is a negative value. This is how it is defined in the class `Tuner`, the portable system-independent interface to a tuner:
~~~
//!
//! Get the signal strength.
//! @param [in,out] report Where to report errors.
//! @return The signal strength in percent (0=bad, 100=good).
//! Return a negative value on error.
//!
int signalStrength(Report& report);
~~~
And this is how it is implemented in the Linux variant of the class `Tuner`:
~~~
uint16_t strength = 0;
if (::ioctl(_guts->frontend_fd, ioctl_request_t(FE_READ_SIGNAL_STRENGTH), &strength) < 0) {
    const ErrorCode err = LastErrorCode();
    // Silently ignore deprecated feature.
    if (err != DVB_ENOTSUPP) {
        report.error(u"error reading signal strength on %s: %s", {_guts->frontend_name, ErrorCodeMessage(err)});
    }
    return -1;
}
~~~
When the driver does not support reporting the strength, it returns a error. In all cases of error, we return -1. We do not report "unsupported strength" as an error message but we still return -1. When the signal strength is zero, this means that the driver supports strength and the strength is zero (which means "no signal", a very normal situation when scanning the network).

So, this is a broken driver which pretends to support the strength but always return zero. A driver is perfectly allowed to not support the signal strength but it must return an error. Old drivers all supported the signal strength. I have seen that some recent drivers don't support it. But they all correctly report an error code. This is a very specific broken driver. This is the first time I see this. And I have used _a lot_ of tuner drivers on Linux.

I will try to find a way to manage this in `tsscan`, let me think about it.

@nuclidjp, could you try `tsscan` with this tuner on a Windows machine? The Windows DirectShow framework is awful, much worse than LinuxTV. But, on the other hand, Windows drivers tend to be more reliable and accurate than Linux drivers. The reason is simple, Windows drivers are provided by the tuner vendor with full hardware spec of the device. Most Linux drivers are developed by free contributors, based on reverse engineering of Windows drivers. They usually nicely address nominal cases but ofter fail to address specific details. Some tuner vendors provide Linux drivers but most don't.

lelegard, 2020-05-13:
OK, I think I found a easy workaround for your case.

In practice, the signal strength and quality are used only when scanning offsets. In some countries, the actual signal can be -3 to +3 times 166kHz from the central frequency of a channel, not the same offset from one channel to another. So, a full scan means scanning all offsets around the central frequency of each channel. In practice, depending on the tuner capabilities, you can get the signal on 4 or 5 different offsets. But the actual signal is on the offset with the best strength / quality.

In your case, you don't use offsets (no `--use-offsets`). So, reading the strength and quality is useless to make a decision, it is just an information. 

I have pushed a modification where signal strength and quality are ignored (not even read) in the absence of offset scanning. The only drawback is that you no longer have the signal strength information with a good tuner / driver. Not a big issue.

Of course, trying to scan offsets with such a broken tuner driver is not possible. But, in Japan, you don't seem to use offsets.

Could you please try the updated code?

APTX, 2020-05-14:
@lelegard I think you misunderstood my comment. It was not supposed to reflect what TSDuck does. It was supposed to describe the intent of the change I made. It mentions the **difference** fromTSDuck's behavior. `tsscan` checks for a signal lock before checking the strength, and, as far as I understand it, if you get a signal lock you have to have some signal, so signal strength of `0` indicates a bug in the driver and/or that it actually does not support reading the signal strength.

In the change you made you simply avoided checking the signal strength altogether.

nuclidjp, 2020-05-14:
@lelegard 
I want to run `tsscan` on Windows but I couldn't build the latest TSDuck.
Lots of errors C2220.
I used Visual Studio 2019.
Please get the latest pre-built TSDuck.

lelegard, 2020-05-14:
> In the change you made you simply avoided checking the signal strength altogether.

Yes, that is the whole purpose. If a tuner or driver cannot reliably report signal strength and quality, then scanning offsets does not make sense anyway since it needs to compare signal strength across all offsets around the channel central frequency. So, the idea is to completely avoid checking signal strength when offset scanning is disabled (the default).

> I want to run tsscan on Windows but I couldn't build the latest TSDuck.
Lots of errors C2220.
I used Visual Studio 2019.

Strange, I use Visual Studio 2019 as well, always with the latest update installed. Other users use down to VS 2015. It consistently builds on most versions. Maybe there is a missing dependency (run the PowerShell script `install-prerequisites.ps1` if necessary).

Anyway, building on Windows is awfully slow, 5 to 10 times slower than on Linux with GCC on equivalent hardware. So, using the latest nightly builds is better.

> Please get the latest pre-built TSDuck.

Nightly builds are automatically made available on tsduck.io in the pre-release area (direct link: https://tsduck.io/download/prerelease/) You can get the build of the current repo state for Windows and Ubuntu.

nuclidjp, 2020-05-14:
@lelegard 

Thank you for the pre-release.
I confirmed that `tsscan` works properly on windows10.

lelegard, 2020-05-14:
> I confirmed that tsscan works properly on windows10.

Thanks. You mean with the S270, right?

It would be interesting to test on Linux with the latest fix (drop checking signal strength when no offset scanning is used).

nuclidjp, 2020-05-14:
The result for Windows 10 is S270.
There seems to be no Pre-releases for arm, so it will take a long time to build.
When the build is over, test and report.

lelegard, 2020-05-14:
> There seems to be no Pre-releases for arm, so it will take a long time to build.

Yes, it takes quite a long time to build on a Raspberry Pi, I have one. Don't you have a PC with Linux on it? On a recent laptop, using `make -j10`, TSDuck is built in 1 to 2 minutes.

The pre-releases are automatically built and published every night using GitHub Actions (the CI/CD framework on GitHub). The GitHub runner systems only support Windows, Ubuntu (x64) and macOS. So, TSDuck nightly builds are available for Windows and Ubuntu only (TSDuck on macOS is deployed using Homebrew). Official releases are built on multiple VM's on my Mac (except for the Rapsberry Pi) and more options are available.

APTX, 2020-05-14:
> If a tuner or driver cannot reliably report signal strength and quality, then scanning offsets does not make sense anyway since it needs to compare signal strength across all offsets around the channel central frequency.

Thank you for the explanation @lelegard , that does make sense. I thought checking the signal strength was something `tsscan` did in all cases.

nuclidjp, 2020-05-14:
Raspberry Pi and S270.
Raspberry pie

`tsscan --verbose --hf-band-region japan --save-channels channels.xml`

Scan channel 13,473,142,857 Hz
* UHF channel 13 (473.142 MHz), intensity: 0%

The scan will not continue with this display.

lelegard, 2020-05-14:
There is probably a residual display issue. But it should not stop because of an invalid strength.

> The scan will not continue with this display.

What do you mean? Does it terminate? Or hang? Or continue with other channels?

Please post the complete log of `tsscan --debug --japan -g`.

If possible, please post the complete log of the same command on Windows with the same tuner.

nuclidjp, 2020-05-14:
tsscan --debug --japan -g
tsscan: Debug: debug level set to 1
scanning channel 13, 473,142,857 Hz
tsscan: Debug: trying offset 0
tsscan: Debug: using ISDB-T as default delivery system
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 25 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 8 (0x00000008)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 473142857 (0x1C339649)
tsscan: Debug: [2] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [3] cmd = 5 (DTV_BANDWIDTH_HZ), data = 6000000 (0x005B8D80)
tsscan: Debug: [4] cmd = 39 (DTV_TRANSMISSION_MODE), data = 1 (0x00000001)
tsscan: Debug: [5] cmd = 38 (DTV_GUARD_INTERVAL), data = 0 (0x00000000)
tsscan: Debug: [6] cmd = 19 (DTV_ISDBT_SOUND_BROADCASTING), data = 0 (0x00000000)
tsscan: Debug: [7] cmd = 20 (DTV_ISDBT_SB_SUBCHANNEL_ID), data = 0 (0x00000000)
tsscan: Debug: [8] cmd = 22 (DTV_ISDBT_SB_SEGMENT_COUNT), data = 13 (0x0000000D)
tsscan: Debug: [9] cmd = 21 (DTV_ISDBT_SB_SEGMENT_IDX), data = 0 (0x00000000)
tsscan: Debug: [10] cmd = 41 (DTV_ISDBT_LAYER_ENABLED), data = 7 (0x00000007)
tsscan: Debug: [11] cmd = 18 (DTV_ISDBT_PARTIAL_RECEPTION), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [12] cmd = 23 (DTV_ISDBT_LAYERA_FEC), data = 9 (0x00000009)
tsscan: Debug: [13] cmd = 24 (DTV_ISDBT_LAYERA_MODULATION), data = 6 (0x00000006)
tsscan: Debug: [14] cmd = 25 (DTV_ISDBT_LAYERA_SEGMENT_COUNT), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [15] cmd = 26 (DTV_ISDBT_LAYERA_TIME_INTERLEAVING), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [16] cmd = 27 (DTV_ISDBT_LAYERB_FEC), data = 9 (0x00000009)
tsscan: Debug: [17] cmd = 28 (DTV_ISDBT_LAYERB_MODULATION), data = 6 (0x00000006)
tsscan: Debug: [18] cmd = 29 (DTV_ISDBT_LAYERB_SEGMENT_COUNT), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [19] cmd = 30 (DTV_ISDBT_LAYERB_TIME_INTERLEAVING), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [20] cmd = 31 (DTV_ISDBT_LAYERC_FEC), data = 9 (0x00000009)
tsscan: Debug: [21] cmd = 32 (DTV_ISDBT_LAYERC_MODULATION), data = 6 (0x00000006)
tsscan: Debug: [22] cmd = 33 (DTV_ISDBT_LAYERC_SEGMENT_COUNT), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [23] cmd = 34 (DTV_ISDBT_LAYERC_TIME_INTERLEAVING), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [24] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: using ISDB-T as default delivery system
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 25 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 8 (0x00000008)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 473142857 (0x1C339649)
tsscan: Debug: [2] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [3] cmd = 5 (DTV_BANDWIDTH_HZ), data = 6000000 (0x005B8D80)
tsscan: Debug: [4] cmd = 39 (DTV_TRANSMISSION_MODE), data = 1 (0x00000001)
tsscan: Debug: [5] cmd = 38 (DTV_GUARD_INTERVAL), data = 0 (0x00000000)
tsscan: Debug: [6] cmd = 19 (DTV_ISDBT_SOUND_BROADCASTING), data = 0 (0x00000000)
tsscan: Debug: [7] cmd = 20 (DTV_ISDBT_SB_SUBCHANNEL_ID), data = 0 (0x00000000)
tsscan: Debug: [8] cmd = 22 (DTV_ISDBT_SB_SEGMENT_COUNT), data = 13 (0x0000000D)
tsscan: Debug: [9] cmd = 21 (DTV_ISDBT_SB_SEGMENT_IDX), data = 0 (0x00000000)
tsscan: Debug: [10] cmd = 41 (DTV_ISDBT_LAYER_ENABLED), data = 7 (0x00000007)
tsscan: Debug: [11] cmd = 18 (DTV_ISDBT_PARTIAL_RECEPTION), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [12] cmd = 23 (DTV_ISDBT_LAYERA_FEC), data = 9 (0x00000009)
tsscan: Debug: [13] cmd = 24 (DTV_ISDBT_LAYERA_MODULATION), data = 6 (0x00000006)
tsscan: Debug: [14] cmd = 25 (DTV_ISDBT_LAYERA_SEGMENT_COUNT), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [15] cmd = 26 (DTV_ISDBT_LAYERA_TIME_INTERLEAVING), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [16] cmd = 27 (DTV_ISDBT_LAYERB_FEC), data = 9 (0x00000009)
tsscan: Debug: [17] cmd = 28 (DTV_ISDBT_LAYERB_MODULATION), data = 6 (0x00000006)
tsscan: Debug: [18] cmd = 29 (DTV_ISDBT_LAYERB_SEGMENT_COUNT), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [19] cmd = 30 (DTV_ISDBT_LAYERB_TIME_INTERLEAVING), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [20] cmd = 31 (DTV_ISDBT_LAYERC_FEC), data = 9 (0x00000009)
tsscan: Debug: [21] cmd = 32 (DTV_ISDBT_LAYERC_MODULATION), data = 6 (0x00000006)
tsscan: Debug: [22] cmd = 33 (DTV_ISDBT_LAYERC_SEGMENT_COUNT), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [23] cmd = 34 (DTV_ISDBT_LAYERC_TIME_INTERLEAVING), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: [24] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
* UHF channel 13 (473.142 MHz), strength: 0%

The cursor is blinking.
It doesn't seem to end.
Or is it right to end it?

`tsscan --verbose --hf-band-region japan`
The first two lines are displayed and `tsscan` does not continue.
The cursor is blinking.

Then test Windows.

nuclidjp, 2020-05-14:
Windows 10 and S270
The scan was successful.

`tsscan --verbose --hf-band-region japan`

scanning channel 13, 473,142,857 Hz
scanning channel 14, 479,142,857 Hz
* UHF channel 14 (479.142 MHz), strength: 100%, quality: 100%
scanning channel 15, 485,142,857 Hz
* UHF channel 15 (485.142 MHz), strength: 100%, quality: 50%
scanning channel 16, 491,142,857 Hz
scanning channel 17, 497,142,857 Hz
* UHF channel 17 (497.142 MHz), strength: 100%, quality: 50%
  Transport stream id: 0, 0x0000
scanning channel 18, 503,142,857 Hz
* UHF channel 18 (503.142 MHz), strength: 100%, quality: 0%
scanning channel 19, 509,142,857 Hz
scanning channel 20, 515,142,857 Hz
* UHF channel 20 (515.142 MHz), strength: 100%, quality: 0%
  Transport stream id: 32468, 0x7ED4
scanning channel 21, 521,142,857 Hz
* UHF channel 21 (521.142 MHz), strength: 100%, quality: 50%
  Transport stream id: 32464, 0x7ED0
scanning channel 22, 527,142,857 Hz
* UHF channel 22 (527.142 MHz), strength: 100%, quality: 50%
  Transport stream id: 32465, 0x7ED1
scanning channel 23, 533,142,857 Hz
scanning channel 24, 539,142,857 Hz
* UHF channel 24 (539.142 MHz), strength: 100%, quality: 0%
  Transport stream id: 32464, 0x7ED0
scanning channel 25, 545,142,857 Hz
scanning channel 26, 551,142,857 Hz
* UHF channel 26 (551.142 MHz), strength: 100%, quality: 0%
  Transport stream id: 32466, 0x7ED2
scanning channel 27, 557,142,857 Hz
scanning channel 28, 563,142,857 Hz
* UHF channel 28 (563.142 MHz), strength: 100%, quality: 50%
  Transport stream id: 32466, 0x7ED2
scanning channel 29, 569,142,857 Hz
* UHF channel 29 (569.142 MHz), strength: 100%, quality: 0%
  Transport stream id: 32467, 0x7ED3
scanning channel 30, 575,142,857 Hz
scanning channel 31, 581,142,857 Hz
* UHF channel 31 (581.142 MHz), strength: 100%, quality: 50%
  Transport stream id: 32465, 0x7ED1
scanning channel 32, 587,142,857 Hz
scanning channel 33, 593,142,857 Hz
scanning channel 34, 599,142,857 Hz
scanning channel 35, 605,142,857 Hz
scanning channel 36, 611,142,857 Hz
scanning channel 37, 617,142,857 Hz
scanning channel 38, 623,142,857 Hz
scanning channel 39, 629,142,857 Hz
* UHF channel 39 (629.142 MHz), strength: 100%, quality: 50%
  Transport stream id: 32467, 0x7ED3
scanning channel 40, 635,142,857 Hz
scanning channel 41, 641,142,857 Hz
scanning channel 42, 647,142,857 Hz
scanning channel 43, 653,142,857 Hz
* UHF channel 43 (653.142 MHz), strength: 100%, quality: 0%
  Transport stream id: 32468, 0x7ED4
scanning channel 44, 659,142,857 Hz
scanning channel 45, 665,142,857 Hz
scanning channel 46, 671,142,857 Hz
scanning channel 47, 677,142,857 Hz
* UHF channel 47 (677.142 MHz), strength: 100%, quality: 0%
scanning channel 48, 683,142,857 Hz
scanning channel 49, 689,142,857 Hz
* UHF channel 49 (689.142 MHz), strength: 100%, quality: 10%
scanning channel 50, 695,142,857 Hz
scanning channel 51, 701,142,857 Hz
* UHF channel 51 (701.142 MHz), strength: 100%, quality: 0%
scanning channel 52, 707,142,857 Hz
* UHF channel 52 (707.142 MHz), strength: 100%, quality: 0%
scanning channel 53, 713,142,857 Hz
scanning channel 54, 719,142,857 Hz
scanning channel 55, 725,142,857 Hz
scanning channel 56, 731,142,857 Hz
scanning channel 57, 737,142,857 Hz
scanning channel 58, 743,142,857 Hz
scanning channel 59, 749,142,857 Hz
scanning channel 60, 755,142,857 Hz
scanning channel 61, 761,142,857 Hz
scanning channel 62, 767,142,857 Hz

nuclidjp, 2020-05-14:
It is a continuation of Windows and S270.

`tsscan --debug --japan -g`

This is a text file.
[tsscan --debug --japan -g.txt](https://github.com/tsduck/tsduck/files/4628567/tsscan.--debug.--japan.-g.txt)

[20 more replies]