tsduck/tsduck #580: zap plugin fails to find service discovered by tsscan

APTX, 2020-05-10:
I'm running `tsp` like this:
```
tsp -I dvb -c "日テレ1" -P zap "日テレ1" -O file 日テレ1.ts
* Error: zap: service 日テレ1 not found in SDT
```
I am guessing, that since the encoding had to be set when using `tsscan` the `zap` plugin also would require setting the encoding, as it looks for the service in SDT. Unfortunately `zap` neither accepts `--japan` or `--default-charset ARIB-STD-B24`.

lelegard, 2020-05-10:
> I am guessing, that since the encoding had to be set when using tsscan the zap plugin also would require setting the encoding, as it looks for the service in SDT. Unfortunately zap neither accepts --japan or --default-charset ARIB-STD-B24.

You are definitely right. I will add the charset options here too. Since ISDB uses the same basic signalisation as DVB but using a different way of specifying character sets, the consequences emerge one after the other.

In the meantime, you can use the service id instead of the service name in the `zap` plugin.

lelegard, 2020-05-11:
I have added the charset options to all plugins using service names at some point. And there were a lot of them.

Such options should be repeated in almost all plugins in your case. This is not very convenient. There are many options which influence the way to interpret the stream: character sets, standards (DVB, ATSC, ISDB), CAS ids, HF band regions. In some cases, it is useful to specify different options in different plugins of the same chain. So, they must remain plugin-specific options. Moreover, not all plugins use all of them. It depends on what the plugin does.

To simplify this, I also added all these options at `tsp` level (before the first plugin specification). These options have no effect on `tsp` itself. They just serve as default for plugins which implement these options. So, in your case, using `tsp --japan -I ... -P ... -O ...` is equivalent to repeat option `--japan` in all plugins which support it. See section "Default options for plugin" in the `tsp` documentation.

Finally, you may have not noticed the "TSDuck configuration file". It allows the user to specify defaults which apply to selected TSDuck commands. Have a look at Appendix A in the [user's guide](https://tsduck.io/download/docs/tsduck.pdf). I have added section A.3.1 with a convenient configuration file for Japanese users.

APTX, 2020-05-12:
Thank you very much for implementing this @lelegard ! This simplifies working with japanese TS greatly.

I've tried using the config file (using `tsp` and `tsscan`), setting `--japan` on `tsp`, and directly on the `zap` plugin.

`zap` can also find the services by name.

closed 2020-05-12 by APTX