tsduck/tsduck #581: Approach for notifying about issues that are discovered during calls to OutputPlugin::send() that don't effect the main behavior of send()

alevinsn, 2020-05-12:
Let's say I wanted to notify the caller of DektecOutputPlugin::send(), or some other entity that is interested in such notifications, that the FIFO has been drained, for example.  Currently, the only way this is made apparent at all is the existence of the verbose log message "Pausing transmission temporarily in order to maintain preload."  Currently, to detect this situation programmatically, it would be necessary to compare the text of every verbose log message with the above text, or perhaps a partial string compare would be sufficient, but regardless, this is not a good way to be informed of such types of problems.  Plus, if log messages are ever localized in the future, this approach wouldn't work at all.

This is the sort of issue that has no impact on the primary behavior of send().  There are other examples of this sort of thing within DektecOutputPlugin::send() and likely other output plug-ins as well.  It is likely also relevant for some input plug-ins and perhaps some processing plug-ins.

Possibly, the TSPacketMetadata label mechanism could be used to indicate these sorts of issues to the caller of send(), but this isn't really a good fit, since the issue being notified may not have anything to do with specific packets.  Plus, the label mechanism seems to be more geared to processing plug-ins than output plug-ins.

One approach that would work generically for all plug-ins would be to provide plug-ins with the ability to notify one or more interested recipients of conditions and issues that are specific to a plug-in.  Interested recipients would register with a plug-in that advertises this notification capability, and interested recipients would implement an interface that would be invoked when any notifications are generated by a plug-in.  Each relevant plug-in would define its own constants pertaining to specific plugin-specific issues/conditions.  The use of this mechanism also wouldn't be restricted to applications within methods like OutputPlugin::send().  That is, if the Dektec output plug-in were changed to, say, have its own thread that queried the FIFO to know exactly when the FIFO is drained, as opposed to detecting it after the fact in a call to send(), the notification mechanism could be used independent of calls to send().

This sort of thing would mostly be relevant for programmatic uses of plug-ins, although it could potentially be exposed via tsp.  But, there is already a reasonable and already existing mechanism for informing tsp users of these sorts of issues--logging.

I'd be happy to implement such a general notification mechanism, but I wanted to put it out there and see what the interest level is and see what others think about this sort of approach.  Or, perhaps there are other, superior solutions.

lelegard, 2020-05-13:
Hi Aaron,

This is interesting. Let's think twice first to implement it properly.

The obvious API is to define an interface (an abstract class with one pure virtual method) that is implemented by some application-defined object. The exact definition of this interface could include:
- The plugin name.
- The plugin index (in case the same plugin is used several times).
- Maybe the plugin instance pointer, to be able to do a dynamic_cast on application-defined plugins.
- A condition (integer? string? byte block? a combination of them? an std::initializer_list of ArgMixIn?) describing the condition.

The real problems, for me, are the following:
- Object life time: Being sure that the object implementing the interface is still valid when the plugin invokes it.
- Multi-thread synchronization. The plugin executes in one thread. The object in registered from one application thread and invoked in a plugin thread.

As a first proposal:
- I can define the interface ts::PluginEventHandlerInterface.
- Assume / impose that the application uses the class TSProcessor (high-level API for tsp).
- Register handlers (instances which implement PluginEventHandlerInterface) in TSProcessor.
- Define an API in base class ts::TSP to allow plugins to send events without being aware of the presence of handlers.
- A handler shall be registered _before starting_ the plugin chain.
- A handler shall remain valid as long as the processing runs (not deleted before waitForTermination()).
- A handler can be registered for either:
  - One plugin by name (can be several instances of them).
  - One plugin by index (only one instance).
  - For all plugins.
  - Shall we allow registering a handler for one event type? How to define an event type?
  - An handler object can be registered for several plugins.
  - A plugin can have several registered handlers.
- Shall we give the possibility to unregister a handler? In all cases, unregistration should be rejected while the plugin chain is running.

What's you opinion?

lelegard, 2020-05-15:
Hi Aaron,

I support your idea and I propose an implementation in the current state of the repo. We can amend details if you like.

Here are the principles.

A plugin can signal an event using `tsp->signalPluginEvent(code, &data)` where `code` is just a plugin-specific integer value to indicate the type of event. Some constants should probably be defined in the interface of the plugin for the supported events. The `data` is optional and is the address of some plugin-specific subclass of `ts::Object` to pass additional information to the event handlers, if necessary.

So, if you want to signal events in some critical parts of the `dektec` output plugin, just add such calls.

An application which wants to receive events from plugins implements the interface `ts::PluginEventHandlerInterface` in some class. The virtual method `handlePluginEvent()` is invoked with a `ts::PluginEventContext` object containing all details of the event.

The classes `ts::TSProcessor` and `ts::InputSwitcher` (the API's for `tsp` and `tsswitch` respectively) are now subclasses of a common superclass `ts::PluginEventHandlerRegistry` which defines `registerEventHandler()` and `unregisterEventHandler()` methods. 

By default, the event handler will be called for all events in all plugins. But the registration can be made with a combination of criteria: event type, plugin name or instance. Any number of event handlers can be concurrently present. The implementation is currently basic. It will be fine with a dozen event handlers. But if the application registers thousands of event handlers, the performances will probably be catastrophic !

The [doxygen documentation](https://tsduck.io/doxy/) will be automatically updated during the next nightly build.

There is a working sample in file `src/utest/utestTSProcessor.cpp`. It is written in unitary test style, not a real application (this is just a way to check that no bug will appear later). But all mechanisms are used and it is working.

So, you may experiment in your application and check if you can pass events and associated information to your application classes. Please provide your feedback here.

alevinsn, 2020-05-21:
Sorry for the long delay in responding.  Overall, there is much to like in the design and implementation, but I would personally prefer a simpler solution that is entirely plug-in based and not dependent on a whole bunch of infrastructure.  That is, event handlers would be registered directly with plug-in objects, as opposed to registered at a global level as is done in your solution using PluginEventHandlerRegistry and other components.  A plug-in based approach implies that ts::Plugin would be enhanced, or, a new interface could be provided that plug-ins may implement.  This approach would also make the solution more amenable to custom pipelines that are not based on ts::TSProcessor or ts::InputSwitcher, as was the case with the pipeline I implemented in sample/sample-app/myexec.cpp a number of commits ago (but has since been replaced with an implementation that uses ts::TSProcessor).  Even in a custom pipeline, providing a "tsp" object is still a requirement, if only for logging, and as such, such a class could implement signalPluginEvent() and do something independent of all the new infrastructure that you added, but using signalPluginEvent() is a little clunky for this purpose, and it would be preferable to call each registered event handler directly from the plug-in itself (likely using a helper method to do so).  But, I do like the simple approach of passing in code and &data, although I would prefer it to be *data so a null pointer could be passed in instead of requiring that a ts::Object be passed in.

I know that individual plug-in objects are not available in (the current implementation of) myexec.cpp, since the plug-ins are used indirectly via TSProcessor.  That, however, doesn't prevent the event handler registration mechanism from being done at the plug-in level, however.

I would guess that the addition of the PluginEventHandlerRegistry was geared towards the object lifetime issue that you raised in your first response, but I don't see how it prevents event handlers from going away before they are actually used.  The code in utestTSProcessor.cpp could easily be altered to cause the TestEventHandler objects to go out of scope before the plug-in signals any events.  It's possible to remedy this situation, to a certain extent, by adding code in the destructor of PluginEventHandlerInterface's destructor to unregister with the registry, but doing so requires significant changes to the base PluginEventHandlerInterface design and implementation to make it aware that it has been registered with the registry in the first place.  However, realistically, there is no way to solve object lifetime issues at this level, and such issues already exist within the TSDuck project anyway (such as with tsp--plug-ins regularly use the tsp object but have no way to guarantee that the tsp object is still valid when they use it).  Unless you want to add something like reference counting to TSDuck, I don't believe that it is the job of TSDuck to manage this.  I also don't believe that it is the job of TSDuck to manage multi-thread synchronization, another concern that you raised.  Technically, the current solution does nothing to address multi-thread synchronization as far as I can tell, and I think the only thing really to do is document the situation.

I should also add that managing this at the plug-in level solves the possible performance issue that you raised with having too many event handlers registered.  The filtering mechanism that you added is nice, but if this is done at the plug-in level, I don't think any further filters are necessary, and with this approach, each registered event handler would be invoked regardless, and it would be the event handler's responsibility to disregard events that it isn't interested in.

There is one object lifetime consideration that may not be a real issue but is something to be aware of.  As currently implemented, the semi-optional data argument has a very limited lifetime and will likely be destroyed soon after calling signalPluginEvent().  Implementations of handlePluginEvent() need to be aware of this and only use the object within handlePluginEvent().  I think it is sufficient to handle this with documentation.

Finally, I'm a little confused by a comment from utestTSProcessor.cpp at the start of the implementation of handlePluginEvent():  "We cannot assert here, log a debug message. Since messages will no be logged, the posst-processing assertions will fail and the problems wil be reported."  I'm not sure why this would be--perhaps it has something to do with the test being run from the unit test mechanism?  Also "post-processing" and "will" are misspelled, and "no be logged" should be replaced with "not be logged".

lelegard, 2020-05-21:
Hi Aaron,

> I would personally prefer a simpler solution that is entirely plug-in based and not dependent on a whole bunch of infrastructure. That is, event handlers would be registered directly with plug-in [...]

I understand. That would basically move the handler registration code from `PluginEventHandlerRegistry` to the base class `Plugin`.

I see a few inconvenients with that approach. First you need access to the individual plugin instances. In some applications, that could be no problem. But, in case those plugin instances are hidden somewhere, this is a problem. More generally, weak coupling of objects are always better than tight coupling. Said otherwise, it is better to have the possibility to keep the plugin machinery and the application-specific event handlers completely independent.

> This approach would also make the solution more amenable to custom pipelines that are not based on ts::TSProcessor or ts::InputSwitcher

If you have a look at those two pipelines, they were almost unchanged except being now subclasses of `PluginEventHandlerRegistry`.

Since everything about handler registration is encapsulated here, you just need to have one instance of `PluginEventHandlerRegistry` in your plugin pipeline. Depending on your application structure or personal preference, you may prefer composition or association over inheritance, but the idea remains the same.

> Even in a custom pipeline, providing a "tsp" object is still a requirement, if only for logging, and as such, such a class could implement signalPluginEvent() and do something independent of all the new infrastructure that you added

Yes, the `TSP` instance needs to get a reference to the `PluginEventHandlerRegistry`. But, since your custom `TSP` subclass has probably already some pointer to your custom plugin pipeline, this is why making your pipeline a subclass of `PluginEventHandlerRegistry` is convenient. Then, the implementation of `signalPluginEvent()` in your custom `TSP` just needs to call `callEventHandlers()` in the custom plugin pipeline (inherited from `PluginEventHandlerRegistry`).

> but using signalPluginEvent() is a little clunky for this purpose, and it would be preferable to call each registered event handler directly from the plug-in itself (likely using a helper method to do so).

This helper method is precisely `PluginEventHandlerRegistry::callEventHandlers()`.

> But, I do like the simple approach of passing in code and &data, although I would prefer it to be *data so a null pointer could be passed in instead of requiring that a ts::Object be passed in.

You can already pass a null pointer as data:
~~~~
virtual void signalPluginEvent(uint32_t event_code, Object* plugin_data = nullptr) const = 0;
~~~~

Using the generic `Object` type is simply a clean C++ way to avoid passing a `void*` pointer which is later cast to some application type without guarantee that the cast is correct. In C++, we shouldn't use and cast `void*`. We should oblige the application types to derive from some very generic class (here `Object` since it already existed) and then use a `dynamic_cast` on the pointer. If the conversion is invalid (not the same subclass of `Object`), then the cast pointer will be zero and this can be safely checked in the application. This is illustrated in `utestTSProcessor`.

> I would guess that the addition of the PluginEventHandlerRegistry was geared towards the object lifetime issue that you raised in your first response, but I don't see how it prevents event handlers from going away before they are actually used.

Yes, you are right. I did not find a solution for that. The life-time problem is endemic to C++ (unlike Java). The generic solution I use in TSDuck is the `SafePtr` template class. But using it for event handlers was too restrictive for me. That would have meant that the all objects which implement `PluginEventHandlerInterface` would have to be allocated and managed by a `SafePtr` on type `PluginEventHandlerInterface`. But, if we generalize the method to all handlers, then we run into conflicts where objects implementing several handler interfaces would have to be allocated and managed by `SafePtr` on several independent interfaces.

So, I used the same approach as all demux classes (`SectionDemux`, `MPEDemux`, `T2MIDemux`): let the handler classes be managed independently and let the application make sure that the handlers are unregistered before being destroyed. This is why, in the documentation of `unregisterEventHandler()`, I added _"This is typically used in the destructor of an event handler"._ This is the (unsatisfactory) C++ way: cleanup everything explictly in destructors. The same constraint exists in classes inheriting from `Thread` for instance. If they forget to call their own `waitForTermination()` in their destructor, they will likely run into trouble.

The other problem I mentioned was the thread safety. I solved it in a brutal way using one single `Mutex`. In the general case, this is not good for performances to have mutexes with a large coverage. But, in that specific case, having a more complicated fine-grained synchronization was simply not worth it (I think).

> It's possible to remedy this situation, to a certain extent, by adding code in the destructor of PluginEventHandlerInterface's destructor to unregister with the registry, but doing so requires [...]

Yes, that is the idea: use the destructor of the subclass to let the application handle it.

[... cut ...]  in this specific test, we don't assert / assume test conditions in the handlers (they are called from plugin threads). We just log events in a buffer. The test is designed in such a way that logged messages are expected in a specific order and content. So, when back in the test fixture, we can safely assert on the logged messages. Ugly, but efficient unit testing...

> Also "post-processing" and "will" are misspelled, and "no be logged" should be replaced with "not be logged".

Thanks, fixed ;)

alevinsn, 2020-05-21:
> I understand. That would basically move the handler registration code from `PluginEventHandlerRegistry` to the base class `Plugin`.

Well, with slight modifications, since you won't need to keep track of the plug-in name.

> I see a few inconvenients with that approach. First you need access to the individual plugin instances. In some applications, that could be no problem. But, in case those plugin instances are hidden [...]

The information in the ts::Criteria object would be sufficient to register with specific plug-ins if TSProcessor is used.  This would be no different than how command-line arguments are passed to specific plug-ins, which works even though the actual application doesn't have access to the actual plug-in objects when TSProcessor is used.  And, if the interface is on actual plug-ins, as opposed to the loose coupling that you prefer, this approach also works for custom pipelines that don't utilize all the infrastructure that you added but have access to plug-in objects.  That is, developers could register directly with the plug-in object if they want, or they could go through the event registry approach if they are using TSProcessor.  Both approaches would be feasible.

However, as I think about it some more, for both approaches to be feasible, it still is necessary to have some storage repository to keep track of event handlers and their associated Criteria objects for later use when the plug-in objects are actually created if TSProcessor is used (and something similar is already done for per-plug-in command-line arguments).

> > This approach would also make the solution more amenable to custom pipelines that are not based on ts::TSProcessor or ts::InputSwitcher
> 
> If you have a look at those two pipelines, they were almost unchanged except being now subclasses of `PluginEventHandlerRegistry`.

True, but PluginEventHandlerRegistry is overkill if all you want to do in your pipeline is pay attention to events from a single plug-in, which I think will be the most likely situation.  The ability to register a single event handler to listen to events from every plug-in is not a realistic situation.

> The other problem I mentioned was the thread safety. I solved it in a brutal way using one single `Mutex`. In the general case, this is not good for performances to have mutexes with a large [...]

You would effectively get fine-grained locking with little effort if registration were managed at the plug-in level. rather than within a global object that applies to all plug-ins.

> Yes, this is possible. But I think that could make application more error-prone.

Disregarding events that are of no interest is a very common situation that developers already have to do in many situations outside of the scope of TSDuck.  The situations in which developers would want to use this functionality are already pretty limited and will likely be constrained to paying attention to events from a single plug-in or two in a pipeline.  And, for any event handling to work, the developer will have to read the documentation for a specific plug-in to know what events are of interest and what data is passed in for them.  Since that's already required, I don't see how not having the filter capability makes things more error-prone.  But, it will make things simpler, particularly if registration is done at the plug-in level.  All callEventHandlers() would have to do is iterate through all the registered event handlers and call handlePluginEvent() on each.

> The intermediate `PluginEventContext` has a limited lifetime indeed. It is just of a "bag of data" which avoids having too many parameters in the handler method. But the real user-data (the subclass [...]

But, effectively, there must be some restrictions on the lifetime of the user data.  If it's allocated using new, for example, but there are multiple event handlers, which would be responsible for deallocating the object?

closed 2026-04-24 by github-actions[bot]