tsduck/tsduck #582: ISDB-S scanning and tuning

APTX, 2020-05-16:
The [Japanese television channel frequency page on wikipedia](https://ja.wikipedia.org/wiki/%E3%83%86%E3%83%AC%E3%83%93%E5%91%A8%E6%B3%A2%E6%95%B0%E3%83%81%E3%83%A3%E3%83%B3%E3%83%8D%E3%83%AB#%E8%A1%9B%E6%98%9F%E5%9F%BA%E5%B9%B9%E6%94%BE%E9%80%81%E3%81%AEIF%E3%81%A8%E8%AB%96%E7%90%86%E3%83%81%E3%83%A3%E3%83%B3%E3%83%8D%E3%83%AB) contains very useful information about how ISDB-S works (At least in Japan).

It turns out that all the physical channels are fixed, so a scan like the UHF scan should work.

There are BS and CS satellites, both at 110 degrees. There is only left or right polarization. The left column has the right polarized channels (右 means right and 左 means left).

The frequencies in the table are offsets from a base frequency (which I assume is what _LNB_ is). The frequencies are: `10.678 GHz` for right polarization, and `9.505 GHz` for left polarization.

Channel bandwidth is always `34.5 MHz`. Multiple services share a single physical channel.

There is a [diagram](https://ja.wikipedia.org/wiki/%E6%97%A5%E6%9C%AC%E3%81%AB%E3%81%8A%E3%81%91%E3%82%8B%E8%A1%9B%E6%98%9F%E6%94%BE%E9%80%81#BS-IF/CS-IF%EF%BC%88%E5%91%A8%E6%B3%A2%E6%95%B0%E5%A4%89%E6%8F%9B%EF%BC%89) showing the frequencies at which UHF/BS/CS channels are on in sockets in apartments.

Current [BS physical channel allocation](https://ja.wikipedia.org/wiki/%E6%97%A5%E6%9C%AC%E3%81%AB%E3%81%8A%E3%81%91%E3%82%8B%E8%A1%9B%E6%98%9F%E6%94%BE%E9%80%81#%E7%89%A9%E7%90%86%E3%83%81%E3%83%A3%E3%83%B3%E3%83%8D%E3%83%AB%EF%BC%88BS%E3%83%87%E3%82%B8%E3%82%BF%E3%83%AB%E6%94%BE%E9%80%81%EF%BC%89) and [CS physical channel allocation](https://ja.wikipedia.org/wiki/%E6%97%A5%E6%9C%AC%E3%81%AB%E3%81%8A%E3%81%91%E3%82%8B%E8%A1%9B%E6%98%9F%E6%94%BE%E9%80%81#%E7%89%A9%E7%90%86%E3%83%81%E3%83%A3%E3%83%B3%E3%83%8D%E3%83%AB%EF%BC%88%E6%9D%B1%E7%B5%8C110%E5%BA%A6CS%E3%83%87%E3%82%B8%E3%82%BF%E3%83%AB%E6%94%BE%E9%80%81%EF%BC%89).

In practice all "left" polarized channels are ISDB-S3, which means that I probably can't test those at all.

I compared the `satellite_delivery_system_descriptor`'s TSDuck found in the .ts file I shared and they all match the listed frequencies. For example:
```
      <satellite_delivery_system_descriptor frequency="11,727,480,000" orbital_position="110.0" west_east_flag="east" polarization="right" modulation_system="ISDB-S" modulation_type="ISDB-S" symbol_rate="28,860,000" FEC_inner="ISDB-S"/>
      <satellite_delivery_system_descriptor frequency="11,765,840,000" orbital_position="110.0" west_east_flag="east" polarization="right" modulation_system="ISDB-S" modulation_type="ISDB-S" symbol_rate="28,860,000" FEC_inner="ISDB-S"/>
      <satellite_delivery_system_descriptor frequency="12,149,440,000" orbital_position="110.0" west_east_flag="east" polarization="right" modulation_system="ISDB-S" modulation_type="ISDB-S" symbol_rate="28,860,000" FEC_inner="ISDB-S"/>
```
are `BS-1`, `BS-3`, and `BS-23` respectively.

lelegard, 2020-05-16:
Hi @APTX,

This is quite interesting, very different from what is seen in Europe and North America.

First, since Wikipedia is open to all contributors, I would strongly recommend that you update or create the English versions of those pages yourself! The English version of the first page lacks the satellite information about Japan. The other pages don't even have an English version. The whole world - outside Japan - would be grateful to you 😄 

The diagram you mention (copy below) is particularly interesting.
[image]

My understanding is that there are only 4 sets of frequencies, BS/CS with L/R, and those sets are sufficiently narrow to be transposed altogether from the dish to the tuner. So, there is no need to _dynamically_ re-program the LNB in the dish each time you tune to a TS. Am I right?

So, I took for granted concepts that you may not have at all in Japan. In Europe and North America, there are so many different frequencies on each satellite on a so wide spectrum that it is impossible to transpose them all at the same time from the dish down to the tuner. We usually split the frequency spectrum in two parts, "low band", "high band". With the combination of horizontal and vertical polarity in each band, that makes 4 different sets - or _quadrants_ - of TS and there is room for only one quadrant at a time on the downstream cable. So, when you tune to a TS, you need to send, from the tuner to the LNB (the receiving head in the dish), a command to switch to the right quadrant. The dish reconfigures its transposition scheme and, suddenly, the complete content of the downstream cable changes. Then, we compute the transposed frequency and we can program the tuner on that frequency.

Sending the quadrant selection command from the tuner to the dish is performed either using tone & voltage (old method) or using the DiSEqC protocol (modern method).

So, my understanding is that there is no such command in Japanese satellite reception. From the BS or CS and then left or right polarity, you just pick up a fixed base transposed frequency and that's all. Could you confirm?

If this is correct, then the complete tone / voltage / DiSEqC sequence should be avoided. Maybe I could define a new syntax for a more generic form of `--lnb` (or a newer more generic option) to integrate the BS and CS Japanese schemes.

APTX, 2020-05-16:
> My understanding is that there are only 4 sets of frequencies, BS/CS with L/R, and those sets are sufficiently narrow to be transposed altogether from the dish to the tuner. So, there is no need to dynamically re-program the LNB in the dish each time you tune to a TS. Am I right?

That is also my understanding. This may not be the case everywhere, but there is a shared satellite antenna where I live, so tuners may simply have no way of configuring it. Not sure if it would be different if I had my own antenna.

If you do the math with the offsets and get the frequencies on which the channels are transmitted from the satellite you can see that the channels overlap each other: BS-1 is on `11,727,480,000 Hz`, BS-2 is on `11,746,660,000 Hz`, making them only `19,180,000 Hz` (`19.18 MHz`) apart, which is much less then their bandwidth of `34.5 MHz` (but they do differ in polarization). BS-24 `12,168,620,000 Hz`, so all 24 channels "take up" `~450MHz`.

CS starts after a larger gap from BS, ND-1 is `12,271,000,000 Hz`.

Only on the coax cable socket do you get the channels spread out as on the diagram, so you get all of it (including terrestrial) in one socket.

> From the BS or CS and then left or right polarity, you just pick up a fixed base transposed frequency and that's all. Could you confirm?

As far as I understand, yes. When I modified TSDuck to not do the tone/DiSEqC calls I could get `tsscan` to tune to the "satellite transmission frequencies" (like the `11,727,480,000 Hz`). TSDuck reversed this calculation and got the offset frequency (but I'm not certain it is done completely correctly).

This also lines up with the initial scan file I found for `dvbv5` which just has the offset frequencies from the wiki page, in kHz.

I'm not sure how useful that would be for other regions, but it would be nice if TSDuck understood the physical channel names, like "13" (terrestrial), or "BS-2".

lelegard, 2020-05-16:
> here is a shared satellite antenna where I live, so tuners may simply have no way of configuring it.

Here, we have DiSEqC switches to solve this. Dishes for shared usage have "quad-head" LNB's, with 4 cables going down. Each head is hard-coded to a quadrant. Each cable transports its own quadrant. The 4 cables are plugged to the 4 inputs of a switch. The switch has "some number" of outputs, depending on the model, one for each user. When the switch receives a tone / voltage / DiSEqC  command from a tuner, it routes the appropriate entry to the user.

This is of course more expensive but outputs thousands of TV channels.

> it would be nice if TSDuck understood the physical channel names, like "13" (terrestrial), or "BS-2".

I will find some sort of solution. But you will have to test it.

lelegard, 2020-05-17:
Hi @APTX,

> If you do the math with the offsets and get the frequencies on which the channels are transmitted from the satellite ...

Thanks. From your explanations, I rebuilt an Excel table of all frequencies from the base frequencies, channel widths and local oscillator frequencies ([see here](https://github.com/tsduck/tsduck/tree/master/doc)). And all results match the table from the Wikipedia article you linked to. So, I now completely understand how it works. I can now find a generic solution to integrate this into TSDuck.

This is indeed very different from what we see in Europe and America. Terrestrial frequencies are centrally managed as they are in Japan. But, once frequency ranges are allocated to commercial satellite operators, these operators do what they want within their allocator range. In Japan, it seems that satellite frequencies are centrally controlled mostly the same way as terrestrial ones.

Do you confirm that ISDB-S reception in Japan is limited to the two BS and CS networks? Nothing else?

APTX, 2020-05-18:
@lelegard BS and CS is "[backbone broadcasting](https://ja.wikipedia.org/wiki/%E5%9F%BA%E5%B9%B9%E6%94%BE%E9%80%81%E4%BA%8B%E6%A5%AD%E8%80%85)". There is also "[general broadcasting](https://ja.wikipedia.org/wiki/%E4%B8%80%E8%88%AC%E6%94%BE%E9%80%81)" (which is just defined as everything but backbone broadcasting). There is at least one broadcast company that uses [different broadcasting parameters](https://ja.wikipedia.org/wiki/%E3%82%B9%E3%82%AB%E3%83%91%E3%83%BC!%E3%83%97%E3%83%AC%E3%83%9F%E3%82%A2%E3%83%A0%E3%82%B5%E3%83%BC%E3%83%93%E3%82%B9#%E6%94%BE%E9%80%81%E6%B3%A2%E5%8F%97%E4%BF%A1%E3%81%AE%E4%BB%95%E7%B5%84%E3%81%BF%E3%81%A8%E9%85%8D%E7%B7%9A). The page says that this service uses DVB-S. For this system the antenna can not be shared as the tuner does send some signal to the antenna, so this might be exactly like the system you know. I'm not sure you can normally buy the hardware to receive this service from third parties (though maybe a DVB-S tuner would work).

lelegard, 2020-05-21:
Hi @APTX,

I have made a few modifications to improve the Japanese user experience:

- The LNB management has been reworked, based on `libdvb5` evolutions since early versions of TSDuck. Instead of the fixed transposition scheme, multi-band and polarization-controlled dishes are supported. Some predefined LNB's are listed in an XML file named `tsduck.lnbs.xml`. Your type of LNB is named "Japan 110BS/CS" with aliases "Japan", "110BS", "110CS". Very similar to `libdvb5`. So, you can now use `--lnb japan`.
- In the user-defined TSDuck configuration file, the parameter `default.lnb` has been added. So, instead of using option `--lnb` everywhere, you will probably add `default.lnb = japan` in your configuration file.
- In the case of a polarization-controlled dish like yours ("stacked" transposed bands), no tone/voltage/DiSEqC command is sent to the LNB.
- The HF bands XML configuration file has been extended to include predefined satellite bands. So, in region "japan", the bands "BS" and "CS" have been added.
- The option `--channel-transponder` (or `-c`) which uses TV channel names to retrieve the tuning information from an XML file has been extended. Before looking up the TV channel names XML file, it checks if the name has the form _band-number_ where _band_ is a valid HF band name for the default region. In that case, the tuning information (frequency and optionally polarity) are taken from that channel. Thus, "UHF-22" (ISDB-T) or "BS-8" (ISDB-S) are recognized as valid names in region "japan". Note that CS-band channels are named "CS-8", for instance, not "ND-8" as found in some documents.

So, when you sum up everything, a satellite user in Japan will:
- Add `default.region = japan` and `default.lnb = japan` in the `$HOME/.tsduck`
- Tune using commands such as `tsp -I dvb -c BS-8 ....`

Please note that I have not tested this, neither on ISDB-S (for obvious reasons) nor on DVB-S (the current restrictions due to the corona-virus crisis prevent me from accessing my usual sat reception in the office, I only have DVB-T reception at home).

Could you please test it with ISDB-S? Do not forget other modulation parameters such as `--symbol-rate`. The band name such as "BS-8" can only provide the frequency and polarity. The frequency and polarity are characteristics of the physical channel as defined by some central authority. The other parameters are characteristics of the signal which is inside the channel and are probably not predefined forever.

I will modify `tsscan` to be able to scan channels in bands other that UHF or VHF so that scanning bands BS and CS will be possible in Japan. But, before doing this, I have a question for you. Scanning DVB-T (and probably ISDB-T) bands is easy. The experience shows that tuners do not really care about other parameters such as guard interval or FEC. Just give the right frequency and the tuner will find the signal and adapt to the right modulation parameters. However, with DVB-S (and ISDBD-S?), the modulation parameters shall be precise. If you provide the right frequency and polarity (the "channel" characteristics) but incorrect symbol rate or FEC, you won't receive anything. This is my experience with DVB-S. So, how to scan BS or CS bands? We can try each channel, they aren't so many. But which modulation parameters shall we use? We can provide them as command-line parameters. But, in that case, they should be identical on all channels. My experience with DVB-S shows that different transponders use different modulation parameters, even inside one single network of one single commercial operator. What are your thought on this?

APTX, 2020-05-23:
I updated the config and tried scanning:
```
$ tsscan --verbose -c BS-1 -g --save-channels tsduck-isdb-s-pt3.xml

* Frequency: 11,727,480,000 Hz R
  Transport stream id: 16400, 0x4010
* Frequency: 11,727,480,000 Hz R
  Transport stream id: 16400, 0x4010
* Frequency: 11,727,480,000 Hz R
  Transport stream id: 16400, 0x4010
* Frequency: 11,765,840,000 Hz R
  Transport stream id: 16432, 0x4030
* Frequency: 11,765,840,000 Hz R
  Transport stream id: 16432, 0x4030
* Frequency: 11,880,920,000 Hz R
  Transport stream id: 16528, 0x4090
* Frequency: 11,880,920,000 Hz R
  Transport stream id: 16528, 0x4090
* Frequency: 11,880,920,000 Hz R
  Transport stream id: 16528, 0x4090
* Frequency: 11,957,640,000 Hz R
  Transport stream id: 16592, 0x40D0
* Frequency: 11,996,000,000 Hz R
  Transport stream id: 16625, 0x40F1
* Frequency: 11,996,000,000 Hz R
  Transport stream id: 16625, 0x40F1
* Frequency: 11,804,200,000 Hz R
  Transport stream id: 17488, 0x4450
* Frequency: 11,804,200,000 Hz R
  Transport stream id: 17488, 0x4450
* Frequency: 11,765,840,000 Hz R
  Transport stream id: 16432, 0x4030
* Frequency: 11,919,280,000 Hz R
tsscan: No PAT found, services are unknown
* Frequency: 11,919,280,000 Hz R
tsscan: No PAT found, services are unknown
* Frequency: 11,957,640,000 Hz R
  Transport stream id: 16592, 0x40D0
* Frequency: 12,072,720,000 Hz R
  Transport stream id: 18224, 0x4730
* Frequency: 12,072,720,000 Hz R
  Transport stream id: 18224, 0x4730
* Frequency: 12,072,720,000 Hz R
  Transport stream id: 18224, 0x4730
* Frequency: 12,111,080,000 Hz R
  Transport stream id: 18256, 0x4750
* Frequency: 12,111,080,000 Hz R
  Transport stream id: 18256, 0x4750
* Frequency: 12,111,080,000 Hz R
  Transport stream id: 18256, 0x4750
* Frequency: 12,149,440,000 Hz R
tsscan: No PAT found, services are unknown
* Frequency: 12,149,440,000 Hz R
tsscan: No PAT found, services are unknown
Name       Provider   ServId TSId   ONetId Type PMTPID
---------- ---------- ------ ------ ------ ---- ------
BS朝日1      ビーエス朝日     0x0097 0x4010 0x0004 0x01 0x0101
BS朝日1      ビーエス朝日     0x0097 0x4010 0x0004 0x01 0x0101
BS朝日1      ビーエス朝日     0x0097 0x4010 0x0004 0x01 0x0101
BS朝日2      ビーエス朝日     0x0098 0x4010 0x0004 0x01 0x0201
BS朝日2      ビーエス朝日     0x0098 0x4010 0x0004 0x01 0x0201
BS朝日2      ビーエス朝日     0x0098 0x4010 0x0004 0x01 0x0201
BS朝日3      ビーエス朝日     0x0099 0x4010 0x0004 0x01 0x0203
BS朝日3      ビーエス朝日     0x0099 0x4010 0x0004 0x01 0x0203
BS朝日3      ビーエス朝日     0x0099 0x4010 0x0004 0x01 0x0203
WOWOWプライム  WOWOW      0x00BF 0x4030 0x0004 0x01 0x0101
WOWOWプライム  WOWOW      0x00BF 0x4030 0x0004 0x01 0x0101
WOWOWプライム  WOWOW      0x00BF 0x4030 0x0004 0x01 0x0101
ご案内チャンネル   WOWOW      0x0317 0x4030 0x0004 0xC0 0x0402
ご案内チャンネル   WOWOW      0x0317 0x4030 0x0004 0xC0 0x0402
[... 368 lines cut ...]
|  Transport Stream Id: ............. Unknown  |  Services: .............. 0  |
|  Bytes: ........................ 34,118,240  |  PID's: Total: .......... 1  |
|  TS packets: ...................... 181,480  |         Clear: .......... 1  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ........... 1,105  |         With PCR's: ..... 0  |
|     Suspect and ignored: ................ 2  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ...................... Unknown          Unknown  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ................................................. Unknown  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================
```

`dvbv5-zap` still tuned even though I used the entry with the "wrong" `STREAM_ID`, so I'm not sure how much that value really matters.

I also tried it with `--symbol-rate 28860000` (it's the only value that appears in the TSDuck scan file, `dvbv5` doesn't store a symbol rate), but it does not really change the result.

I've also noticed that, when tuning to an ISDB-S channel with `tsp`, the output .ts file always has packets with transport errors (as listed by `tsanalyze`). I've tried this with reading from `/dev/dvb/adapter0/dvr0` when tuned by `dvbv5-zap` and it did not happen. `tsdump` says the errors are only in the first packets.

lelegard, 2020-05-23:
Hi @APTX,

Thanks for your tests.

In fact, your command does not do a BS band scanning (it is not yet supported). Since you specify some modulation parameter (`-c BS-1`), it does a NIT-based scanning, starting to read the NIT on transponder BS-1. So, it scans only the frequencies which are listed in the NIT.

You can see that `tsscan` exactly follows the NIT structure in a rather stupid way. For instance, it scans the same frequency 11,727,480,000 Hz three times in a row. Then 11,765,840,000 Hz twice. Then, 11,880,920,000 Hz three times. Etc... Of course, each time it reads a frequency, it finds the same TS.

And if you have a look at [this PSI dump from the BS stream you prevously sent me](https://tsduck.io/streams/japan-nippontv/BS-Nippon-Television.psi.txt), you can see that the first three TS in the NIT (TS ids. 16400, 16401, 16402) are in fact on this same frequency 11.72748 GHz with the same modulation parameters. Then, the following two TS (ids 16432, 16433) are on the same frequency 11.76584 GHz. The next three TS share the same 11.88092 GHz. And so on...

There are two frequencies on which we find no PAT: 11,919,280,000 Hz and 12,149,440,000 Hz (both scanned twice, both repeated in the NIT). All modulation parameters are identical in the NIT.

I do not know what `dvbv5` is doing but look at the excerpt of its input file that you included:
~~~ 
[BS01_0]
        DELIVERY_SYSTEM = ISDBS
        FREQUENCY = 1049480
        STREAM_ID = 16400
[BS01_1]
        DELIVERY_SYSTEM = ISDBS
        FREQUENCY = 1049480
        STREAM_ID = 16401
[BS01_2]
        DELIVERY_SYSTEM = ISDBS
        FREQUENCY = 1049480
        STREAM_ID = 16402
~~~
It shows the same duplication as in the NIT: three different TS ids on the same frequency. A TS can have only one TS id.

So, assuming that these informations are correct, that leads to the question: Do you have the possibility to mix different TS in the same carrier frequency in ISDB-S (like DVB-T2 PLP or DVB-S2 multi-stream)? In that case, there should be some identification of a virtual substream.

Interestingly, in the Linux TV interface, there is a parameter named `DTV_ISDBS_TS_ID_LEGACY` which is redefined as `DTV_STREAM_ID`,
the usual command for multistream encapsulation in other standards.

So, now, how do we identify each substream? For instance, the DVB-T2 PLP (physical layer pipe) id is an 8-bit value. When several TS are encapsulated in the same DVB-T2 frequency, the tuning parameters for each TS include the PLP id (0 to 255).

What would be the equivalent for ISDB-S? More important, where is it listed? How does a receiver know which parameter value it should use? The ISDB satellite delivery descriptor does not mention this. I see no additional descriptor in the NIT. When the NIT says that TS ids 16400, 16401, 16402 share the same frequency and same parameters, how can we select them?

Maybe the TS id is used as substream id but this is not really rational. A TS id is a logical value with no physical meaning. It is used to organize a network. A multi-stream "stream id" is something different, it has a physical meaning. If the physical stream id is 16 bits wide (DVB-T2 PLP's are only 8 bits wide), then some convention can be established to use the same value for the physical stream id and the logical TS id of the TS which is inside that physical channel. But, does ISDB-S use this convention? Where is it specified?

Note that if this assumption is correct, then it will be impossible to do blind BS-band scanning as we do on the UHF-band with ISDB-T. Doing blind band scanning means that you need to be able to read each existing TS on a frequency without knowing if it exists and which TS id it uses. With ISDB-S, if (still an hypothesis) we need to know the TS id of a stream in advance to read it, then scanning cannot be blind. It must start from a prior map of the network, as provided in the NIT.

Maybe there is some information in the channel allocation table you mentioned. But since it is in Japanese, I cannot find any useful information (Google Translate is good only when you know where to look at and then you submit one specific paragraph).

With `dvbv5`, are you able to read selectively TS ids 16400, 16401, 16402, all of them being on the same frequency 11.72748 GHz? If this is the case, then I will have a look at `dvbv5` source code.

lelegard, 2020-05-23:
> then I will have a look at dvbv5 source code.

OK, I couldn't resist and had a look in their source code just right now. My assumption was correct. They use multi-stream decapsulation using `DTV_STREAM_ID` as command and the inner TS id as parameter. The same `DTV_STREAM_ID` command is already used to specify DVB-T2 PLP and DVB-S2 ISI (unlike ISDB-S inner stream ids, PLP and ISI are 8-bit physical stream identifiers which are not related to the logical TS id of the inner streams).

I have made a couple of modifications to support this. This is untested as usual, for lack of ISDB-S network in France. Could you please test?

- For direct access to a TS (`dvb` plugin), the option `--stream-id` has been added to specify the TS id inside that frequency. By default, none is specified and you get the current behaviour: the "first" inner TS is used (whatever "first" means here).

- During scanning, the TS id of the various NIT descriptions are used in the scanning phase. So, now, scanning the same frequency multiple times should result in distinct TS.

- When using `-c tv-channel-name`, the TS id of the service is used to jump to the correct sub stream.

- But note that using `-c band-channel-number` (such as `BS-1`) is not sufficient. The band channel describes the physical channel containing multiple TS (frequency and polarisation). If you target one specific TS inside the BS/CS channel, you must select it using `--stream-id`. This is not necessary, however, for `tsscan` which just needs an initial NIT. In a correctly configured network, the NIT should be the same on all TS.

So, it seems that ISDB-S uses the logical TS as physical stream identifier. I do not know if this is an intrinsic builtin property of ISDB-S or a Japanese convention. I am thinking of other ISDB-S networks from the Philippines or South America. Do they use the same convention? We would need local testers to make sure.

APTX, 2020-05-24:
I tried to do band scanning before, but it wasn't supported. As it turned out, the _stream id_ is important and not being provided lowers the usefulness of the satellite physical channels, but they should be usable as a list of "well-known transponders".

I've run the scan using `-c BS-1` again and I'm  happy to report that it's working perfectly:
```
$ tsscan --verbose -c BS-1 -g --save-channels tsduck-isdb-s-test3.xml

* Frequency: 11,727,480,000 Hz R
  Transport stream id: 16400, 0x4010
* Frequency: 11,727,480,000 Hz R
  Transport stream id: 16401, 0x4011
* Frequency: 11,727,480,000 Hz R
  Transport stream id: 16402, 0x4012
* Frequency: 11,765,840,000 Hz R
  Transport stream id: 16432, 0x4030
* Frequency: 11,765,840,000 Hz R
  Transport stream id: 16433, 0x4031
* Frequency: 11,880,920,000 Hz R
  Transport stream id: 16528, 0x4090
* Frequency: 11,880,920,000 Hz R
  Transport stream id: 16529, 0x4091
* Frequency: 11,880,920,000 Hz R
  Transport stream id: 16530, 0x4092
* Frequency: 11,957,640,000 Hz R
  Transport stream id: 16592, 0x40D0
* Frequency: 11,957,640,000 Hz R
  Transport stream id: 16593, 0x40D1
* Frequency: 11,996,000,000 Hz R
  Transport stream id: 16625, 0x40F1
* Frequency: 11,996,000,000 Hz R
  Transport stream id: 16626, 0x40F2
* Frequency: 11,804,200,000 Hz R
  Transport stream id: 17488, 0x4450
* Frequency: 11,804,200,000 Hz R
  Transport stream id: 17489, 0x4451
* Frequency: 11,765,840,000 Hz R
  Transport stream id: 17970, 0x4632
* Frequency: 11,919,280,000 Hz R
  Transport stream id: 18097, 0x46B1
* Frequency: 11,919,280,000 Hz R
  Transport stream id: 18098, 0x46B2
* Frequency: 11,957,640,000 Hz R
  Transport stream id: 18130, 0x46D2
* Frequency: 12,072,720,000 Hz R
  Transport stream id: 18224, 0x4730
* Frequency: 12,072,720,000 Hz R
  Transport stream id: 18225, 0x4731
* Frequency: 12,072,720,000 Hz R
  Transport stream id: 18226, 0x4732
* Frequency: 12,111,080,000 Hz R
  Transport stream id: 18256, 0x4750
* Frequency: 12,111,080,000 Hz R
  Transport stream id: 18257, 0x4751
* Frequency: 12,111,080,000 Hz R
  Transport stream id: 18258, 0x4752
* Frequency: 12,149,440,000 Hz R
  Transport stream id: 18288, 0x4770
* Frequency: 12,149,440,000 Hz R
  Transport stream id: 18289, 0x4771

Name       Provider   ServId TSId   ONetId Type PMTPID
---------- ---------- ------ ------ ------ ---- ------
BS朝日1      ビーエス朝日     0x0097 0x4010 0x0004 0x01 0x0101
BS朝日2      ビーエス朝日     0x0098 0x4010 0x0004 0x01 0x0201
BS朝日3      ビーエス朝日     0x0099 0x4010 0x0004 0x01 0x0203
BS-TBS     BS-TBS     0x00A1 0x4011 0x0004 0x01 0x0101
BS-TBS     BS-TBS     0x00A2 0x4011 0x0004 0x01 0x0102
BS-TBS     BS-TBS     0x00A3 0x4011 0x0004 0x01 0x0103
BS-TBS     BS-TBS     0x02FE 0x4011 0x0004 0xC0 0x0706
Gガイド       BS-TBS     0x0300 0x4011 0x0004 0xC0 0x195F
ＢＳテレ東      ＢＳテレビ東京    0x00AB 0x4012 0x0004 0x01 0x01F0
[... 274 lines cut ...]
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ...... 18097 (0x46B1)  |  Services: .............. 3  |
|  Bytes: ........................ 30,125,308  |  PID's: Total: ......... 21  |
|  TS packets: ...................... 160,241  |         Clear: ......... 17  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 4  |
|     With transport error: ........... 1,066  |         With PCR's: ..... 1  |
|     Suspect and ignored: ................ 2  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ............... 17,389,999 b/s   18,869,999 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: .................................... 13 sec (0 mn 13 sec)  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x00F1  BSスカパー! ...................................... S   14,775,546 b/s  |
|  0x0348  スカパー!ガイド ..................................... C      312,983 b/s  |
|  0x0349  BSスカパー!データ ................................... C          Unknown  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================
```
`dvb` can tune to the channels.

lelegard, 2020-05-24:
Hi @APTX,

Thanks for the good news. So, now TSDuck seems to have a pretty good support for ISDB-S on Linux. And if it can find more channels than `dvbv5`, this is even better...

Concerning the transport errors, you mentioned that they are located in the first packets. I have seen that in the past essentially on Windows where the DirectShow framework is so heavy that the starting phase was sometimes unreliable. If this is endemic with your tuner, then you may use something like `tsp -I dvb ... -P skip 1000 ...` to skip the first 1000 packets (you may test to find the best number of packets to skip).

Concerning, the _channel "BS-TBS" not found_ error, I am a bit puzzled. It should work. It looks for the first service named "BS-TBS" (case-insensitive, spaces ignored) in a TS with a delivery system which is supported by the tuner.

In my case, I have DVB-T and DVB-S tuners. My channel file contains terrestrial and satellite networks. The large national services are available in both networks but, depending on the currently used tuner, it finds the appropriate TS. Of course, since I have no ISDB-S tuner, I can't test your case. I had a look at the source code, in case I found something obvious about ISDB-S in channel files. I found nothing. Maybe there is some silly coding bug that I missed and which I cannot reproduce with DVB tuners.

I have added a few debug messages in the last commit. Could you retry with your channel file? See the example below:

~~~
$ grep -wi arte ~/.tsduck.channels.xml 
      <service id="0x0407" name="Arte" provider="Multi4" LCN="7" PMTPID="0x012C" type="0x19" cas="false"/>
      <service id="0x283E" name="arte HD" provider="ARD" PMTPID="0x13F6" type="0x19"/>
      <service id="0x7034" name="arte" provider="ARD" PMTPID="0x0190" type="0x01"/>
      <service id="0x2081" name="ARTE" provider="CSAT" PMTPID="0x0834" type="0x19"/>
$
~~~
The same channel "Arte" is found in 1) French terrestrial network, 2) and 3) a German satellite, 4) a French satellite.

I use a DVB-T tuner:
~~~
$ tslsdvb
/dev/dvb/adapter0 (Silicon Labs Si2168, DVB-T, DVB-T2, DVB-C)
$ tsp --debug -I dvb -c arte -P until -p 1 -O drop
tsp: Debug: debug level set to 1
* Debug: debug level set to 1
* Debug: trying to load "/home/thierry/tsduck/src/tsplugins/release-x86_64/tsplugin_dvb.so"
* Debug: dvb: debug level set to 1
* Debug: drop: debug level set to 1
* Debug: trying to load "/home/thierry/tsduck/src/tsplugins/release-x86_64/tsplugin_until.so"
* Debug: until: debug level set to 1
* Debug: dvb: loading XML file /home/thierry/.tsduck.channels.xml
* Debug: dvb: loading XML file /home/thierry/tsduck/src/libtsduck/dtv/tsduck.channels.model.xml
* Debug: dvb: searching channel "arte" for delivery systems DVB-T, DVB-T2, DVB-C in /home/thierry/.tsduck.channels.xml
* Debug: dvb: searching channel "arte" in TS id 0x0003, delivery system DVB-T
* Debug: dvb: searching channel "arte" in TS id 0x0002, delivery system DVB-T
* Debug: dvb: searching channel "arte" in TS id 0x0008, delivery system DVB-T
* Debug: dvb: searching channel "arte" in TS id 0x0004, delivery system DVB-T
* Debug: dvb: found channel "arte" in TS id 0x0004
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dvb: using /dev/dvb/adapter0 (DVB-T, DVB-T2, DVB-C)
* Debug: dvb: starting discarding frontend events
* Debug: dvb: finished discarding frontend events
* Debug: dvb: tuning on /dev/dvb/adapter0/frontend0
...
~~~

APTX, 2020-05-24:
Thank you for your help @lelegard .

The new logs did help, I accidentally used the wrong tuner (ISDB-T instead of ISDB-S):
```
* Debug: dvb: searching channel "BS-TBS" for delivery systems ISDB-T in /home/aptx/tsduck-isdb-s-test3.xml
```
I'm not sure how it happened, the tuner in the command in my comment is correct (that is `-a 0` is the ISDB-S tuner). I might have fixed it in the comment and not on the command line...

When using the correct tuner `dvb` finds the channel without issue.

Is there any way of listing the channels in a _tuning file_, like `-g` from `tsscan` does?

----

This reminded me that `tsscan` doesn't seem to be checking the tuner type:
```
tsscan --verbose -a 0 -g --save-channels tsduck-isdb-s-test4.xml
```
(without the `-c BS-1`)

`tsscan` will try the terrestrial frequencies and print lots of:
```
scanning channel 13, 473,142,857 Hz
tsscan: satellite frequency 473,142,857 Hz cannot be transposed using LNB type Japan 110BS/CS
```

----

There is also CS left to try, but I don't think I can receive CS in my apartment or at least neither `dvbv5-scan` nor `tsscan` (using all `-c CS-*`) failed to find anything. I might be able to try it in a different location.

lelegard, 2020-05-24:
> This reminded me that tsscan doesn't seem to be checking the tuner type:
> tsscan --verbose -a 0 -g --save-channels tsduck-isdb-s-test4.xml
> (without the -c BS-1)
> tsscan will try the terrestrial frequencies and print lots of:

This is because, by default, without tuning parameter to a given frequency (such as the removed `-c BS-1`), `tsscan` performs a UHF band scan.

The three possible modes:
- `tsscan --uhf-band` -> scan UHF band
- `tsscan --vhf-band` -> scan VHF band
- `tsscan --frequency ....` (or equivalent tuning option to some specific carrier) -> NIT-based scan using the NIT from the specified frequency.

This is documented but maybe counter-intuitive. I may remove the default and request an explicit option to avoid confusion.

I may also check the tuner type in the case of UHF or VHF scanning. But this can be done by more than one tuner type: DVB-T, DVB-T2, ATSC, ISDB-T. Recent cable tuners are also able to receive terrestrial signals. Only satellite tuners are probably excluded.

NIT-based scanning is acceptable on any type of network.

lelegard, 2020-05-24:
Now, everything seems to work as expected on Linux. So, it is time to say a rude word: "Windows"...

@APTX, do you have any possibility to try your ISDB-S tuner on a Windows system? You may use the daily nightly builds from https://tsduck.io/download/prerelease/

Honestly, I don't expect much from Windows here. ISDB is not explicitly supported on Windows. There is no specific DirectShow objects for that. I had to tweak DVB tuning spaces for that. I was indeed surprised that @nuclidjp was able to successfully use his ISDB-T tuner on Windows right away. I anticipated more trouble.

For ISDB-S, the situation seems worse. Windows does not seem to support any kind of multi-stream encapsulation, neither with DVB-T2 nor DVB-S2. At least, I did not see any suitable parameter (but the documentation for the DirectShow framework is obsolete and incomplete). We have experienced that multi-stream was mandatory to receive all BS channels. But, at least, you could try the "initial" TS of each BS channel.

APTX, 2020-05-25:
I have a computer with windows, and it's probably easier to put the tuner into that computer, than it is to put windows on the linux computer. This will take some time.

APTX, 2020-05-26:
I tried the PT3 tuner on windows, but it does not seem to work with TSDuck at all. `tslsdvb` lists no devices.

The tuner appears in device manager, and a sample application can use it (tune and capture a .ts). The driver comes with it's own SDK, so it probably does not use DirectShow at all, at least the source code of the sample application does not use it.

APTX, 2020-05-26:
I also noticed that the CS physical channels are actually called `ND`, not `CS`. So it should be `ND-1`, not `CS-1` (and so on). I am not really sure why that is.

I still haven't had a chance to try CS.

lelegard, 2020-05-26:
> I also noticed that the CS physical channels are actually called ND, not CS. So it should be ND-1, not CS-1 (and so on). I am not really sure why that is.

To be generic, the format _"name-number"_ is interpreted as a band name and a channel number in that band.

In the file `tsduck.hfbands.xml`, the band names are "UHF", "VHF", "BS", "CS":
~~~
  <hfband type="VHF">
    <region name="japan"/>
    <channels first_channel="1" last_channel="3" base_frequency="93,000,000" channel_width="6,000,000"/>
    <channels first_channel="4" last_channel="12" base_frequency="173,000,000" channel_width="6,000,000"/>
  </hfband>

  <hfband type="UHF">
    <region name="japan"/>
    <channels first_channel="13" last_channel="62" base_frequency="473,142,857" channel_width="6,000,000"/>
  </hfband>

  <hfband type="BS">
    <region name="japan"/>
    <channels first_channel="1" last_channel="24" base_frequency="11,727,480,000" channel_width="19,180,000"
              even_polarity="left" odd_polarity="right"/>
  </hfband>

  <hfband type="CS">
    <region name="japan"/>
    <channels first_channel="1" last_channel="24" base_frequency="12,271,000,000" channel_width="20,000,000"
              even_polarity="right" odd_polarity="left"/>
  </hfband>
~~~

This is why a channel in band CS is named CS-number. If you duplicate or rename the last entry with `type="ND"`, then channel names ND-number become valid. But "ND" is not a band name. It this really important?

Why did they call ND-1, in the first place, a channel in band CS while BS-1 is a channel in band BS?

lelegard, 2020-05-26:
> I tried the PT3 tuner on windows, but it does not seem to work with TSDuck at all. tslsdvb lists no devices.
>
> The tuner appears in device manager, and a sample application can use it (tune and [...]

If it does not use DirectShow, it cannot be used with plugin `dvb`. It there is an SDK, then it could be possible to develop a dedicated input plugin. But that plugin would be for Windows only since the same tuner uses the LinuxTV framework and, thus, can be used by plugin `dvb` on Linux.

I can understand that. DirectShow is an awful nightmare. Additionally, there is no support for ISDB. Windows drivers for ISDB tuners probably use tweaks to appear as DVB tuners and reuse the DVB modulation parameters in an ISDB way (ISDB parameters are essentially a subset of DVB parameters). I would be curious to get `tslsdvb -v` with the ISDB-T tuner which was successfully used by @nuclidjp on Windows.

APTX, 2020-05-27:
I've looked through the SDK source code and found out that the tuner may support more features than the linux driver exposes. The source code is not available for all of the SDK, it has some binary only parts, so it may not be easy at all to try and implement any of them.

For example, there is a voltage setting:
```
		enum LnbPower {
			LNB_POWER_OFF,	// オフ
			LNB_POWER_15V,	// 15V 出力
			LNB_POWER_11V	// 11V 出力
		};

		// [機能] LNB 電源制御
		// [説明] チューナーの電源とは独立に制御可能です。デフォルト値は LNB_POWER_OFF です。
		// [返値] STATUS_DEVICE_IS_NOT_OPEN_ERROR → デバイスがオープンされていない
		//        STATUS_INVALID_PARAM_ERROR      → 引数 lnbPower が NULL (GetLnbPower のみ)
		virtual status SetLnbPower(LnbPower  lnbPower)       = 0;
		virtual status GetLnbPower(LnbPower *lnbPower) const = 0;
```
There are functions mentioning frequency offsets and C/N AGC (signal strength? quality?).

There is a list of supported channels:
```
		// (ISDB-S)
		// +----+------+---------+ +----+------+---------+ +----+------+---------+
		// | ch | TP # | f (MHz) | | ch | TP # | f (MHz) | | ch | TP # | f (MHz) |
		// +----+------+---------+ +----+------+---------+ +----+------+---------+
		// |  0 | BS 1 | 1049.48 | | 12 | ND 2 | 1613.00 | | 24 | ND 1 | 1593.00 |
		// |  1 | BS 3 | 1087.84 | | 13 | ND 4 | 1653.00 | | 25 | ND 3 | 1633.00 |
		// |  2 | BS 5 | 1126.20 | | 14 | ND 6 | 1693.00 | | 26 | ND 5 | 1673.00 |
		// |  3 | BS 7 | 1164.56 | | 15 | ND 8 | 1733.00 | | 27 | ND 7 | 1713.00 |
		// |  4 | BS 9 | 1202.92 | | 16 | ND10 | 1773.00 | | 28 | ND 9 | 1753.00 |
		// |  5 | BS11 | 1241.28 | | 17 | ND12 | 1813.00 | | 29 | ND11 | 1793.00 |
		// |  6 | BS13 | 1279.64 | | 18 | ND14 | 1853.00 | | 30 | ND13 | 1833.00 |
		// |  7 | BS15 | 1318.00 | | 19 | ND16 | 1893.00 | | 31 | ND15 | 1873.00 |
		// |  8 | BS17 | 1356.36 | | 20 | ND18 | 1933.00 | | 32 | ND17 | 1913.00 |
		// |  9 | BS19 | 1394.72 | | 21 | ND20 | 1973.00 | | 33 | ND19 | 1953.00 |
		// | 10 | BS21 | 1433.08 | | 22 | ND22 | 2013.00 | | 34 | ND21 | 1993.00 |
		// | 11 | BS23 | 1471.44 | | 23 | ND24 | 2053.00 | | 35 | ND23 | 2033.00 |
		// +----+------+---------+ +----+------+---------+ +----+------+---------+
```
Note that the odd `ND` channel frequencies don't match the current ones. The tuner is not capable of tuning the left-polarized frequencies as per the [diagram](https://camo.githubusercontent.com/6482b10d2332d4dcb969930de440309f82f127fb/68747470733a2f2f75706c6f61642e77696b696d656469612e6f72672f77696b6970656469612f636f6d6d6f6e732f7468756d622f312f31362f4a6170616e6573655f42726f6164636173745f49465f4672657175656e63792e7376672f3238383070782d4a6170616e6573655f42726f6164636173745f49465f4672657175656e63792e7376672e706e67) as it only goes up to `2150 MHz`.

I have also found that the TS ID for ISDB-S comes from TMCC:
```
		// ISDB-S TMCC 情報
		// (参考) STD-B20 2.9 TMCC情報の構成 ～ 2.11 TMCC情報の更新
		struct TmccS {
			uint32 Indicator;	// 変更指示 (5ビット)
			uint32 Mode[4];		// 伝送モードn (4ビット)
			uint32 Slot[4];		// 伝送モードnへの割当スロット数 (6ビット)
								// [相対TS／スロット情報は取得できません]
			uint32 Id[8];		// 相対TS番号nに対するTS ID (16ビット)
			uint32 Emergency;	// 起動制御信号 (1ビット)
			uint32 UpLink;		// アップリンク制御情報 (4ビット)
			uint32 ExtFlag;		// 拡張フラグ (1ビット)
			uint32 ExtData[2];	// 拡張領域 (61ビット)
		};

		// [機能] ISDB-S の TMCC 情報を取得
		// [返値] STATUS_DEVICE_IS_NOT_OPEN_ERROR → デバイスがオープンされていない
		//        STATUS_INVALID_PARAM_ERROR      → 引数 tuner が 1 より大きいか引数 tmcc が NULL
		virtual status GetTmccS(uint32 tuner, TmccS *tmcc) = 0;
```
I assume that the tuner itself reads TMCC and uses it for tuning (There is also TMCC in ISDB-T). With this information a blind scan should be possible. TMCC is not part of MPEG-2 TS, so I don't know if this is at all useful for TSDuck.

____

> Why did they call ND-1, in the first place, a channel in band CS while BS-1 is a channel in band BS?

I realize it is not consistent, and I have no idea why that was chosen, but that's how the physical channels are referred to. The same can be seen on the [channel allocation table for CS](https://ja.wikipedia.org/wiki/%E6%97%A5%E6%9C%AC%E3%81%AB%E3%81%8A%E3%81%91%E3%82%8B%E8%A1%9B%E6%98%9F%E6%94%BE%E9%80%81#%E7%89%A9%E7%90%86%E3%83%81%E3%83%A3%E3%83%B3%E3%83%8D%E3%83%AB%EF%BC%88%E6%9D%B1%E7%B5%8C110%E5%BA%A6CS%E3%83%87%E3%82%B8%E3%82%BF%E3%83%AB%E6%94%BE%E9%80%81%EF%BC%89).

lelegard, 2020-05-27:
I am not familiar with TMCC. As far as I can read, with ISDB-T, it identifies the layers A, B, C. Those layers A/B/C concepts are exposed by the Linux TV API for ISDB-T. But the same driver API does not include them in ISDB-S.

I read through two academic papers about [ISDB-T](http://www2.elo.utfsm.cl/~elo352/Experiencia%20TVD%20web/PaperTVD.pdf) and [ISDB-S](http://www.tijbc.com/pruebas-7419/I0782925.pdf), respectively. They are clearly focused on modulation. TMCC is described in both cases but identifying layers seems explicit with ISDB-T only.

APTX, 2020-05-27:
@lelegard if you are curious TMCC For ISDB-T is described in _ARIB STD-B31 3.15 and 3.13_. For ISDB-S it's _ARIB STD-B20 2.9-2.11_. I think the papers you linked explain it better than the ARIB specification.

trollcop, 2020-06-19:
I'm pretty sure I'm late to the party but I just noticed this and have some experience with ISDB broacasts.

> So, assuming that these informations are correct, that leads to the question: Do you have the possibility to mix different TS in the same carrier frequency in ISDB-S (like DVB-T2 PLP or DVB-S2 multi-stream)? In that case, there should be some identification of a virtual substream.

Yeah, what happens is on BS/CS, a single 34MHz-wide channel with 8PSK modulation is ~50mbit of bandwidth, gets split into 48 "slots" which are then assigned to different providers under same TSID.

So you'll have for example:
Provider 1 tsid 0x0001 18 slots
Provider 2 tsid 0x0002 18 slots
Provider 3 tsid 0x0003 12 slots

iirc for PT3 / Toshiba demod, if you don't specify a TSID, the 1st one in the transponder will be used.
When doing NIT scanning, TSID *must* be set to get expected results during scan.

I don't know if that was the problem (i skimmed through the replies here) but if it was, that's the reason.

> I tried the PT3 tuner on windows, but it does not seem to work with TSDuck at all. tslsdvb lists no devices.

Yeah, it's not directshow at all, just a custom Jungo WinDriver interface to PCI device.

lelegard, 2020-06-19:
> I don't know if that was the problem (i skimmed through the replies here) but if it was, that's the reason.

It was indeed. After adding option `--stream-id` in the `dvb` plugin (still named DVB sorry), users in Japan were able to access any TS inside the sat channel. The same principle was applied to NIT-based scanning and it worked.

This works on Linux only. Windows / DirectShow has no way to specify a stream id inside a frequency (DVB-T2, DVB-S2, ISDB-S, none). Or at least I could not find a way.

If you are experienced in both ISDB and Windows DirectShow, could you explain a bit more how it works? Linux has an extensive support for all modulation options, with regular updates of the API with new features. DirectShow, on the other hand, seems to be frozen for at least 10 years with poor or no support for recent evolutions or even ISDB entirely. DirectShow works fine on DVB and ATSC (except multi-stream and some other recent features). And that's all.

trollcop, 2020-06-19:
> Windows / DirectShow has no way to specify a stream id inside a frequency (DVB-T2, DVB-S2, ISDB-S, none). Or at least I could not find a way.

This is card vendor-specific. Usually some IInterface is exposed which you query which then has methods for controlling hardware-specific stuff. You can see an example here for controller ISDB-S capable card from Digital-Devices:

https://github.com/radi-sh/BDASpecial-DD/blob/master/BdaSpecials/DD/DDSpecials.cpp#L501

I don't know any other PCI-based ISDB-S card but anyway, its up to the vendor to enable such support. You then have to check against known support hardware/driver list and enable such feature.

By the way about the LNB, yeah, current 2K broadcasting uses only one polarization and no band switching, so it all fits over a single cable. No signalling or voltage difference needed. Newly started 4K/8K satellite band stacks opposite polarization above 2150MHz, going all the way to 3.2GHz, so it again allows building-wide shared LNBs which do not require DiSeQc switches or voltage control of band/polarity.

lelegard, 2020-06-19:
> This is card vendor-specific.

OK, so that works only for proprietary TV-watching applications. TSDuck wants to be tuner vendor-agnostics and relies only on common public API's from the operating system (ie. LinuxTV and DirectShow). It seems that Windows will become a less and less DTV-friendly system.

> Also, authoritative source on 110BS/CS channel, frequency, slot, etc allocation is here, 

Japanese only 😞 Like all recent versions of all ARIB documents. Japanese TV seems to protect investment for Japanese industries only (or companies which can pay a Japanese translator).

>  Blind scan in ISDB-S domain is largely useless as the frequencies

It also proved inapplicable because of multi-stream. We can blindly scan a single-TS frequency, not a multi-TS one without prior knowledge of the TS ids inside. Or is there a way?

trollcop, 2020-06-19:
> OK, so that works only for proprietary TV-watching applications. TSDuck wants to be tuner vendor-agnostics and relies only on common public API's from the operating system (ie. LinuxTV and DirectShow). It seems that Windows will become a less and less DTV-friendly system.

Well, vendors will usually publish the API, such as digital devices, TBS, DVBWorld etc. I wonder if there's any better support for newer modulations/standards in MediaFoundation. I haven't looked.

> Japanese only 😞 Like all recent versions of all ARIB documents. Japanese TV seems to protect investment for Japanese industries only (or companies which can pay a Japanese translator).

This has been the case always, this is the classic case of "Not Invented Here".

> It also proved inapplicable because of multi-stream. We can blindly scan a single-TS frequency, not a multi-TS one without prior knowledge of the TS ids inside. Or is there a way?

TSID information for a transponder is in TMCC data:

[image]

When a transponder is tuned, this should be available via demod interface. Then you can pick the correct TSID to select.

lelegard, 2020-06-19:
> Well, vendors will usually publish the API, 

I haven't seen any.

> When a transponder is tuned, this should be available via demod interface. Then you can pick the correct TSID to select.

Which demod interface? DirectShow? LinuxTV? I haven't seen any.

trollcop, 2020-06-19:
> I haven't seen any.

I linked one example (digital-devices), there's 2 more for TBS/DVBWorld https://github.com/radi-sh/BonDriver_BDA/tree/master/BdaSpecials

> Which demod interface? DirectShow? LinuxTV? I haven't seen any.

Not sure since I work with embedded where I have direct access to hardware. But TMCC is a part of ISDB that's not going away, so if not already exposed by lunixtv it probably should be.

APTX, 2021-08-07:
I recently tried to scan BS again (I still don't have access to CS) on a new system and got the `FE_SET_TONE` error again. I reread this issue and found the `--lnb` option, but the error would remain:
```
tsscan --japan --lnb japan -a 0 -c BS-1 --nit-scan -g --update-channels tsduck_channels.xml
tsscan: DVB frontend FE_SET_TONE error: Unknown error 524
```

There was also mention of putting the value in `.tsduck` so I tried it again with this in the config file:
```
default.region = japan
default.lnb = japan
```
This got rid of the error. Is the way I passed the argument on the command line wrong?

I also noticed that the ISDB-S channels always have `0` frequency in the xml file:
```xml
  <network id="0x0004" type="ISDB-S">
    <ts id="0x4010" onid="0x0004">
      <isdbs frequency="0" symbolrate="28,860,000" polarity="vertical" FEC="2/3"/>
      <service id="0x0097" name="BS朝日1" provider="ビーエス朝日" PMTPID="0x0101" type="0x01" cas="false"/>
      <service id="0x0098" name="BS朝日2" provider="ビーエス朝日" PMTPID="0x0201" type="0x01" cas="false"/>
      <service id="0x0099" name="BS朝日3" provider="ビーエス朝日" PMTPID="0x0203" type="0x01" cas="false"/>
    </ts>
  <!-- more entries -->
  </network>
```
With frequency `0` it's not possible to tune to any of the ISDB-S channels. ISDB-T scanning still works as before.

I checked the config file from the system I did the scans previously and that config **does** have frequencies set. Either something broke since or it's some weird arm64 issue.

----

TMCC is kind of neat. It simplifies tuner configuration and can make channel scanning a lot faster. It's much easier to detect the TMCC signal (It's always BPSK IIRC). If it's not there there the frequency can be skipped altogether. I've only seen the PT3 SDK give access to TMCC data, but the SDK is windows only and does not use DirectShow.

I don't know of any ISDB tuner that can be used on windows that uses DirectShow. There's also a bunch of tuners that have linux drivers that do not use linux DVB APIs.

PT3 (the only ISDB-S tuner that uses the linux dvb API that I know) tunes to the first TSID in TMCC if it's not set (and I would guess all tuners behave like that). From there a NIT scan can find all the other services.

ISDB tuners have to use TMCC information even if they do not expose it as it carries transmission parameters (the layers in ISDB-T and services in ISDB-S among other things). The linux DVB API exposes lots of [ISDB-T layer properties](https://www.kernel.org/doc/html/v5.14-rc1/userspace-api/media/dvb/fe_property_parameters.html#dtv-isdbt-layer-a-c-parameters), but I don't know if any tuner uses them as it's all in TMCC (it could be used by a modulator?).

lelegard, 2021-08-07:
Hi @APTX,

Please open a new issue per new problem. This thread was used one year ago and was extremely useful to add ISDB support. I would prefer to discuss new problems separately.

- Considering the inconsistency of the `--japan` option between the command line and configuration file, which version of TSDuck are you using? Did you try the latest state of the repo?

- Considering TMCC, I just discovered that it means "transmission and modulation configuration control" after some googling. 
How does a tuner receive it? What kind of software API can be used?

APTX, 2021-08-08:
I could reproduce the issues with a build from current git HEAD, so I created #840 and #841 for the scanning issues.

As for how a tuner receives the TMCC signal: As far as I understand (and I don't really understand signals and modulation so this could be wrong) the TMCC signal is there on the same frequency as the broadcast signal. On a [constellation diagram](https://en.wikipedia.org/wiki/Constellation_diagram) TMCC and the broadcast signal do not overlap, so it's possible to receive both at the same time.

As for obtaining TMCC information: The only public API I am aware of is the [PT-3 SDK](https://earthsoft.jp/PT3/download_SHA-2.html) (Relevant source code is the `PT3-Example-400.zip` download). I wrote a TSDuck tsp plugin that uses the SDK to use PT3 with TSDuck on windows. I don't know if I can share it because, while the SDK source code is available to anyone, the SDK has no license information whatsoever.

DekTec tuners might give access to TMCC information. DekTec modulators expect TMCC information to be part of the TS, so maybe a DekTec tuner would write ts files like that (I don't have a DekTec tuner).

I guess it would also be possible to get TMCC information with an SDR.

trollcop, 2021-08-08:
> I could reproduce the issues with a build from current git HEAD, so I
> created #840 <https://github.com/tsduck/tsduck/issues/840> and #841
> <https://github.com/tsduck/tsduck/issues/841> for the [...]
correct

> As for obtaining TMCC information: The only public API I am aware of is
> the PT-3 SDK <https://earthsoft.jp/PT3/download_SHA-2.html> (Relevant
> source code is the PT3-Example-400.zip download). I [...]
TMCC is easily accessible from the demodulator registers after tuning, the
only problem is of course those are generally not public or portable. no
other standard has a similar hierarchical metadata available even if tuning
of the main modulation fails so I don't think any API actually supports it
other than through extensions.
TMCC in satellite is useful because you can get the neighbor tsid values,
but it's usefulness in terrestrial is questionable unless you are trying to
understand how the signal is framed - which layers are partial reception
which layers are QAM64 and the code rate and similar stuff.

DekTec tuners might give access to TMCC information. DekTec modulators
> expect TMCC information to be part of the TS, so maybe a DekTec tuner would
> write ts files like that (I don't have a DekTec tuner).
dektek does not have any isdbs tuner. I think they do have something with
isdbt but yeah.

> I guess it would also be possible to get TMCC information with an SDR.
I have actually done this for terrestrial before, not TMCC but the AC which
is the earthquake warning data which is also encoded in bpsk. you don't
even need to receive the full 8 megahertz band you just need to receive
enough to get the narrow encoded bpsk stuff.

APTX, 2021-08-09:
> I have actually done this for terrestrial before, not TMCC but the AC which
is the earthquake warning data which is also encoded in bpsk.

@trollcop is this code publicly available?

> TMCC in satellite is useful because you can get the neighbor tsid values, but it's usefulness in terrestrial is questionable unless you are trying to understand how the signal is framed

Another useful thing might be the emergency bit which says if there is an emergency broadcast on this frequency (but I have no idea if it's used in practice). 

TMCC also allows to switch the tuning parameters during broadcast. There is some defined procedure where TMCC will change first and notify that the parameters will change. I have no idea if this is used in the real world.