tsduck/tsduck #583: error make debian  stretch

dayser, 2020-05-20:
`make
  [CXX] utest.cpp
In file included from utest.cpp:43:0:
tsunit.h:649:165: error: specialization of 'template<class T> static void tsunit::Assertions::equal(const T*, const T*, const string&, const string&, const char*, int)' in different namespace [-fpermissive]
 inline void tsunit::Assertions::equal<char>(const char* expected, const char* actual, const std::string&, const std::string&, const char* sourcefile, int linenumber)
                                                                                                                                                                     ^
tsunit.h:560:21: error:   from definition of 'template<class T> static void tsunit::Assertions::equal(const T*, const T*, const string&, const string&, const char*, int)' [-fpermissive]
         static void equal(const T* expected, const T* actual, const std::string& estring, const std::string& vstring, const char* sourcefile, int linenumber);
                     ^~~~~
tsunit.h:655:177: error: specialization of 'template<class T> static void tsunit::Assertions::equal(const T*, const T*, const string&, const string&, const char*, int)' in different namespace [-fpermissive]
 inline void tsunit::Assertions::equal<char16_t>(const char16_t* expected, const char16_t* actual, const std::string&, const std::string&, const char* sourcefile, int linenumber)
                                                                                                                                                                                 ^
tsunit.h:560:21: error:   from definition of 'template<class T> static void tsunit::Assertions::equal(const T*, const T*, const string&, const string&, const char*, int)' [-fpermissive]
         static void equal(const T* expected, const T* actual, const std::string& estring, const std::string& vstring, const char* sourcefile, int linenumber);
                     ^~~~~
../../Makefile.common:909: recipe for target 'release-x86_64/utest.o' failed
make[2]: *** [release-x86_64/utest.o] Error 1
Makefile:43: recipe for target 'default' failed
make[1]: *** [default] Error 2
Makefile:59: ошибка выполнения рецепта для цели «default»
make: *** [default] Ошибка 2`
pls help me

lelegard, 2020-05-20:
Hi,

I no longer have any Debian or Raspbian 9 (stretch). Which version of GCC does it use? 6.3 maybe?

I currently test without problem on these platforms:
- Raspbian 10 (buster) : GCC 8.3
- Ubuntu 20.04 : GCC 9.3
- Fedora 32 : GCC 10.0
- Windows 10 : Visual Studio 2019 (16.5)

As a test, I installed package `g++-6` on Ubuntu 20.04 and got a parallel install of GCC 6.4. I rebuilt TSDuck using command `make CXX=g++-6 CC=gcc-6 -j10`. And I see the same compilation issue. It seems linked to GCC 6.x.

I have made a modification to avoid that compilation issue with GCC 6.x. You may retry with the latest repo. I hope it won't trigger another issue with some other compiler.

dayser, 2020-05-20:
> 

> Hi,
>
> I no longer have any Debian or Raspbian 9 (stretch). Which version of GCC does it use? 6.3 maybe?
>
> I currently test without problem on these platforms:
>
> * Raspbian 10 (buster) : GCC 8.3
> * [...]

gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)

dayser, 2020-05-20:
> Hi,
>
> I no longer have any Debian or Raspbian 9 (stretch). Which version of GCC does it use? 6.3 maybe?
>
> I currently test without problem on these platforms:
>
> * Raspbian 10 (buster) : GCC 8.3
> * [...]

tnx! make completed.

closed 2020-05-21 by lelegard