tsduck/tsduck #584: Request for feedback: completing ISDB Japan support for v3.21

lelegard, 2020-05-24:
Dear TSDuck users in Japan, especially @masnagam, @nuclidjp, @APTX,

Since last released version 3.20, a number of works have been done to improve TSDuck for Japanese users, including:
- ARIB STD-B24 character sets
- ARIB-defined ISDB signalisation (mostly done but not fully completed)
- ISDB-T tuners tested on Linux and Windows (reception and UHF band scanning)
- ISDB-S tuners tested on Linux (multi-stream reception and NIT-based scanning), feasibility to be confirmed for Windows
- More general definitions of LNB's, including polarisation-controlled LNB's as used with BC/CS bands
- Definition of various shortcuts such as synthetic option `--japan` (see also sample configuration file in appendix A.3 of user's guide).

At some point, this deserves the release of a new version. Before releasing version 3.21, do you see anything to add?

Thanks for your feedback

nuclidjp, 2020-05-25:
@lelegard 

If possible, support B-CAS.
B-CAS is indispensable for broadcasting in Japan.
I may want to check the TS file with `tsp -O play`.

lelegard, 2020-05-25:
Hi @nuclidjp 

> If possible, support B-CAS.

I understand the need but you should not expect any implementation of B-CAS in TSDuck anytime soon, unfortunately, for many good reasons:

- General reasons:
  - B-CAS is certainly required to watch TV in Japan but TSDuck is not a TV-watching application. This is a technical toolbox for engineers to test and manipulate TV systems.
  - Outside CAS vendors and TV operators, descrambling TV signals is rarely done for legitimate purposes. TSDuck will never support hackers' operations.

- Technical reasons: There are two independent parts in such operations, content (de)scrambling and smartcard operations.
  - As far as I understand, Japanese scrambling algorithm is Multi-2. I don't have any Multi-2 implementation or specification. Although being an experienced crypto user, I am not a cryptographer. And experienced crypto users certainly know that you shouldn't mess with cryptography when you aren't a cryptographer.
  - In practice, implementing that kind of system requires simultaneous coding and debugging, using a real smartcard in a real environment. The developer must be physically located in Japan and must have access to real networks.

APTX, 2020-05-26:
Is there any technical limitation in TSDuck that makes it impossible to make a TV-watching application out of it (besides the scrambling)?

As far as I know [1seg](https://en.wikipedia.org/wiki/1seg) is not scrambled so @nuclidjp could try those.

Speaking of 1seg: It would be nice if TSDuck knew that `0xC0` service type is 1seg.

Example `tsanalyze` output for a 1seg service:
```
|=============================================================================|
|  Service: 0x05A0 (1440), TS: 0x7FE4 (32740), Original Netw: 0x7FE4 (32740)  |
|  Service name: フジテレビ, provider: (unknown)                                   |
|  Service type: 0xC0 (Data service)                                          |
|  TS packets: 20,925, PID's: 9 (clear: 9, scrambled: 0)                      |
|  PMT PID: 0x1FC8 (8136), PCR PID: 0x0150 (336)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Data service ................................. C      400,973 b/s  |
|  0x0150  PCR (not otherwise referenced) ............... C        6,496 b/s  |
|  0x0151  AVC video (320x180, baseline profile, level 1  C      216,803 b/s  |
|  0x0152  MPEG-2 AAC Audio (Audio layer 0, @32,000 Hz,   C       50,128 b/s  |
|  0x0154  MPEG-2 PES private data ...................... C          153 b/s  |
|  0x0850  DSM-CC Sections .............................. C       46,986 b/s  |
|  0x0851  DSM-CC Sections .............................. C       77,396 b/s  |
|  0x085A  DSM-CC Sections .............................. C            0 b/s  |
|  0x085B  DSM-CC Sections .............................. C            0 b/s  |
|  0x1FC8  PMT .......................................... C        3,008 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================
```

____

It would also be nice if options provided on the command line did not interfere with options from the config file. I added `--japan` to the config file and if I also use it on the command line an error appears:
```
$ tsanalyze --japan file.ts
tsanalyze: too many option --japan
```
This prevents you from running commands as they appear in GitHub issues/comments as you either have to remove the config or edit the command.

It would also be useful if the tools could print the options they actually use (when run with `--verbose` or `--debug`) as it may not be immediately obvious as some of the options are in the config file.

lelegard, 2020-05-26:
> Is there any technical limitation in TSDuck that makes it impossible to make a TV-watching application out of it (besides the scrambling)?

TSDuck can be used to watch TV. Just try `tsp -I dvb ... -P zap ... -O play`. It is just much less friendly than the applications which usually come pre-packaged with tuners to watch TV on your PC. The output plugin `play` is usually a trivial way to verify test results, visually.

I meant that watching TV has never been a primary objective for TSDuck. So, unlike a pure TV watching application, implementing descrambling is useless to most engineers using TSDuck for their tests. There is a `descrambler` plugin but it is designed for technical tests using specific keys for specific test cases. It can also use "fake ECM" from `tsecmg` to emulate a CAS and test the CAS integration capabilities of a real MUX for instance. The other way round, the plugins `scrambler` and `datainject` emulate a MUX to test the integration of a real CAS. We are a bit far away from a leisure application to watch TV :)

There is no open-source descrambling plugin for TSDuck which interfaces a real commercial CAS product. You can still develop one as closed source if you know the specification of the smart card, although all modern CAS include CW encryption which requires knowledge of secret keys and algorithms. Having worked for 4 CAS vendors (and still working for one), I have developed a few closed-source plugins to test proprietary products. So, this is technically feasible but not open source for obvious security reasons.

> It would also be nice if options provided on the command line did not interfere with options from the config file. I added --japan to the config file and if I also use it on the command line an error appears

You are right. Although this is not related to Japan ISDB, I should probably allow multiple usage of the same boolean option because it is just redundant and harmless.

> It would also be useful if the tools could print the options they actually use (when run with --verbose or --debug) as it may not be immediately obvious as some of the options are in the config file.

Another good idea. Will do...

APTX, 2020-05-27:
@lelegard thank you for your explanation.

I checked that `0xC0` is defined as "Data Service" in _ARIB STD-B10 Part 2 6.2.13 Service  descriptor_. I haven't found a reference to 1seg and that service type, but the PIDs contained in the service match what should be in 1seg.

In general TSDuck fully supports 1seg as there isn't really anything different in it from the TS perspective. It's just a much lower bandwidth (and quality) service, so this really is just about identifying that a service is 1seg. I thought that the service type would be enough, but it isn't.

All the [ISDB-T test streams in the TSDuck stream repository](https://tsduck.io/streams/?name=japan-dttv) have a 1seg service. A 1seg service always seems to have: type `0xC0`/_Data Service_, audio/video, clear (unscrambled) data.

`AHK.ts` also has a _Data Service_ that is not 1seg (Gガイド). The only difference is that it doesn't have audio/video, it's only _DSM-CC_.

1seg plays in VLC, so it's easy for anyone to try. It may be necessary to select the 1seg service in Playback->Program or using `zap` to remove other services.

nuclidjp, 2020-05-27:
@lelegard 

Oops, I forgot that One Seg isn't scrambled.
B-CAS support is not important if I can confirm with 1Seg.
The B-CAS support request was a mistake.

lelegard, 2020-05-27:
@APTX, data services are used to transport applications or software updates. All their PID's are typically DSM-CC ou specialized sections. They are seen as technical services, not proper "TV channels" and are not displayed to the users (set-top boxes or TV sets). I assume that the 1seg service is for mobile reception and should not be displayed in the channel list of the TV set. Is that right? In that case, declaring it as a data service is a cheap and easy way to hide it from the channel list. Mobile receivers probably have dedicated software to identify 1seg.

> In general TSDuck fully supports 1seg as there isn't really anything different in it from the TS perspective.

Yes, TSDuck is a technical toolbox to manipulate transport streams. Some users don't care about TV channels at all.

lelegard, 2020-05-27:
> It would also be nice if options provided on the command line did not interfere with options from the config file. I added --japan to the config file and if I also use it on the command line an error appears

> It would also be useful if the tools could print the options they actually use (when run with --verbose or --debug) as it may not be immediately obvious as some of the options are in the config file.

Both implemented.

lelegard, 2020-05-29:
If there is nothing more, I will release version 3.21 over the week-end.

lelegard, 2020-05-30:
Just released 3.21

closed 2020-05-30 by lelegard