tsduck/tsduck #585: What is the difference between the "master" and "switch" branch ?

bobben1no, 2020-05-26:
Hi !

I probably have a question which is easy to answer.  I have tried to make some "new" plugin in 3.19-1411 by editing existing plugin and try to add some extra logging functions and it works.  But when I downloaded the latest 3.20 version tsduck-master I noticed that the plugin source do not look identical. Then I downloaded the tsduck-switch branch which had identical plugin source to the one I build on my Ubuntu system.  With tsduck-switch source I could add the "new" plugin that worked in 3.19. I am using Ubuntu. 

Also when I have downloaded non-modified source on Windows I have only success building & build installers by using the tsduck-master branch. tsduck-switch branch only causes errors in the powershell. Not a big problem as I prefer Linux Ubuntu but I am curious what the difference is between the branches.

Example pes plugin tsduck-master:
`TS_REGISTER_PROCESSOR_PLUGIN(u"pes", ts::PESPlugin);`

Example pes plugin tsduck-switch:
`TSPLUGIN_DECLARE_VERSION
TSPLUGIN_DECLARE_PROCESSOR(pes, ts::PESPlugin)`

lelegard, 2020-05-26:
The branch `switch` is obsolete. It was an attempt to change the architecture of the `tsswitch` utility in order to fix a long-standing bug. I never completed it. I keep it for reference when I come back to that issue some day, to remember what I did at that time.

The compilation issue is due to recent changes I made in the code. The old plugin interface was inherited from version 1.0 back in 2005 (!) when TSDuck was written in C instead of C++. The old interface exported predefined symbols which were called by the main application. In the meantime, additional forms of automatic registrations were added (tables, descriptors, section filters, extensions) using more modern techniques. For consistency and robustness, the plugin registration now uses the same kind of technique.

The sample applications (subdirectory `sample`) were updated.

I should have defined compatibility macros. I will do. But using the new registration is better anyway.

lelegard, 2020-05-26:
> I should have defined compatibility macros

Done, the old plugin registration macros have been redefined and redirect to the new syntax.