tsduck/tsduck #586: CA_mode in EIT vs SDT

tarelda, 2020-05-27:
I am working through similiar issue to https://github.com/tsduck/tsduck/issues/558 . 
I have two streams main and let's call it "merged". First one doesn't have EPG information of any sort and I am getting it from device that already handles decryption. On the other hand second has elementary streams scrambled, but EIT is not encrypted.
My workflow for "merged stream" consists of following steps:
1. Zap SID that I am interested in and adjust SID to the one in main stream.
2. Update TSID and ONID to match main stream. 
3. Filter PID 18.
After that streams are merged and to further adjust the stream I apply SDT transformations.
I mark my SID as one accompanied with PF EIT and schedule EIT. 
So far so good, before and after DVBT modulation (I don't use tsduck for that), EPG can be used by receivers. But there is one catch. When checking with eit plugin, tsduck reports that there is EPG for 0 days. In contrary to original "merged" stream that has report of 7 days worth of EPG. I dumped EIT with tables plugin and found out that only field that doesn't match my service definition is CA_mode. I obviously tried changing SDT table with sdt plugin before merging "merged" stream, but it doesn't update EIT information. 
It seems to be more chicken and egg kind of situation than issue, but still wanted to report it. Since it looks like I will be doing a lot more of this kind of stuff and all my "merged" stream transformations are only to adjust EIT, I am thinking about investing my time into developing EIT merging plugin that does all this stuff on its own and allows for changing CA_Mode too.  What do you think about that?

lars18th, 2020-06-01:
Hi @tarelda ,

The idea of one "EIT merging" plugin seems very interesting. From my point of view the current implementation of TSDuck is not sufficient in the EIT area. For example, please consider this "simple" scenario:

- You want to "smooth" the EIT tables in a custom created MPTS. The problem is that the bitrate of the EIT information is very high for your wandwitdh, and you want to _reduce_ the bitrate of EIT tables to adapt it to the target bitrate of your MPTS. This can be done reading all incoming EIT tables and "re-scheduling" them with a low repetition rate.

This problem is very common when you want to merge different services from different sources to modulate one DVB-T signal. However, with the current TSDuck I don't have a solution to this problem.

Futhermore, inside the code of the TSDuck it's already implemented some kind of "PSI/SI generic table parser-scheduler". That's a class tha can function as a DB (database) of tables, that can read data at any point and write when necessary. I feel it will be useful to use/extend this class to develop the scheduler EIT plugin. This plugin will can then do a lot of useful functionalities: smooth, merge, process, etc.

So if you want to develop such plugin it will be welcome. I feel @lelegard don't have implemented it before because lack of time and interest. But I'm sure he will accept with open arms for this contribution.

Regards.

tarelda, 2020-06-03:
@lars18th In my case this is not useful feature for TSDuck since my DVB modulators have it implemented. 
Also I found out why I had report of 0 days of EPG in eit plugin. For some reason despite passing TDT during merge it was not descending into target stream. Preventing psi merge fixed it.

lars18th, 2020-06-05:
Hi @tarelda ,

> @lars18th In my case this is not useful feature for TSDuck since my DVB modulators have it implemented.

Then you're using professional DVB modulators. In my case I have only a cheap HiDes USB, so I do all with TSDuck. That's a huge difference!

> Also I found out why I had report of 0 days of EPG in eit plugin. For some reason despite passing TDT during merge it was not descending into target stream. Preventing psi merge fixed it.

OK. Then your problem is fixed. Anyway I feel the EIT plugin can be improved in several areas. And if you can report how to reproduce the problem and solve it, then @lelegard will can fix the code. You agree?

Regards.