tsduck/tsduck #587: pcrverify: Weird jitter values reported

cbernander, 2020-05-28:
Hi!

Whenever I try to run _pcrverify_ on MPEG-TS UDP multicast streams, I get a lot of extreme jitter values reported which to me suggests there is some bug in the jitter calculation.

I can also reproduce this locally using _ffmpeg_ to generate a multicast stream:

`ffmpeg -re -i bbb_sunflower_1080p_60fps_normal.mp4 -r 30 -t 00:00:10 -c:v libx264 -preset ultrafast -tune zerolatency -x264-params "nal-hrd=cbr" -b:v 1M -minrate 1M -maxrate 1M -bufsize 2M -f mpegts "udp://224.0.0.100:3000?pkt_size=1316"`

```
$ tsp -I ip --receive-timeout 5000 224.0.0.100:3000 -P pcrverify -t -j 100000 -P pcrextract --pcr -o pcr.csv.log -O drop
* pcrverify: 2020/05/28 13:02:25, PID 256 (0x0000000000000100), PCR jitter: 12,070,644,220,459 = 447,060,897,054 micro-seconds = 454,263,511 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:26, PID 256 (0x0000000000000100), PCR jitter: 12,070,647,648,234 = 447,061,024,008 micro-seconds = 454,263,640 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:26, PID 256 (0x0000000000000100), PCR jitter: 12,070,647,913,953 = 447,061,033,850 micro-seconds = 454,263,650 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:26, PID 256 (0x0000000000000100), PCR jitter: 12,070,647,887,381 = 447,061,032,865 micro-seconds = 454,263,649 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:26, PID 256 (0x0000000000000100), PCR jitter: 12,070,649,986,561 = 447,061,110,613 micro-seconds = 454,263,728 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:26, PID 256 (0x0000000000000100), PCR jitter: 12,070,651,368,300 = 447,061,161,788 micro-seconds = 454,263,780 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:26, PID 256 (0x0000000000000100), PCR jitter: 12,070,650,385,140 = 447,061,125,375 micro-seconds = 454,263,743 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:26, PID 256 (0x0000000000000100), PCR jitter: 12,070,650,145,993 = 447,061,116,518 micro-seconds = 454,263,734 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:27, PID 256 (0x0000000000000100), PCR jitter: 12,070,650,704,003 = 447,061,137,185 micro-seconds = 454,263,755 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:27, PID 256 (0x0000000000000100), PCR jitter: 12,070,651,580,875 = 447,061,169,662 micro-seconds = 454,263,788 packets + 152 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:28, PID 256 (0x0000000000000100), PCR jitter: 10,446,956,039,877 = 386,924,297,773 micro-seconds = 454,263,789 packets + 113 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:28, PID 256 (0x0000000000000100), PCR jitter: 10,446,956,016,880 = 386,924,296,921 micro-seconds = 454,263,788 packets + 113 bytes + 5 bits
* pcrverify: 2020/05/28 13:02:30, PID 256 (0x0000000000000100), PCR jitter: 10,446,955,964,694 = 386,924,294,988 micro-seconds = 454,263,786 packets + 63 bytes + 1 bits
* pcrverify: 2020/05/28 13:02:30, PID 256 (0x0000000000000100), PCR jitter: 10,446,950,540,367 = 386,924,094,087 micro-seconds = 454,263,550 packets + 88 bytes + 3 bits
* Error: ip: error receiving from UDP socket: Resource temporarily unavailable
* pcrverify: 86 PCR OK, 14 with jitter > 2,700,000 (100,000 micro-seconds), 1 unchecked
```
See attached _pcrextract_ output.
[pcr.csv.log](https://github.com/tsduck/tsduck/files/4695130/pcr.csv.log)

lelegard, 2020-06-01:
Hi @cbernander,

Thank you for reporting this. This is a bug (an intermediate integer overflow). I just fixed it. I added a non-regression test (number 64 in the [test suite](https://github.com/tsduck/tsduck-test)).

The output is now:
~~~
* pcrverify: PID 256 (0x0100), PCR jitter: -4,985,013 = 184,630 micro-seconds = 204 packets + 62 bytes + 0 bits
* pcrverify: PID 256 (0x0100), PCR jitter: -3,691,979 = 136,739 micro-seconds = 151 packets + 62 bytes + 0 bits
* pcrverify: PID 256 (0x0100), PCR jitter: -7,107,540 = 263,242 micro-seconds = 291 packets + 62 bytes + 0 bits
* pcrverify: PID 256 (0x0100), PCR jitter: -3,399,216 = 125,896 micro-seconds = 139 packets + 62 bytes + 0 bits
* pcrverify: PID 256 (0x0100), PCR jitter: -6,006,997 = 222,481 micro-seconds = 246 packets + 41 bytes + 2 bits
* pcrverify: 95 PCR OK, 5 with jitter > 2,700,000 (100,000 micro-seconds), 1 unchecked
~~~

Note that the PCR verification is based on a constant evaluated bitrate (typically a broadcast stream). When the stream is VBR, there will always be some jitter since the reference is constant and the reality is variable.

The evaluation is done on PCR only, without any influence from wall clock time. So, the easiest way to reproduce it was to save `ffmpeg` output into a TS file and use `tsp` on that file. Thus, we get rid of the IP setup. This TS file is now stored in the test suite for non-regression. The `ffmpeg` command enforced a min and max video bitrate of 1 Mb/s. The actual video bitrate (based on PCR's) is evaluated to 1,162,901 b/s.

cbernander, 2020-06-02:
Thanks for the fix and for the clarification regarding the PCR verification method used.

It would have been really nice to have a PCR verification method based on a monotonic clock (since it seems VBR is becoming increasingly common in IP multicast systems), and if this was combined with a PCR interval check (which I think there already exists a feature request for) it would be a real killer feature! Oh well, maybe I'll create a feature request for it some day (or try to learn C++ - again... ;-)

lelegard, 2020-06-02:
@cbernander 

> It would have been really nice to have a PCR verification method based on a monotonic clock (since it seems VBR is becoming increasingly common in IP multicast systems), 

This would be a dedicated real-time application with dedicated IP input. This is possible to develop such an application on top of the TSDuck library.

This is not possible with `tsp` which is a generic framework, working on offline files and real time streams equally, heavily multi-threaded and buffered. Inside `tsp`, there is no direct relation between the exact time a TS packet is received and the time it is processed in a plugin somewhere in the chain.

lars18th, 2020-06-05:
Hi @lelegard ,

First, thank you for fixing this error. I see this error too some time ago, but I don't reported it, as the data reported by `pcrvefiry` is quite difficult to understand. So I use other tools that reports the simple nanoseconds drifts. Perhaps, you will want to include some small change in the UI to add another simple view of the data. You agree?

Anyway, I have _another_ suggestion related to this:

> This is not possible with tsp which is a generic framework, working on offline files and real time streams equally, heavily multi-threaded and buffered. Inside tsp, there is no direct relation [...]

I remember that perhaps a year ago I've suggested to support M2TS timestamping. You refused to do it (and I accept it). But, perhaps this approach can help to solve this problem: If TSDuck supports _internally_ the M2TS format and not only the MPEG2-TS, then it will be easy to timestamp arriving packets in realtime. Then the processing will know the exact time on a TS packet is received.

Please consider it. And remember that I prefer to not reinvent the wheel every time, so the M2TS is a good solution to link TS packets with realtime timestamps.

Regards.

lelegard, 2020-06-05:
> Please consider it. And remember that I prefer to not reinvent the wheel every time, so the M2TS is a good solution to link TS packets with realtime timestamps.

It could be possible to let the input plugin set some form of timestamps in the packet metadata. However, there would be no guarantee on the presence or accuracy of these time stamps. Typically, an input plugin which privileges performance over accuracy using very large buffers would generate one single timestamp for all packets which are read in one global "read" or "receive" operation.

My main problem with TSDuck is one point I initially viewed as an advantage which finally turns into a drawback : the extremely generic and flexible structure. Generally speaking, you must choose between specialized tools which do one thing extremely well and generic tools which do many many things but nothing as accurate as a specialized tools. The problem with the second class (to which TSDuck clearly belongs) is that users are first pleased with the number of features and then, when using one of them intensively, they complain that it is not very accurate. Accurate jitter measurement must be done using a specialized application, the more specialized, the more accurate.

I will think about it.

cbernander, 2020-06-05:
@lelegard 
On Linux you have socket options (SO_TIMESTAMP[NS]) to timestamp packets on reception in the kernel, so the accuracy of those timestamps should be good enough to be practically useful.

lars18th, 2020-06-05:
Hi @lelegard ,

Thank you! Your explanation is very clear, and sincere. In any case, I personally like the current model of the TSDuck.

> I will think about it.

I only want to contribute with a simple comment: If you can incorporate internal timestamps with _low_ effort, then please try to do it. After that we'll see how to use them.

Regards.

lelegard, 2020-06-11:
@cbernander, @lars18th,

I have started some underground work in that direction. I have added a time stamp to the packet metadata. This time stamp is optional and filled by the input plugin when available.

To exercise this, I have added support for M2TS file in the `file` plugin (input, output and packet processor). M2TS files are automatically detected on input and the time stamp of each packet is added to its metadata. These metadata are propagated and accessible (read/write) to all plugins. On output, `-O file --format m2ts` generates an M2TS file using the time stamp in the metadata of the packet. Test 65 of the test suite shows that an M2TS file crosses the complete chain and is recreated identically on output, meaning that the time stamps were correctly read and propagated.

Taking the opportunity to redesign the `TSFile` class, I added a proprietary TSDuck format which embeds all packet metadata in a 14-byte header per packet (4 bytes in the case of M2TS). Use `-O file --format duck` to create such a file. This format is also auto-detected in input `file` plugin. This format can, of course, only be used in pipes `tsp | tsp` or `tsswitch | tsp` but it allows the transmission of the original time stamps, packet labels and a few other properties.

Other things to do include:

- Getting UDP receive timestamps (Linux only, not available on macOS and Windows).
- Provide time stamp from system clock in the tsp input thread if the input plugin does not provide one. This will be less precise but at least a later processing in another plugin will have some sort of actual input time stamp.
- Use the input time stamp in the `pcrverify` plugin.
- Other ?

lars18th, 2020-06-11:
Hi @lelegard ,

This is a huge improvement, thank you!
When I have more time I'll take a deep look. The pipe support of labels seems very interesting. :wink: 

In the meantime, some questions:
- Is this new format compatible with the plugins `timeshift` and `encap`/`decap`? For the first the idea is to have an accurate behaviour even when the input stream fails. With sufficient support it could be used for a realtime "+1" service generation capturing from a multicast stream. And regarding the second, the idea is to encapsulate not only 188 TS packets, but 14+188 "duck" packets.
- These new timestamps will be used for the DVB Modulators outputs? My concern is about the problem of the _precise_ bitrates. The problem is when the modulator includes one empty NULL packet time to time to compensate and the input bitrate has _less_ bitrate than the target modulation. You will try to target this problem too?

Thank you @lelegard for listening to us! Your toolkit is amazing for us.
And please remember to reopen this issue to continue discussiong about this topic (at time it lists as "closed").
Regards.

lelegard, 2020-06-11:
> Is this new format compatible with the plugins timeshift 

Yes

> and encap/decap?

No.

> These new timestamps will be used for the DVB Modulators outputs? 

No, and I have no idea how it could be done. The modulator device takes the packets as input and I see no way to provide individual time stamps, packet by packet. This is of course a matter of proprietary API of the modulator.

> And please remember to reopen this issue 

Done

cbernander, 2020-06-11:
This sounds great!! I have nothing to add other than to reiterate my previous request for (eventually) including a (timestamp-based) PCR interval check in the pcrverify plugin.

Once again, thank you for all your work on this great toolkit!

lars18th, 2020-06-11:
Hi @lelegard ,

> > and encap/decap?
> 
> No.

So in this case perhaps a patch for a v2 should be developed _in the future_ (not now).
Please, you want to add a lable for a future enhancement of this?

> > These new timestamps will be used for the DVB Modulators outputs?
> 
> No, and I have no idea how it could be done. The modulator device takes the packets as input and I see no way to provide individual time stamps, packet by packet. This is of course a matter of proprietary API of the modulator.

Ummmm... That's my idea: Tuner modulators read the stream in _realtime_, but with buffering. So if the buffer size is known, then the _time_ inserting the packets could have some influence. What you think? It has sense to use these timestamp marks in the plugins that feeds the modulators? The problem that I see is that DVB-T standard modulations use factional bitrates that is not possible to handle at time, so this generates a drift over time. Any idea to overcome this problem?

bobben1no, 2020-06-11:
Is this PCR jitter based on bitrate calculated from CBR bitrate between the first two PCRs of a program ?  

I am testing writing plugin based on pcrverify. It calculates CBR ts average bitrate between all PCRs for a program and calculate program PCR accuracy based on this average CBR bitrate. It is not ready yet but the basics is working.  

The accuracy gets better the more PCRs received as the avarage CBR TS bitrate get's more correctly estimated.  This kind of PCR accuracy is typically way below +/-500ns for CBR transport streams.    I know VBR ts (no null packets ) results in PCR accuracy in microsecond (or worse) but I assume the example given which shows micro second jitter is for CBR ts ?

Example PCR accuracy, TS cbr bitrates based on per program pcr and all PCRs:
```
PID;max negative pcr accuracy [ns];max positive pcr accuracy [ns];average pcr accuracy [ns];avrage bitrate per PCR[bits/s];avrage bitrate all PCRs[bits/s];pcr min rep[ms];pcr max rep[ms];pcr avr rep[ms]
1048;-259;185;0;39512085;39512066;39512225;1.256000;36.198002;32.521000
1077;-259;111;0;39512247;39512066;39512225;0.532000;36.046001;34.305000
1156;-222;148;0;39512087;39512066;39512225;0.647000;31.631001;28.098000
1186;-222;148;0;39512208;39512066;39512225;1.103000;30.870001;29.488001
1347;-259;148;0;39512066;39512066;39512225;0.380000;36.846001;32.629002
1549;-259;148;0;39512097;39512066;39512225;0.380000;36.351002;32.466000
1711;-222;148;0;39512067;39512066;39512225;1.941000;36.730999;32.457001
```

lelegard, 2020-06-12:
> Is this PCR jitter based on bitrate calculated from CBR bitrate between the first two PCRs of a program ?

No. By default, the bitrate is periodically re-evaluated based on PCR's. So, the considered bitrate is "constant by periods". See the tsp documentation in the user's guide.

lelegard, 2020-06-12:
I have completed a second round of modifications.

- UDP kernel timestamps are used on Linux (socket option `SO_TIMESTAMPNS`) as input timestamps.
- RTP and SRT timestamps are used as input timestamps.
- When the input plugin does not provide its own timestamps, `tsp` uses its system time at the time the input plugin returns a bulk of packets.
- The plugin `pcrverify` can now use the original input timestamps as reference. See option `--input-synchronous`. This is not the default since it wouldn't make sense with an offline source.

I haven't tested `pcrverify` in that configuration because I know that some of you are waiting for it. So, I let you test it 😉 

I still need to propagate packet metadata (similar to `-O file --format duck`) to some relevant plugins such as `fork`.

Please provide your tests feedback here.

lars18th, 2020-06-12:
> I still need to propagate packet metadata (similar to `-O file --format duck`) to some relevant plugins such as `fork`.

Please, don't forget the plugin `regulate` !

lelegard, 2020-06-12:
> Please, don't forget the plugin `regulate` !

Before using the plugin `regulate`, we must answer two questions:
- Why do we want to regulate?
- What is the time reference for regulation?

First, why do we want to regulate? Because the source is unregulated... Unregulated sources are usually files or other offline sources. DVB, UDP, RTP, SRT, all of them are self-regulated sources and using `regulate` can only makes things worse.

Second, what is the time reference for regulation? On input timestamps? No...

Two cases:
- By default, for input plugins which cannot provide their own input timestamps, `tsp` uses the system clock at the time of arrival of the packet. If you want to regulate, this is because your source is not regulated. And in that case, that means that input arrival times are not correctly regulated and cannot be used as regulation reference.
- Some input plugins can provide their own input timestamps: UDP, RTP, SRT. Good... But those sources are already regulated. So you do not need to regulate.

OK, I see one single use case for regulation based on input timestamps: playing an M2TS file. All the rest is inapplicable.

As a conclusion, input timestamps are not silver bullets. They may _look_ attractive for regulation but in fact this is a trap (except for M2TS files).

lelegard, 2020-06-13:
The propagation of the time stamps and metadata have been generalised to plugins `fork` and `merge`. So, now, in a graph of multiple instances of `tsp`, the original time stamps and labels are propagated everywhere.

The principle can be easily generalised because all stuff about file format / packet read / write / time stamps / metadata are now encapsulated in a class `TSPacketStream` which relies on generic stream read and stream write interfaces. Thus, by association, you can magically promote any class implementing those interfaces as "TS format aware".

@cbernander, could you re-test `pcrverify` with option `--input-synchronous` on your UDP stream? If this is on Linux, all UDP packets will be tagged with a kernel timestamp.

cbernander, 2020-06-13:
I've only been able to test on RTP streams so far (where everything looks good as far as I can tell) - I have to wait until Monday to test on a UDP stream.

If possible, it would be nice to have a flag (in the ip input plugin?) to prefer network timestamps over RTP timestamps to get a unified experience for UDP and RTP streams (i.e. taking network latency+jitter into account for both types) - especially when used in conjunction with TS tools/decoders that simply discards/ignores the contents of the RTP header.

lars18th, 2020-06-14:
> If possible, it would be nice to have a flag (in the ip input plugin?) to prefer network timestamps over RTP timestamps to get a unified experience for UDP and RTP streams (i.e. taking network [...]

You took the words right out of my mouth!
An option to select between RTP timestamps and network timestamps it's necessary when reading a  RTP stream. Futhermore a plugin to _compare_ between the two will be desirable. :wink:

lars18th, 2020-06-14:
> > Please, don't forget the plugin `regulate` !
> 
> Before using the plugin `regulate`, we must answer two questions:
> 
> * Why do we want to regulate?
> * What is the time reference for regulation?
> 
> ...
> 
> As a conclusion, input timestamps are not silver bullets. They may _look_ attractive for regulation but in fact this is a trap (except for M2TS files).

OK. You're right.
However, the base of my request is based on this use case:

- One UDP source with MPEG-TS. This source has a high jitter. Then the source can't be considered fully regulated. If we use the incoming timestamps in the output, then the jitter will not be corrected. Futhermore, the processing based on chunks/buffers inside TSDuck (necessary in any case) can increase the jitter troubles if we want to output using the original timestamps.

So in this case, what's the best solution to overcome the problem? That's to "remove" the jitter problem. Any idea?

lelegard, 2020-06-14:
> So in this case, what's the best solution to overcome the problem? That's to "remove" the jitter problem. Any idea?

You mean an RTP stream with correct RTP time stamps but erratic reception time stamps? In that case, it would make sense to regulate on the RTP time stamps. But this works only if you have an external clock reference such as RTP time stamps. It is useless on bare UDP packets.

> If possible, it would be nice to have a flag (in the ip input plugin?) to prefer network timestamps over RTP timestamps

Will do. It should also apply to SRT source time.

lelegard, 2020-06-15:
>> If possible, it would be nice to have a flag (in the ip input plugin?) to prefer network timestamps over RTP timestamps
>
> Will do. It should also apply to SRT source time.

Added `--timestamp-priority` option to `ip` and `srt` input plugins.

lars18th, 2020-06-16:
Hi @lelegard ,

> You mean an RTP stream with correct RTP time stamps but erratic reception time stamps? In that case, it would make sense to regulate on the RTP time stamps. But this works only if you have an [...]

I feel that's not my question.
To better understand, I list here the different cases when receiving one RTP stream:

1. The RTP timestamps are valid, and the reception has low jitter. Then nothing to do; all is OK!
2. The RTP timestamps are valid, and the reception has a huge jitter. Then trusting the RTP timestamps with the new support the troubles caused by the jitter are corrected if the _output_ plugin uses the timestamps.
3. The RTP timestamps have errors (empty, fake, etc.), and the reception has low jitter. Then it's qual to receive a plain UDP stream. With the new parameter `--timestamp-priority` we can discard the RTP timestamps and all works.
4. The RTP timestamps have errors, and the reception has huge jitter. In that case, what we need to do to restore the correct timestamps?

As you can see my question relates to point 4. Futhermore, I want also to confirm that _all_ realtime outputs of TSDuck will use the timestamps in the point 2. For example, I'm not sure about the modulators behaviour.

Can you please provide more information? I'm not suggesting to implement anything else. I'm just trying to understand the current behavior. So your help is welcome.
Thank you for this amazing project!
Regards.

lelegard, 2020-06-16:
> The RTP timestamps have errors, and the reception has huge jitter. In that case, what we need to do to restore the correct timestamps?

Then, the only remaining time reference is the PCR. In the case of a CBR stream, the bitrate is easily computed from at least 2 PCR values and then you can regulate all packets (already implemented).

Now, in the case of VBR, you have zero reliable reference between two PCR values. In technical terms, you're screwed...

lars18th, 2020-06-16:
Hi @lelegard ,

> Then, the only remaining time reference is the PCR. In the case of a CBR stream, the bitrate is easily computed from at least 2 PCR values and then you can regulate all packets (already implemented).

CBR is an easy case to solve...

> Now, in the case of VBR, you have zero reliable reference between two PCR values. In technical terms, you're screwed...

Why? It's not the objective of the timestamps to solve the problem?
Let me to explain what I think about this problem (the problem of a VBR streaming or CBR with partial content... that's without NULLs or with some services removed):

- The PCR marks can be assumed correct. If not, then the MPEG-TS is invalid, as it's mandatory to have valid clock marks at small intervals. In this case, only a tool oriented to AV content can handle the data. So this case is out-of-scope of the TSDuck.
- With consecutive PCR valid marks we have the 27MHz clock source reference of the sender. And this it's independent of the network _clock_, so for this reason we can receive packets in bursts with unwanted jitter errors.
- Then between two PCR marks, the time elapsed between these packets it's known and reliable.
- So the information that we lost is the intra-lapses of the TS packets inside the interval.

My question is how to solve this problem with the new timestamp support. My first idea is this: support the recreation of the timestamp marks using a simple technique: Inside each PCR interval we can count the PUSI marks of each pid. Assuming that a PUSI mark is the start of a PES packet, then we can derivate that every PUSI mark could be equidistand in time. For example, with a 25FPS video stream, one PUSI appears every 40ms. So we have sufficient information to mark the distribution of the packets inside  every PCR marks.

What you think about this idea?

In fact what I'm requesting is a method/plugin/whatever to convert a VBR or incomplete CBR stream in a full CBR stream with correct timestamps. With this _complete_ CBR stream we can do whatever processing that we need, and if it's necessary remove the padding before the output.

I hope you can explain if this could be achieved.
Regards.

lelegard, 2020-06-16:
> Assuming that a PUSI mark is the start of a PES packet, then we can derivate that every PUSI mark could be equidistand in time. 

No. The frame rate is constant, not the bitrate. The PTS are equidistant, not the PCR. That is the difference between CBR and VBR.

> For example, with a 25FPS video stream, one PUSI appears every 40ms.

Not necessarily. There is no guarantee that there one PES packet per video frame. Nowadays, most video streams I have recently seen have one PES packet per frame. But, I clearly remember that MPEG-2 video broadcast streams used to have one PES packet _per GOP_ (12 frames), so having only 2 PES packets, 2 PUSI and 2 PCR per second. What you describe is a current practice _today_, not a strict requirement. We cannot rely on it.

But the most annoying problem is that `tsp` is a streaming processor. It always goes forward, never backward. So, once you get the second PCR of an interval, it is too late to adjust past timestamps between that PCR and the previous one.

lars18th, 2020-06-16:
> > Assuming that a PUSI mark is the start of a PES packet, then we can derivate that every PUSI mark could be equidistand in time.
> 
> No. The frame rate is constant, not the bitrate. The PTS are equidistant, not the PCR. That is the difference between CBR and VBR.

Well, the frame rate _today_ doesn't need to be constant either. :wink: 
In any case: Even without using the PTS marks, every PUSI mark implies a PTS inside the packet (if the pid is clock referenced). And applying some restriction (based on parameters, or derivated from other data), we can stablish a simple rule: PUSI marks equidistant in time will generate valid streams. I'm not suggesting to compute entirely the restriction, but to provide a tool the enforce this equidistance.

> Not necessarily. There is no guarantee that there one PES packet per video frame. Nowadays, most video streams I have recently seen have one PES packet per frame. But, I clearly remember that MPEG-2 [...]

OK. But this assumption can be valid anyway: Inside every interval defined by two consecutive PCR marks, we can group the TS packets in chunks based on the PUSI marks, and these chunks should have fixed time equidistance. For sure the duration of the entire interval can vary, but the PCR timestamps are sufficient information to know the durantion of the interval.

> But the most annoying problem is that `tsp` is a streaming processor. It always goes forward, never backward. So, once you get the second PCR of an interval, it is too late to adjust past timestamps between that PCR and the previous one.

Here the new timestamping enhancemet can help. Using the implementation of the `timeshift` we have the capability to buffer/delay some processing. Then we can first calculate the PCR difference between an incoming PCR mark and the next. This information can be then shared _forward_ with a following plugin that reads _after_ the `timeshift` (the buffer). Take note that new timestamp marks will guarantee that the values correspond to the correct packets. And the whole process can be done.

What you think about this "tricky" forward processing? It has sense to you, or it's too complex?

cbernander, 2020-06-18:
@lelegard 
I've tested the `pcrverify` plugin with the `--input-synchronous` option on both RTP and UDP inputs for a couple of days now and it seems to work great!

Regarding the `--timestamp-priority` option, I wonder if it would be possible to output _which_ type of timestamp measurement (rtp/kernel/tsp) actually is used so the displayed values can be accurately interpreted?

lelegard, 2020-06-18:
> I've tested the pcrverify plugin with the --input-synchronous option on both RTP and UDP inputs for a couple of days now and it seems to work great!

Good to hear. Thank you for the feedback.

> Regarding the --timestamp-priority option, I wonder if it would be possible to output which type of timestamp measurement (rtp/kernel/tsp) actually is used so the displayed values can be accurately interpreted?

This is not immediately possible. At the plugin level, we can only check if there is an input timestamp and what is its value. But the source of the timestamp was lost long ago. I need to think about a way to propagate that information.

lelegard, 2020-06-19:
>  I need to think about a way to propagate that information.

Done, please test it.

cbernander, 2020-06-19:
Looks really good, thanks a lot!