tsduck/tsduck #588: Warning: Updated build location in source tree

lelegard, 2020-05-31:
Dear TSDuck users,

This post is for developers who routinely rebuild TSDuck and execute it from the build location.

Now that version 3.21 is out, we start version 3.22 with a new way of building TSDuck on Linux and macOS. This is a response to several requests from developers in the past few months.

Previously, TSDuck used a classical (you may say _old_) hierarchical build. All binaries were produced in a subdirectory of the source directory, at various levels. The result was that libraries, plugins and utilities were built in distinct places. To run them from their respective places, it was necessary to define multiple environment variables (`LD_LIBRARY_PATH`, `TSPLUGINS_PATH`). This was not very convenient. A helper script named `setenv.sh` was built in each binary directory to be "sourced" in _bash_. But developers using other shells than _bash_ complained that the `setenv.sh` did not work with their shell.

Now, everything is built in the same binary directory under the directory `bin/<target>-<arch>-<hostname>` where `<target>` is either `release` or `debug`. Note that the use of the host name in the directory makes it possible to use the same shared source tree to build from different systems or virtual machines. All builds will coexist using distinct names under `bin`.

Because everything is built in the same place, all utilities can be run directly without defining any environment variable. For convenience, the new script `build/setenv.sh` rebuilds the binary directory name and adds it in the`PATH`. Of course, since this is a "sourced" script, it works with _bash_ only. If users of other shells would like to provide equivalent scripts for other shells, feel free to submit a pull request.

For consistency, the Windows builds are now also located under `bin` instead of `build\msvc`.

This is explained in the [build instructions](https://tsduck.io/doxy/building.html#runbuild) (the page will be automatically updated at the next nightly build, please be patient).

Although the builds and installers have been tested on Linux, macOS and Windows, it is possible that problems remain. Please report them here.

bobben1no, 2020-06-11:
I have not changed to version 3.22 yet and I had completely forgot the method for run from build in linux for version 3.21 and older. However I found it here https://srtlab.github.io/srt-cookbook/apps/tsduck/

As the "old" run from build method is not documented on https://tsduck.io/doxy/building.html#runbuild  I hope it is ok to post it here in case I'm not the only one that needs this info :)
```
Depending on your target (release vs. debug, 32 bits vs. 64 bits, Intel vs. ARM), execute one of:

source src/tstools/release-x86_64/setenv.sh
source src/tstools/debug-x86_64/setenv.sh
source src/tstools/release-i386/setenv.sh
source src/tstools/debug-i386/setenv.sh
source src/tstools/release-arm/setenv.sh
source src/tstools/debug-arm/setenv.sh
```

lelegard, 2020-06-12:
Or simply `make doxygen` on your side. You will get the documentation (including build instructions) for _your_ exact version.

bobben1no, 2020-06-12:
Great ! Did not know about that possibility.

lelegard, 2020-06-12:
This is the way the programming doc is generated in the tsduck.io web site.

Originally, I manually ran `make doxygen` on my Mac from time to time and pushed the result to the site.

Now, everything is automatically managed by the GitHub CI/CD system. Every night, a job is triggered on GitHub runners. The current state of the repo is used to rebuild the binaries for Windows and Ubuntu. In the same job, `make doxygen` is run. At the end, everything, docs & binaries are updated in tsduck.io. This way, the doc on tsduck.io is always up-to-date with the repo.

closed 2025-02-24 by lelegard