tsduck/tsduck #589: TsDuck library sample-app code giving exception in 64-bit

satyamsameer, 2020-06-03:
Dear Team

I have installed the "TSDuck-Win64-3.21-1819.exe" with development enabled for 64- bit.
The sample-app code is successfully compiled in this configuration but I am unable to run this app in debug mode a well in release mode. It gives me the following exception in file tsBlockCipher.h. Here is the screenshot.

[image]

I am using Visual Studio 2015 Pro version for development.
Anyone of you knows what I am doing wrong here?

lelegard, 2020-06-03:
Hi @satyamsameer 

I cannot reproduce it. I installed TSDuck-Win64-3.21-1819.exe on Windows 10 and recompiled sample-app.exe. It works. I use Visual Studio 2019 16.6.0.

satyamsameer, 2020-06-03:
Thanks for the reply @lelegard 
I noticed that my c++ compiler is unable to recognize "constexpr" variables that is where it is giving me exception. Do you have any idea how can I resolve this?

lelegard, 2020-06-03:
The compiler recognises `constexpr` because it compiles. Maybe the generated code is incorrect. Maybe. This must be confirmed. If this is confirmed, then you should upgrade your compiler because this kind of problem is so dangerous that any (correct) application may fail some day, including in production. To be precise, I have personnally run into MSC compiler code generation bugs twice. Not this one. But twice. Just to let you know.

satyamsameer, 2020-06-04:
You are right @lelegard I found that in older versions of MSC compilers the constexpr is undefined I might have to upgrade my visual studio to a newer version to make it work. I applied some work around in my code now it is compiling and running in release mode perfectly but now it gives  exception "Stack cookie instrumentation code detected a stack-based buffer overrun." in debug mode. At last if I am unable to rectify this problem then I will switch to newer version of visual studio.

closed 2020-06-17 by lelegard