tsduck/tsduck #590: Possibility to remove  Audio PID inside PMT using TSDuck

Jazzykool85, 2020-06-03:
Hi Team,

Let me tell you, i have started using this tool from last 5 months and its excellent.
I have a question.
Is there any way available with TSduck to remove  Audio PID inside PMT for a TS File.
i explored the document and found svremove , But it just saying about the service removal .
Could you help me with this?

lelegard, 2020-06-03:
The plugin `pmt` tweaks a PMT. It has a lot of options, including `--remove-pid` which removes any PID reference from the PMT. Then, the PID contents can be removed using the plugin `filter`. Be sure to read all available options for these plugins in the user's guide.

Jazzykool85, 2020-06-06:
Thanks a lot Legard.
I will check and try on my side and get back to you.

closed 2020-06-06 by Jazzykool85