tsduck/tsduck #593: Need help demuxing a data stream from a TS file.

thomasfiveash, 2020-06-06:
Hi @lelegard!

I have a TS file with a single program and two streams. I'm working on an application using the C++ library to try to demux the second stream's data so that I can use it in the application. Currently, I have a really hacky method of parsing the TS stream to get me my data, but I would much prefer to use a library as my hacky method is really unstable and slow. I would like to read in the TS file and iterate over each frame grabbing the demuxed TS payload from a specific pes, do something with it, and then repeat this until the end of the file.
I was easily able to dump the data I wish to obtain with two separate TSDuck commands:

`tsdump --payload -p PES_ID --no-headers -l input.ts > output.txt`
and
`tsp -I file input.ts -P pes -p PES_ID --payload -o output.txt -O drop`

I've spent way too many hours trying to figure out how to try to use tsp to do this in code, and I don't think I'll get anywhere unless I ask for help. I started looking into using the various ts:: classes to read the file in, find a packet, grab the stream, etc. Unfortunately, I haven't gotten very far at all.

I was able to easily to this in FFmpeg, avformat_open_input, avformat_find_stream_info, av_read_frame, etc. But I would much rather prefer to use TSDuck to do this due to the licensing issues and FFmpeg. I'd also like to become more familiar with TSDuck, as the knowledge will be rather useful to me in the future.

Any guidance on how to do this is **very** much appreciated. I doubt it's a hard thing to do, I'm just new to TSDuck and am completely lost in the sea of classes.

Thanks!

lelegard, 2020-06-06:
I am not 100% sure to understand what you want to do. If you want to get the PES packets (audio or video) of an elementary stream from an application, the right class to use is [PESDemux](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tsPESDemux.h). Your application shall implement [PESHandlerInterface](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tsPESHandlerInterface.h). Each time a new complete PES packet is available, your application class will be invoked with that packet. As an example, have a look at the [pes plugin](https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_pes.cpp).

t0mf, 2020-06-06:
I am just wanting to retrieve the binary payload of a pes (with a specified pid) for every frame. It's not audio/video, but I don't think that matters foe PESDemux. Looking at the pes plugin, I think I should be able to use it as an example to get me going. It looks pretty straight forward. I'll report back if I got it or not.

Also, I don't know why I didn't think of looking at the plugin source... I guess I just got lost in the API instead of realizing I have a ton of examples right in front of me.

lelegard, 2020-06-06:
> I am just wanting to retrieve the binary payload of a pes (with a specified pid) for every frame. It's not audio/video, but I don't think that matters foe PESDemux.

Indeed, it does not matter for `PESDemux`. It returns complete PES packets, whatever the content is. If the content format is known, then it can return additional info about audio and video attributes. But returning the complete PES packet always works.

I was confused by the word "frame". With video, in the general case, there are several frames per PES packet. Sometimes, there is only one. Sometimes, each PES packet contains a complete GOP.

t0mf, 2020-06-06:
Oh my bad, I didn't even think of that. I haven't done anything with video/audio frames, only stuff with encoded metadata. Which means I don't think of GOP, I just think of each individual frame. 

While you're here, what plugin/class would be good to look at to figure out how to dump a stream from IP Address/Port straight to file, without minimal processing overhead?

t0mf, 2020-06-07:
I think I accomplished my initial question! Now to optimize it..

lelegard, 2020-06-07:
> what plugin/class would be good to look at to figure out how to dump a stream from IP Address/Port straight to file, without minimal processing overhead?

See the [sample application](https://github.com/tsduck/tsduck/blob/master/sample/sample-app/myexec.cpp).

Your can also use the class `UDPReceiver` to read UDP input and class `TSFile` to write output TS file.

> I think I accomplished my initial question! Now to optimize it.. (figure out best way to read all the TSPackets in and pass the the PESDemuxer).

It depends on your input source. Either loop on receive/read from specialised classes such as `UDPReceiver` or `TSFile` or use a plugin-based application as illustrated [here](https://github.com/tsduck/tsduck/blob/master/sample/sample-app-custom/myexec.cpp).

But if you just need a specific treatment or display or extraction on PES payload, you may write your own plugin and use `tsp` to run it on any kind of input and output. See the [sample plugin](https://github.com/tsduck/tsduck/tree/master/sample/sample-plugin) with all building instructions and, of course, read the `pes` plugin source code for PES packet processing.

TSDuck was really designed for that purpose: just write a small plugin to get your specific processing in any kind of environment.

t0mf, 2020-06-07:
> It depends on your input source. Either loop on receive/read from specialised classes such as `UDPReceiver` or `TSFile` or use a plugin-based application as illustrated [here](https://github.com/tsduck/tsduck/blob/master/sample/sample-app-custom/myexec.cpp).

Gotcha, I just didn't know if there was some other bulk read. I'm currently passing a buffer of like 100 TSPackets to the read and then feeding them to the Demuxer. It seems to be fast. The number of 100 is rather arbitrary though, don't know the optimal number to read at once.

> But if you just need a specific treatment or display or extraction on PES payload, you may write your own plugin and use `tsp` to run it on any kind of input and output. See the [sample [...]

I think my next thing I tackle will be using tsp to grab from UDP and dump to a file. That looks easy enough. Maybe I could incorporate what I currently have into a plug-in. Although, that's relying on using a callback function to my main class (outside TSDuck) whenever the PESHandlerInterface handlePESPacket callback happens to pass it the packet's payload so I can grab the data, parse it, and log.

Now I just need to figure out how to build everything for static linking so I don't have that DLL hanging around. I don't want to deal with installers to keep track of dlls for this little tool, just want the executable. That should be easy enough though.

Thanks for all the help! Once I got a few things figured out, it's not as bad as I thought. I just got stuck on little nuisances like.. no default constructor for PESDemux, figuring out how to actually utilize that bitset for pid filtering, figuring out how to convert a std::string to a ts::UString (there's a bug in my version of VS with the codecvt_utf8_utf16<char16_t> and the char16 symbol not being defined which took way too long to figure out how to get around that), and a few other small but seemingly easy things now that I look back.

I am definitely impressed with TSDuck and will continue to develop things with it for all my TS needs!

lelegard, 2020-06-07:
> no default constructor for PESDemux

This is done on purpose. A `PESDemux` object, just like many other TSDuck objects manipulating MPEG structures, needs a `DuckContext` object.

All plugin objects have such an object named `duck` (see the source code of the `pes` plugin).

If you define a standalone application, just define an instance of `DuckContext` and use it.

Otherwise, a `DuckContext` object can be very powerful.
- It can define standard command line parameters to define local preferences for instance and make sure that those preferences are used everywhere.
- All demux objects collect information from what they demux and propagate that information through the `DuckContext` object. For instance, when a section demux encounters an ATSC table, then it adds to the context the information that the TS seems to be an ATSC one. Then, later, when a table is deserialised using the same context, if a descriptor tag is used in different standards with different structures, then the deserialisation will correctly interpret the descriptor as the ATSC version of the tag. Etc.

Again, look at the source code for examples.

> figuring out how to actually utilize that bitset for pid filtering

It's a C++ standard [`std::bitset`](https://en.cppreference.com/w/cpp/utility/bitset). The PID values in the set indicate which PIDs should be demuxed (you probably won't need to demux all PIDs in the TS).

> figuring out how to convert a std::string to a ts::UString

See the definition of `UString`. `UString` is UTF-16. All `std::string` and `char*` are supposed to contain UTF-8 characters. There are constructors and conversion methods between `UString` and UTF-8 in both directions. Also note that UString literals (well, string literals which can be directly converted into `UString`) can be used with a leading `u` as in `u"abcd"`. This is standard C++. Again, see the source code for examples.

> Is there a way to use link in tsduck.lib instead of using the .dll? 

The file `tsduck.lib` is the export library for `tsduck.dll`. So, linking against  `tsduck.lib` means that you use the DLL.

There is a static library named `tsducklib.lib` which is built with the rest of TSDuck. But it is not installed in the development part of the binary installer. Using it can leads to unexpected problems. The source code of TSDuck is very modular (this is the secret for its longevity). Hundredths of tables, descriptors, filters and plugins are defined in their own source code module. But they are referenced nowhere. They automatically register themselves on initialisation. In other words, when you add a new descriptor, for instance, just add its source code and that's all. 

When using the DLL, it works correctly because everything is in the DLL and automatically registered on startup.

When using the static library, you have two major problems:
- The linker pulls only what is referenced from your application. The vast majority of tables, descriptors, etc. are directly referenced nowhere and are consequently not inserted in the final application. And when you run the application, you will note that most signalisation or plugins are missing.
- You cannot use plugins which are defined in their own independent DLL (most plugins are). Since the plugins are linked again the `tsduck.dll`, that means that the TSDuck code would be duplicated between your executable (because of the static library) and `tsduck.dll` which is activated by the plugins.

So, yes, the static library exists but, no, don't use it (or you will get into troubles).

t0mf, 2020-06-07:
Woah, Thanks for the awesome response!

> This is done on purpose. A `PESDemux` object, just like many other TSDuck objects manipulating MPEG structures, needs a `DuckContext` object.

I figured out that I needed it, but I had no idea what it did. Great information to know!

> It's a C++ standard [`std::bitset`](https://en.cppreference.com/w/cpp/utility/bitset). The PID values in the set indicate which PIDs should be demuxed (you probably won't need to demux all PIDs in the TS).

Yeah I had never used them before, so I had to figure that one out. I have certain PIDs that I wanted to filter on. This worked exactly how I expected once I figured out how to actually set the `std::bitset`. I'll definitely use those in the future for other projects though!

> See the definition of `UString`. `UString` is UTF-16. All `std::string` and `char*` are supposed to contain UTF-8 characters. There are constructors and conversion methods between `UString` and [...]

Everything I work on uses `std::string`, thus I had to convert some input info from `std::string` to to the `ts::Ustring` so I could use in TSDuck, I just ran into problems that are fixed in VS 2019 16.2, I'm on 16.0.2 and can't update. 

Specifically, Microsoft forgot to define the symbol: `std::locale::id std::codecvt<char16_t>, char, _Mbstatet>::id`. Because of this, my linker throws an unresolved external error when converting from a `std::string` to a `ts::Ustring` using the following: `std::wstring_convert<std::codecvt_utf8_utf16<char16_t> char16_t>{}.from_bytes([std::string to convert]`)

> The file `tsduck.lib` is the export library for `tsduck.dll`. So, linking against `tsduck.lib` means that you use the DLL.
>
> There is a static library named `tsducklib.lib` which is built with the [...]

Understood! I'll just make sure I always have access to the DLL.

Again, thank you for your work and all the help you have provided. It is very appreciated.

t0mf, 2020-06-08:
@lelegard 

Random quick question, if I have a application that's using the TSP plugins ip, until, and file (similar to your new simpler sample program), does `tsduck.dll` contain everything I need to run or will it be calling into another `tsplugin_.dll`? Only `tsplugin_until.dll` is in the plugins folder, so I would assume the others are built into `tsduck.dll`? Do I need to include `tsplugin_until.dll` with my application?

Thanks!

lelegard, 2020-06-09:
The answer to your question in the page ["Using the TSDuck Library"](https://tsduck.io/doxy/usinglibrary.html).

To use the TSDuck library on Windows, everything you need is preset in a property file named `tsduck.props`. So, the _only_ thing you need to do is to add the following line to your Visual Studio project file:
~~~
<Import Project="$(TSDUCK)\tsduck.props" />
~~~
If you don't want to manually modify it, there is a way to add a "property sheet" to your project in the zillions of menus of Visual Studio. As far as I am concerned, I always manually craft project files.

t0mf, 2020-06-09:
Oh I was talking when I eventually build an installer and build my program files directory. There's no way I can say "just install TSDuck". Would I be able to put just the `tsduck.dll` and the relevant plug-in dlls in my app folder directly? Or is there more modular/external dependancy calling outside of that?

lelegard, 2020-06-09:
> I know the .props automatically copies the tsduck.dll to the build output directory, but not all the plugins I want to use. So I'd have to collect all those?

To run your exe, probably not because the plugins are in `%TSDUCK%\bin` which is itself in `%Path%` (if you used the TSDuck installer). So, the plugins will be found.

If you package your application, you need to redistribute in your package `tsduck.dll` and the `tsplugin_*.dll` you use. You may do that, as long as the terms of the BSD 2-clause license are respected (which is a liberal license).

t0mf, 2020-06-09:
> To run your exe, probably not because the plugins are in `%TSDUCK%\bin` which is itself in `%Path%` (if you used the TSDuck installer). So, the plugins will be found.
>
> If you package your [...]

Gotcha. Noted about the BSD-2.

So I built an application which simulates your sample program essentially. I can run the tsp command `tsp -I ip <valid port> -O file "test.ts"` and it works perfectly. If I set the exact same situation up following the code example the tsproc.start() command always fails and I have no idea why.. any thoughts?

lelegard, 2020-06-09:
> The Asynchronous Report says it got errors, but I can't find anywhere to actually retrieve them.

Start it in debug mode:
~~~
    // Use an asynchronous logger to report errors, logs, debug, etc.
    // Make it display all messages up to debug level (default is info level).
    ts::AsyncReport report(ts::Severity::Debug);
~~~
Also use `ts::Args` or a subclass of it to define your command line arguments. Options such as `--verbose` and `--debug` are automatically integrated. Look at the source code of the various tools for examples.

t0mf, 2020-06-09:
I was using severity of debug, still nothing that lets me know what's going wrong.

lelegard, 2020-06-11:
>  Any recommendations on how to get a smooth file output for streams <= 40 Mbps?

I have already processed real-time DVB-S2 streams over 60 MB/s using tsp, with tsp using less that 5% of the CPU on a regular multi-core PC. So, this is a matter of system tuning and application architecture. There is no intrinsic limitation with TSDuck. I can't answer your question, it depends on your system and application.

t0mf, 2020-06-14:
> I have already processed real-time DVB-S2 streams over 60 MB/s using tsp, with tsp using less that 5% of the CPU on a regular multi-core PC. So, this is a matter of system tuning and application [...]

So at minimum I had to give a decent size input buffer. But after that, I think you're right - I think it was the PC's problem. Once I switched to a different one it seemed to work perfectly (Still had to have the buffer though).