tsduck/tsduck #594: How do you embed data as UDP packets in a TS file using the command line?

VDelport, 2020-06-07:
Hello, I would like to embed a data file (no MPEG data) in the form of UDP packets with an IP multicast address as the destination address and the local, private IP address as the source address in a transport stream using mpeinject. The result should be saved as a * .ts file. Unfortunately I cannot find an example for the command line call of tsp. Would you please give an example of a corresponding command line? Best regards.

lelegard, 2020-06-07:
There is a complete example of MPE injection in section 5.2.21 of the user's guide.

VDelport, 2020-06-07:
Thank you for your quick reply.
Am I correct that a transport stream from another source must always exist for the embedding of the UDP packets? In the example in section 5.2.21 of the user's guide, the transport stream is generated by a DVB tuner. How can you replace the DVB tuner with software to create a dummy transport stream? Is there a TSDuck command for this?

lelegard, 2020-06-15:
The easiest way is to create a null TS at a given bitrate, regulate it at playout speed and then inject all PSI, starting with the PAT, using XML files. That would be something like that:
~~~
tsp --bitrate 10,000,000 \
    -I null \
    -P regulate \
    -P inject --pid 0 --bitrate 15000 --stuffing pat.xml \
    ...
~~~

VDelport, 2020-06-15:
@lelegard  Thanks for the useful hint. Thanks for the useful hint. I will try it. I first created a video with ffmpeg into a transport stream and embedded the data there.