tsduck/tsduck #595: Options for inserting content in splices

moffitj, 2020-06-08:
I've tried to find a combination of plugins that would accomplish the inverse of the rm_splice plugin but haven't been successful so far.

Does anyone know of a combination of plugins that would let me insert content in place of a splice that was removed with the rm_splice plugin? I was thinking piping the output of rm_splice to ts_switch as the primary, and then triggering ffmpeg to pipe into a very large buffer on the first input of tsswitch on demand. But I found that rmsplice only removes the elementary streams of the service, so ts_switch doesn't detect that the primary stream has dropped.

Would an option on rm_splice that would stop all output when spliced be suggested? What other options exist?

Thanks,

lelegard, 2020-06-09:
This is currently not possible, at least directly. If the replacement content (typically "local ads" in the original spirit of SCTE 35) is in an SPTS file with same PID layout, the best solution would be to enhance the `rmsplice` plugin with an option specifying the replacement file. Instead of dropping packets, the plugin would replace them with packets from the replacement file, selecting only packets belonging to one of the removed PIDs and ignoring others. But this is new code to be done.

lelegard, 2020-06-15:
> That makes sense. From what I understand that would only work if the source and the insertion were CBR at the same bitrates, right?

At least, the insertion must not have a higher bitrate.

This is the general constraint with video splicing: same bitrate, same encoding, same encoding profile, same clock. Video splicing should be called "video transplant" and is just like "organ transplant": if the organ is not fully compatible, it is rejected.

arikwiesel, 2020-07-06:
I'm trying ti inject scte35 but it doesn't work. I see in verbose that it ok (using immediate) but it's not exist in the file.
I'm using input live output file.
this it from the verbose:
spliceinject: loaded file inject-scte.xml, 474 bytes
* 
spliceinject: enqueuing SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate
* 
appreciate any help
Arik

lelegard, 2020-07-07:
> I see in verbose that it ok (using immediate) but it's not exist in the file.

Failing to insert data is usually due to a lack of input stuffing. A transport stream is like a file system, you can insert new data only if there is some free space (ie. null packets, aka stuffing). If there is not enough free space, you need to create it, either by removing existing PID's or services or by artificially inflating the input stuffing. Please read the user's guide, especially the reference section for `tsp`  in chapter 3.

arikwiesel, 2020-07-07:
Thank you very much. Yes, you are right, I've added input stuffing and it works.
Now I'm looking for a notification when I receive SCTE35 from the input. I see log but can I get a notification specifically for the SCTE35 at the input?

lelegard, 2020-07-07:
> Now I'm looking for a notification when I receive SCTE35 from the input. 

Have a look at plugin `tables`. It intercepts tables or sections and can display them, save them (binary or XML) or even send them in UDP packets for some monitoring application.

kirillbushev, 2021-09-16:
b@moffitj  Did you find a solution to fill space provided by rmsplice?

kisso, 2023-04-21:
> This is currently not possible, at least directly. If the replacement content (typically "local ads" in the original spirit of SCTE 35) is in an SPTS file with same PID layout, the best solution [...]

Does this possibility exist yet?

closed 2023-04-21 by moffitj

kisso, 2023-04-21:
so what is the purpose of scte35 and rmsplice, when I cannot insert new ad?

> > Does this possibility already exist?
> 
> No, the replacement file would need to be absolutely in sync with the original removed slice, same variation of bitrate at the same time, identical (or close to identical) OCR, etc. That use case is unrealistic. Too many people would try to replace the removed slice by video slice not in sync. And would complain here that it does not work well. So I decided to not implement it.

so what is the purpose of scte35 and rmsplice, when I cannot insert new ad?

lelegard, 2023-04-21:
> so what is the purpose of scte35 and rmsplice, when I cannot insert new ad?

The purpose of `rmsplice` is to remove these annoying ads on recorded content. Provided that the broadcaster lets the splice information in the stream, which is often the case. This is a viewer-friendly feature, not a feature for broadcasters.

For broadcasters, the insertion of splice _information_ is done using `spliceinject` and the corresponding monitoring using `splicemonitor`.

The insertion of splice _audio/video content_ (your request) is a complex remux operation which needs a specialized equipment. At least in pure broadcasting context, IP streaming is different since it uses splice information to switch source instead of replacing content on the same TS. And TSDuck is not a MUX.

kisso, 2023-04-24:
our idea to insert ad, is to prepare transcoded .ts ad according to stream format, and then replace video and audio packets