tsduck/tsduck #596: Linux Mult-Tuner Detection

SS-FF, 2020-06-11:
Hi Thierry,

for Linux, there is a naming scheme for addressing a specific physical tuner:

https://github.com/tsduck/tsduck/blob/f82d2f3d6097a89982e0eef347ed42e910a2f140/src/libtsduck/dtv/tsTuner.h#L63-L67

but tslsdvb does not actually detect all the available physical tuners, just the adapters, so you'll never know how many (and which kinds of) physical tuners are available.

Is that just a missing bit or intentional for some reason?

lelegard, 2020-06-11:
> Is that just a missing bit or intentional for some reason?

No particular reason. This dates back from a time where all tuner devices has only one tuner. This needs to be changed.

When multi-tuner devices started to appear, they exposed two distinct adapters with only one front-end in each. I remember my first dual-tuner, an Hauppauge DVB-T PCI tuner from around 2008 or 9 which did that way: `/dev/dvb/adapter0` and `/dev/dvb/adapter1`.

This is an implementation choice in the driver. Apparently, most multi tuner devices expose one adapter per tuner. But some of them expose one adapter and multiple front-ends.

Could you indicate:
- The model of the device and its characteristics.
- `ls -lR /dev/dvb`
- `tslsdvb -v /dev/dvb/adapter0:0`
- `tslsdvb -v /dev/dvb/adapter0:1`
- ...
- Does the reception work on the second tuner when specifying `/dev/dvb/adapter0:1`?

I would like to know if this is only a matter of listing all tuners which misses some of them or if there is a deeper issue with secondary front-ends.

SS-FF, 2020-06-12:
I have a really interesting case here:

```
root@pc:/mnt/hdd/xxx_tv_3.18_armv7# ls -la /dev/dvb/
drwxr-xr-x    3 root     root            60 Jan  1  1970 .
drwxr-xr-x   17 root     root          2640 Mar 21 22:11 ..
drwxr-xr-x    2 root     root          2360 Mar 21 22:11 adapter0
root@pc:/mnt/hdd/xxx_tv_3.18_armv7# ls -la /dev/dvb/adapter0/
drwxr-xr-x    2 root     root          2360 Mar 21 22:11 .
drwxr-xr-x    3 root     root            60 Jan  1  1970 ..
crw-rw----    1 root     video     212,  71 Jan  1  1970 audio0
crw-rw----    1 root     video     212, 111 Mar 21 22:11 audio1
crw-rw----    1 root     video     212, 112 Mar 21 22:11 audio2
crw-rw----    1 root     video     212, 113 Mar 21 22:11 audio3
crw-rw----    1 root     video     212,  88 Jan  1  1970 ca0
crw-rw----    1 root     video     212,  89 Jan  1  1970 ca1
crw-rw----    1 root     video     212,  98 Jan  1  1970 ca10
crw-rw----    1 root     video     212,  99 Jan  1  1970 ca11
crw-rw----    1 root     video     212, 100 Jan  1  1970 ca12
crw-rw----    1 root     video     212, 101 Jan  1  1970 ca13
crw-rw----    1 root     video     212, 102 Jan  1  1970 ca14
crw-rw----    1 root     video     212, 103 Jan  1  1970 ca15
crw-rw----    1 root     video     212, 104 Jan  1  1970 ca16
crw-rw----    1 root     video     212, 105 Jan  1  1970 ca17
crw-rw----    1 root     video     212, 106 Jan  1  1970 ca18
crw-rw----    1 root     video     212, 107 Jan  1  1970 ca19
crw-rw----    1 root     video     212,  90 Jan  1  1970 ca2
crw-rw----    1 root     video     212, 108 Jan  1  1970 ca20
crw-rw----    1 root     video     212, 109 Jan  1  1970 ca21
crw-rw----    1 root     video     212, 110 Jan  1  1970 ca22
crw-rw----    1 root     video     212,  91 Jan  1  1970 ca3
crw-rw----    1 root     video     212,  92 Jan  1  1970 ca4
crw-rw----    1 root     video     212,  93 Jan  1  1970 ca5
crw-rw----    1 root     video     212,  94 Jan  1  1970 ca6
crw-rw----    1 root     video     212,  95 Jan  1  1970 ca7
crw-rw----    1 root     video     212,  96 Jan  1  1970 ca8
crw-rw----    1 root     video     212,  97 Jan  1  1970 ca9
crw-rw----    1 root     video     212,   0 Jan  1  1970 demux0
crw-rw----    1 root     video     212,   3 Jan  1  1970 demux1
crw-rw----    1 root     video     212,  30 Jan  1  1970 demux10
crw-rw----    1 root     video     212,  33 Jan  1  1970 demux11
crw-rw----    1 root     video     212,  36 Jan  1  1970 demux12
crw-rw----    1 root     video     212,  39 Jan  1  1970 demux13
crw-rw----    1 root     video     212,  42 Jan  1  1970 demux14
crw-rw----    1 root     video     212,  45 Jan  1  1970 demux15
crw-rw----    1 root     video     212,  48 Jan  1  1970 demux16
crw-rw----    1 root     video     212,  51 Jan  1  1970 demux17
crw-rw----    1 root     video     212,  54 Jan  1  1970 demux18
crw-rw----    1 root     video     212,  57 Jan  1  1970 demux19
crw-rw----    1 root     video     212,   6 Jan  1  1970 demux2
crw-rw----    1 root     video     212,  60 Jan  1  1970 demux20
crw-rw----    1 root     video     212,  63 Jan  1  1970 demux21
crw-rw----    1 root     video     212,  66 Jan  1  1970 demux22
crw-rw----    1 root     video     212,   9 Jan  1  1970 demux3
crw-rw----    1 root     video     212,  12 Jan  1  1970 demux4
crw-rw----    1 root     video     212,  15 Jan  1  1970 demux5
crw-rw----    1 root     video     212,  18 Jan  1  1970 demux6
crw-rw----    1 root     video     212,  21 Jan  1  1970 demux7
crw-rw----    1 root     video     212,  24 Jan  1  1970 demux8
crw-rw----    1 root     video     212,  27 Jan  1  1970 demux9
crw-rw----    1 root     video     212,   1 Jan  1  1970 dvr0
crw-rw----    1 root     video     212,   4 Jan  1  1970 dvr1
crw-rw----    1 root     video     212,  31 Jan  1  1970 dvr10
crw-rw----    1 root     video     212,  34 Jan  1  1970 dvr11
crw-rw----    1 root     video     212,  37 Jan  1  1970 dvr12
crw-rw----    1 root     video     212,  40 Jan  1  1970 dvr13
crw-rw----    1 root     video     212,  43 Jan  1  1970 dvr14
crw-rw----    1 root     video     212,  46 Jan  1  1970 dvr15
crw-rw----    1 root     video     212,  49 Jan  1  1970 dvr16
crw-rw----    1 root     video     212,  52 Jan  1  1970 dvr17
[... 66 lines cut ...]

  Device Path:
  Bit error rate ................................. 0 (0%)
  Signal/noise ratio ............................. 0 (0%)
  Signal strength ................................ 0 (0%)
  Uncorrected blocks ............................. 0
  Frequencies:
    Current ............................ 330,000,000 Hz
    Min ................................. 47,000,000 Hz
    Max ................................ 862,000,000 Hz
    Step .................................... 62,500 Hz
    Tolerance .................................... 0 Hz
  Symbol rates:
    Current .............................. 6,900,000 sym/s
    Min .................................... 870,000 sym/s
    Max ................................. 11,700,000 sym/s
    Tolerance .................................... 0 sym/s
  Spectral inversion ........................... off
  FEC(inner) .................................. auto
  Modulation ............................... 256-QAM

  Capabilities: inversion auto, FEC auto, 16-QAM, 32-QAM, 64-QAM, 128-QAM,
    256-QAM
```
The device is this one: http://www.vuplus.com/sub/sub0116.php and what's interesting is that it has two PCI cards, each having 8 tuners, but it's all under adapter0 - there's no adapter1..

PR #599 (2020-06-17): Fix multi-tuner detection on Linux and support Android device naming
Fixes #596
closed 2020-06-17 by lelegard