tsduck/tsduck #598: What does the error message 'mpeinject: error receiving from UDP socket: Bad file descriptor' mean?

VDelport, 2020-06-13:
Hello, I'm trying an MPE exactly according to user manual 5.2.21. The transport stream comes from ffmpeg. The embedded data is delivered to a multicast address via netcat. In debug mode I see that tsp receives the UDP data packets. However, no data is embedded. Among other things, tsp displays the following error message: 'Error: mpeinject: error receiving from UDP socket: Bad file descriptor'
What does the error message mean? Unfortunately I couldn't find anything.

lelegard, 2020-06-13:
There are two things.

First, nothing is inserted in the stream. Do you have enough stuffing (free space) to insert your data? Do you have stuffing at all? A transport stream is like a file system. When it's full (no free space), you can't add anything. Be sure to read and understand the `tsp` section in the TSDuck user's guide, chapter 3, especially the section named "Modifying, inserting and deleting packets". Note that the example in section 5.2.21 removes one service from the input stream and replaces it with stuffing in order to get some free space in the stream.

Concerning the "error receiving from UDP socket: Bad file descriptor", does it appear after you hit Ctrl-C to interrupt `tsp`? If yes, then this can happen if the UDP socket is closed at the same time a receive operation is started.

VDelport, 2020-06-13:
I have found the solution. Thank you for your answer and the project.

closed 2020-06-13 by VDelport