tsduck/tsduck #600: How do you create .ts files from videos in which you can later embed IP / UDP packets with tsp?

VDelport, 2020-06-14:
Hello, if I use one of your .ts file examples to embed IP / UDP data packets with the MPE function of tsp, the embedding works very well.

However, when I try to create my own .ts file with ffmpeg and then embed it via MPE IP / UDP data packets, the embedding fails.

I create the .ts file with the following command:
ffmpeg -i sample.mp4 -f mpegts -b: v 500k -maxrate 500k -minrate 500k -> sample.ts

Could you please give me a command line that can be used to create a .ts file with ffmpeg that can later be embedded with tsp IP / UDP data packets?

Or is there another tool that can be used to generate .ts files?

cbernander, 2020-06-14:
You probably need to add the _muxrate_ option - try adding something like `-muxrate 550k` to the command line. You may need to specify the _bufsize_ option as well if you get errors when transcoding.

Try searching the web for something like `ffmpeg mpegts cbr` for more examples with additional useful options.

VDelport, 2020-06-14:
Thank you very much for your useful tip.
I have now used the following command line and it works. :)
ffmpeg -i sample.ts -muxrate 1700k  -b:v 500k -maxrate 500k -minrate 500k -bufsize 1400k  
-f mpegts sample.ts

closed 2020-06-14 by VDelport