tsduck/tsduck #601: Cannot build deb package without libpcsclite1 dependency

cbernander, 2020-06-16:
I'm trying to build the TSDuck deb package without smartcard support by adding the NOPCSC make variable, but the deb package is still built with a dependency on the libpcsclite1 package. Is it possible to build the package without this dependency?

lelegard, 2020-06-16:
Hi @cbernander,

I have fixed the makefiles so that .rpm and .deb packages are built with not more dependencies than specified on the make command line.

However, the various `make NOxxx` variables were essentially made for custom builds in a restricted environment. When you generate a package for a distro, it is better to take full advantage of the distro and let the package manager resolve all dependencies for you.

If you do not want to take care of dependencies before _building_ TSDuck, the script `build/install-prerequisites.sh` is here to help.

I do not like the idea of different binary packages for the same version of TSDuck offering distinct sets of features. If you build a .deb package without PCSC support (or without CURL or Dektec or whatever), please keep it for your private usage. Do not distribute it. This would create some confusion.

cbernander, 2020-06-16:
Understood - it will only be used internally.

Thank you.

closed 2020-06-20 by cbernander