tsduck/tsduck #602: tsARIBCharset::Encoder not inserting LS1 into generated strings

trollcop, 2020-06-17:
Not sure if this is the right place to ask/mention this but,

when encoding a string like L"日本共産党放送"
the resulting B24-encoded string is "46 7c 4b 5c 36 26 3b 3a 45 5e 4a 7c 41 77"
However the LS1 (0x0e) shift is not prepended to the string. When looking at descriptors in ISDB-T broadcasts, for example in SDT,
```
            48      descriptor_tag
            12      descriptor_length
                01 00 0F 
                0E 4E 48 4B 45 1D 46 .... 
```

You can see that text starts with 0x0E locking shift.
Should this be fixed? Should I be manually prepending LS1?

lelegard, 2020-06-17:
What makes you think that the string "日本共産党放送" should start with LS1?

According to ARIB STD-B24, LS1 is a character with a specific function: "assign G1 to GL". So, an encoded string shall start with LS1 only if you need to "assign G1 to GL". This is not always the case.

The initial state of an ARIB STD-B24 encoder/decoder is:

- G0: Kanji 2-byte encoding
- G1: Alphanumeric 1-byte encoding
- G2: Hiragana 1-byte encoding
- G3: Katakana 1-byte encoding
- GL = G0
- GR = G2

So, when a string starts with an alphanumeric sequence (latin characters), we must use "Alphanumeric 1-byte encoding". By chance, it is already in G1. But G1 is not in the two active sets, GL and GR. So, you need to "assign G1 to GL" (LS1) and then start to encode your latin characters. Note that you could also "assign G1 to GR" (LS1R) if you prefer, there are several ways to encode the same string.

In other words, an encoded string shall start with LS1 _only_ if the first character is alphanumeric / latin.

In the case of your descriptor
~~~
$ tscharset --japan -d '0E 4E 48 4B 45 1D 46'
NHKEテ
$ 
~~~

The string starts with 0x0E (LS1) because the first four characters are latin: NHKE -> 4E 48 4B 45. Then, 0x1D (SS3) means "assign G3 to GL", effectively using "Katakana 1-byte encoding". Are the following characters really Katakana?

In the case of your initial string, if we try to decode the binary without LS1:
~~~
$ tscharset --japan -d '46 7c 4b 5c 36 26 3b 3a 45 5e 4a 7c 41 77'
日本共産党放送
~~~

Being French and European, I unfortunately don't know a word of Japanese and I can hardly recognize Japanese characters. But the decoding looks good to me if I visually compare it with your string. Google Translation gives "Japanese Communist Party Broadcast". Is that correct? A very different type of service compared to NHKE, right ?

Could you confirm if the characters are Kanji? If this is the case, then we use "Kanji 2-byte encoding", already in G0, which in turn is already assigned to GL. So, we can directly start to encode characters without any control sequence.

Reference: http://www.arib.or.jp/english/html/overview/doc/6-STD-B24v6_4-1p3-E1.pdf
See table 7-1 page 37.

trollcop, 2020-06-18:
Hey,

You are absolutely right.
Reason all this came up is I was trying to verify the generated descriptors/etc in TSReader, and was getting garbage encoding, so I thought there was something missing.

> $ tscharset --japan -d '0E 4E 48 4B 45 1D 46'
> NHKEテ

> Are the following characters really Katakana?
Yes, they are. I cut off the remainder of the descriptor hex, but it says ＮＨＫＥテレ福岡 which is NHK-E channel, and テレ are 2 katakana characters. And by your explanation, needing LS1 at start followed by SS3 there makes sense.

> Could you confirm if the characters are Kanji?
Yep, entire string is kanji, so it makes sense to start encoding them right away without shift with default encoder state.

Anyway, the whole reason this came up, is I've asked TSReader developer how the ARIB string is detected, and he is checking for the control codes at beginning of a string, such as LS1/LS0/SS2/SS3, and then switching string decoding to B24. 

So this "works" most of the time because transport name descriptor usually starts with 3 wide alpha letters, or sometimes katakana, as per ARIB TR-B14 part 5, Table 9-3 "TS Names". ref: http://www.arib.or.jp/english/html/overview/doc/8-TR-B14v6_0-5p5-E1.pdf
However some do start with kanji, such as "東海テレビ", where this logic would break.

Anyway, sorry for the long rant, but yeah it seems to be tsARIBCharset is working properly and there's just a misunderstanding with the tools that analyze generated content. I've re-analyzed the ts with slightly more expensive tools (DekTec StreamXpert) and it's appearing correctly:

[image]

Thanks a lot for such a detailed reply!

trollcop, 2020-06-18:
But, wait a second. If I try and encode a string like "NCPテレビ"
the string is get is "1b 7e ce c3 d0 25 46 25 6c 25 53"
which is not LS1 but ESC + ?? and encodes ascii as some 8-bit chars.

what is tscharset by the way, I don't see that tool available in the standard release downloads

lelegard, 2020-06-18:
> Anyway, the whole reason this came up, is I've asked TSReader developer how the ARIB string is detected, and he is checking for the control codes at beginning of a string, such as LS1/LS0/SS2/SS3, and then switching string decoding to B24.

I understand the encoding dilemma because I had it too. The problem with ARIB signalisation is that they re-used part of the DVB syntax with sometimes a different semantics: B24 character sets instead of DVB encoding, JST instead of UTC, etc. This is an extremely poor design choice in an international environment. Either you choose a standard and rigorously use it or invent a new standard. For instance, ATSC chose a radically different way. They created a new set of tables and descriptors and use a specific character encoding inside them. When you decode an ATSC structure, you know which character encoding to use. On the contrary, when exploring DVB tables and descriptors, DVB character encoding is used everywhere on Earth, except in Japan. And there is no obvious proof that the stream comes from Japan. So, by default, DVB is DVB for TSDuck.

TSReader tries to guess the encoding from some hints that work in some cases, but not in others. TSDuck also tries to guess the standard, globally: if we encounter specific ATSC or ISDB signalisation in the TS, then the whole TS is tagged with a standard. There was already a `--default-charset` option in TSDuck because some DVB operators use an incorrect signalisation. They assume that the most useful charset for their language is the default one (incorrect according to DVB spec). I assume that their proprietary receivers make the same assumption. But at TSDuck level, we need to indicate which charset is used.

So I added option `--default-charset` long ago and now you have to use `--default-charset ARIB-STD-B24` on Japanese streams. Given the number of incorrect usages of DVB data in ARIB, I added the synthetic option `--japan` to set all suitable defaults for a Japanese stream. If you work only with Japanese streams, you should set in the TSDuck configuration file (see Appendix 1 of the user's guide).

> what is tscharset by the way, I don't see that tool available in the standard release downloads

It has been added in version 3.21, the version which now correctly handles Japanese streams (thanks to the help of two Japanese users). 

> If I try and encode a string like "NCPテレビ"
> the string is get is "1b 7e ce c3 d0 25 46 25 6c 25 53"
> which is not LS1 but ESC + ?? and encodes ascii as some 8-bit chars.

~~~
$ tscharset --japan -d 1b7ecec3d02546256c2553
NCPテレビ
~~~

Looks good to me. 1B 7E is LSR1, "assign G1 to GR". As I wrote before, there are more than one way to encode the same string. TSDuck is probably not the best optimizing encoding tools for ARIB STB-B24. There is probably a way to gain one or two bytes here and there. But, honestly, one or two bytes... As long as the encoding is correct, this is fine with me.

trollcop, 2020-06-18:
> This is an extremely poor design choice in an international environment.

Yeah, tell me about it. TSReader doesn't do proper EIT offset for ISDB due to the UTC/JST reason, since everone else in the world does it properly.

> It has been added in version 3.21, the version which now correctly handles Japanese streams

Hm? I grabbed windows-x64-portable zip from /releases and that's not one of the binaries.

> Looks good to me. 1B 7E is LSR1, "assign G1 to GR". As I wrote before, there are more than one way to encode the same string. TSDuck is probably not the best optimizing encoding tools for ARIB [...]

Yep, I understand. But I'm curious, and since you know the codebase better, do you think if I init with _GL_last = false in the encoder, is that going to break anything? It seems to result in LS1 + ascii being used for ascii encoding which saves a bit of space and makes descriptor readable, and doesn't seem to break any other test cases (kanji string, mixed kanji+kana, etc). Space *is* kinda a concern because TR-B14 specifies fairly limited byte sizes for descriptor lengths, so...

lelegard, 2020-06-18:
> Hm? I grabbed windows-x64-portable zip from /releases and that's not one of the binaries.

Ooops, you are right, thank you for noticing this. I just found a mistake in the MSBuild file for tscharset. On Windows, when the build was "made for an installer", tscharset was not rebuilt. For my tests, I rebuild installers on top of the dev environment and tscharset has been usually rebuilt recently, so the binary was already there and embedded in the package. For the official release, I clean up everything first and rebuild from scratch. And there, tscharset was not rebuilt. This is the same for the nightly builds, they are built on a fresh empty GitHub runner VM system. So, tscharset is not in the Windows nightly builds either.

The problem does not exist on macOS and Linux since the build system is completely different. 

I just fixed this. So, the next nightly builds for Windows will include tscharset. I will also rebuild and update the release 3.21.

> do you think if I init with _GL_last = false in the encoder, is that going to break anything?

No, I don't think so. I will update and test. But this is just a flag which ensures that we alternate the use of GL and GR. The initial state is not important. It will change the encoding but in a still valid way.

> Space is kinda a concern because TR-B14 specifies fairly limited byte sizes for descriptor lengths, so...

Yes, a descriptor is limited to 255 bytes in all standards. That is a master MPEG specification.

Honestly, more or less one byte will not change much in practice. The only descriptor in which the limitation has an impact is the extended event descriptor. But this one is designed to chain more than one instance.

Talking about the extended event descriptor, I noticed that NHK uses it incorrectly (seen in an ISDB-T capture). They encode the string text _first_ and then they split the binary content over the various extended event descriptors. In at least one occurrence, I see that the split occurred in the middle of a 2-byte character. The second descriptor started "in the middle of a character" (and without charset-switching control sequence of course). This is illegal. The text field of the descriptor must be a character string _"coded using the character sets and methods described in Annex A"_ as they say. In that case, the character string is incorrect and the beginning of the string was decoded as garbage, up to the next charset-switching control sequence.

trollcop, 2020-06-18:
> I just fixed this. So, the next nightly builds for Windows will include tscharset. I will also rebuild and update the release 3.21.

Nice, thank you!

> Yes, a descriptor is limited to 255 bytes in all standards. That is a master MPEG specification.

True, but it is even further limited by operating specifications, TR-B14, for example Network Name Descriptor specifies:

[image]

Tho I suppose that means actually visual presentation and not how many bytes it takes to encode the same thing, but still.

Anywho, it seems to run with last=false and results in slightly more "user-friendly" encoding of ASCII as actually ASCII, so I'm OK with that, as long as it has no other side-effects.

> they split the binary content over the various extended event descriptors

I'll be getting to EIT soon, but I thought sections were split up, then assembled into a single long thing, and then parsed after reassembly, buut yeah I wouldn't be too surprised if some things were broken there... thankfully ISDB-S3 done away with this godawful encoding and switched to UTF-8 finally... Probably same will be in ISDB-T2 if that ever gets finalized.

lelegard, 2020-06-18:
> TR-B14, for example Network Name Descriptor specifies:
> ... 10 two-byte or 20 one-byte characters ...

This is a particularly insane requirement. What about a network name mixing one-byte latin chars and two-byte kanji chars? And even though, it does not make sense to limit the network name.

I updated the encoder strategy so that it now uses GL at the first charset switch. I had, of course, to update the test suites. Interestingly, the encoded strings in the unitary test suite were either the same size or _longer_ than previously. But in the system test suite, the encoded strings were either the same size or _shorter_. So, basically, it depends. The gain in space is not obvious, it depends on the context.

lelegard, 2020-06-19:
> > I just fixed this. So, the next nightly builds for Windows will include tscharset. I will also rebuild and update the release 3.21.
>
> Nice, thank you!

I confirm that `tscharset.exe` for Windows is now in the latest nightly builds here: https://tsduck.io/download/prerelease/

I still had not yet time to rebuild release 3.21 with the fix (this must be done manually while the nightly builds are fully automated).

trollcop, 2020-06-19:
> I updated the encoder strategy so that it now uses GL at the first charset switch.

Excellent. Yeah I've ran some tests here and it looked OK but it's good to have confirmation.

> I confirm that tscharset.exe for Windows is now in the latest nightly builds here:

Nice, thank you. That will do for now, no problem.

> This is a particularly insane requirement. What about a network name mixing one-byte latin chars and two-byte kanji chars? And even though, it does not make sense to limit the network name.

I think this is aimed at receivers with limited ram/storage space, as TR-B14 specs cover these kinda cases, for example the simple logo descriptor is limited to 5 characters (because it might be rendered by mobile phone or similar limited-capacity device).

And I guess since we did resolve the LS1 stuff + ended up with a slightly nicer looking encoder output, this bug can be closed as well.

Thanks for a quick reply on this.

closed 2020-06-19 by trollcop