tsduck/tsduck #603: Linux Tuner Identification

SS-FF, 2020-06-18:
Hi Thierry,

this is some kind of follow-up to  https://github.com/tsduck/tsduck/issues/596.

Now we detect and address multi-tuners, but this still relies on the device indexes under `/dev`.

But I would assume that those are not guaranteed to remain constant over time. Especially when it comes to USB tuners which - like all USB devices - can be plugged in and out any time.

Do you have any idea (besides the device name) how a device could be identified somewhat better?

It's about being able to answer the question:

"Is device `/dev/dvb/adapter2/frontend0` the same that it was yesterday?"

Thanks

lelegard, 2020-06-18:
> It's about being able to answer the question:
> "Is device /dev/dvb/adapter2/frontend0 the same that it was yesterday?"

I don't think there is a general answer. That is the fate of all USB devices. If you have more than one tuner, you should probably start with a script analyzing the output of `tslsdvb`. Or we could introduce some new syntax in TSDuck to address a tuner by "device info", the string which is displayed by `tslsdvb` and returned in field `name` of `struct dvb_frontend_info`.

But if you have two tuners of the same model (same device info) and connected to distinct antennas or dishes, that won't solve the problem.

Maybe there is some solution using udev. But this is a question that should be asked in the V4L mailing list.

SS-FF, 2020-06-24:
> Maybe there is some solution using udev. But this is a question that should be asked in the V4L mailing list.

Yea, I thought about something like a serial number - but I suppose the DVB API does not have a way to retrieve it?

lelegard, 2020-06-24:
>  I thought about something like a serial number - but I suppose the DVB API does not have a way to retrieve it?

As device specific info, there is only the "device info" string, for instance for an AverMedia TD310 USB device;
~~~
$ tslsdvb 
/dev/dvb/adapter0 (Silicon Labs Si2168, DVB-T, DVB-T2, DVB-C)
$ 
~~~
The device info string is "Silicon Labs Si2168". So, it won't work with distinct tuners of the same model.

I suppose you need an initial script before starting the application which determines which USB tuner device is which `/dev/dvb` adapter.

The output from `dmeg` at the time the device is plugged is:
~~~
[   75.622787] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[   75.951754] usb 1-1: New USB device found, idVendor=07ca, idProduct=1871, bcdDevice= 1.00
[   75.951757] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   75.951758] usb 1-1: Product: TD310 Device
[   75.951759] usb 1-1: Manufacturer: AVerMedia
[   75.951760] usb 1-1: SerialNumber: 202866000009
~~~
Try to correlate device number, here `usb 1-1`, with the adapter number.

Or with `lsusb`:
~~~
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 07ca:1871 AVerMedia Technologies, Inc. TD310 Device
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ 
$ sudo lsusb -s 1:2 -v

Bus 001 Device 002: ID 07ca:1871 AVerMedia Technologies, Inc. TD310 Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x07ca AVerMedia Technologies, Inc.
  idProduct          0x1871 
  bcdDevice            1.00
  iManufacturer           1 AVerMedia
  iProduct                2 TD310 Device
  iSerial                 3 202866000009
  bNumConfigurations      1
...
~~~
For some reason, the `iSerial` info is not displayed without `sudo`.

Hope this helps...

SS-FF, 2020-06-27:
> As device specific info, there is only the "device info" string, for instance for an AverMedia TD310 USB device;
>
> ```
> $ tslsdvb
> /dev/dvb/adapter0 (Silicon Labs Si2168, DVB-T, DVB-T2, DVB-C)
> $
> ``` [...]

Thanks for your detailed response. Yes, the above it what I'm doing right now.

BTW, I forgot to mention that what you had written above:

> But if you have two tuners of the same model (same device info) and connected to distinct antennas or dishes, that won't solve the problem.

I exactly the concern that I'm having. It's wrong to assume that antenna connections are interchangeable, that's why I'm requiring  that each tuner needs to do its own channel scan. Cable, connectors, antennas, antenna position, moon phase (😆) can all make a difference.
Actually, the rule is not channel-scan "per-tuner" but rather "per-antenna connection". Unfortunately, this is not possible to detect. You can only make assumptions based on heuristics...

> I suppose you need an initial script before starting the application which determines which USB tuner device is which `/dev/dvb` adapter

On Windows I got it working nicely now - the application is monitoring for device change system events and dynamically detects all changes...

> The output from `dmesg` at the time the device is plugged is:
> 
> ```

Hm, parsing a text log seems messy..

> 
> Or with `lsusb`:
> For some reason, the `iSerial` info is not displayed without `sudo`.
> 
> Hope this helps...

Thanks for the hint, yes that seems feasible, although it does not fit into the comfortable situation I'm having right now: 
I have a single binary for aarch64 ARM that runs on all (at least most) ARM Linux distributions and even on Android (same for AMD64). That's why I had hoped for something in the DVB API or something that I could call from inside the tsDuck binary. 
Could there be something useful in the /sys/* file system that could be queried from tslsdvb?

lelegard, 2020-06-29:
The best method I could find so far:
~~~
$ find /sys/devices -name 'dvb*frontend*'
/sys/devices/pci0000:00/0000:00:0c.0/usb1/1-1/dvb/dvb0.frontend0
$
$ dmesg | grep 'usb 1-1' | grep -i serialnumber | tail -1
[  259.192271] usb 1-1: SerialNumber: 202866000009
~~~

From the first command, you can derive a USB address and a DVB adapter name. Using this USB address, you can find the tuner serial number. This way, you can associate a specific DVB adapter to a known serial number (ie. known tuner and antenna source).

Or, more realistically, to look up the DVB adapter name for a given known serial number:
~~~
$ SERIAL=202866000009
$ 
$ USBADDR=$(dmesg | grep -i "usb.*serial.* $SERIAL" | tail -1 | sed -e 's/^.* usb *//' -e 's/:.*//')
$ ADAPTER=$(basename $(find /sys/devices -path "*/usb*/$USBADDR/dvb/*" -name "dvb*frontend*" | head -1) | sed -e 's|^dvb|/dev/dvb/adapter|' -e 's|\.frontend|:|')
$ 
$ echo $ADAPTER 
/dev/dvb/adapter0:0
~~~

However, the amount of information in `dmesg`  and the structure of USB devices unders `/sys/devices` may depend on tuner driver or kernel version (depends on what is logged in `dmesg`).

So, since there is no guarantee on what could be extracted as unique DVB adapted information and no guarantee that it will remain stable over time, I will not include that in TSDuck code. No way.

For a given system setup such as yours, once you know which tuner serial number is on which source, you should write a script which uses the above method and then run your application with the right DVB adapter name as argument.

Since your may have to adapt depending on driver or kernel upgrade, it is better to have a script that you can easily update.

SS-FF, 2020-06-29:
Those were great pointers!

Forget about dmesg - I think I got the solution - it works on Android and Ubuntu:

### Android
```
usr:/ $ find /sys/devices -name 'dvb*frontend*'
/sys/devices/70090000.xusb/usb1/1-2/1-2:1.0/dvb/dvb0.frontend0
usr:/ $ cat /sys/devices/70090000.xusb/usb1/1-2/serial
0014039368
```

### Ubuntu
```
usr@ubuntu2:~$ find /sys/devices -name 'dvb*frontend*'
/sys/devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0/dvb/dvb0.frontend0
usrz@ubuntu2:~$ cat /sys/devices/platform/vhci_hcd.0/usb1/1-1/serial
0014039368
```

What do you think?

lelegard, 2020-06-30:
OK, it's implemented now.

On Linux only (nothing done on Windows), the unique identification information such as the serial number is now included in the tuner info.

Example:
~~~
$ tslsdvb 
/dev/dvb/adapter0 ("Silicon Labs Si2168, AVerMedia TD310 Device 2.00 SN:202866000009", DVB-T, DVB-T2, DVB-C)
/dev/dvb/adapter1 ("DiBcom 7000MA/MB/PA/PB/MC, Hauppauge Nova-T Stick 2.00 SN:4027216814", DVB-T)
$
~~~

Note that there is no guarantee that such unique information is available. It depends on the drivers, the kernel, etc. So, if it's here, you're lucky. If it's not there, too bad...

For the ease of use, the complete tuner description (the string between quotes) can be also used as tuner device name. As an example, the three commands below are equivalent:
~~~
$ tsp -I dvb -d "Silicon Labs Si2168, AVerMedia TD310 Device 2.00 SN:202866000009" ...
$ tsp -I dvb -d /dev/dvb/adapter0 ...
$ tsp -I dvb -a 0 ...
~~~

SS-FF, 2020-06-30:
Implemented?? Wow, that's awesome, I had expected something like "yes, sounds feasible" 😃 

Thank you very much, I really appreciate it.

Let me tell you how it can be done on Windows - hold on..

SS-FF, 2020-06-30:
On Windows, it is rather easy:

From each tuner filter, you can query its "device path". Here's an example:

```
2: "Hauppauge WinTV-soloHD DVBC Tuner" (DVB-C)                                                                                                             
  Device Path:      @device:pnp:\\?\usb#vid_2040&pid_8268#0014039368#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{b50b8116-da24-4f97-80d1-00451702c5f7}         
3: "Hauppauge WinTV-soloHD DVBT Tuner" (DVB-T)                                                                                                             
  Device Path:      @device:pnp:\\?\usb#vid_2040&pid_8268#0014039368#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{7c8095ab-c110-40e5-9f4d-310858bbbf64}                                                                                                                                             
```

The cool thing about that device path is that it remains constant. It doesn't change. I can plug the tuner into any USB port, I can add or remove other devices in the meantime, but the device path never changes.

Looking closer...

@device:pnp:\\?\usb#vid_2040&pid_8268#**0014039368**#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{b50b8116-da24-4f97-80d1-00451702c5f7}    

...you can see that the serial (see my above examples) is included there.

So, all that is needed to identify a tuner is that device path.

SS-FF, 2020-06-30:
Implementation:

I have extended tsTuner by adding a devicePath property:

[image]
(in case of Linux, it contains the `/dev/...` path)

And here's how to get that path:

```c
    // Loop on all enumerated tuners.
    for (size_t dvb_device_current = 0; dvb_device_current < tuner_monikers.size(); ++dvb_device_current) {

        HRESULT hr;
        ::WCHAR* wstring;

        // Get friendly name of this tuner filter
        const UString tuner_name(GetStringPropertyBag(tuner_monikers[dvb_device_current].pointer(), L"FriendlyName", debug_report));
        report.info(u"");
        report.info(u"Found tuner filter \"%s\"", {tuner_name});

        UString device_path;

        hr = tuner_monikers[dvb_device_current]->GetDisplayName(0, 0, &wstring);
        if (ComSuccess(hr, u"IMoniker::GetDisplayName", report)) {
            device_path = ToString(wstring);
            ::CoTaskMemFree(wstring);
        }

        report.info(u"Device path: \"%s\"", {device_path});

        // If a device name was specified, filter this name.
        if (tuner != 0 && !tuner->_device_name.empty()) {
            if (dvb_device_index >= 0 && int(dvb_device_current) != dvb_device_index) {
                // Device specified by index, but not this one, try next tuner
                continue;
            }
            else if (dvb_device_index < 0 && 
                (!(tuner_name.similar(tuner->_device_name) || (!device_path.empty() && device_path.similar(tuner->_device_name))))) {
                // Device specified by name, but not this one, try next tuner
                // Since the filter names are long and complicated, ignore case and blanks, use UString::similar().
                continue;
            }
            // Device found, update device name
            tuner->_device_name = tuner_name;
            tuner->_device_path = device_path;
        }

```

The device path can be used instead of the tuner name for selecting a tuner.

I'm not creating a PR, because I don't know whether or how you would want to do it, and my version is too far away from yours at this time.

lelegard, 2020-06-30:
I will have a look at the Windows option.

Note that, on Linux, not all drivers expose a serial number.

See this on a MyGica S270 tuner:
~~~
$ tslsdvb
/dev/dvb/adapter0 ("Siano Mobile Digital MDTV Receiver, MDTV Receiver 2.00", ISDB-T)
~~~

In fact, there is no `serial` file in the sys device tree for that one:
~~~
$ ls /sys/devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/dvb/dvb0.frontend0/device/
3-2:1.0              bDeviceClass     bMaxPower           descriptors  dvb          manufacturer  product    speed      version
authorized           bDeviceProtocol  bNumConfigurations  dev          ep_00        maxchild      quirks     subsystem
avoid_reset_quirk    bDeviceSubClass  bNumInterfaces      devnum       idProduct    media0        removable  tx_lanes
bcdDevice            bmAttributes     busnum              devpath      idVendor     port          remove     uevent
bConfigurationValue  bMaxPacketSize0  configuration       driver       ltm_capable  power         rx_lanes   urbnum
~~~

So, in some cases, the unique device identification won't be possible in case of multiple tuners of the same model.

SS-FF, 2020-06-30:
```
$ ls /sys/devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/
````
What's in this directory?

SS-FF, 2020-06-30:
Or is that just the same (the last subdirectory linking back to this)?

I doubt that the device doesn't have a serial, but when the driver doesn't expose it - well, then we're out of luck.
(I'm glad that it's at least some vendor/model that I've never heard about ;-)

lelegard, 2020-07-01:
> Or is that just the same (the last subdirectory linking back to this)?

It is indeed. It is easier to use the symbolic link `dvbX.frontendX/device` because its name is always the same while the pointed directory is at different numbers of levels from `/sys/devices` (probably depending on the bus hierarchy in the system).

> I doubt that the device doesn't have a serial, but when the driver doesn't expose it - well, then we're out of luck.

Yes, it is clearly a driver issue, not a hardware one.

> (I'm glad that it's at least some vendor/model that I've never heard about ;-)

It is an ISDB-T tuner that was offered by a Japanese user. Very different from DVB tuners we are used to.

lelegard, 2020-07-01:
@SS-FF, your usage of the Windows device path has been integrated in the code. Displayed with `tslsdvb -v`. Can be used as tuner name: `tsp -I dvb -d "@@device:pnp:\\?\usb#vid..."`. Note the double `@@`. Read the user guide sections 3.1.5 and 6.1.3.2 if you want to know why.

SS-FF, 2020-07-01:
> @SS-FF, your usage of the Windows device path has been integrated in the code. Displayed with `tslsdvb -v`. Can be used as tuner name: `tsp -I dvb -d "@@device:pnp:\\?\usb#vid..."`. 

Great!

> Note the double `@@`. Read the user guide sections 3.1.5 and 6.1.3.2 if you want to know why.

I know why. I've gone through this last year ;-)

SS-FF, 2020-07-01:
Device paths on Windows can come in different variations. PCI device paths look a bit different and there are also "software device", where the device path can look like:

```
@device:sw:{71985F48-1CA1-11D3-9CC8-00C04F7971E0}\VendorName ProductName Tuner 1A7375CF-0
```

But you can still rely on those paths being unique.

SS-FF, 2021-05-03:
Done.

closed 2021-05-03 by SS-FF