tsduck/tsduck #604: _sections.size() < 1000

Galvo87, 2020-06-19:
I am getting a recurring EIT error on a particular frequency.

`tsp: dtv/tsEITProcessor.cpp:417: virtual void ts::EITProcessor::handleSection(ts::SectionDemux&, const ts::Section&): Assertion `_sections.size() < 1000' failed.`

The transport stream comes from a dvb-t2 card, the other channels run fine 24/7.
Is it possible to log the real cause of the issue, or some way to manage .size > 1000 to not let tsp crash?
Thanks.

lelegard, 2020-06-19:
Hi @Galvo87,

The assertion was an initial foolproof guard that should have been replaced later. Having it still in place is clearly a mistake. I just replaced it with some adjustable limit above which the next EIT section is dropped and a warning message is displayed.

You may pull the repo and [rebuild](https://tsduck.io/doxy/building.html). This is trivial on Linux and macOS. If you need a Windows version and don't want to setup a Windows build environment (not very difficult either), you can wait until tomorrow morning (UTC) to get the [latest nightly builds](https://tsduck.io/download/prerelease/).

Now, the source cause remains: an overflow of internally buffered EIT sections.

Some plugins process EIT (remove, rename TS or service, update time, etc.)

It would be interesting to see your complete `tsp` command so that we could see how EIT are processed. And how often does the problem occur?

Generally speaking, `EITProcessor` (named in the assertion failure) is an internal C++ class which processes EIT sections in many ways. A few number of EIT sections may be temporarily buffered inside it. Each EIT section is completely loaded, modified and reinjected, replacing input packets. If the currently reinjected EIT section is very large and, at the same time, a lot of small EIT sections are received, they must be buffered. This is normally a transient situation. Since the number of reinjected sections is at most identical to the input number and since EIT sections are never enlarged, there should be no global overflow over time.

So, getting an overflow (more than 1000 buffered EIT sections!) is not a normal situation.

Trying to find an explanation, I can only think of one. This is very technical about the way sections are packetized in the TS. This is a job which is performed by another internal C++ class named `Packetizer` which can implement different packetization strategies. Sections can be inserted at TS packet boundaries or packed, starting anywhere in the middle of a packet. The EIT processor always use a packed strategy to avoid losing space and because this is the "natural way" of processing EIT's.

However, in all cases, a TSDuck `Packetizer` never splits a section header between two TS packets. This is not required by the MPEG standard but some buggy receivers are known to have problems with splitted headers. So, as a precaution, when a section header (8 bytes) does not fit at the end of a TS packet, a few stuffing bytes are added and the section starts at the beginning of the next TS packet.

If we assume that your stream has packed EIT sections and that the original operator's EIT packetizer is authorized to split section headers, then we will accumulate a few buffered bytes each time the input EIT's have splitted headers. If `tsp` is running 24x7, we may accumulate too many late sections after days or weeks.

This is the only explanation I can find today.

Galvo87, 2020-06-19:
Thank you for the insightful reply.
Yeah tsp instances are running 24/7, and this particular instance is the only one giving me this issue, so maybe I'm not correctly using plugins in the chain?
`sudo tsp -v -I dvb --signal-timeout 10 --adapter 14 --frequency 682000000 -P analyze -w -P svremove 9 -P svremove 10 -P svremove 11 -P svremove 90 -P svremove 106 -P svremove 218 -P svremove 219 -P svremove 220 -P svremove 221 -P rmorphan -P pcrbitrate -P regulate -P continuity --fix -O ip IP:PORT`

lelegard, 2020-06-19:
OK, so there is only `svremove` which processes EIT's. But it only removes EIT sections for the removed service, So the amount of output sections is less than the input one. Unless there is one service for which there is no EIT (a data service maybe). In which case the scenario I described could apply. But only if this is the first service in the chain (id 9). Because, starting with the second plugin, the input EIT PID was already processed in the previous plugin. So, the packetization was performed by TSDuck, without section header split.

Additional remarks:

- Using the root account (`sudo`) to run a complex software which can dynamically activate shared libraries is a bad idea in terms of security.
- The `analyze` plugin seems useless because it produces the analysis report at the end only and the command is running "forever". Maybe you should consider using option `--interval`.

lelegard, 2020-06-19:
Could you also describe how often the problem occurs? This could give a hint on the root cause.

Galvo87, 2020-06-19:
It occurs everyday, only on this particular tsp instance.

Galvo87, 2020-06-22:
Any hints on this?
If it's of any help, the process is killed with this error:
`was killed with signal SIGSEGV","code":"SIGSEGV"`
Thanks

lelegard, 2020-06-22:
> If it's of any help, the process is killed with this error:
was killed with signal SIGSEGV","code":"SIGSEGV"

Do you use the latest update? I removed the assertion and drop overflown sections with a warning message.

If the problem occurs on one TS only, there must be something special in this TS. Could you provide a 2-minute capture of it?

Also add option `--monitor` to your `tsp` command. It monitors CPU time and VM usage of the process. We could see if there is some particularity is in the memory usage.

Galvo87, 2020-06-22:
Ok I'll build the latest update, test it and report back!
If the problem still persists I'll provide you a sample.
Thanks

Galvo87, 2020-06-23:
At the moment the new build is not giving the _sections.size() < 1000 issue anymore, but it's still failing with segmentation fault SIGSEGV... it happens only in one particular frequency, even on different dvb adapters.

lelegard, 2020-06-23:
The SIGSEV is definitely a problem which must be fixed. I should not happen.

Did you try the recommended `--monitor` option and what is the display? Do you see any memory leak (VM always increasing)?

I need to reproduce the problem. If I cannot reproduce it, it cannot be fixed. Could you please try to reproduce it on a file?

- First, capture a few minutes of the frequency which exhibits the problem:
  - `tsp -I dvb ... -P until --second 120 -O file problem.ts`
  - specify the same tuning parameters as you use in the command which crashes.
- Second, try to reproduce it on the captured file:
  - `tsp -I file problem.ts --infinite -P .... -O drop`
  - specify the same suite of `-P` plugins as you use in the command which crashes.
  - The command will run endlessly, until you stop it or it crashes. If it crashes, please post the file in some file sharing service.

Galvo87, 2020-06-25:
I've captured the sample and tested with my usual plugins for more than 24h and everything seems good, no more SIGSEV...
But usually I launch more than 20 tsp instances, I'll test if multiple instances are the problem and report back.

lelegard, 2020-06-25:
If the problem occurs on the same specific TS all the time, this is probably not a matter of system load (your 20 instances of tsp). Otherwise, it would probably occur on a random TS.

If the problem does not occur on a cycled captured segment, the problem is probably due to either one specific data structure in the TS which occurs rarely (and especially not in the captured segment) or an accumulation of different data structures (there were not enough of them in the captured segment).

You still did not mention the result of the `--monitor` option. Is the VM stable or always increasing?

I suggest that you remove the tsp instance on the culprit stream from the machine where you run the 20 instances of tsp and move it alone on another machine.

On this machine, do not use the binary distribution. Clone the repo and build in debug mode (`make -j10 debug`). Be sure to enable core dump (`ulimit -c unlimited`). Use option `--monitor`. After the crash, please provide the complete output of the command, include all monitoring traces. Then, analyse the core and get a backtrace:

~~~
gdb tsp core
(gdb) backtrace
(gdb) quit
~~~

The core file may not be in the current directory, it depends on the distro. So, you have to get it first.

Galvo87, 2020-06-25:
Sure I can try the above on a second machine.
I never used the binary distribution, I always compiled from github repo.
When compiling with `make -j$(nproc) debug` with I get:

`/usr/bin/ld: /home/developer/tsduck/bin/debug-x86_64-moi-pro-2/libtsduck.so: undefined reference to `ts::EITProcessor::MIN_BUFFERED_SECTIONS'
collect2: error: ld returned 1 exit status
make[3]: *** [../../Makefile.common:953: /home/developer/tsduck/bin/debug-x86_64-moi-pro-2/tsanalyze] Error 1
make[3]: *** Waiting for unfinished jobs....
/usr/bin/ld: /home/developer/tsduck/bin/debug-x86_64-moi-pro-2/libtsduck.so: undefined reference to `ts::EITProcessor::MIN_BUFFERED_SECTIONS'
collect2: error: ld returned 1 exit status
make[3]: *** [../../Makefile.common:953: /home/developer/tsduck/bin/debug-x86_64-moi-pro-2/tsbitrate] Error 1
/usr/bin/ld: /home/developer/tsduck/bin/debug-x86_64-moi-pro-2/libtsduck.so: undefined reference to `ts::EITProcessor::MIN_BUFFERED_SECTIONS'
collect2: error: ld returned 1 exit status
make[3]: *** [../../Makefile.common:953: /home/developer/tsduck/bin/debug-x86_64-moi-pro-2/tscharset] Error 1
make[2]: *** [Makefile:43: default] Error 2
make[1]: *** [Makefile:59: default] Error 2
make: *** [Makefile.common:785: debug] Error 2`

lelegard, 2020-06-26:
Hi @Galvo87 

Sorry about that. I just fixed it and you may rebuild. The explanation is [here](https://tsduck.io/doxy/ts_platform_8h.html#aa5f5dfb0a0f2879ad295da2225e8a9f9).

There is an inconsistency between C++ compilers where something _must_ be defined with GCC without optimization (debug for instance) and _must not_ be defined with MSC. GCC with optimization and clang (all modes) do not care if the definition is made or not. Since my main development platform is macOS (clang), I rarely build on Linux in debug mode. The CI/CD automated process builds and tests on Linux, but not in debug mode.

Galvo87, 2020-07-25:
Hi, I managed to get a dump file after a crash.
How can I safely share it to you?
Thanks

lelegard, 2020-07-25:
Hi @Galvo87,

Thanks for your efforts. You can use any file transfer service if the file is too big to be attached to the GitHub issue. But I probably won't be able to download the file in the next two weeks, so please take care of the file expiration date in services such as wetransfer. 

Please also provide the log file with option `--monitor` that I recommended several times. It is important to get a timeline of the usage of virtual memory, in case the problem is a memory leak.

Galvo87, 2020-08-12:
Hi, [here](https://www.dropbox.com/sh/uud6ooecztq3kmt/AACEFm8JDcDjQfrDt3FNXIXza?dl=0) you can access the dumps, please let me know when you've downloaded them.
Thank you.

lelegard, 2020-08-12:
Thanks, I got them. Analyzing core dumps requires the executable which generated them. Could you provide the exact `tsp` binary which was used? Do you also have the the log with `--monitor` option?

Galvo87, 2020-08-13:
Sure, you can find them [here](https://www.dropbox.com/sh/2flbqtothniwui8/AABRxUS-go3EDvDYracuPdpPa?dl=0).
tsp version is 3.22-1882

lelegard, 2020-08-13:
Alright, I meant all TSDuck binaries, not only `tsp`, since most code in inside shareable libraries.

~~~
Core was generated by `tsp --monitor -v -I dvb --signal-timeout 10 --adapter 12 --frequency 498000000'.
~~~
And _please_, also provide the process log, especially for the `--monitor` output.

Galvo87, 2020-08-13:
Alright, I must have misunderstood something... how can I generate the process log using --monitor?
How can I find all TSDuck binaries and sharead libs?
Thanks