tsduck/tsduck #605: Support for HiDes UT-200C

nuclidjp, 2020-06-20:
Hi, @lelegard 

Please consider supporting HiDes UT-200C.

UT-200C
http://www.hides.com.tw/product_ut200_eng.html

Official driver
https://www.dropbox.com/sh/gm1m9mcuwucilr5/AABI9SNeYgasC02GmrVbgKCTa?dl=0

Other Linux drivers
https://github.com/Zapelin-Media/ite

I hope the official UT-200C driver does not have the same problems as the official UT-100C driver.

My Toshiba TV receives the UT-200C RF output every time without any problems.
However, the USB tuner may or may not be able to receive signals.
I've tried Windows, Linux, and other software, but the situation remains the same.
This may once again be a problem with my environment or settings.

I don't think there is a problem with the signal strength.
Verified on my TV and level checker at an electronics store.
```
scanning channel 13, 473,142,857 Hz
* UHF channel 13 (473.142 MHz), strength: 100%, quality: 100%
  Transport stream id: 32464, 0x7ED0

  Name      ServId TSId   ONetId Type PMTPID
  --------- ------ ------ ------ ---- ------
  NHK総合1・秋田 0x4800 0x7ED0 0x7ED0 0x01 0x01F0
  NHK総合2・秋田 0x4801 0x7ED0 0x7ED0 0x01 0x03F0
            0xFFF0 0x7ED0 0x7ED0      0x1CF0

```

lelegard, 2020-06-21:
Hi @nuclidjp 

I don't think I will do that in a near future. There are several reasons for that:

- It is extremely time consuming and I have few time.
- This is an ISDB-T modulator and I only have DVB receivers, so I will be impossible to test for me.
- The software support for HiDes device is poor. I had long email exchanges with ITE guys on what should be done to improve the situation and make the software part of HiDes device easier to integrate, maintain and upgrade. The exchanges were quite cordial but I saw no improvement since then. So, this is a bit discouraging. Spending a lot of time for little or no result.

nuclidjp, 2020-06-21:
Hi @lelegard 

I've acknowledged.
Thank you for your efforts.

closed 2020-06-21 by nuclidjp