tsduck/tsduck #606: tscharset in Windows console output garbled

trollcop, 2020-06-21:
Heya,

I think in order to properly print UTF-8 in win10 console you need to add

```SetConsoleOutputCP(65001);``` or ```SetConsoleOutputCP(CP_UTF8);```

before calling any printf() stuff.

It doesn't (to my knowledge ?) break anything if you just print ascii or whatever, but will properly handle UTF. I've tested this by doing chcp 65001 then re-running tscharset:

[image]

It's not a deal breaker since I could just manually chcp but I think if you're planning to print UTF-8 better change console codepage anyway?

lelegard, 2020-06-21:
Hi @trollcop 

Thanks for the advice, I will have a try. Nowadays, most operating systems use UTF-8 by default.  Thus, all languages can be correctly displayed. On macOS and Linux terminals, all languages are OK and Japanese looks correct to me (as far as I can visually recognise the characters).

I know that Windows cannot be avoided as an operating system and this is the reason why TSDuck wants to support it decently. But I try to avoid it as much as I can for many reasons:

- Compilation speed : look at the "Actions" tab in the GitHub repo of the project, the compilation time is 7 or 8 mn on Linux and macOS and 45 mn on Windows. I have the same order of magnitude on physical machines.
- Compiler bugs. I found another one no later than yesterday ([see here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/base/tsEnumUtils.h#L39)) when I tried to improve the type safety of some enumeration types. Also [see here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/base/tsArgMix.h#L40) for the most vicious one I found and which took more than one year to be fixed.
- DirectShow silently becoming abandonware.

And when I use Windows, I never - _never_ - use the old MS DOS console from the 80's. I use either a Cygwin or Msys shell or a PowerShell windows. If you want to be Windows'ish, use PowerShell (there are many PowerShell scripts in TSDuck).

If TSDuck forces a switch to UTF-8 (no other language-specific charset) on the console, then I need to check:

- Compatibility with all languages
- Impact on other "more civilised" command windows: PowerShell, Cygin, Msys.

I will try and let you know.

trollcop, 2020-06-21:
Thank you, I was just bringing it to your attention with a possible solution, which worked for me.

> use PowerShell

actually powershell has exactly same problem as cmd.exe

i suspect your compilation time problem is due to not using precompiled headers. properly setup msvc should be comparable or faster than gcc.

lelegard, 2020-06-21:
> Thank you, I was just bringing it to your attention with a possible solution, which worked for me.

It does not work for me. I started the application with:
~~~
    report.info(u"previous input CP: %d, output CP: %d", {::GetConsoleCP(), ::GetConsoleOutputCP()});

    if (::SetConsoleCP(CP_UTF8) == 0) {
        report.error(u"SetConsoleCP error: %s", {ErrorCodeMessage()});
    }

    if (::SetConsoleOutputCP(CP_UTF8) == 0) {
        report.error(u"SetConsoleOutputCP error: %s", {ErrorCodeMessage()});
    }

    report.info(u"new input CP: %d, output CP: %d", {::GetConsoleCP(), ::GetConsoleOutputCP()});
~~~

On the old CMD console:
[image]

On PowerShell console:
[image]

On Msys console (`mintty`) with bash shell, already in UTF8 mode anyway:
[image]

But explicitly changing the CP from outside does not help either on CMD console:
[image]

Maybe some fonts for CMD console are missing (this is an English version of Windows).

But anyway, it works all the time with `mintty`. So, the solution is : use a civilised console 😄

trollcop, 2020-06-21:
Hi,

The output with questionmark-boxes is correct behavior if console font is not capable of Unicode.

However not changing codepage still results in junk output:

[image]

If you try tscharset without codepage changing mod, your output should be same as 1st run in my screenshot, without chcp. Are you seeing the same?

lelegard, 2020-06-21:
All right, I made the update. Please check it. Now, all TSDuck command line tools switch the input and output code pages to UTF-8 on initialisation and restore the previous code pages on exit.

Installing Japanese locale and switching to MS Gothic font, I can see the Japanese characters. However, having all backslash characters transformed into a Yen symbol is very disturbing. If the output were a true UTF-8, it wouldn't do that. The two characters have distinct Unicode code points. So, there must be something wrong in those fonts. I see that your screen snapshots have the same problem. How can you live on Windows with all directory separators being displayed as Yen?

trollcop, 2020-06-21:
> Please check it. 

I'm grab the nightly build soon and update once checked.

> How can you live on Windows with all directory separators being displayed as Yen?

Haha, this has been the case since forever, one of Microsoft guys wrote about the reasons back in 2005 or so but this literally dates back to msdos days.

https://web.archive.org/web/20090305082113/http://blogs.msdn.com/michkap/archive/2005/09/17/469941.aspx

lelegard, 2020-06-22:
> Haha, this has been the case since forever, one of Microsoft guys wrote about the reasons back in 2005 or so but this literally dates back to msdos days.

Amazing! The guy really made fun of a whole country. Two countries, apparently.

And they still dare to call it "UTF-8 code point". In 2020, seriously ?

This is even more weird as Msys or Cygwin command windows correctly display backslashes _and_ Japanese characters at the same time in the default configuration. Why are you waiting to switch to a civilized shell 😄 ?

trollcop, 2020-06-22:
Looks like nightly bulids are updated and I can confirm utf8 output is working as intended:

[image]

This can be considered closed now, thanks a lot!

closed 2020-06-22 by trollcop