tsduck/tsduck #607: TSDuck fails to read TS file that does not start with a sync byte

APTX, 2020-06-25:
I got a TS file and wanted to use it with DekTec. The `file` input plugin and `tsanalyze` would print:
```
cannot detect TS file format
```
and `tsanalyze` would claim there are no ts packets in the file. Providing the format `--format TS` did not help.

I tried a [DVB Inspector](http://www.digitalekabeltelevisie.nl/dvb_inspector/) and it could open the file without issue. I found the offset at which the first valid packet is in the file and then tried:
```
tsp -I file -b <offset> file.ts -P analyze -O drop
```
This worked and I got the _analyze_ output. The offset was less than 188 bytes, so it's probably an incomplete packet at the beginning of the file.

I don't know how to do this with `tsanalyze` as `-b` there is for _bitrate_, not _byte offset_.

Is TSDuck supposed to support such TS files? Can it work with partially corrupt files? In general the MPEG2-TS format should allow for recovery after corrupted data.

lelegard, 2020-06-25:
The tool `tsresync` can fix that kind of problem. There are many options, check the user's guide.

APTX, 2020-06-25:
Is there no way to have TSDuck handle this on the fly? This looks like a severe limitation that forces the user to not only know about this limitation, but also make "fixed" copies of ts files.

lelegard, 2020-06-26:
>  This looks like a severe limitation that forces the user to not only know about this limitation, but also make "fixed" copies of ts files.

Well, in any kind of environment or product, rejecting corrupted input is rarely called a _limitation_, even less a _severe limitation_.

And having a tool to cleanup corrupted input is not a common feature. So, one could say that TSDuck gives more opportunity to handle corrupted input than many other tools.

For instance, I used `tsresync` several times to extract a TS from Wireshark captured data. I mean without analysing the data based on the knowledge of the pcap format, just extracting the TS from a mass of raw binary data of any format. This requires some options and capabilities to go back and forth that cannot be generalized to all commands.

APTX, 2020-06-26:
> Well, in any kind of environment or product, rejecting corrupted input is rarely called a limitation, even less a severe limitation.

If you phrase it as: "rejecting all input due to recoverable corruption" it starts to look more like a limitation (and definitely not a feature).

I also tried `vlc` and `DtPlay` and both successfully read the same TS file TSDuck can't. If other tools that read TS files can read such files, then it not unreasonable to conclude that it is a limitation of TSDuck. This is unfortunate as I consider TSDuck a much more useful tool than any other TS related tools I've tried.

`tsresync` sounds like it can do a lot more than is needed for most cases. It would be very useful to be able to use that from `tsp`, but (as far as I understand TSDuck plugins) it can't be a processing plugin.

`tsanalyze` should at least say that there is a sync error, to give the user a chance at figuring out what to do.

The file I tried to use is a test file from a broadcaster. I haven't seen sync errors in any TS files I've captured myself, so it _may_ be that the corruption is there for a reason (though that reason is beyond me).

APTX, 2020-07-01:
I realized it's possible to do:
```
tsresync file.ts | tsanalyze
```
This allows `tsanalyze` to work on partially corrupt .ts files, even though the way to call it is non-obvious. Piping `tsresync` output works with all the TSDuck tools including `tsp` (where standard input can be used instead of an input plugin).

This is almost all I wanted, the only thing missing are some of the `file` input plugin features, like looping over the file. `tsresync` itself does not provide those features, and it's not really possible to do them in `tsp` when standard input is used.

It's still possible to save the `tsresync` output in a file (essentially a fixed copy of the original .ts without the sync errors) and use that, but I specifically want to avoid making copies of .ts files, just to work with them.