tsduck/tsduck #608: DTAPI_E_INSUF_LOAD when outputting to DekTec modulator

APTX, 2020-06-25:
I don't know if this is TSDuck related at all.

Sometimes I get this error:
```
* dektec: DTU-215 output FIFO load is 6,710,848 bytes, starting transmission
* Error: dektec: output device start send error: DTAPI_E_INSUF_LOAD (DTAPI status 4130)
* dektec: terminating DTU-215 output
* dektec: DTU-215 output terminated
```
When this happens I disconnect/reconnect the DekTec device and the error goes away:
```
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 13,421,696 bytes.
* dektec: DTU-215 output FIFO load is 13,421,696 bytes, starting transmission
```

The FIFO size is different when it works. The same TSDuck call is used for both calls.

Is this something that tends to happen with DekTec devices?

lelegard, 2020-06-25:
This is probably due to packet starvation. The producer must probably not provide packets at a sufficient bitrate for the target modulation parameters. You could try option `--stuffing` to the output plugin `dektec`. In case of packet starvation, null packets are automatically generated. But I think that this works only on some modulators with hardware stuffing capabilities. I used to have an old PCI DVB-T modulator on which this did not work. You should read the data sheet of your modulator (or simply try the option).

APTX, 2020-06-25:
Do you have any idea why packet starvation might occur? I use file as input (the same file even!). Shouldn't it always be able to provide enough data?

APTX, 2020-06-25:
I will try the `DtPlay` tool if I see the error again to see if it behaves any differently.

APTX, 2020-07-01:
I got the issue to reproduce and tried it `DtPlay` to see what it would do.

The same error happens with `DtPlay`. This can not be a TSDuck issue, it's either a problem with the DekTec SDK or the device I have (so a DekTec issue either way). Disconnecting and reconnecting the device made it work again with both `DtPlay` and `TSDuck`.

I noticed I did not copy the FIFO size log, so I updated the previous comment. From the log it looks like TSDuck adjusts the load to the FIFO size reported by the DekTec device, which sometimes reports a value that is too low.

closed 2020-07-01 by APTX