tsduck/tsduck #611: Question about section organization in EIT Schedule table

charluy, 2020-06-29:
Hello everyone, 

I have been making a DVB table generator using TSDuck classes, and had no issues until the EIT schedule table.
When i generate an EIT schedule table (actual or other) whit a few events, can see that uncomplete segments are filled whit empty sections and empty segments are filled whit 8 empty sections, all this until the "last section number".

Acording to this recomendation: https://www.etsi.org/deliver/etsi_ts/101200_101299/101211/01.12.01_60/ts_101211v011201p.pdf in the section "EIT Schedule structure" on page 16 it's says:

> "Entirely empty segments shall be represented by an empty section, (i.e. a section which does not contain any
> loop over events) with the value s0 + 0 encoded in the field segment_last_section_number."

So i wonder if there is any way to generate the EIT schedule table following that recommendation, adding only one empty section per empty segment and not adding empty sections at end of uncomplete segments (i think this last one its not in the recommendation).

I tried to use the static function ts::EIT:Fix whit all 3 possibilities like this: 

```
// previusly defined and filled ts::EIT eit.
ts::BinaryTable bin_eit;
eit.serialize(duck, bin_eit);
ts::EIT::Fix(bin_eit, ts::EIT::FIX_EXISTING); // Also try whit FILL_SEGMENTS and ADD_MISSING.
this->packetizer.removeAll();
this->packetizer.addTable(bin_eit, /*repetition_rate*/(ts::MilliSecond)0);
```

and got the same output.

To clarify what i mean, i left a wireshark capture of the UDP output made directly whit the abstract table:

`this->packetizer.addTable(duck, eit, /*repetition_rate*/(ts::MilliSecond)0);`

[eit_schedule_sid_1101.zip](https://github.com/tsduck/tsduck/files/4848075/eit_schedule_sid_1101.zip)

Best regards, Diego Sánchez.

charluy, 2020-07-10:
Hello again,

I came up whit a fix to work around this issue and i would like to shared in case someone need it.

First of all, in the file tsEIT.cpp it is necesary to add a new mode in the FixMode enum called REMOVE_EMPTY and then make the following changes:

```
void ts::EIT::serializeContent(DuckContext& duck, BinaryTable& table) const
{
    ...

        // If we cannot at least add the fixed part, open a new section.
        // Add empty sections only at the begginin of a segment.
        while (remain < EIT_EVENT_FIXED_SIZE || section_number < target_section) {
            if(remain < EIT_EVENT_FIXED_SIZE) {
                addSection(table, section_number, payload, data, remain);
            }
            else if((section_number%8) == 0) {
                addSection(table, section_number, payload, data, remain);
            }
            else {
                section_number++;
            }
        }

    ...

    Fix(table, REMOVE_EMPTY);
}
```

```
void ts::EIT::Fix(BinaryTable& table, FixMode mode)
{
    ...

    // Now add or fix sections.
    for (size_t si = 0; si < table.sectionCount(); si++) {

        // Section pointer.
        const SectionPtr& sec(table.sectionAt(si));
        // Segment number of this section:
        const size_t seg = si / SECTIONS_PER_SEGMENT;
        // Identified last section in this segment:
        const uint8_t seg_last = segment_last_section_number[seg];

        // Process non-existent section.
        if (sec.isNull()) {
            // Create an empty section if required.
            if ((mode != REMOVE_EMPTY) && (mode > FILL_SEGMENTS || si > seg_last)) {
                empty_payload[4] = seg_last;
                table.addSection(new Section(tid,
                                             is_private,
                                             table.tableIdExtension(),
                                             table.version(),
                                             is_current,
                                             uint8_t(si),  // section_number
                                             last_section, // last_section_number
                                             empty_payload,
                                             EIT_PAYLOAD_FIXED_SIZE));

            }
        }
        else if (((mode == REMOVE_EMPTY)||(mode == FIX_EXISTING)) && sec->isValid() && sec->payloadSize() >= EIT_PAYLOAD_FIXED_SIZE) {
            // Patch section: update last section in segment and last table id.
            uint8_t* const pl = const_cast<uint8_t*>(sec->payload());
            if (pl[4] != seg_last || pl[5] != last_table_id) {
                pl[4] = seg_last;
                pl[5] = last_table_id;
                sec->recomputeCRC();
            }
        }
    }
}
```

Next in tsCyclingPacketizer.cpp:

```
void ts::CyclingPacketizer::addTable(const BinaryTable& table, MilliSecond rep_rate)
{
    for (size_t i = 0; i < table.sectionCount(); ++i) {
        if((table.tableId() >= 0x50) && (table.tableId() <= 0x6F)) {
            if(!table.sectionAt(i).isNull()) {
                addSection(table.sectionAt(i), rep_rate);
            }
        }
        else {
            addSection(table.sectionAt(i), rep_rate);
        }
    }
}
```

Hope someone found this usefull.

Best regards, Diego Sánchez.

lelegard, 2020-07-11:
Hello @charluy 

Sorry for the late reply;

This was a long standing issue in TSDuck. The core engine of TSDuck was designed to automatically process binary tables of any kind. But EIT's are not real "tables" as defined by ISO 13818-1 (MPEG-TS). The usage rules from ETSI TS 101 211 are such that EIT sections are grouped into segments but there should be no section between segments when they are not adjacent. In the MPEG definition, a table is a set of _all_ sections with same table id, table id extension and version, _all_ sections from zero to _last_section_number_. In the case of EIT's, there are missing sections and the result cannot be qualified as a _table_.

The class `BinaryTable` which is used by the method `EIT::Fix()` contains a binary _table_. You can add sections one by one but the object is marked as valid only when all sections are present. So, a conformant set of EIT sections (according to ETSI TS 101 211) will never be representable as a `BinaryTable`. This is why `EIT::Fix()` is a dead end. It will never implement ETSI TS 101 211 correctly.

So, I decided to fix that once for all using a new static method named `EIT::ReorganizeSections()`. Unlike `EIT::Fix()`, it uses a `SectionPtrVector`, a list of arbitrary sections, not a `BinaryTable`. There is no other parameter. It just takes a list of sections and completely reorganises them to implement ETSI TS 101 211. Non-EIT sections are left unmodified.

On output, all sections can be simply injected in a packetizer.

At command line level, the new option `--eit-normalization` can be used where a section file or XML file is loaded. After loading all sections or XML structures, `EIT::ReorganizeSections()` is invoked on them. So, it is now easier to inject fully conformant EIT's using plugins such as `-P inject --eit-normalization eit.xml`. You don't have to worry about the organization of events inside EIT's. Just put events in `<EIT>` any number of structures (you just need to specify the correct identification of the service) and the option `--eit-normalization` will do the rest.

There are some explanations [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tables/tsEIT.h#L43) and [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tables/tsEIT.h#L331).

charluy, 2020-07-15:
Hello @lelegard 

Thanks you very much for the reply and for the new feature,

Although i'm concerned about efficiency when it's necessary to make a little change in the content of some table. For example, if i generating the EIT of an transport stream in a network with one hundred services and all of them have several events, then i understand that it is necessary to: 

- Re serialize the given EIT into the binary table.

- Re collect in a vector pointer all EIT sections of all services in the network.

- Apply the static method EIT::ReorganizeSections() to the vector pointer.

- Erase the content of the cycling packetizer and re inject all the binary.

And all this will happen when any change it's made on some event. Therefore i was thinking that it may be useful to have a EIT constructor that take the value of last_table_id as parameter so there's no need to invoke the reorganizeSections() method over all the sections, it will be only necesary to do it on each EIT separatly. Then each EIT table will be packetized in a different CyclingPacketizer and his continuity counter has to be restamped before sending.

I think that this last step is not covered by tsduck but I think it is useful for those who use it as an external library.

I don't really know if the first option will be inefficient so i like to know your opinion.

Best regards, Diego Sánchez.

lelegard, 2020-07-15:
Hi @charluy 

I understand your concern. I would suggest the following approach which (I think) gives some good balance between efficiency and accuracy:

- Segment your event processing per service. Modifying an event may potentially need to regenerate all EIT schedule sections for that service, without impact on other services.
- Use one `SectionFile` per service. You do not need to "save" it as a file, just use an instance of the class.
- Build your EIT's for a service using one or more instances of `EIT` class. You can use one instance per event or all events in one instance, we do not care. Use safe pointer to `EIT` (`AbstractTablePtr`).
- Store all your `EIT` instances for that service in the `SectionFile` for that service using `section_file.add(eit_ptr)`.
- Call `section_file.reorganizeEITs(reftime)`. The `SectionFile` now contains the final EIT sections, ready for broadcast.
- Erase all EIT sections for that service from the packetizer: `packetizer.removeSections(table_id, service_id)`. You have to do that for all `table_id` in the range `TID_EIT_S_ACT_MIN` to `TID_EIT_S_OTH_MAX`.
- Load the updated EIT sections in the packetizer: `packetizer.addSections(section_file.sections())`

In the general case, modifying events in a service may need to rebuild a large number of EIT sections for that service (especially if the start time is modified). So, I think that what is described above is the minimum work to do.

If you want to fine tune the system and limit the modification to the very minimum, then you will end up with a very complex code and many specific cases. This will be error-prone, long to write and a burden to maintain.

charluy, 2020-07-17:
Hi @lelegard 

Thanks you very much for this suggestion, i'm already working on implement it.
Although i have one questions (hope not being very annoying): this method will not have 100% accuracy?

Best regards, Diego Sánchez.

lelegard, 2020-07-17:
This method will provide 100% accuracy in terms of section content and organization in segments.

charluy, 2020-07-22:
Excellent, thank you very much for all the help.

Best regards, Diego Sánchez.

closed 2020-07-22 by charluy