tsduck/tsduck #612: Can you make tsp ignore TS packets with wrong sync word?

VDelport, 2020-07-03:
First of all, sorry for asking this question about the issues.

I built a simple radio transmission system with GNU Radio. The data is not protected with an FEC error protection.

I want to transmit a transport stream with embedded UDP segments. The TS data is evaluated on the receiving side with the following tsp command:

/home/delport/.blocksat/src/tsduck/bin/release-x86_64-ftkt-nb04/tsp --realtime --buffer-size-mb 1.0 --max-flushed-packets 10 --max-input-packets 10 -P bitrate_monitor -p 1 --min 0 -I file ts_input_fifo -P mpe --pid 32-33 --udp-forward --local-address 127.0.0.1 -O drop

If I simulate the radio transmission system (no noise), everything goes well. When I work with SDR devices and antennas, i.e. with a disturbed radio channel, tsp stops because the synchronization word was not found.

* Error: file: synchronization lost after 1,045 packets, got 0x20 instead of 0x47

Is there an option for tsp that tsp simply ignores TS packets for which it has not found a synchronization word?

lelegard, 2020-07-03:
This is the role of a demodulator to check errors and verify TS sync bytes. In case of error detection and partial recovery, the _transport_error_indicator_ bit shall be set in the TS header.

This what a physical tuner does. This is why the input plugin `dvb` always receives correct TS packets. If the signal is noisy, packets are dropped or corrected by the tuner / demodulator.

An SDR-based software demod shall do the same.

VDelport, 2020-07-03:
Ok, thank you for your answer.

closed 2020-07-03 by VDelport