tsduck/tsduck #613: When trying to replace PMT section using the inject plugi, no PCR's are available on the output file

mitso23, 2020-07-05:
Hi Thierry,

Thanks a lof for the amazing work you have done. 

I am trying to replace a PMT on an existing stream using the following command,
tsp -I file INPUT_FILE  -P inject pmt.xml --replace --pid 256  -O file OUTPUT_FILE

The PMT is successfully replaced, however there are no PCR's present on the output file. Is that expected behaviour ?
My expectation would be re-adjusted on the output stream.

Thanks a lot,
Dimitrios

lelegard, 2020-07-05:
> The PMT is successfully replaced, however there are no PCR's present on the output file. Is that expected behaviour ?

It is indeed.

TSDuck is a toolbox, the screwdriver, hammer and saw of transport streams. Each tool does what it does, no more, no less. Building a system with these tools is left to the human engineer. The plugin `inject` injects sections from a binary or XML file. That is what it does. It does not interpret the content of the file, it can even contain incomplete tables or invalid sections (in the case of a binary file), the crappy sections will be inserted without further check (which can be extremely useful in some test cases).

> My expectation would be re-adjusted on the output stream.

What do you mean by "re-adjust" the output stream? Replacing the content of a PID does not change the clock of the stream, the PCR shall remain unchanged.

mitso23, 2020-07-05:
Hi Thierry,

Thanks for the reply. Sorry what I wrote doesn't make sense. What I mean is that if we replace an existing pid then at least the PCR in the input file should be preserved as is in the output file. I think even if we add an extra section then at bare minimum the output PCR should match the input PCR (in my case I want to inject some static ECM's that shouldn't impact the overall bitrate of the stream). 

Could you explain the reasoning of removing the PCR from all the pids on the output file ? Maybe it would be better to remove only the PCR pid (as this represents the overall bitrate of the stream and keep the remaining PCR as is)

Many thanks again.
Dimitrios

lelegard, 2020-07-05:
> Could you explain the reasoning of removing the PCR from all the pids on the output file ? 

Your command only replaces PID 256. All other PID's, including their PCR's, are left untouched.

If you had PCR's on PID 256, they are of course removed. But this is normal. PCR's are usually present on video or audio PES PID's and sometimes on dedicated PID's. Most of the time, PCR's are present in the video PID of a service. PCR's are never present on section PID's. Well, it is technically feasible but never used in practice because it does not make sense.

Also remember that, in the general case, there is not _one single clock reference_ in a TS. There is one per service. So, when you replace a PES PID with a PMT PID, the previous PID cannot belong to the service the PMT of which you are inserting. So, even though it would be technically possible to copy the PCR from input to output in a replaced PID, it would not make sense at all.

Also remember that the PCR's are not part of the TS header. They are part of the adaptation field of the TS packet. So, having a PCR in a TS packet changes the size of the TS payload. The insertion of PCR shall consequently be done ahead of packetization (typically in the encoder in the case of a video or audio PID). So, when you _replace a PID_, you erase everything, PCR included.

lelegard, 2020-07-05:
> Could you explain the reasoning of removing the PCR from all the pids on the output file ?

They are not removed. In you case, all PID's (except PID 256) are left unmodified, PCR included.

mitso23, 2020-07-05:
> > Could you explain the reasoning of removing the PCR from all the pids on the output file ?
> 
> They are not removed. In you case, all PID's (except PID 256) are left unmodified, PCR included.

Unfortunately this is not what happens. When i tsdump the generated output file there is not a single PCR reference present. Same result I get when I call tsbitrate, it reports "cannot compute transport bitrate, insufficient PCR". 

In my case I am replacing a PMT pid with another PMT pid (none of them has PCR as you said). My problem is that the PCR gets removed from the other PIDS as well. I must be either doing something wrong or there is a bug somewhere. 

Do you see anything wrong in this command "tsp -I file INPUT_FILE -P inject pmt.xml --replace --pid 256 -O file OUTPUT_FILE" ?

When I use tscmp I can see that for some reasons all the pcr are removed. 
Packet 48 differ at offset 5, 7 bytes differ, PID 4097, packet 0 in PID
  Packet from boa_clear.ts:
      ---- TS Header ----
      PID: 4097 (0x1001), header size: 188, sync: 0x47
      Error: 0, unit start: 0, priority: 0
      Scrambling: 0, continuity counter: 0
      Adaptation field: yes (184 bytes), payload: no (0 bytes)
      Discontinuity: 0, random access: 0, ES priority: 0
      PCR: 0x003C4E5E1E8
      ---- Full TS Packet Content ----
      47 10 01 20 B7 10 01 9B B0 97 7E 00 FF FF FF FF  G.. ......~.....
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF              ............
  Packet from boa_clear_premium_pmt.ts:
      ---- TS Header ----
      PID: 4097 (0x1001), header size: 188, sync: 0x47
      Error: 0, unit start: 0, priority: 0
      Scrambling: 0, continuity counter: 0
      Adaptation field: yes (184 bytes), payload: no (0 bytes)
      Discontinuity: 0, random access: 0, ES priority: 0
      ---- Full TS Packet Content ----
      47 10 01 20 B7 00 FF FF FF FF FF FF FF FF FF FF  G.. ............
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF              ............
  Differing area from boa_clear.ts:
      10 01 9B B0 97 7E 00                             .....~.
  Differing area from boa_clear_premium_pmt.ts:
      00 FF FF FF FF FF FF                             .......

lelegard, 2020-07-05:
> > > Could you explain the reasoning of removing the PCR from all the pids on the output file ?

> > They are not removed. In you case, all PID's (except PID 256) are left unmodified, PCR included.

> Unfortunately this is not what happens. When i tsdump the generated output file there is not a single PCR reference present. 

Well, if such a simple command which does nothing but injecting a PMT in replacement of another one had the side effect of removing all PCR's, then many users would have run into it. I just tried on a sample TS and, as expected, only the PMT PID was replaced. All other PID's are unmodified.

If you are absolutely positive about the disappearance of the PCR's in your file, please post the input TS file and your XML file to some file sharing service and I will have a look.

As additional info, could you please post your system and TSDuck version? Did you install binaries or did you recompile TSduck?

mitso23, 2020-07-06:
Hi Thierry,

Thanks for your replay. Unfortunately I can't share the stream. The problem seems to occur only with this particular stream. And  in my case it looks like VLC is still able to stream it correctly (defaults to DTS timestamps). Once I find some time I will debug this problem myself. I got the source code and I already start looking into it. 

Many thanks again for your help. I will close the issue now.