tsduck/tsduck #614: Scte35 Splice Insert over a video file

gdavila, 2020-07-08:
First of all, thanks for this wonderful tool. It is a MUST for all video/media engineers!!!

I'm using `tsduck` to stream a video file in loop. On the receiver of this video loop, I need to detect each time the loop begins (I can accept a couple of seconds of error) so I believe that inserting SCTE-35 at the beginning of the video as markers could help me. In this way, I'm playing with `spliceinject` option and here arises my question. When I define  (within the xml file) the `splice_insert` with `splice_immediate="false"` the SCTE-35 is rightly injected. However, when I use `splice_immediate="true"` the injections doesnt seem to work. ¿Any clue about what am I misunderstanding? 

Below, is the command I'm using  in order to create a video sample with SCTE-35 injection: 

```sh
tsp -v --add-input-stuffing 1/10 \
 -I file BBB_reference_10s.ts \
 -P pmt --service 1 --add-programinfo-id 0x43554549 --add-pid 600/0x86 \
 -P spliceinject --service 1 --files scte35.xml --wait-first-batch  \
 -O file BBB-scte35.ts
```
The previous command produces the following output:

```
* spliceinject: waiting for first batch of commands
* spliceinject: loaded file scte35.xml, 528 bytes
* spliceinject: enqueuing SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate
* spliceinject: received first batch of commands
* file: initial input bitrate is 8,640,187 b/s
* pmt: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* pmt: PMT version 0 modified
* spliceinject: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* spliceinject: injecting SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate, current PTS: 0x0000203A0
```
According to `tsanalyze`output, the video created (BBB-scte35.ts) doesnt seem to have the SCTE35 markers.

```
tsanalyze BBB-scte35.ts
|=============================================================================|
|  PID: 0x0258 (600)                                     SCTE 35 Splice Info  |
|  Service: 0x0001 (1) Big Buck Bunny, Sunflower version                      |
|-----------------------------------------------------------------------------|
|  0xFC (252, SCTE 35 Splice Information)                                     |
|      Repetition: ....... 0  ms  Section cnt: ........ 1                     |
|      Min repet.: ....... 0  ms                                              |
|      Max repet.: ....... 0  ms                                              |
|=============================================================================|

```
The xml file I'm using:

scte35.xml
```xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <splice_information_table protocol_version="0" pts_adjustment="0" tier="0x0FFF">
    	<splice_insert splice_event_id="0x00000012" splice_event_cancel="false"  out_of_network="true" splice_immediate="true"  unique_program_id="0x00" avail_num="0" avails_expected="0">
      	<break_duration auto_return="true" duration="360,000"/>
    	</splice_insert>
    <splice_avail_descriptor identifier="0x43554549" provider_avail_id="0x00000012"/>
  </splice_information_table>
</tsduck>
```

lelegard, 2020-07-09:
> According to tsanalyzeoutput, the video created (BBB-scte35.ts) doesnt seem to have the SCTE35 markers.

Why do you say that there is no marker? Note the `Section cnt: ........ 1`. So, `tsanalyze` has found one section in the PID, which seems normal for one "immediate" section. Of course, only one section is not enough to evaluate the repetition rate (hence the zero).

Try command `tstables BBB-scte35.ts -a -p 600` to view the section.

gdavila, 2020-07-09:
Thanks! 
`tstables BBB-scte35.ts -a -p 600` shows the scte35 marker inserted. My misunderstanding was due to the `repetition` value. I thought it always had to be greater than 0. Now I kown that for some reason my encoder on the receiver side is not reading the marker with `splice_inmediate=true`. I will try with another encoder.

BTW, if I use `splice_inmediate=false`, I noticed that the `pts_time` must always to be greater than 2 seconds (180000), otherwise, the marker is not inserted and I see the following log:
```
* spliceinject: waiting for first batch of commands
* spliceinject: loaded file scte35_v2.xml, 532 bytes
* spliceinject: enqueuing SpliceInsert out @0x00002BB38, start: 0x1FFFFFC18, end: 0x00002BB38, 2 times
* spliceinject: received first batch of commands
* file: initial input bitrate is 8,640,187 b/s
* pmt: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* pmt: PMT version 0 modified
* spliceinject: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* spliceinject: dropping SpliceInsert out @0x00002BB38, start: 0x1FFFFFC18, end: 0x00002BB38, 2 times, obsolete, current PTS: 0x00002EE00
```

 As far as I understand, this is because two copies of the `splice_insert` command are injected: the first copy would be injected two seconds earlier of the specified `pts_time`. Am I right? Is it possible to change this behaviour?

lelegard, 2020-07-10:
> Am I right? Is it possible to change this behaviour?

Yes and yes.

Please read the user's guide, reference section for plugin `spliceinject`. TSDuck has many plugins with many options. It is impossible to use without the user's manual. The manual itself is pretty big but just read the table of contents and jump to the appropriate section.

gdavila, 2020-07-14:
Thanks a lot,
I will close the issue given that the misunderstanding was mine, as I expected =)

closed 2020-07-14 by gdavila